fwupd/fwupd #6871: fwupd interferes with usb device behind nested hubs

wshanks, 2024-03-04:
**Describe the bug**
A Kensington USB mouse ([this one](https://www.newegg.com/kensington-ci20/p/N82E16826125045) if it is relevant) stopped working after I applied the most recent updates on my Fedora 39 laptop (Lenovo ThinkPad T14 Gen 2). When I look at the system logs, I see

    fwupd[4250]: 14:56:59.731 FuUsbDevice          failed to load BOS descriptor from USB device: USB error on device 047d:102e : Operation timed out [-7]

That message led me to [this Debian mailing list post](https://www.mail-archive.com/debian-bugs-dist@lists.debian.org/msg1958725.html) from a few days ago which describes a similar issue and goes into more detail of attempted debugging steps. For me, I found that the mouse just did not work at all, rather than partially working like reported there. I tried `systemctl stop fwupd.service` and confirmed that the mouse started working again after unplugging it and plugging it back in. `systemctl start fwupd.service` stopped the mouse from working once again. Checking the history of dnf updates, I see that the `fwupd` packages were updated from 1.9.13 to 1.9.14:

```
    Upgrade  fwupd-1.9.14-1.fc39.x86_64                          @updates
    Upgraded fwupd-1.9.13-1.fc39.x86_64                          @@System
    Upgrade  fwupd-plugin-flashrom-1.9.14-1.fc39.x86_64          @updates
    Upgraded fwupd-plugin-flashrom-1.9.13-1.fc39.x86_64          @@System
    Upgrade  fwupd-plugin-modem-manager-1.9.14-1.fc39.x86_64     @updates
    Upgraded fwupd-plugin-modem-manager-1.9.13-1.fc39.x86_64     @@System
    Upgrade  fwupd-plugin-uefi-capsule-data-1.9.14-1.fc39.x86_64 @updates
    Upgraded fwupd-plugin-uefi-capsule-data-1.9.13-1.fc39.x86_64 @@System
```

and I do not notice other relevant packages being updated (like something with `usb` in the name).

Like the mailing list post points out, the most suspicious line in the release notes for 1.9.14 seems to be "Fix DS-20 descriptors by opening the GUsbDevice earlier" and I noticed that is some previous discussion about difficulties with nested USB hubs here (the only thing that comes up when I search for "BOS descriptor").

One other note -- when I first noticed this, I actually had the mouse plugged into a StarTech USB hub (which lsusb detects as "Genesys Logic, Inc. GL3523 Hub") which was then plugged into a Lenovo Thunderbolt 4 dock which was plugged into the Thunderbolt 4 port of the laptop. However, I then tested and got the same issue both using the mouse plugged directly into the dock and using the mouse plugged directly into a usb port on the laptop.

**Steps to Reproduce**
This is mostly described above:

1. `systemctl stop fwupd.service`
2. Plug a Ci20 Kensington optical mouse into the laptop
3. Mouse should work
4. `systemctl start fwupd.service`
5. Mouse should not work any more

In my standard set up, I have a webcam and keyboard plugged into the StarTech hub plugged into the Thunderbolt 4 dock as well and they are not affected by fwupd.

**Expected behavior**
`fwupd.service` should not stop the mouse from working.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version

compile   org.freedesktop.fwupd         1.9.14
compile   com.hughsie.libxmlb           0.3.15
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.14
runtime   org.kernel                    6.7.6-200.fc39.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): dnf

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

**NOTE**: I ran this without the Thunderbolt 4 dock or the StarTech hub for simplicity, so it should just be a standard Thinkpad T14 Gen 2 output with a USB mouse plugged in.

```shell
fwupdmgr get-devices --show-all-devices
LENOVO 20W1SB9V00
│
├─Unknown Device:
[... 241 lines cut ...]
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               4b3d2033-ae0c-4f25-bc3f-96f0ed347569
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Lenovo Ltd. PK CA 2012
      Vendor:             Lenovo Ltd.
      GUID:               71599d14-9b31-5270-b3bd-74c494585820 ← UEFI\CRT_9AEF2123F4DE7C19AFABD909BB2C8CAC4411E07E

```

</details>

**Additional questions**

- Operating system and version: Fedora Linux 39
- Have you tried rebooting? Yes
- Is this a regression? Yes. No issue was noticed with 1.9.13.

hughsie, 2024-03-04:
Ugh, so I really didn't think _getting_ the BOS descriptor would affect anything. 2/2 of the reports are Kensington mice, which might be helpful. Out of interest, does `sudo gusbcmd save` also cause a warning and for the mouse not to work too?

wshanks, 2024-03-04:
`sudo gusbcmd save` does cause the mouse to stop working. I don't see a warning on the console. This is what I see in `journalctl -xe`:

```
Mar 04 11:53:29 wshankspurple sudo[13308]:  wshanks : TTY=pts/1 ; PWD=/home/wshanks ; USER=root ; COMMAND=/usr/bin/gusbcmd save
Mar 04 11:53:29 wshankspurple audit[13308]: USER_CMD pid=13308 uid=1000 auid=1000 ses=3 subj=unconfined_u:unconfined_r:unconfined_t:s0-s0:c0.c1023 msg='cwd="/home/wshanks" cmd=67757362636D642073617665 exe="/usr/bin/sudo" terminal=pts/1 r>
Mar 04 11:53:29 wshankspurple audit[13308]: CRED_REFR pid=13308 uid=1000 auid=1000 ses=3 subj=unconfined_u:unconfined_r:unconfined_t:s0-s0:c0.c1023 msg='op=PAM:setcred grantors=pam_env,pam_fprintd acct="root" exe="/usr/bin/sudo" hostname>
Mar 04 11:53:29 wshankspurple gnome-terminal-[5272]: void terminal_screen_shell_preexec(VteTerminal*): assertion '!priv->between_preexec_and_precmd' failed
Mar 04 11:53:29 wshankspurple sudo[13308]: pam_unix(sudo:session): session opened for user root(uid=0) by wshanks(uid=1000)
Mar 04 11:53:29 wshankspurple audit[13308]: USER_START pid=13308 uid=1000 auid=1000 ses=3 subj=unconfined_u:unconfined_r:unconfined_t:s0-s0:c0.c1023 msg='op=PAM:session_open grantors=pam_keyinit,pam_limits,pam_keyinit,pam_limits,pam_syst>
Mar 04 11:53:30 wshankspurple kernel: usb 3-7: usbfs: process 13310 (gusbcmd) did not claim interface 0 before use
Mar 04 11:53:30 wshankspurple kernel: usb 3-1: usbfs: process 13310 (gusbcmd) did not claim interface 0 before use
Mar 04 11:53:30 wshankspurple sudo[13308]: pam_unix(sudo:session): session closed for user root
```

From internet searching, it seems like the mouse may have been manufactured in 2006, so it is pretty old. It hasn't otherwise given me any reason to replace it though.

I changed the issue description because I had written it back when I thought it was important that the mouse was behind a dock and a hub.

hughsie, 2024-03-04:
> sudo gusbcmd save does cause the mouse to stop working

@superm1 okay if I add the VID/PID to an internal no-bos-descriptor flag or something like that?

superm1, 2024-03-04:
I mean it makes sense certainly as a bandaid; but are we missing something in the design?  Should we maybe only be opting in devices with a manufacturing date that's new enough?

hughsie, 2024-03-04:
> I mean it makes sense certainly as a bandaid; but are we missing something in the design? 

Maybe!? Getting a USB descriptor shouldn't brick the device, and I'm also wondering how this doesn't break in Windows too.

> Should we maybe only be opting in devices with a manufacturing date that's new enough?

How do we get the manufacturing date tho?

hughsie, 2024-03-04:
@wshanks can you get us the `sudo lsusb -vvv` output please

superm1, 2024-03-04:
> How do we get the manufacturing date tho?

I was naively hoping there was something in the descriptor to go on.  If not; then lets see if there are any better heuristics.

wshanks, 2024-03-04:
Here is the `sudo lsusb -vvv` output:

<details>

```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0002 2.0 root hub
  bcdDevice            6.07
  iManufacturer           3 Linux 6.7.7-200.fc39.x86_64 xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 0000:00:0d.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0019
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x000a
    No power switching (usb 1.0)
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0100 power
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
[... 1635 lines cut ...]
    bmSublinkSpeedAttr[15]   0x00c600ab
      Speed Attribute ID: 11 198Mb/s Symmetric TX SuperSpeed
Hub Descriptor:
  bLength              12
  bDescriptorType      42
  nNbrPorts             4
  wHubCharacteristic 0x000a
    No power switching (usb 1.0)
    Per-port overcurrent protection
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  bHubDecLat          0.0 micro seconds
  wHubDelay             0 nano seconds
  DeviceRemovable    0x00
 Hub Port Status:
   Port 1: 0000.02a0 5Gbps power Rx.Detect
   Port 2: 0000.02a0 5Gbps power Rx.Detect
   Port 3: 0000.02a0 5Gbps power Rx.Detect
   Port 4: 0000.02a0 5Gbps power Rx.Detect
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

</details>

closed 2024-03-05 by hughsie

heyakyra, 2024-03-05:
For googlers, this also seems to affect Ledger Nano X

kdj0c, 2024-03-11:
FYI This also affect Nostromo n52 gamepad.

This makes it disconnect/reconnect in loop:

[  610.101511] hid-generic 0003:050D:0815.0386: input,hidraw1: USB HID v1.10 Mouse [Honey Bee  Nostromo SpeedPad2 ] on usb-0000:05:00.3-1/input1
[  610.117114] usb 3-1: USB disconnect, device number 76
[  610.446931] usb 3-1: new low-speed USB device number 77 using xhci_hcd
[  610.594508] usb 3-1: New USB device found, idVendor=050d, idProduct=0815, bcdDevice= 2.10
[  610.594516] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  610.594520] usb 3-1: Product: Nostromo SpeedPad2 
[  610.594524] usb 3-1: Manufacturer: Honey Bee 
[  610.646861] input: Honey Bee  Nostromo SpeedPad2  as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:05:00.3/usb3/3-1/3-1:1.0/0003:050D:0815.0387/input/input922
[  610.698538] hid-generic 0003:050D:0815.0387: input,hidraw0: USB HID v1.10 Keyboard [Honey Bee  Nostromo SpeedPad2 ] on usb-0000:05:00.3-1/input0
[  610.703784] input: Honey Bee  Nostromo SpeedPad2  as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:05:00.3/usb3/3-1/3-1:1.1/0003:050D:0815.0388/input/input923
[  610.755706] hid-generic 0003:050D:0815.0388: input,hidraw1: USB HID v1.10 Mouse [Honey Bee  Nostromo SpeedPad2 ] on usb-0000:05:00.3-1/input1
[  662.668052] usb 3-1: USB disconnect, device number 77
[  663.053234] usb 3-1: new low-speed USB device number 78 using xhci_hcd
[  663.197720] usb 3-1: New USB device found, idVendor=050d, idProduct=0815, bcdDevice= 2.10
[  663.197727] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  663.197731] usb 3-1: Product: Nostromo SpeedPad2 
[  663.197734] usb 3-1: Manufacturer: Honey Bee 
[  663.238075] input: Honey Bee  Nostromo SpeedPad2  as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:05:00.3/usb3/3-1/3-1:1.0/0003:050D:0815.0389/input

hughsie, 2024-03-11:
@kdj0c does that work in Windows? That's got bcdDevice= 2.10 and so should accept getting the BOS descriptors. I'm guessing stopping fwupd and then doing "sudo gusbcmd save" makes the device act in the same way?

kdj0c, 2024-03-11:
> @kdj0c does that work in Windows? That's got bcdDevice= 2.10 and so should accept getting the BOS descriptors. I'm guessing stopping fwupd and then doing "sudo gusbcmd save" makes the device act in the same way?

I don't have a windows installation here, so I don't know if it still works.

Also here is the log when I run:

fwupdtool get-devices --verbose --verbose

10:30:09.672 FuUsbDevice          ignoring missing BOS descriptor: USB error on device 050d:0815 : Input/Output Error [-1]

On Fedora 39, I fixed the issue by downgrading fwupd to fwupd-1.9.5-2.fc39.

ofourdan, 2024-03-11:
Same here, latest fwupd 1.9.14 breaks my old trustee Wacom Bamboo One.

The device gets removed then re-added, then removed again continuously with fwupd 1.9.14.

Reverting to 1.9.13 fixes that issue.

```
[    2.071746] input: Wacom Bamboo1 Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:056A:0069.0002/input/input10
[    2.071792] wacom 0003:056A:0069.0002: hidraw1: USB HID v1.10 Mouse [WACOM CTF-430 V2.0-0] on usb-0000:00:14.0-2/input0
```

Attaching [lsusb-vvv.txt…]()

hughsie, 2024-03-11:
@ofourdan on a hunch, can you try the https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/ build please

superm1, 2024-03-11:
Or you can grab snap/flatpak artifacts if it's easier depending on your distro.

ofourdan, 2024-03-11:
> @ofourdan on a hunch, can you try the https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/ build please

Works for me! :+1:

hughsie, 2024-03-11:
> Works for me! 👍

Okay, so I think the last commit was unnecessary -- I think we just need a release with https://github.com/fwupd/fwupd/commit/f92ea06110f5075d7b94c90ad1a637f6b43a647c included. @superm1 I'm tempted to leave it *in* as an example on what to do -- and so we can quirk without a recompile if a broken USB 2.1 device turns up in the real world. Or maybe we remove the .quirk file and leave the code change?

superm1, 2024-03-11:
Let's leave both. I agree it's easier to quirk later if the infra is there. Who knows with ancient USB what we'll find!

smunaut, 2024-07-21:
Damn, it took me forever to find this issue ...

My FTDI FT232 USB Serial adapter wasn't working properly ( data transfers were still working but any control transfer to set baudrate etc were crashed, the whole EP0 was unresponsive after trying to get BOS desc) and this was caused by fwupd ( which is 1.9.14 on stable gentoo ).

hughsie, 2024-07-21:
> which is 1.9.14 on stable gentoo

This is terribly old. I thought Gentoo was famous for getting the latest versions of everything?

smunaut, 2024-07-21:
1.9.20 is available but it's not the default, it has to be selected explicitely. I opened a bug on gentoo tracker to bump the stable/default version.