fwupd/fwupd #6873: update error: partition kind required

noraj, 2024-03-05:
**Describe the bug**

When updating the firmware I get `partition kind required`.

**Steps to Reproduce**

```
➜ sudo fwupdmgr update                                                                                         
Devices with no available firmware updates: 
 • DELL0B20:00 04F3:3146
 • Fingerprint Sensor
 • GL3224 SD reader [0x0749]
 • Internal SPI Controller
 • TPM
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • PC SN530 NVMe WDC 512GB
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.32.1 to 1.35.0?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Fixed the issue where the system stops responding at the BIOS              ║
║ Auto-Recovery screen when you update the BIOS.                               ║
║                                                                              ║
║ Latitude 3420 must remain plugged into a power source for the duration of    ║
║ the update to avoid damage.                                                  ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Decompressing…           [                                       ] Less than one minute remaining…
partition kind required
```

**Expected behavior**

Updating firmware without error.

**fwupd version information**

```shell
➜ fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.15
compile   org.freedesktop.Passim        0.1.5
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.Passim        0.1.5
runtime   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.7.8-arch1-1
runtime   org.freedesktop.fwupd         1.9.14
```

Installed from pacman ([ArchLinux](https://archlinux.org/packages/extra/x86_64/fwupd/)).

<details>

<summary>**fwupd device information**</summary>

```shell
➜ fwupdmgr get-devices --show-all-devices
Dell Inc. Latitude 3420
│
├─DELL0B20:00 04F3:3146:
│     Device ID:          38444342e1ea2c0d3626b6bb88dab4b8cb09ddc1
│     Summary:            Touchpad
│     Current version:    0x0005
│     Bootloader Version: 0x0001
[... 741 lines cut ...]
  Name:                 uefi_recovery
  Flags:                require-hwid
  Order:                2

09:21:28.199 FuEngine             resetting update motd timeout

Selected volume: /org/freedesktop/UDisks2/block_devices/nvme0n1p1
/boot/intel-ucode.img
/boot/vmlinuz-linux-lts
/boot/vmlinuz-linux
/boot/initramfs-linux-lts.img
/boot/initramfs-linux-lts-fallback.img
/boot/initramfs-linux.img
/boot/initramfs-linux-fallback.img
/boot/refind_linux.conf
/boot/fw/fwupd-f29927be-430b-4da0-8b80-b8de79353de2.cap
/boot/fwupdx64.efi
```

</p>
</details> 

**Related**

The only issue with the same error message I found is https://github.com/fwupd/fwupd/issues/6826, but I'm not sure it's the same cause.

noraj, 2024-03-06:
<details><summary>sudo fwupdtool get-devices --plugins uefi_capsule -vv</summary>
<p>

```
➜ sudo fwupdtool get-devices --plugins uefi_capsule -vv
12:40:12.218 FuDebug              verbose to debug (on console 1)
12:40:12.218 FuEngine             starting fwupd 1.9.14…
12:40:12.219 FuMain               locked /run/lock/fwupdtool
12:40:12.220 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…                 [ -                                     ]12:40:12.220 FuConfig             loading config /etc/fwupd/fwupd.conf
12:40:12.220 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
12:40:12.220 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
12:40:12.220 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
12:40:12.220 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x347
12:40:12.220 FuConfig             setting config [fwupd] DisabledPlugins=test;test_ble
12:40:12.220 FuConfig             default config, ignoring [fwupd] OnlyTrusted=true
12:40:12.220 FuConfig             default config, ignoring [fwupd] AllowEmulation=false
12:40:12.220 FuConfig             setting config [uefi_capsule] OverrideESPMountPoint=/efi
12:40:12.220 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
12:40:12.220 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
12:40:12.220 FuConfig             ::configuration loaded
12:40:12.220 FuRemoteList         loading /usr/share/fwupd/metainfo
12:40:12.220 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
12:40:12.220 XbSilo               file: 84e6c8d7-72bb-88af-9889-086c4e1c4479, current:84e6c8d7-72bb-88af-9889-086c4e1c4479, cached: (null)
12:40:12.220 XbSilo               loading silo with file contents
12:40:12.220 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
12:40:12.220 FuRemoteList         ::remote_list changed
12:40:12.220 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
12:40:12.220 FuRemoteList         ::remote_list changed
12:40:12.220 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
12:40:12.220 FuRemoteList         ::remote_list changed
12:40:12.220 FuEngine             priority local and vendor-directory is not download remote, so bumping
12:40:12.220 FuRemoteList         ordering lvfs-testing=lvfs+1
12:40:12.220 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…                 [                                       ]12:40:12.220 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:40:12.220 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
12:40:12.221 Jcat                 loaded 1 certificates
12:40:12.221 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
12:40:12.221 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
12:40:12.221 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:40:12.221 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
12:40:12.221 Jcat                 loaded 1 certificates
12:40:12.221 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
12:40:12.221 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
12:40:12.221 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]12:40:12.221 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
12:40:12.221 FuHistory            got schema version of 11
Loading…                 [*                                      ]12:40:12.224 FuPlugin             load(/usr/lib/fwupd-1.9.14/libfu_plugin_modem_manager.so)
12:40:12.225 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
12:40:12.225 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
12:40:12.225 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
12:40:12.225 XbSilo               file: 128fa0a4-9b9a-6264-ee9e-0661d2f9b12c, current:128fa0a4-9b9a-6264-ee9e-0661d2f9b12c, cached: (null)
12:40:12.225 XbSilo               loading silo with file contents
Loading…                 [*                                      ]12:40:12.226 FuConfig             loading config /etc/fwupd/fwupd.conf
12:40:12.226 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
12:40:12.226 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
12:40:12.226 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
12:40:12.226 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x347
12:40:12.226 FuConfig             setting config [fwupd] DisabledPlugins=test;test_ble
12:40:12.226 FuConfig             setting config [fwupd] OnlyTrusted=true
12:40:12.226 FuConfig             setting config [fwupd] AllowEmulation=false
12:40:12.226 FuConfig             setting config [uefi_capsule] OverrideESPMountPoint=/efi
12:40:12.226 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
[... cut ...]
12:40:12.708 FuPluginTest         destroy

12:40:12.708 FuProgress           raw timing data was { 0,001, 0,000, 0,000, 0,000 } -- steps were set as [ 1 1 3 95 ] but should have been [ 69 3 23 5 ] at ../fwupd-1.9.14/libfwupdplugin/fu-context.c:985
12:40:12.709 FuProgress           raw timing data was { 0,000, 0,000, 0,004, 0,000, 0,002, 0,001 } -- steps were set as [ 63 1 8 1 26 1 ] but should have been [ 0 1 53 4 30 12 ] at ../fwupd-1.9.14/plugins/uefi-capsule/fu-uefi-capsule-plugin.c:1100
```

</p>
</details>

hughsie, 2024-03-06:
Aha, so I think this might be fixed with the 1_9_x or main branches. Would it be possible to try one of those?

noraj, 2024-03-07:
Is there somewhere I can retrieve a dev build or pre-compiled rc or something? Because compiler the main branch seems tricky

[image]

superm1, 2024-03-07:
We have artifacts on the actions.  Here's a recent one: https://github.com/fwupd/fwupd/actions/runs/8188443836

noraj, 2024-03-07:
I can't test the binary like that since the new client is incompatible with the old demon.

[image]

I'm not sure I want to `sudo pacman -U fwupd-2.0.0-1-x86_64.pkg.tar` and risk messing my system with an `--overwrite`.

[image]

And https://github.com/fwupd/fwupd/blob/main/contrib/PKGBUILD is not very portable so I can build it.

noraj, 2024-03-07:
Yeah buts no.

```
sudo pacman -R fwupd fwupd-efi 
sudo pacman -U fwupd-2.0.0-1-x86_64.pkg.tar
```

Installing the old fwupd before or not has the same effect that upgrading directly.

Because this fwupd-2.0.0 conflicts not with the other fwupd package but with external packages like flashroom and libjcat.

```
➜ sudo pacman -Fy /usr/bin/flashrom          
:: Synchronizing package databases...
 core                                                                                              1073,3 KiB  3,03 MiB/s 00:00 [------------------------------------------------------------------------------] 100%
 extra                                                                                               39,3 MiB  5,43 MiB/s 00:07 [------------------------------------------------------------------------------] 100%
usr/bin/flashrom is owned by extra/flashrom 1.2-4

/tmp/toto 
➜ sudo pacman -Fy /usr/lib/libjcat.so                 
:: Synchronizing package databases...
 core is up to date
 extra is up to date
usr/lib/libjcat.so is owned by extra/libjcat 0.2.1-1
```

So I guess this works in a minimal environment with nothing installed but on a real system it conflicts a lot.

I could try to modify the AL PKGBUILD to use git instead to have a better build compatible with AL system and existing packages.

https://gitlab.archlinux.org/archlinux/packaging/packages/fwupd/-/blob/main/PKGBUILD?ref_type=heads

superm1, 2024-03-07:
I guess this is because we subproject those two things in our builds but real arch doesn't.  Could you help us fix our PKGBUILD to not do that?

noraj, 2024-03-07:
I came up with this experimental PKGBUILD but it wouldn't pass the test suite.

```pkgbuild
# Test PKGBUILD

pkgname=(fwupd fwupd-docs)
_pkgname=fwupd
pkgver=1.9.10.r504.g67a911b62
pkgrel=1
pkgdesc="Simple daemon to allow session software to update firmware"
arch=(x86_64)
url='https://github.com/fwupd/fwupd'
license=(LGPL-2.1-or-later)
depends=(
  bluez
  curl
  efivar
  flashrom
  fwupd-efi
  gcab
  gnutls
  hicolor-icon-theme
  libarchive
  libcbor
  libgudev
  libgusb
  libjcat
  json-glib
  libmm-glib
  libqmi
  libsmbios
  libxmlb
  passim
  polkit
  protobuf-c
  python
  shared-mime-info
  tpm2-tss
)
makedepends=(
  bash-completion
  gi-docgen
  gnu-efi-libs
  gobject-introspection
  meson
  noto-fonts
  noto-fonts-cjk
  pandoc
  python-cairo
  python-gobject
  python-pillow
  vala
  valgrind
  git
  python-dbusmock
  libdrm
  xz
)
checkdepends=(umockdev)
source=(
  "git+https://github.com/fwupd/fwupd.git"
  "https://gitlab.archlinux.org/archlinux/packaging/packages/fwupd/-/raw/main/fwupd.sysusers"
)
sha512sums=('SKIP'
            '637203080b55eda74a659f58c853a9a723a2dad5da70915b2b0e036c6145a649468ebec700cc83975d9cb5378b9dced8b3a3b26bdbcc75ddc774837355e75deb')
b2sums=('SKIP'
        'e65ca7da22a20a40882cfc1fe4479643f9a38c90a4f2c3e71e6e5e3de1d6db212a0f17d600097619fe3cdb0a9b860422f8b0b9a9d45441518e51a7eb12a918bb')

pkgver() {
    cd $_pkgname
[... 518 lines cut ...]
║ • Fixed the issue where the system stops responding at the BIOS              ║
║ Auto-Recovery screen when you update the BIOS.                               ║
║                                                                              ║
║ Latitude 3420 must remain plugged into a power source for the duration of    ║
║ the update to avoid damage.                                                  ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Waiting…                 [***************************************] Less than one minute remaining…
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Devices with the latest available firmware version:
 • PC SN530 NVMe WDC 512GB
 • UEFI dbx
Devices with no available firmware updates: 
 • DELL0B20:00 04F3:3146
 • 0000:00:1f.5
 • Fingerprint Sensor
 • GL3224 SD reader [0x0749]
 • TPM
 • UEFI Device Firmware
 • UEFI Device Firmware
An update requires a reboot to complete. Restart now? [y|N]: 
```

Seems it works.

superm1, 2024-03-07:
Great thanks!  Would you mind a PR to our PKGBUILD the deltas that make sense?

noraj, 2024-03-07:
I wouldn't mind if I was knowing what I'm doing. I just took AL PKGBUILD and hacked it. I switched from tarball release to git, removed pgp signature, added missing make dependencies by comparing with your PKGBUILD, removed `plugin_intel_spi` option because it was display as nonexistent while compiling, and a few arrangements. It's just a patch-up job. I'm not confident PRing as I don't understand the implication of the required changes. Also, I tried to glue up two PKGBUILD while one is made for 1.x.x and the other for 2.0.0, who knows what breaking changes I didn't take into consideration. Not even saying that I had to disable the tests that were failing. Just because I was able to compile it doesn't mean it's a good patch. And I have a hard time trying to understand what going on in [arch.sh](https://github.com/fwupd/fwupd/blob/main/contrib/ci/arch.sh) so that the CI would continue to build. I don't know anything about C compiling and fwupd seems complex.

All I could offer are some advices about PKGBUILD if you have some questions, and giving https://wiki.archlinux.org/title/Meson_package_guidelines as a pointer.

superm1, 2024-03-08:
Hmm, well I tried to add flashrom and libjcat so they don't subproject; but it fails CI.  Any ideas?
https://github.com/fwupd/fwupd/pull/6898

superm1, 2024-03-08:
I think the failure is related to the fact that arch makes you run the build as non-root.  Not really sure how to help it though.
```
FAILED: src/tests/auth/metadata.xml.gz.jcat 
/usr/sbin/jcat-tool --basename --appstream-id localhost self-sign src/tests/auth/metadata.xml.gz.jcat ../fwupd/src/tests/auth/metadata.xml.gz
Failed to create file “/root/.local/share/libjcat/pki/secret.key.IKBBK2”: Permission denied
```

superm1, 2024-03-22:
udisks is mandatory now, and partition detection happens on both main and 1_9_X branches.

closed 2024-03-22 by superm1