fwupd/fwupd #6879: redfish - no error report when firmware update fail occured

vincent-tang1, 2024-03-06:
**Precondition**
- Use ubuntu 23.10 
- Use snap to install 'fwupd' with parameter "--channel=stable"
- redfish.quirk apply as below
```
[REDFISH\VENDOR_Advantech&ID_BMC]
Flags = manager-reset,no-manager-reset-request
RedfishResetPreDelay = 100000
```

**Describe the bug**
Using redfish plugin to update BMC FW, and we try do firmware update fail testing. After failed, console can show error message, but furthermore check "get-history" command, it shows incorrect status (still be success state).

**Steps to Reproduce**
1. Get devices
```
├─FWA6172 Platform Management:
│     Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
│     Summary:            Redfish host bus adapter
│     Current version:    1.01.0000
│     Vendor:             Advantech (REDFISH:ADVANTECH)
│     GUIDs:              10598c6b-6f24-518d-9065-bb948baa4907 ← REDFISH\VENDOR_Advantech&SOFTWAREID_894feb65-2ca8-4378-a5d3-41c78ff5b59f-BMC
│                         69ec5e09-0466-5679-83f5-0dc75e3ff4d5 ← REDFISH\VENDOR_Advantech&ID_BMC
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
```
2. Upgrade image and maunally do firmware update fail, the console can show error message.
```
root@FWA-6172-efi:~# fwupdmgr update
Devices with no available firmware updates:
 • BIOS
 • BL
 • BMCONF
 • FPGA
 • NVRAM
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade FWA6172 Platform Management from 1.01.0000 to 1.03.0000?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This unstable release fixes the following issues:                            ║
║                                                                              ║
║ • Fix bug for test 1                                                         ║
║ • Improve performance for test 1                                             ║
║                                                                              ║
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ • Add support for test 1                                                     ║
║                                                                              ║
║ FWA6172 Platform Management and all connected devices may not be usable      ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Waiting…                 [************************************** ] Less than one minute remaining…
failed to parse https://10.234.147.1:443/redfish/v1/TaskService/Tasks/0: The URI /redfish/v1/TaskService/Tasks/0 was not found.
```

3. but using Get history command, no seen any error report, and uploaded result still be 'Success' status (not fail)
```
root@FWA-6172-efi:~# fwupdmgr get-history
Advantech FWA-6172
│
└─FWA6172 Platform Management:
  │   Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
  │   Previous version:   1.01.0000
  │   Update State:       Success
  │   Last modified:      2024-03-06 10:14
  │   GUID:               10598c6b-6f24-518d-9065-bb948baa4907
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │
  └─FWA6172 Platform Management BMC Update:
        New version:      1.03.0000
        Remote ID:        lvfs
        Release ID:       82544
        Summary:          Firmware for the Advantech FWA6172 BMC
        Variant:          sku:10598c6b
        License:          Proprietary
        Size:             14.7 MB
        Created:          2024-01-26
        Urgency:          High
        Vendor:           Advantech
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:
        This unstable release fixes the following issues:

        • Fix bug for test 1
        • Improve performance for test 1

        Some new functionality has also been added:

        • Add support for test 1
        Issues:           CVE-2022-36392
                          CVE-2022-29871
        Checksum:         16908423072412b16252f52ae313d34697452940583ed887936f9cc8408dfeba
```

**Expected behavior**
Should have fail status/message from "get-history" command if firmware update fail

**Root cause**
I think i might should know why to lead tihs problem. 
for above case, when fw update fail, fu_redfish_legacy_device_write_firmware() will return FALSE and then carry the error message "The URI /redfish/v1/TaskService/Tasks/0 was not found"
```
static gboolean
fu_redfish_legacy_device_write_firmware(FuDevice *device,
                    FuFirmware *firmware,
                    FuProgress *progress,
                    FwupdInstallFlags flags,
                    GError **error)
{
...
    return fu_redfish_device_poll_task(FU_REDFISH_DEVICE(self), location, progress, error);
}
```
And then it will invoke fu_redfish_plugin_cleanup(). But because it clean up all devices info incluging error status, so lead to this issue. 
```
static gboolean
fu_redfish_plugin_cleanup(FuPlugin *plugin,
              FuDevice *device,
              FuProgress *progress,
              FwupdInstallFlags flags,
              GError **error)
{
    FuRedfishPlugin *self = FU_REDFISH_PLUGIN(plugin);
    guint64 reset_timeout = 0;
    g_autofree gchar *restart_timeout_str = NULL;
    g_autoptr(FuRedfishRequest) request = fu_redfish_backend_request_new(self->backend);
    g_autoptr(GPtrArray) devices = NULL;

    /* nothing to do */
    if (!fu_device_has_private_flag(device, FU_REDFISH_DEVICE_FLAG_MANAGER_RESET))
        return TRUE;

    /* progress */
    fu_progress_set_id(progress, G_STRLOC);
    fu_progress_add_step(progress, FWUPD_STATUS_DEVICE_BUSY, 1, "manager-reboot");
    fu_progress_add_step(progress, FWUPD_STATUS_DEVICE_RESTART, 5, "pre-delay");
    fu_progress_add_step(progress, FWUPD_STATUS_DEVICE_RESTART, 67, "poll-manager");
    fu_progress_add_step(progress, FWUPD_STATUS_DEVICE_RESTART, 18, "post-delay");
    fu_progress_add_step(progress, FWUPD_STATUS_DEVICE_BUSY, 9, "recoldplug");
...
    /* remove all the devices */
    devices = fu_backend_get_devices(FU_BACKEND(self->backend));
    for (guint i = 0; i < devices->len; i++) {
        FuDevice *device_tmp = g_ptr_array_index(devices, i);
        fu_backend_device_removed(FU_BACKEND(self->backend), device_tmp);
    }
...
```

**Potential solution:**
When error is found during firmware update. Later, calling fu_redfish_plugin_cleanup() should nothing to do that aviod device error staus disappear. (Actually, we don't have to wait BMC auto reboot or execute BMC reboot if firmware update failed, right ?)

 **Summary**
**Not sure if you agree the root cause and possible solution. and then could you help fix it ? Thanks** 

**fwupd version information**
```
fwupdmgr --version

compile   com.hughsie.libxmlb           0.3.6
compile   io.snapcraft.fwupd            5749
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.9.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.5.0-10-generic
runtime   org.freedesktop.fwupd         1.9.14
```

hughsie, 2024-03-06:
So, maybe -- the update actually succeeded -- but the polling is broken and should ignore when the URI isn't (yet?) available? If so, does the hughsie/redfish-poll-task branch work better?

vincent-tang1, 2024-03-06:
Sorry, my words might caused you misunderstanding and took your efforts in mismatch solution.
It is really update fail as my expectation.
To manually generate a fw update fail and validate it, I triggered a RESTful API command as **"curl DELETE  URI /redfish/v1/Tasks/0"**(it will abort update operation in BMC internal, and really kill this resource) during fwupd fw update. but from my expection, it should have error report, but in fact is NOT.

Not only this error in comment 1, should say that for all error type message, fwupd will not show error report from get-history command.
Following concept patch , not sure if it is ok for you ?
```
static gboolean
fu_redfish_plugin_cleanup(FuPlugin *plugin,
              FuDevice *device,
              FuProgress *progress,
              FwupdInstallFlags flags,
              GError **error)
{
    FuRedfishPlugin *self = FU_REDFISH_PLUGIN(plugin);
    guint64 reset_timeout = 0;
    g_autofree gchar *restart_timeout_str = NULL;
    g_autoptr(FuRedfishRequest) request = fu_redfish_backend_request_new(self->backend);
    g_autoptr(GPtrArray) devices = NULL;

+  /* check if fu_redfish_legacy_device_write_firmware is success, if NOT, should nothing to do and return directly.
+   * otherwise it will call subsequent code about "/* remove all the devices */" to cause this problem
+   * In the other hand, if the write update is fail, BMC will not auto reboot or BMC do not need to execute reboot,
+   * so we can break it in the beginning of this function. 
+   */
+  if (is_write_firmware_success == FALSE)   // can use the "GError **error" variable check if having any error ?
+        return TRUE;

    /* nothing to do */
    if (!fu_device_has_private_flag(device, FU_REDFISH_DEVICE_FLAG_MANAGER_RESET))
        return TRUE;
```

In fact, for POC, i use temporarily use below hard-code patch for testing. Final i can see error report as my expectation
```
static gboolean
fu_redfish_plugin_cleanup(FuPlugin *plugin,
              FuDevice *device,
              FuProgress *progress,
              FwupdInstallFlags flags,
              GError **error)
{
    FuRedfishPlugin *self = FU_REDFISH_PLUGIN(plugin);
    guint64 reset_timeout = 0;
    g_autofree gchar *restart_timeout_str = NULL;
    g_autoptr(FuRedfishRequest) request = fu_redfish_backend_request_new(self->backend);
    g_autoptr(GPtrArray) devices = NULL;

+ return;
```
and get-history command content is my expectation ( shows "Update State: Failed", and "Update Error: ..." )
```
Advantech FWA-6172
│
└─FWA6172 Platform Management:
  │   Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
  │   Previous version:   1.01.0000
  │   Update State:       Failed
  │   Update Error:       failed to parse https://10.234.147.1:443/redfish/v1/TaskService/Tasks/0: The URI /redfish/v1/TaskService/Tasks/0 was not found.
  │   Last modified:      2024-03-06 15:26
  │   GUID:               10598c6b-6f24-518d-9065-bb948baa4907
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │
  └─FWA6172 Platform Management BMC Update:
        New version:      1.03.0000
        Remote ID:        lvfs
        Release ID:       82544
        Summary:          Firmware for the Advantech FWA6172 BMC
        Variant:          sku:10598c6b
        License:          Proprietary
        Size:             14.7 MB
        Created:          2024-01-26
        Urgency:          High
        Vendor:           Advantech
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:
        This unstable release fixes the following issues:

        • Fix bug for test 1
        • Improve performance for test 1

        Some new functionality has also been added:

        • Add support for test 1
        Issues:           CVE-2022-36392
                          CVE-2022-29871
        Checksum:         16908423072412b16252f52ae313d34697452940583ed887936f9cc8408dfeba

```

Will appreciate any feedback and discussion ^^

vincent-tang1, 2024-03-07:
> So, maybe -- the update actually succeeded 
1. No, this update is really failed. To do update fail testing, i manually performed a operation "DELETE /redfish/va/Task/0" during the update such that update precedure abort in BMC.

> -- but the polling is broken and should ignore when the URI isn't (yet?) available?
2. No, this is not polling problem, and we also should NOT ignore that URI isn't available due to above (1.) comment  

> If so, does the hughsie/redfish-poll-task branch work better?
3. After verified this patch, as result is not my expectation, my expectation is when any update error occur, fwupd utility should return right away. and can report error in "get-history" command.

Here, I have a opinion, not sure if it is appropriate as below concept patch for you reference ?
```
static gboolean
fu_redfish_plugin_cleanup(FuPlugin *plugin,
              FuDevice *device,
              FuProgress *progress,
              FwupdInstallFlags flags,
              GError **error)
{
    FuRedfishPlugin *self = FU_REDFISH_PLUGIN(plugin);
    guint64 reset_timeout = 0;
    g_autofree gchar *restart_timeout_str = NULL;
    g_autoptr(FuRedfishRequest) request = fu_redfish_backend_request_new(self->backend);
    g_autoptr(GPtrArray) devices = NULL;

+  /* check if fu_redfish_legacy_device_write_firmware is success, if NOT, should nothing to do and return right away.
+   * otherwise it will call subsequent code about "/* remove all the devices */" to cause this problem
+   * In the other hand, if the write update is fail, BMC will not auto reboot or BMC do not need to be executed reboot operation in fwupd.
+   * so we can break it in the beginning of this function if this condition occured
+   */
+  if (found_update_fail() )   // somehow check if fu_redfish_legacy_device_write_firmware() update is fail
+        return FALSE;

    /* nothing to do */
    if (!fu_device_has_private_flag(device, FU_REDFISH_DEVICE_FLAG_MANAGER_RESET))
        return TRUE;
```

vincent-tang1, 2024-03-07:
Here is my POC, i use below dirty hard-code patch for testing. Final i can see error report as my expectation
```
static gboolean
fu_redfish_plugin_cleanup(FuPlugin *plugin,
              FuDevice *device,
              FuProgress *progress,
              FwupdInstallFlags flags,
              GError **error)
{
    FuRedfishPlugin *self = FU_REDFISH_PLUGIN(plugin);
    guint64 reset_timeout = 0;
    g_autofree gchar *restart_timeout_str = NULL;
    g_autoptr(FuRedfishRequest) request = fu_redfish_backend_request_new(self->backend);
    g_autoptr(GPtrArray) devices = NULL;

+ return;
```
and get-history command content can appears error report ( "Update State: Failed", and "Update Error: ..." )
```
Advantech FWA-6172
│
└─FWA6172 Platform Management:
  │   Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
  │   Previous version:   1.01.0000
  │   Update State:       Failed
  │   Update Error:       failed to parse https://10.234.147.1:443/redfish/v1/TaskService/Tasks/0: The URI /redfish/v1/TaskService/Tasks/0 was not found.
  │   Last modified:      2024-03-06 15:26
  │   GUID:               10598c6b-6f24-518d-9065-bb948baa4907
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │
  └─FWA6172 Platform Management BMC Update:
        New version:      1.03.0000
        Remote ID:        lvfs
        Release ID:       82544
        Summary:          Firmware for the Advantech FWA6172 BMC
        Variant:          sku:10598c6b
        License:          Proprietary
        Size:             14.7 MB
        Created:          2024-01-26
        Urgency:          High
        Vendor:           Advantech
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:
        This unstable release fixes the following issues:

        • Fix bug for test 1
        • Improve performance for test 1

        Some new functionality has also been added:

        • Add support for test 1
        Issues:           CVE-2022-36392
                          CVE-2022-29871
        Checksum:         16908423072412b16252f52ae313d34697452940583ed887936f9cc8408dfeba

```

vincent-tang1, 2024-03-07:
Another examle for fw update fail testing, somehow i trigger manually it during the fwupd update
```
root@FWA-6172-efi:~/tmp/hughsie-redfish-poll-task/fwupd# fwupdmgr update
Devices with no available firmware updates:
 • BIOS
 • BL
 • BMCONF
 • FPGA
 • NVRAM
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade FWA6172 Platform Management from 0.77.0000 to 1.03.0000?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This unstable release fixes the following issues:                            ║
║                                                                              ║
║ • Fix bug for test 1                                                         ║
║ • Improve performance for test 1                                             ║
║                                                                              ║
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ • Add support for test 1                                                     ║
║                                                                              ║
║ FWA6172 Platform Management and all connected devices may not be usable      ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Waiting…                 [************************************** ] Less than one minute remaining…
Transfer of image 'firmware.bin' to 'BMC' failed.
```

As you can see, it return the error message and come from Exception state to be detected, as below as noted code
```
static gboolean
fu_redfish_device_poll_task_once(FuRedfishDevice *self, FuRedfishDevicePollCtx *ctx, GError **error)
{
    FuRedfishDevicePrivate *priv = GET_PRIVATE(self);
    JsonObject *json_obj;
    const gchar *message = "Unknown failure";
    const gchar *state_tmp;
    g_autoptr(FuRedfishRequest) request = fu_redfish_backend_request_new(priv->backend);
    ...
    if (g_strcmp0(state_tmp, "Exception") == 0 ||
        g_strcmp0(state_tmp, "UserIntervention") == 0) {
        g_set_error_literal(error, FWUPD_ERROR, ctx->error_code, message);
        return FALSE;
    }
   ..
```

However, the issue as, "get-history" command have not seen error report
```
root@FWA-6172-efi:~/tmp/hughsie-redfish-poll-task/fwupd# fwupdmgr get-history
Advantech FWA-6172
│
└─FWA6172 Platform Management:
  │   Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
  │   Previous version:   0.77.0000
  │   Last modified:      2024-03-07 05:50
  │   GUID:               10598c6b-6f24-518d-9065-bb948baa4907
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │
  └─FWA6172 Platform Management BMC Update:
        New version:      1.03.0000
        Remote ID:        lvfs
        Release ID:       82544
        Summary:          Firmware for the Advantech FWA6172 BMC
        Variant:          sku:10598c6b
        License:          Proprietary
        Size:             14.7 MB
        Created:          2024-01-26
        Urgency:          High
        Vendor:           Advantech
        Release Flags:    • Trusted metadata
                          • Is upgrade
       ....
```

vincent-tang1, 2024-03-07:
Here is my POC, i use below dirty hard-code patch for testing. Final i can see error report as my expectation
```
static gboolean
fu_redfish_plugin_cleanup(FuPlugin *plugin,
              FuDevice *device,
              FuProgress *progress,
              FwupdInstallFlags flags,
              GError **error)
{
    FuRedfishPlugin *self = FU_REDFISH_PLUGIN(plugin);
    guint64 reset_timeout = 0;
    g_autofree gchar *restart_timeout_str = NULL;
    g_autoptr(FuRedfishRequest) request = fu_redfish_backend_request_new(self->backend);
    g_autoptr(GPtrArray) devices = NULL;

+ return;
```
and get-history command content can appears error report ( "Update State: Failed", and "Update Error: ..." )
```
Advantech FWA-6172
│
└─FWA6172 Platform Management:
  │   Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
  │   Previous version:   1.01.0000
  │   Update State:       Failed
  │   Update Error:       failed to parse https://10.234.147.1:443/redfish/v1/TaskService/Tasks/0: The URI /redfish/v1/TaskService/Tasks/0 was not found.
  │   Last modified:      2024-03-06 15:26
  │   GUID:               10598c6b-6f24-518d-9065-bb948baa4907
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │
  └─FWA6172 Platform Management BMC Update:
        New version:      1.03.0000
        Remote ID:        lvfs
        Release ID:       82544
        Summary:          Firmware for the Advantech FWA6172 BMC
        Variant:          sku:10598c6b
        License:          Proprietary
        Size:             14.7 MB
        Created:          2024-01-26
        Urgency:          High
        Vendor:           Advantech
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:
        This unstable release fixes the following issues:
        ...
```

vincent-tang1, 2024-03-07:
Hello,

> @hughsie
[Keep polling in fu_redfish_device_poll_task() even for task errors](https://github.com/fwupd/fwupd/commit/67a911b62b71cabce72e3e68fe2c6de423d0d127) …

sorry, i am afriad that this patch is not a solution.

hughsie, 2024-03-07:
> sorry, i am afriad that this patch is not a solution.

Ahh yes, sorry -- I didn't mean to auto-close this. It's a valid problem, but it doesn't actually fix *this* problem. I'll look at this later, unless you beat me to it.

vincent-tang1, 2024-03-07:
It is ok, ^^

> [Keep polling in fu_redfish_device_poll_task() even for task errors](https://github.com/fwupd/fwupd/commit/67a911b62b71cabce72e3e68fe2c6de423d0d127) …

Sorry, could you first revert this commit from master because it might lead to some problem, you can refer to following scenario.

Considering a use case, when a user is updating fw via fwupd and then BMC will create a 
URI /redfish/v1/Tasks/0 to monitor update progress. At this monent, suddenly user do not want to update it, so he use DELETE /redfish/v1/Tasks/0 to abort this update procedure. But because this commit, user need wait for 1 hours to leave in fwupd utility. Therfore, i think, for normal behavior, for this case, fwupd utility should can return "this task can not found" error message right away.

vincent-tang1, 2024-03-10:
> It's a valid problem, but it doesn't actually fix this problem. I'll look at this later

Hello, not sure if you need other information or clarify something from above comments ? ^^

vincent-tang1, 2024-03-11:
> It is ok, ^^, and thanks your help
> 
> > [Keep polling in fu_redfish_device_poll_task() even for task errors](https://github.com/fwupd/fwupd/commit/67a911b62b71cabce72e3e68fe2c6de423d0d127) …
> 
> Sorry, could you first revert this commit [[67a911b](https://github.com/fwupd/fwupd/commit/67a911b62b71cabce72e3e68fe2c6de423d0d127)] from main trunk because it might lead to some problem ? You can refer to following scenario:
> 
> Considering a use case, when a user is updating fw via fwupd and then BMC will create a URI /redfish/v1/Tasks/0 to monitor update progress. At this monent, suddenly user do not want to update it, so he use DELETE /redfish/v1/Tasks/0 to abort this update procedure. But because this commit, user need wait for 1 hours to leave in fwupd utility. Therfore, i think, for normal behavior, for this case, fwupd utility should can return "this task can not found" error message right away.

About this, not sure if you have other opinion ? ^^

vincent-tang1, 2024-03-15:
it is ok :), not sure if you have some update ?

vincent-tang1, 2024-03-19:
I think that i might have a potential solution as below
```
diff --git a/plugins/redfish/fu-redfish-device.c b/plugins/redfish/fu-redfish-device.c
index 68abcc0c8..be093ce42 100644
--- a/plugins/redfish/fu-redfish-device.c
+++ b/plugins/redfish/fu-redfish-device.c
@@ -696,6 +696,7 @@ fu_redfish_device_poll_task_once(FuRedfishDevice *self, FuRedfishDevicePollCtx *
                                    FWUPD_ERROR,
                                    FWUPD_ERROR_INVALID_FILE,
                                    "no TaskState for task manager");
+               fu_device_remove_private_flag(FU_DEVICE(self), FU_REDFISH_DEVICE_FLAG_MANAGER_RESET);
                return FALSE;
        }
        state_tmp = json_object_get_string_member(json_obj, "TaskState");
@@ -706,11 +707,13 @@ fu_redfish_device_poll_task_once(FuRedfishDevice *self, FuRedfishDevicePollCtx *
        }
        if (g_strcmp0(state_tmp, "Cancelled") == 0) {
                g_set_error_literal(error, FWUPD_ERROR, FWUPD_ERROR_INTERNAL, "Task was cancelled");
+               fu_device_remove_private_flag(FU_DEVICE(self), FU_REDFISH_DEVICE_FLAG_MANAGER_RESET);
                return FALSE;
        }
        if (g_strcmp0(state_tmp, "Exception") == 0 ||
            g_strcmp0(state_tmp, "UserIntervention") == 0) {
                g_set_error_literal(error, FWUPD_ERROR, ctx->error_code, message);
+               fu_device_remove_private_flag(FU_DEVICE(self), FU_REDFISH_DEVICE_FLAG_MANAGER_RESET);
                return FALSE;
        }
```

And after verified, it can work to show error report as below

**Step 1 : I manually trigger a fail during update fw**
```
(fwupd) root@FWA-6172-efi:~#fwupdtool update
...
WARNING: This package has not been validated, it may not work properly.
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade FWA6172 Platform Management from 0.77.0000 to 1.03.0000?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This unstable release fixes the following issues:                            ║
║                                                                              ║
║ • Fix bug for test 1                                                         ║
║ • Improve performance for test 1                                             ║
║                                                                              ║
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ • Add support for test 1                                                     ║
║                                                                              ║
║ FWA6172 Platform Management and all connected devices may not be usable      ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Loading…                 [ -                                     ]
WARNING: This package has not been validated, it may not work properly.
Writing…                 [*************************************  ]
Transfer of image 'firmware.bin' to 'BMC' failed.
``` 

**Step 2: check error report, it can correctly show error state and message (see Update State, Update Error)** 
```
fwupd) root@FWA-6172-efi:~# fwupdtool get-history
...
WARNING: This package has not been validated, it may not work properly.
Advantech FWA-6172
│
└─FWA6172 Platform Management:
  │   Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
  │   Previous version:   0.77.0000
  │   Update State:       Failed
  │   Update Error:       Transfer of image 'firmware.bin' to 'BMC' failed.
  │   Last modified:      2024-03-19 10:52
  │   GUID:               10598c6b-6f24-518d-9065-bb948baa4907
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │
  └─FWA6172 Platform Management BMC Update:
        New version:      1.03.0000
        Remote ID:        lvfs-testing
        Release ID:       82544
        Summary:          Firmware for the Advantech FWA6172 BMC
        Variant:          sku:10598c6b
        License:          Proprietary
        ...
```

From above, the 'fwupdtool' is used to verify and result is workable.
But i always can not execute **"fwupdmgr"** using **compling version**, seems like it can not be initialied normally, 
so i am not sure if it is workable in **'fwupdmgr'**.

Do you know whether in each time **"fwupdmgr update"** command,  FU_REDFISH_DEVICE_FLAG_MANAGER_RESET flag status can be recovery into redfish.quirk define ?

Could you review if this is an appropriate solution ? if yes, you can modify into contents you would expect

Thanks!

vincent-tang1, 2024-03-20:
@hughsie
could you please help to take a look latest comment, thanks!!

vincent-tang1, 2024-03-21:
Because this ticket already be mark as fixed, I am a little worry that it might have no any message update notification.
Thereforce, I create another smilar ticket to trace and re-organize key information into it.

vincent-tang1, 2024-03-22:
> [Keep polling in fu_redfish_device_poll_task() even for task errors](https://github.com/fwupd/fwupd/commit/67a911b62b71cabce72e3e68fe2c6de423d0d127) 

About this commit, considering a scenario, when a user is updating fw via fwupd and then BMC will create a URI /redfish/v1/Tasks/0 to monitor update progress. At this monent, suddenly user do not want to update it, so he use DELETE /redfish/v1/Tasks/0 to abort this update procedure. But because this commit, user need wait for 1 hours to leave in fwupd utility. Therfore, i think, for normal behavior, for this case, fwupd utility should can return "this task can not found" error message right away.

In the other hand, this commit also found other problem #6954, i suggested that we might require first revert it.

hughsie, 2024-03-22:
@vincent-tang1 I've reverted 67a911b62b71cabce72e3e68fe2c6de423d0d127

closed 2024-03-22 by superm1