fwupd/fwupd #6884: fwupd-refresh.timer causing HDD to wakeup

axet, 2024-03-07:
**Describe the bug**

Periodic starting of /lib/systemd/system/fwupd-refresh.timer causing HDD to wake up from IDLE state.

**Steps to Reproduce**

Put your HDD into sleep, wait until fwupd-refresh.timer to tigger (or manually start it).

**Expected behavior**

fwupd should not read smart attributes from devices in IDLE state. Please check drive state using `hdparm -C /dev/sda`

**fwupd version information**

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.1.0-18-amd64
runtime   org.freedesktop.fwupd         1.8.12

```

```
apt install fwupd
```

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
Gigabyte Technology Co., Ltd. H610M H DDR4
│
├─Unknown Device:
│     Device ID:          0ca7d9ea83b2d18d8523bb2eed1753317fe1a969
│     GUID:               06d8ee08-fa01-5ed7-be1c-b77936b6f43a ← GPIO\ID_INTC1056:00
│   
├─12th Gen Intel Core™ i3-12100F:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0000002c
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         615995d1-bdd0-5048-a291-83eaef0de820 ← CPUID\PRO_0&FAM_06&MOD_97
│                         dbcb113c-bc96-576d-8891-49dea0d60b12 ← CPUID\PRO_0&FAM_06&MOD_97&STP_5
│     Device Flags:       • Internal device
│   
├─HFM512GD3JX013N:
│     Device ID:          0fba4a8bedba53076fc9b147655e59455bc25a87
│     Summary:            NVM Express solid state drive
│     Current version:    41000C20
│     Vendor:             SK hynix (NVME:0x1C5C)
│     Serial Number:      FYA9N059910509U18
│     GUIDs:              0ee57466-ca9d-5f10-a005-ec6bacf6fb82 ← NVME\VEN_1C5C&DEV_174A
│                         4406b8dc-20c5-5f3a-bc9c-8afdcee48684 ← NVME\VEN_1C5C&DEV_174A&REV_00
│                         34e06406-f620-5b9d-a9cd-7f3a4e648633 ← NVME\VEN_1C5C&DEV_174A&SUBSYS_1C5C174A
│                         34f9d6fa-e1d6-5d7d-909e-0406644298c5 ← NVME\VEN_1C5C&DEV_174A&SUBSYS_1C5C174A&REV_00
│                         68a7e984-1f8b-54a1-8025-ef9c3ba4cec0 ← HFM512GD3JX013N
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
[... 97 lines cut ...]
│     Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
│     Summary:            GIGABYTE
│     GUID:               3184bd50-b815-5656-a3dc-6b591607879d ← UEFI\CRT_897FB5900210D26D610F43FC9C3174791A800621
│   
└─UEFI dbx:
      Device ID:          362301da643102b9f38477387e2193e57abaa590
      Summary:            UEFI revocation database
      Current version:    371
      Minimum Version:    371
      Vendor:             UEFI:Linux Foundation
      Install Duration:   1 second
      GUIDs:              c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
                          f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
      Device Flags:       • Internal device
                          • Updatable
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Only version upgrades are allowed
                          • Signed Payload
    
```

</details>

- Operating system and version: Debian 12 Bookworm

hughsie, 2024-03-07:
> fwupd should not read smart attributes from devices in IDLE state

Does it? The refresh timer might cause fwupd to read a file or worst case download and save a file, but we shouldn't be touching the SMART stuff at all.

superm1, 2024-03-07:
Maybe it's a case of the daemon getting dbus activated and that happens in the probe sequence?

axet, 2024-03-07:
When I tried to debug it, I see 'fatrace' read actions on sda mount points. But it is hard to catch, since fwupd only doing it once per 12h.

After disabling fwupd update timers my HDD no longer wakes during the day.

superm1, 2024-03-07:
Can you see if in the journal that the daemon was started at same time too?

In your case a workaround would be to disable the plugin your drive uses on the fwupd.conf file.

axet, 2024-03-07:
It is disabled for a week. HDD stays idle. Just before report this issue I ran `systemctl start fwupd-refresh.timer` and HDD wakes. 99% it is related to fwupd.

Lets wait until tomorrow, since it only happens once a day.

superm1, 2024-03-07:
We have a " feature " in fwupd that we will stop an idle daemon after some time to avoid consuming resources for nothing.

These are the kinds of things that make an argument not to do that.

As your drive doesn't have any updates on the LVFS I think for your situation it's better to just disable the ata plugin in fwupd.conf.

axet, 2024-03-08:
Today around 20:51 my HDD wakes up. Logs showing:

```
2024-03-08T20:50:50.057753+03:00 axet-desktop NetworkManager[932]: <info>  [1709920250.0570] dhcp4 (enp5s0): state changed new lease, address=10.10.5.80
2024-03-08T20:50:55.010977+03:00 axet-desktop udisksd[914]: Mounted /dev/sda3 at /media/root/EFI on behalf of uid 0
2024-03-08T20:50:55.496702+03:00 axet-desktop systemd[1]: media-root-EFI.mount: Deactivated successfully.
2024-03-08T20:50:55.496991+03:00 axet-desktop udisksd[914]: Cleaning up mount point /media/root/EFI (device 8:3 is not mounted)
2024-03-08T20:50:55.498639+03:00 axet-desktop udisksd[914]: Unmounted /dev/sda3 on behalf of uid 0
2024-03-08T20:50:55.648066+03:00 axet-desktop fwupd[1808526]: 17:50:55.647 FuPluginUefiCapsule  skipping device that failed coldplug: ESRT GUID '00000000-0000
-0000-0000-000000000000' was not valid
2024-03-08T20:50:56.183075+03:00 axet-desktop dbus-daemon[885]: [system] Successfully activated service 'org.freedesktop.fwup
d'
2024-03-08T20:50:56.183140+03:00 axet-desktop systemd[1]: Started fwupd.service - Firmware update daemon.
2024-03-08T20:50:56.196573+03:00 axet-desktop fwupd[1808526]: 17:50:56.196 FuPluginLinuxSwap    could not parse /proc/swaps: no volumes for devnum 26
2024-03-08T20:50:56.202338+03:00 axet-desktop fwupdmgr[1808519]: Updating lvfs
2024-03-08T20:50:57.935739+03:00 axet-desktop fwupdmgr[1808519]: Downloading…             [ -                                     ]Downloading…             [ -                                     ]Downloading…             [  -                                    ]Downloading…             [   -                                   ]Downloading…             [    \                                  ]Downloading…             [     \                                 ]Downloading…             [      \                                ]Downloading…             [       \                               ]Downloading…             [        |                              ]Downloading…             [         |                             ]Downloading…             [          |                            ]Downloading…             [           |                           ]Downloading…             [            /                          ]Downloading…             [ -                                     ]Downloading…             [ -                                     ]Downloading…             [*                                      ]Downloading…             [**                                     ]Downloading…             [***                                    ]Downloading…             [*******                                ]Downloading…             [************                           ]Downloading…             [*****************                      ]Downloading…             [*****************************          ]Downloading…             [*****************************          ]
2024-03-08T20:50:57.936171+03:00 axet-desktop fwupdmgr[1808519]: Successfully downloaded new metadata: 0 local devices supported
2024-03-08T20:50:57.937641+03:00 axet-desktop systemd[1]: fwupd-refresh.service: Deactivated successfully.
2024-03-08T20:50:57.937670+03:00 axet-desktop systemd[1]: Finished fwupd-refresh.service - Refresh fwupd metadata and update motd.
2024-03-08T20:51:00.133024+03:00 axet-desktop systemd[1]: Started session-4989.scope - Session 4989 of User axet.

```

superm1, 2024-03-11:
I don't know it is correct for fwupd 1.8.x. you might need to test newer fwupd base.

axet, 2024-03-11:
new logs including fatrace:

```
(fwupdmgr)(377129): CO  /media/axet/1TB
(fwupdmgr)(377129): CO  /boot/efi
(fwupd)(377134): CO  /boot
(fwupd)(377134): CO  /boot
(fwupd)(377134): CO  /boot
(fwupd)(377134): O   /boot
(fwupd)(377134): CO  /boot/efi
(fwupd)(377134): CO  /boot
(fwupd)(377134): CO  /boot/grub
(fwupd)(377134): CO  /boot
(fwupd)(377134): CO  /boot/grub
(fwupd)(377134): O   /boot
(fwupd)(377134): C   /boot
udisksd(924): +   /media/root
udisksd(924): D   /media/root
```

and more fwupd service

```
2024-03-11T17:03:52.552876+03:00 axet-desktop systemd[1]: Starting fwupd.service - Firmware update daemon...
2024-03-11T17:03:52.950180+03:00 axet-desktop dbus-daemon[888]: [system] Activating via systemd: service name='org.bluez' unit='dbus-org.bluez.service' requested by ':1.114' (uid=0 pid=377134 comm="/usr/libexec/
fwupd/fwupd")
2024-03-11T17:03:52.952174+03:00 axet-desktop systemd[1]: bluetooth.service - Bluetooth service was skipped because of an unmet condition check (ConditionPathIsDirectory=/sys/class/bluetooth).
2024-03-11T17:03:54.502088+03:00 axet-desktop fwupd[377134]: 14:03:54.501 FuPluginUefiCapsule  skipping device that failed coldplug: ESRT GUID '00000000-0000-0000-0000-000000000000' was not valid
2024-03-11T17:04:00.492119+03:00 axet-desktop udisksd[924]: Mounted /dev/sda3 at /media/root/EFI on behalf of uid 0
2024-03-11T17:04:00.975793+03:00 axet-desktop systemd[1]: media-root-EFI.mount: Deactivated successfully.
2024-03-11T17:04:00.976870+03:00 axet-desktop udisksd[924]: Cleaning up mount point /media/root/EFI (device 8:3 is not mounted)
2024-03-11T17:04:00.979162+03:00 axet-desktop udisksd[924]: Unmounted /dev/sda3 on behalf of uid 0
2024-03-11T17:04:01.641931+03:00 axet-desktop dbus-daemon[888]: [system] Successfully activated service 'org.freedesktop.fwupd'
2024-03-11T17:04:01.642002+03:00 axet-desktop systemd[1]: Started fwupd.service - Firmware update daemon.
2024-03-11T17:04:01.656010+03:00 axet-desktop fwupd[377134]: 14:04:01.655 FuPluginLinuxSwap    could not parse /proc/swaps: no volumes for devnum 26
2024-03-11T17:04:01.662597+03:00 axet-desktop systemd[1]: fwupd-refresh.service: Deactivated successfully.
2024-03-11T17:04:01.662680+03:00 axet-desktop systemd[1]: Finished fwupd-refresh.service - Refresh fwupd metadata and update motd.
```

logging standby:

```
while true; do A=$(sudo hdparm -C /dev/sda | grep state); [ "$OLD" == "$A" ] || echo $(date) - $A; OLD=$A; sleep 1; done
Вс 10 мар 2024 21:29:58 MSK - drive state is: active/idle
Вс 10 мар 2024 21:39:39 MSK - drive state is: standby
Пн 11 мар 2024 07:18:37 MSK - drive state is: active/idle
Пн 11 мар 2024 07:34:39 MSK - drive state is: standby
Пн 11 мар 2024 17:04:00 MSK - drive state is: active/idle
Пн 11 мар 2024 17:09:39 MSK - drive state is: standby
```

superm1, 2024-03-11:
Can you please test 1_9_X branch? You can also put the patch on top of the version in Debian testing.

axet, 2024-03-12:
This is not how bugs should be addressed.

But ok. Here is a 1.9.14 output + patch, unfortunately, drive was active during this fatrace logging. Need to wait one more day for another try. But you can see already it touching the drive.

```
08:31:49.292663 (fwupdmgr)(188378): O   /media/axet/1TB
08:31:49.292663 (fwupdmgr)(188378): C   /media/axet/1TB
08:31:49.292751 (fwupdmgr)(188378): CO  /boot/efi
```

hughsie, 2024-03-12:
> This is not how bugs should be addressed.

I'm not sure what you mean here, but I really didn't appreciate the tone. We're trying to help as an upstream project. If you have a Ubuntu or Red Hat support contract then someone (perhaps even me!) can fix bugs like this with a SLA and with official signed builds backported to stable branches.

I think the immediate workaround is to not have fwupd auto-quit on idle, and then it won't have to re-probe the disks on startup in the cron job. To debug the root cause we need a `sudo fwupdtool get-devices --plugins uefi_capsule -vv` of 1.9.14 plus the patch from above.

axet, 2024-03-13:
I keep seeing a lot of fatrace events but drive keep sleeping. Bug I've seen no "Downloading..." message. I'll wait few more days.

```
15:44:26.468516 (fwupdmgr)(891903): O   /media/axet/1TB
15:44:26.468516 (fwupdmgr)(891903): C   /media/axet/1TB
15:44:26.468537 (fwupdmgr)(891903): CO  /boot/efi
16:19:37.856477 (fwupdmgr)(909001): CO  /boot/efi
16:19:37.856477 (fwupdmgr)(909001): CO  /media/axet/1TB
17:55:58.816464 (fwupdmgr)(955621): CO  /media/axet/1TB
17:55:58.816476 (fwupdmgr)(955621): CO  /boot/efi
18:41:39.844651 (fwupdmgr)(977525): CO  /boot/efi
18:41:39.844664 (fwupdmgr)(977525): CO  /media/axet/1TB
19:43:58.324452 (fwupdmgr)(1007287): CO  /media/axet/1TB
19:43:58.324470 (fwupdmgr)(1007287): CO  /boot/efi
20:05:26.464463 (fwupdmgr)(1017874): CO  /media/axet/1TB
20:05:26.464474 (fwupdmgr)(1017874): CO  /boot/efi
21:13:29.828570 (fwupdmgr)(1050549): CO  /boot/efi
21:13:29.828570 (fwupdmgr)(1050549): CO  /media/axet/1TB
```

```
Ср 13 мар 2024 06:10:11 MSK - drive state is: active/idle
Ср 13 мар 2024 06:24:42 MSK - drive state is: standby
```

```
2024-03-13T21:13:29.826946+03:00 axet-desktop systemd[1]: Starting fwupd-refresh.service - Refresh fwupd metadata and update motd...
2024-03-13T21:13:29.851984+03:00 axet-desktop systemd[1]: fwupd-refresh.service: Deactivated successfully.
2024-03-13T21:13:29.852074+03:00 axet-desktop systemd[1]: Finished fwupd-refresh.service - Refresh fwupd metadata and update motd.
```

closed 2024-03-14 by superm1

axet, 2024-03-15:
You can ignore rotational IDLE devices for automatic executions/daemons. Eventually fwup will update firmware since once device will be running one time during daemon start.

/sys/class/block/sda/queue/rotational

superm1, 2024-03-15:
> You can ignore rotational IDLE devices for automatic executions/daemons. Eventually fwup will update firmware since once device will be running one time during daemon start.
> 
> /sys/class/block/sda/queue/rotational

If the esp is located on this we still need to mount it though to confirm partition type and mount point.