fwupd/fwupd #6888: Ubuntu 22.04.4 - From today it's appear this error "Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 740a74633fc78e329ea701b0c12f8761c24b7988"

DjDiabolik, 2024-03-07:
**Describe the bug**
When i open the Ubuntu Snap Store and try to check update it's appear this error.
Apparently the check for update it's completed correctly.
Reading some thread on forum for resolve this issue need to do this from Terminal...i do this:
`diabolik@NUC-Ubuntu:~$ sudo fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Downloading…             [***************************************]
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 740a74633fc78e329ea701b0c12f8761c24b7988
`

**fwupd version information**
`diabolik@NUC-Ubuntu:~$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.kernel                    6.5.0-21-generic
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
`

Please provide the output of the fwupd devices recognized in your system.
`diabolik@NUC-Ubuntu:~$ fwupdmgr get-devices --show-all-devices
NUC7PJYHN
│
├─GeminiLake [UHD Graphics 605]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    06
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUID:               428afab4-0694-59d7-ab82-09be6ecefa9d
│                         7e675f13-a2d7-52fa-b886-05280bb915a6
│                         52e5049a-b195-5849-9498-e73ee3113abd
│                         6f398ef1-6999-553b-822a-e5798e7d3809
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─HS-SSD-E100 512G:
│     Device ID:          c46944fe84f2b057692258cc87a812cc86364881
│     Summary:            ATA drive
│     Current version:    030fAA20
│     Update Error:       No vendor ID set
│     GUID:               0dc6e288-2ed0-5ef3-8c9f-f2eded1c7c1b
│                         9022c257-5da3-5c5a-a070-e717ec4b185d
│                         372f0676-1fad-5401-b708-00c85066c469
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─Pentium Silver J5040 CPU @ 2.00GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000022
│     Vendor:             Intel
│     GUID:               b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         54817722-b948-55a7-8324-3d4736e90181
│                         f02ec022-4f79-53c2-a68f-08ea48563cb4
│     Device Flags:       • Internal device
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    70
│ │   Minimum Version:    70
│ │   Vendor:             Intel(R) Client Systems (DMI:Intel Corp.)
│ │   Update State:       Success
│ │   GUID:               856758ee-a971-40c7-8ee4-26d67b4f6704
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  77
│       Minimum Version:  77
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUID:             c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
└─TPM:
      Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
      Summary:            TPM 2.0 Device
      Current version:    403.0.0.0
      Vendor:             Intel (TPM:INTC)
      GUID:               ff71992e-52f7-5eea-94ef-883e56e034c6
                          34801700-3a50-5b05-820c-fe14580e4c2d
                          8e1cbc5d-5a11-5149-bfea-b6065d5296ba
                          03f304f4-223e-54f4-b2c1-c3cf3b5817c6
                          52d7b679-db28-5bf7-bd87-41d77aeec600
      Device Flags:       • Internal device
    
`

**Additional questions**

- Operating system and version: Ubuntu 22.04.4 LTS
- Have you tried rebooting? Yes.. so many times
- Is this a regression? Apparently not.. it's appears about right now whitout any apparently errors.

hughsie, 2024-03-07:
What does `sudo fwupdmgr refresh --force --verbose --verbose` say please.

sebastian-brandt, 2024-03-07:
Same issue on my "6.5.0-21-generic #21~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Feb  9 13:32:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux":

```
fwupdmgr --force refresh
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 740a74633fc78e329ea701b0c12f8761c24b7988
```

```
fwupdmgr --version runtime
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.kernel                    6.5.0-21-generic
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```

```
fwupdmgr get-devices --show-all-devices
HP ZBook 15 G4
│
├─Core™ i7-7820HQ CPU @ 2.90GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f4
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         809a0b93-8a12-5338-a571-ad5583acf896
│                         d0f754d5-1395-5573-bc83-85ba955da70a
│     Device Flags:       • Internal device
│   
├─GM206GLM [Quadro M2200 Mobile]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              46a93b53-dbb7-56b8-976d-8dfbd777dbd1
│                         5dd6daa4-8ae2-5df6-ad26-3b035f214c81
│                         ce1b3ea7-8991-5df0-a216-d3ea05fd7441
│                         715f9356-264d-52fb-986f-2cfbf8e1b1ee
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─HP Thunderbolt 3 Dock:
│     Device ID:          9ffb42c609f721d61c8184e97e2fc843ddb8df47
│     Current version:    27.00
│     Vendor:             HP Inc. (THUNDERBOLT:0x00F0, TBT:0x00F0)
│     GUIDs:              de6a84bf-0cc7-566b-8175-551bb293df8b
│                         ffb9d8ee-0303-5ae8-89a3-9765d2079ddc
│                         e671ec89-2d32-548e-8f89-ea2ab2201abf
│                         abbeda4a-e5b9-5214-9f2c-66b014a72f48
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Device stages updates
│                         • Signed Payload
│   
├─MTFDDAK256TBN-1AR1ZABHA:
│     Device ID:          e421b2fc248391f6fe3e55ddbb3c9043be068bd0
│     Summary:            ATA drive
│     Current version:    HPC0014
│     Vendor:             Micron (ATA:0x1344, OUI:00a075)
│     GUIDs:              941e5eb1-7ff6-5c0e-baa8-65110d3507a4
│                         0fea2983-0d11-5276-bcc3-38c78b8129b7
│                         e943e5e7-7469-563c-a618-0fb5c4a6716f
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
[... 72 lines cut ...]
│     Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│     Current version:    7.63.13.6400
│     Vendor:             Infineon (TPM:IFX)
│     GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6
│                         5eebb112-75ad-5536-b173-a11eb3399402
│                         ddf995da-1b32-5a8a-bc1b-8d5af4b38b51
│                         6d81ab63-db2e-50ac-934f-6be9accf5e02
│                         301555de-680d-5ddc-b995-7553fc9138f1
│     Device Flags:       • Internal device
│   
└─Thunderbolt host controller:
      Device ID:          f3afd10f6ee40bdaf2e0540afa42b6ad6e370ba0
      Summary:            Unmatched performance for high-speed I/O
      Current version:    46.00
      Vendor:             HP, Inc. (THUNDERBOLT:0x00F0, TBT:0x00F0)
      GUIDs:              1c9eec61-e620-5940-ac35-ce368078a1c1
                          5dc50c75-5516-50b9-878e-11969f560608
                          b53f064b-f829-5fff-b695-365d52ed6f5e
                          caf6b14f-0953-5368-9f8f-f9659c7a4d82
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Device stages updates
                          • Signed Payload
```

sebastian-brandt, 2024-03-07:
> What does `sudo fwupdmgr refresh --force --verbose --verbose` say please.

Please see attached file:
[verbose.txt](https://github.com/fwupd/fwupd/files/14523437/verbose.txt)

DjDiabolik, 2024-03-07:
> What does `sudo fwupdmgr refresh --force --verbose --verbose` say please.

omg it's so long result.... i try to post here:
`diabolik@NUC-Ubuntu:~$ sudo fwupdmgr refresh --force --verbose --verbose
(fwupdmgr:8973): GLib-DEBUG: 12:50:15.846: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:8973): GLib-GIO-DEBUG: 12:50:15.865: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:8973): dconf-DEBUG: 12:50:15.865: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:8973): dconf-DEBUG: 12:50:15.865: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:8973): dconf-DEBUG: 12:50:15.871: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:8973): dconf-DEBUG: 12:50:15.871: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:8973): dconf-DEBUG: 12:50:15.871: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:8973): GLib-GIO-DEBUG: 12:50:15.871: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:8973): dconf-DEBUG: 12:50:15.875: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:8973): dconf-DEBUG: 12:50:15.875: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:8973): dconf-DEBUG: 12:50:15.875: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:8973): dconf-DEBUG: 12:50:15.875: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:8973): dconf-DEBUG: 12:50:15.875: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:15.879: Emitting ::status-changed() [idle]
Updating lvfs
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:15.886: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:15.886: Emitting ::status-changed() [downloading]
Downloading…             [     \                                 ](fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.079: download progress: 100%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.079: download progress: 100%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.079: download progress: 100%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.079: download progress: 100%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.079: Emitting ::status-changed() [idle]
Downloading…             [***************************************]
Downloading…             [***************************************]
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.079: status-code was 200
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.080: changing metadata URI from https://cdn.fwupd.org/downloads/firmware.xml.gz to https://cdn.fwupd.org/downloads/firmware-07078-stable.xml.gz
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.080: downloading https://cdn.fwupd.org/downloads/firmware-07078-stable.xml.gz
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.080: Emitting ::status-changed() [downloading]
Downloading…             [***************************************]
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.202: download progress: 0%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.202: download progress: 0%
Downloading…             [ -                                     ](fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.202: download progress: 0%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.202: download progress: 0%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.204: download progress: 0%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.204: download progress: 0%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.206: download progress: 0%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.206: download progress: 0%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.208: download progress: 1%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.208: download progress: 1%
Downloading…             [                                       ](fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.209: download progress: 1%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.209: download progress: 1%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.211: download progress: 1%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.211: download progress: 1%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.211: download progress: 1%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.211: download progress: 1%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.213: download progress: 1%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.213: download progress: 1%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.214: download progress: 2%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.214: download progress: 2%
Downloading…             [                                       ](fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.216: download progress: 2%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.216: download progress: 2%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.218: download progress: 2%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.218: download progress: 2%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.218: download progress: 2%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.218: download progress: 2%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.219: download progress: 2%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.219: download progress: 2%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.219: download progress: 2%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.219: download progress: 2%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.220: download progress: 2%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.220: download progress: 2%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.221: download progress: 3%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.221: download progress: 3%
[... cut ...]
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.330: download progress: 100%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.330: download progress: 100%
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.330: Emitting ::status-changed() [idle]
Downloading…             [***************************************]
(fwupdmgr:8973): Fwupd-DEBUG: 12:50:16.330: status-code was 200
Downloading…             [***************************************]
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 0a5a850327f8e8b196b2d79e79ce347cb86037f1
`

hughsie, 2024-03-07:
I'm not sure what's happened, but in the meantime I've kicked off a rebuild just in case.

DjDiabolik, 2024-03-07:
> I'm not sure what's happened, but in the meantime I've kicked off a rebuild just in case.

i'm need to do somethigs on my side ? If yes ask..... consider I'm not very knowledgeable about Linux.

hughsie, 2024-03-07:
wait a few minutes, then `fwupdmgr refresh --force`

DjDiabolik, 2024-03-07:
> wait a few minutes, then `fwupdmgr refresh --force`

oh ok... I noticed this error today. while I was trying to solve another problem with the snap store (ndr persistent notification about software upgrade appears at every reboot).

DjDiabolik, 2024-03-07:
@hughsie 
Now i have retry... now it's changed the hash:
`Downloading…             [************************************** ](fwupdmgr:9806): Fwupd-DEBUG: 13:37:40.257: download progress: 98%
(fwupdmgr:9806): Fwupd-DEBUG: 13:37:40.257: download progress: 98%
(fwupdmgr:9806): Fwupd-DEBUG: 13:37:40.257: download progress: 98%
(fwupdmgr:9806): Fwupd-DEBUG: 13:37:40.257: download progress: 98%
(fwupdmgr:9806): Fwupd-DEBUG: 13:37:40.258: download progress: 98%
(fwupdmgr:9806): Fwupd-DEBUG: 13:37:40.258: download progress: 98%
(fwupdmgr:9806): Fwupd-DEBUG: 13:37:40.258: download progress: 100%
(fwupdmgr:9806): Fwupd-DEBUG: 13:37:40.258: download progress: 100%
(fwupdmgr:9806): Fwupd-DEBUG: 13:37:40.258: download progress: 100%
(fwupdmgr:9806): Fwupd-DEBUG: 13:37:40.258: download progress: 100%
(fwupdmgr:9806): Fwupd-DEBUG: 13:37:40.258: Emitting ::status-changed() [idle]
Downloading…             [***************************************]
(fwupdmgr:9806): Fwupd-DEBUG: 13:37:40.258: status-code was 200
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 0a5a850327f8e8b196b2d79e79ce347cb86037f1`

hughsie, 2024-03-07:
This is what I get:

	[hughsie@hughsie-work ~]$ wget https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat
	--2024-03-07 12:46:46--  https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat
	Resolving cdn.fwupd.org (cdn.fwupd.org)... 151.101.2.49, 151.101.66.49, 151.101.130.49, ...
	Connecting to cdn.fwupd.org (cdn.fwupd.org)|151.101.2.49|:443... connected.
	HTTP request sent, awaiting response... 200 OK
	Length: 2233 (2.2K) [application/octet-stream]
	Saving to: ‘firmware.xml.gz.jcat’
	firmware.xml.gz.jcat                               100%[=============================================================================================================>]   2.18K  --.-KB/s    in 0s      
	2024-03-07 12:46:46 (49.8 MB/s) - ‘firmware.xml.gz.jcat’ saved [2233/2233]

	[hughsie@hughsie-work ~]$ jcat-tool info firmware.xml.gz.jcat 
	JcatFile:
	  Version:               0.1
	  JcatItem:
	    ID:                  firmware-07078-stable.xml.gz
	    AliasId:             firmware.xml.gz
	    JcatBlob:
	      Kind:              sha1
	      Flags:             is-utf8
	      Timestamp:         2024-03-07T11:45:25Z
	      Size:              0x28
	      Data:              e8869797da0642a4564965984f50e78acac7f7ba
	    JcatBlob:
	      Kind:              sha256
	      Flags:             is-utf8
	      Timestamp:         2024-03-07T11:45:25Z
	      Size:              0x40
	      Data:              79afec6b3b95b36995fac4231d1b0bd5d7a4d2124c0b2c4487a115cdc90a3b86
	    JcatBlob:
	      Kind:              pkcs7
	      Flags:             is-utf8
	      Timestamp:         2024-03-07T11:45:25Z
	      Size:              0x8c0
	      Data:              -----BEGIN PKCS7-----
				 MIIGUgYJKoZIhvcNAQcCoIIGQzCCBj8CAQExDTALBglghkgBZQMEAgEwCwYJKoZI
				 hvcNAQcBoIIEOjCCBDYwggKeoAMCAQICDFprhisibP88kP07YjANBgkqhkiG9w0B
				 AQsFADA6MRAwDgYDVQQDEwdMVkZTIENBMSYwJAYDVQQKEx1MaW51eCBWZW5kb3Ig
				 RmlybXdhcmUgUHJvamVjdDAeFw0xODAxMTYwMDAwMDBaFw0yODAxMTYwMDAwMDBa
				 MBkxFzAVBgNVBAMTDlJpY2hhcmQgSHVnaGVzMIIBIjANBgkqhkiG9w0BAQEFAAOC
				 AQ8AMIIBCgKCAQEArR5nseG3+Zs+o41P9LTspiVSGVcp6ifNHhbKKxBvZZXsy0gX
				 +P/VRlsHLiKQrFulQ8GbPODytFv0o+y/0MkiJxv/fY3yEZ2bwNpsSeXFQSGHI6yz
				 jaVNeNCu8lnnDtD7kiC8UNUNHcnA4+2h/Yv4k+KPqYF+Qb6nWAEIID1ObMnjeJUb
				 dbPPvy12aasV3gZcZ+goYNFc0ua6OU/CNEuzAVVCTAJ/EpCdGpll1+6BGU5ImIUG
				 TlMTWq2xmCfCPugakHrmWA66yHWwE6LmC/U7qQDWFemsSNnmzyBB9HPkqsW1DjHr
				 +ZmNUPj3+q2UGnNwP/Cne462XbsZB569w7pnzQIDAQABo4HcMIHZMAwGA1UdEwEB
				 /wQCMAAwNwYDVR0RBDAwLoYXaHR0cHM6Ly9md3VwZC5vcmcvbHZmcy+BE3JpY2hh
				 cmRAaHVnaHNpZS5jb20wEwYDVR0lBAwwCgYIKwYBBQUHAwMwDwYDVR0PAQH/BAUD
				 AweAADAdBgNVHQ4EFgQUMcrnDWk3DM1GJCqJK7EbAycgB+UwHwYDVR0jBBgwFoAU
				 sY3q5COnfgmOte4x4GrdnjQ3ZawwKgYDVR0fBCMwITAfoB2gG4YZaHR0cDovL3d3
				 dy5md3VwZC5vcmcvcGtpLzANBgkqhkiG9w0BAQsFAAOCAYEAjoXBOtVb1qPLuE7I
				 ShdSkk66JMNmzZnODbun3BaViUF3PPhuRiJ2y7Bu7loCnxHwzKpq5Hn4Untg1WhQ
				 SsPEv6H2oc2E4HU1Gds2HEE1UL0VzulPEwaOFX0OZ37LmJ2VFvfgwfNmjAWZ+ngN
				 sjuff3PWeHNsOwbS3To0CBlHqz9LlcA7Gxpxbz4DbUEvgTTKZFbasgUS6/QmbpPy
				 880ThgD93yCg6Q002AooPMw8zW6gGUP7/2D4aKBjgM/IJDrJEicTdKDFtUacsN+d
				 CfIEcn6F1rzPgwo0B/yLYVMpbL8WzxSS7cX5dBlkreH6Nf7g+A2NTryU/6FxoFxo
				 nWJmLssNQCgx42Ywc8CgJRt1yWXxoZXOPap8zPvFAAmrfUlqfZlr2Wzcf2nxpSZJ
				 oQRzGDXDRiYNt28xjX408cJ+Wy+0oWwgnXCkfzMDnJQB8MX6+5Ah4v4mS/XjmOIi
				 Q+hKCT99Yf0Eq3TWzFhyYducqLU904FbrrUSoi07DfxXa/8RMYIB3jCCAdoCAQEw
				 SjA6MRAwDgYDVQQDEwdMVkZTIENBMSYwJAYDVQQKEx1MaW51eCBWZW5kb3IgRmly
				 bXdhcmUgUHJvamVjdAIMWmuGKyJs/zyQ/TtiMAsGCWCGSAFlAwQCAaBpMBgGCSqG
				 SIb3DQEJAzELBgkqhkiG9w0BBwEwHAYJKoZIhvcNAQkFMQ8XDTI0MDMwNzExNDUy
				 NVowLwYJKoZIhvcNAQkEMSIEIHmv7Gs7lbNplfrEIx0bC9XXpNISTAssRIehFc3J
				 CjuGMA0GCSqGSIb3DQEBAQUABIIBAJZu405YxPY+zDOhsmrLo+K4glwiJE8SAl1d
				 8fjUsY8Qo1dQJoam1oZd+eeX1p8S0Uyk315Pp4S6kwE/1fXpGiXKiXo1Hl7b7ULq
				 EYYsWQ8LjFkaBo/9OdzjIa89S+1dB57G+X5q4gC+K/iM5d3s8IT3D8yQnVQUgkDt
				 zK3Xe39qNf5Gfp7QE2s+tBRQI41rRGOLlOqjEGKwgCEU0ehHSuRXlm2Cas+MYZT/
				 XlLJyIdxlpG8e6/0JjjnoT8CEvdT9x+rLWFEBRZHtuQTuKmLJn9yJ7ap5HaRt6Ws
				 kPoN/8Kpp9LEgk3mWjG2fBkLbOR2iY5A8AUcAY3e0G0Ym5Ui87g=
[... 17 lines cut ...]
				 -----END PGP SIGNATURE-----
				 

	[hughsie@hughsie-work ~]$ wget https://cdn.fwupd.org/downloads/firmware-07078-stable.xml.gz
	--2024-03-07 12:47:27--  https://cdn.fwupd.org/downloads/firmware-07078-stable.xml.gz
	Resolving cdn.fwupd.org (cdn.fwupd.org)... 151.101.2.49, 151.101.66.49, 151.101.130.49, ...
	Connecting to cdn.fwupd.org (cdn.fwupd.org)|151.101.2.49|:443... connected.
	HTTP request sent, awaiting response... 200 OK
	Length: 1049053 (1.0M) [application/gzip]
	Saving to: ‘firmware-07078-stable.xml.gz’
	firmware-07078-stable.xml.gz                       100%[=============================================================================================================>]   1.00M  --.-KB/s    in 0.1s    
	2024-03-07 12:47:28 (10.1 MB/s) - ‘firmware-07078-stable.xml.gz’ saved [1049053/1049053]

	[hughsie@hughsie-work ~]$ sha1sum firmware-07078-stable.xml.gz 
	e8869797da0642a4564965984f50e78acac7f7ba  firmware-07078-stable.xml.gz

	[hughsie@hughsie-work ~]$ sha256sum firmware-07078-stable.xml.gz 
	79afec6b3b95b36995fac4231d1b0bd5d7a4d2124c0b2c4487a115cdc90a3b86  firmware-07078-stable.xml.gz

	[hughsie@hughsie-work ~]$ jcat-tool verify firmware.xml.gz.jcat --public-keys /etc/pki/fwupd-metadata/
	firmware-07078-stable.xml.gz:
	    PASSED sha1: OK
	    PASSED sha256: OK
	    PASSED pkcs7: O=Linux Vendor Firmware Project,CN=LVFS CA
	    PASSED gpg: 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2

dcd-arnold, 2024-03-07:
I have the same error: 
```
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 0a5a850327f8e8b196b2d79e79ce347cb86037f1
```
for Ubuntu 20.04:
```
# fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.4
compile   com.hughsie.libjcat           0.1.4
runtime   org.kernel                    5.15.0-76-generic
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.4
```

hughsie, 2024-03-07:
If you follow my same debugging, do you get checksums that match?

aowenson-imm, 2024-03-07:
Same problem here, suddenly from today.

> If you follow my same debugging, do you get checksums that match?

Yes

dcd-arnold, 2024-03-07:
> If you follow my same debugging, do you get checksums that match?

@hughsie I can follow up to this step:

```
# jcat-tool verify firmware.xml.gz.jcat --public-keys /etc/pki/fwupd-metadata/
firmware-07078-stable.xml.gz:
    FAILED sha1: verifying data is not supported
    FAILED sha256: verifying data is not supported
    PASSED pkcs7: O=Linux Vendor Firmware Project,CN=LVFS CA
    PASSED gpg: 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
    FAILED: Validation failed
Validation failed
```

sebastian-brandt, 2024-03-07:
> If you follow my same debugging, do you get checksums that match?

```
sha1sum firmware-07078-stable.xml.gz
e8869797da0642a4564965984f50e78acac7f7ba  firmware-07078-stable.xml.gz
sha256sum firmware-07078-stable.xml.gz
79afec6b3b95b36995fac4231d1b0bd5d7a4d2124c0b2c4487a115cdc90a3b86  firmware-07078-stable.xml.gz
jcat-tool verify firmware.xml.gz.jcat --public-keys /etc/pki/fwupd-metadata/
firmware-07078-stable.xml.gz:
    PASSED sha1: OK
    PASSED sha256: OK
    PASSED pkcs7: O=Linux Vendor Firmware Project,CN=LVFS CA
    PASSED gpg: 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
```

```
fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 0a5a850327f8e8b196b2d79e79ce347cb86037f1

```

diablo248, 2024-03-07:
I also see the same issue

hughsie, 2024-03-07:
Can you attach the output of `sudo fwupdtool refresh --verbose --verbose` please.

dcd-arnold, 2024-03-07:
The error is away for me.

diablo248, 2024-03-07:
Now it started to work in my case

DjDiabolik, 2024-03-07:
@hughsie apparently for me it's fixed....

diablo248, 2024-03-07:
From other setup:
```
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.243: download progress: 95%
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.243: download progress: 95%
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.243: download progress: 95%
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.243: download progress: 96%
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.243: download progress: 96%
Downloading…             [*************************************  ](fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.244: download progress: 98%
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.244: download progress: 98%
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.244: download progress: 98%
Downloading…             [************************************** ](fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.244: download progress: 98%
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.244: download progress: 98%
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.244: download progress: 98%
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.244: download progress: 98%
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.244: download progress: 98%
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.244: download progress: 98%
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.244: download progress: 98%
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.244: download progress: 100%
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.244: download progress: 100%
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.244: download progress: 100%
Downloading…             [***************************************]
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.244: download progress: 100%
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.244: Emitting ::status-changed() [idle]
(fwupdmgr:27039): Fwupd-DEBUG: 13:49:58.244: status-code was 200
Downloading…             [***************************************]
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 0a5a850327f8e8b196b2d79e79ce347cb86037f1

```

afunix, 2024-03-07:
Two of four Ubuntu 22.04 servers started to complain today:
```
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 0a5a850327f8e8b196b2d79e79ce347cb86037f1
```

I'm not sure where this checksum (0a5a850327f8e8b196b2d79e79ce347cb86037f1) comes from since:
```
# jcat-tool info firmware.xml.gz.jcat 
...
Data:              e8869797da0642a4564965984f50e78acac7f7ba
...
Data:              79afec6b3b95b36995fac4231d1b0bd5d7a4d2124c0b2c4487a115cdc90a3b86
...

# sha1sum firmware-07078-stable.xml.gz 
e8869797da0642a4564965984f50e78acac7f7ba  firmware-07078-stable.xml.gz

# sha256sum firmware-07078-stable.xml.gz
79afec6b3b95b36995fac4231d1b0bd5d7a4d2124c0b2c4487a115cdc90a3b86  firmware-07078-stable.xml.gz
root@annie:~# jcat-tool verify firmware.xml.gz.jcat --public-keys /etc/pki/fwupd-metadata/
firmware-07078-stable.xml.gz:
    PASSED sha1: OK
    PASSED sha256: OK
    PASSED pkcs7: O=Linux Vendor Firmware Project,CN=LVFS CA
    PASSED gpg: 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
```

afunix, 2024-03-07:
Btw, firmware-07078-stable.xml.gz is 1.1MB long. I'm not sure if the limit was updated but in [5173](https://github.com/fwupd/fwupd/issues/5173) it was set to 800k.

hughsie, 2024-03-07:
> Btw, firmware-07078-stable.xml.gz is 1.1MB long

Ohh that's it! Super old versions of fwupd had a hardcoded 1MB limit. Ubuntu should backport 0729ab5cb013dd90d72f8010a7d826a38d2ea474 as a priority. I'll see if I can fix something LVFS-side.

paddylandau, 2024-03-07:
> I'll see if I can fix something LVFS-side.

Thank you, @hughsie . Do you need someone to report this to Launchpad for Ubuntu, so that Ubuntu can update its repositories? If so, I'll be happy to file the report.

Alternatively, is compiling from source the only alternative method of getting an updated fwupdmgr until then? (I'm really not good with compiling!)

hughsie, 2024-03-07:
> Do you need someone to report this to Launchpad for Ubuntu

Yes, please. In general I need Canonical to care a lot more about doing updates to the fwupd package. 

> Alternatively, is compiling from source the only alternative method of getting an updated fwupdmgr until then

There's also a snap version too, but be sure to uninstall the deb version if you install the nap one instead.

hughsie, 2024-03-07:
I've pushed (and deployed) https://gitlab.com/fwupd/lvfs-website/-/commit/8908020c1dddaa7ff1e960966c1821d45c820073 and regenerated the legacy metadata -- now:

    [hughsie@hughsie-work docs (master %)]$ du -h firmware-07079-stable.xml.gz
    880K	firmware-07079-stable.xml.gz

which I think is going to work. The next step is to start picking and choosing what updates are presented to such old fwupd versions. Perhaps we can filter out non-security issues or something. The underlying problem is that Canonical isn't keeping fwupd up to date -- 1.7.9 was released on 25th July **2022**.

paddylandau, 2024-03-07:
> Yes, please. In general I need Canonical to care a lot more about doing updates to the fwupd package.

I'll post back here with the report once I've reported this on Launchpad.

> There's also a snap version too, but be sure to uninstall the deb version if you install the nap one instead.

I uninstalled the apt version and installed the snap version. It works, thank you! (I had to close and reopen the terminal because of caching.) So, I think that snap is a better solution than relying on Canonical.

superm1, 2024-03-07:
@KaiChuan-Hsieh can you help here?

paddylandau, 2024-03-08:
I have reported this on Launchpad as a bug in Ubuntu.

If you have this problem, please vote for it: Go to [the bug report](https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/2056548); log in; and select the green writing on the left near the top. You can optionally subscribe to the bug's updates (in the right-hand column).

Thank you

KaiChuan-Hsieh, 2024-03-11:
> @KaiChuan-Hsieh can you help here?

@superm1 

I'll raise it to internal team to evaluate the feasibility.

Thanks,

E1v1s, 2025-03-28:
the same
```
sudo fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Downloading…             [***************************************]
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 7e81a28bfd4d8c1ccfa9c5e54e6f8e3ca7f1126f
```

lukas-fichtner, 2025-03-28:
I believe the issue is related to the EOL status of version [1.7](https://github.com/fwupd/fwupd/blob/main/SECURITY.md) on Ubuntu 22.04. 

Since the service isn't critical for me at the moment, I’ve disabled it using the following commands:
systemctl disable --now fwupd-refresh.timer fwupd-refresh.service
systemctl mask fwupd-refresh.service
systemctl reset-failed fwupd-refresh.service

tipuraneo, 2025-03-28:
same here on Ubuntu 22.04.5
```
Downloading…             [***************************************]
(fwupdmgr:7098): Fwupd-DEBUG: 09:35:35.589: download progress: 100%
(fwupdmgr:7098): Fwupd-DEBUG: 09:35:35.589: Emitting ::status-changed() [idle]
(fwupdmgr:7098): Fwupd-DEBUG: 09:35:35.589: status-code was 200
Downloading…             [***************************************]
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected e2ab73978426e577c02ff02aba20530af88f9bcc
```

Aakhyan, 2025-03-28:
Hi,

This worked for me for fixing the issue:

```
**fwupdtool refresh --verbose --verbose**
**fwupdmgr --force refresh**
**systemctl restart fwupd-refresh.service**
```

Try the above and it should get fixed.

tipuraneo, 2025-03-28:
Jup worked.

wargamez, 2025-03-28:
Same on Ubuntu 20

paddylandau, 2025-03-28:
> This worked for me for fixing the issue: …

Unfortunately, this didn't help me.

DjDiabolik, 2025-03-29:
Omg i don't remember how i resolved this issue... i'm not reported here... strange.

paddylandau, 2025-03-29:
> [@paddylandau](https://github.com/paddylandau) Seems very strange, should help on Ubuntu xxxxx

Thanks. Reading one of the subsequent comments, it seems that the `fwupdtool` should have been used with `sudo`. I've done this, and it's fixed the problem!

Thank you :)

I shall document this for future.

paulliss, 2025-03-29:
Today it's returned

rdiez, 2025-03-29:
On Ubuntu 22.04.5, the workaround with "sudo fwupdtool refresh" etc. worked for a day, and then I got the same "checksum failure" error again. So I had to disable it too.

superm1, 2025-03-29:
This is a duplicate of https://github.com/fwupd/fwupd/issues/8629

closed 2025-03-29 by superm1