fwupd/fwupd #6897: Send time left or time completed in InstallProgress

rishabh-chromium, 2024-03-08:
**Is your feature request related to a problem? Please describe.**
There is currently no way to say how long is left for an update to complete using the daemon. InstallProgress only returns the percentage and not time. 

**Describe the solution you'd like**
Can either mention the time already done or time left so we can use the data to let the user know the amount of time left.

hughsie, 2024-03-08:
Because the downloading is done client-side I'm not sure if the daemon is 100% the right place for this; we'd have to do *both* which seems a bit messy. Would moving the time estimation into libfwupd be acceptable?

rishabh-chromium, 2024-03-08:
I don't think I fully understand what libfwupd does. We usually only work with the daemon. We can jump on a quick call and get to a conclusion for this

hughsie, 2024-11-11:
@rishabh-chromium you still need this?

closed 2024-11-12 by rishabh-chromium