fwupd/fwupd #6903: Hangs after aborting update by pressing *n*

paulmenzel, 2024-03-11:
On a Dell XPS 15 7590 with Debian sid/unstable and *fwupd* 1.9.14-1, I did `fwupdmgr update`, and aborted it by pressing the key *n*, and then it just hung.

```
$ fwupdmgr update
Devices with no available firmware updates: 
 • SSDPEMKF010T8 NVMe INTEL 1024GB
 • TPM
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.25.0 to 1.27.0?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ -This release contains security updates as disclosed in the Dell Security    ║
║ Advisory.                                                                    ║
║                                                                              ║
║ XPS 15 7590 must remain plugged into a power source for the duration of the  ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: n
^C^C^C^C
```

```
$ fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.15
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.fwupd         1.9.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.8.0-rc6+
runtime   org.freedesktop.fwupd         1.9.14
```

GDB backtrace:

```
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
0x00007f7ee265efe0 in __GI__IO_getline_info (fp=fp@entry=0x7f7ee27bdaa0 <_IO_2_1_stdin_>, 
    buf=buf@entry=0x7ffdcc218114 "bU", n=n@entry=3, delim=delim@entry=10, 
    extract_delim=extract_delim@entry=1, eof=eof@entry=0x0) at ./libio/iogetline.c:55
55	./libio/iogetline.c: No such file or directory.
(gdb) set pagination off
(gdb) bt
#0  0x00007f7ee265efe0 in __GI__IO_getline_info (fp=fp@entry=0x7f7ee27bdaa0 <_IO_2_1_stdin_>, 
    buf=buf@entry=0x7ffdcc218114 "bU", n=n@entry=3, delim=delim@entry=10, 
    extract_delim=extract_delim@entry=1, eof=eof@entry=0x0) at ./libio/iogetline.c:55
#1  0x00007f7ee265f0b8 in __GI__IO_getline (fp=fp@entry=0x7f7ee27bdaa0 <_IO_2_1_stdin_>, 
    buf=buf@entry=0x7ffdcc218114 "bU", n=n@entry=3, delim=delim@entry=10, 
    extract_delim=extract_delim@entry=1) at ./libio/iogetline.c:34
#2  0x00007f7ee265ddec in _IO_fgets (buf=buf@entry=0x7ffdcc218114 "bU", n=n@entry=4, 
    fp=0x7f7ee27bdaa0 <_IO_2_1_stdin_>) at ./libio/iofgets.c:53
#3  0x00007f7ee2b97969 in fgets (__stream=<optimized out>, __n=4, __s=0x7ffdcc218114 "bU")
    at /usr/include/x86_64-linux-gnu/bits/stdio2.h:213
#4  fu_console_input_bool (self=self@entry=0x5562f6841090, def=def@entry=1, 
    format=format@entry=0x7f7ee2ba1683 "%s") at ../src/fu-console.c:195
#5  0x00007f7ee2b9a4aa in fu_util_prompt_warning (console=0x5562f6841090, 
    device=device@entry=0x5562f6867160, release=release@entry=0x5562f68687c0, 
    machine=0x5562f6859630 "XPS 15 7590", error=error@entry=0x7ffdcc218320)
    at ../src/fu-util-common.c:398
#6  0x00005562f6051032 in fu_util_update_device_with_release (priv=priv@entry=0x5562f683e310, 
    dev=dev@entry=0x5562f6867160, rel=rel@entry=0x5562f68687c0, error=error@entry=0x7ffdcc218320)
    at ../src/fu-util.c:2866
#7  0x00005562f6056f59 in fu_util_update (priv=0x5562f683e310, values=0x5562f6860dd0, 
    error=0x7ffdcc2184d8) at ../src/fu-util.c:3019
#8  0x00007f7ee2b9aa85 in fu_util_cmd_array_run (array=array@entry=0x5562f683e430, 
    priv=priv@entry=0x5562f683e310, command=0x7ffdcc21a341 "update", values=0x7ffdcc218b48, 
    error=error@entry=0x7ffdcc2184d8) at ../src/fu-util-common.c:521
#9  0x00005562f604c50d in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-util.c:5581
[... 129 lines cut ...]
            arg = G_OPTION_ARG_NONE, arg_data = 0x5562f683e370, 
            description = 0x5562f6059420 "Show devices that are not updatable", 
            arg_description = 0x0}, {long_name = 0x5562f605aba5 "disable-ssl-strict", 
            short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x5562f683e374, 
            description = 0x5562f6059448 "Ignore SSL strict checks when downloading files", 
            arg_description = 0x0}, {long_name = 0x5562f605abb8 "p2p", short_name = 0 '\000', 
            flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7ffdcc2184c0, 
            description = 0x5562f60591a0 "Only use peer-to-peer networking when downloading files", 
            arg_description = 0x0}, {long_name = 0x5562f605abbc "filter", short_name = 0 '\000', 
            flags = 0, arg = G_OPTION_ARG_STRING, arg_data = 0x7ffdcc2184e8, 
            description = 0x5562f6059140 "Filter with a set of device flags using a ~ prefix to exclude, e.g. 'internal,~needs-reboot'", arg_description = 0x0}, {long_name = 0x5562f605abc3 "filter-release", 
            short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_STRING, arg_data = 0x7ffdcc2184f0, 
            description = 0x5562f60590d0 "Filter with a set of release flags using a ~ prefix to exclude, e.g. 'trusted-release,~trusted-metadata'", arg_description = 0x0}, {
            long_name = 0x5562f605b061 "json", short_name = 0 '\000', flags = 0, 
            arg = G_OPTION_ARG_NONE, arg_data = 0x5562f683e378, 
            description = 0x5562f605abd2 "Output in JSON format", arg_description = 0x0}, {
            long_name = 0x5562f605abe8 "no-security-fix", short_name = 0 '\000', flags = 0, 
            arg = G_OPTION_ARG_NONE, arg_data = 0x5562f683e364, 
            description = 0x5562f6059478 "Do not prompt to fix security issues", 
            arg_description = 0x0}, {long_name = 0x5562f605abf8 "no-authenticate", 
            short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7ffdcc2184c8, 
            description = 0x5562f6059090 "Don't prompt for authentication (less details may be shown)", arg_description = 0x0}, {long_name = 0x0, short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, 
            arg_data = 0x0, description = 0x0, arg_description = 0x0}}
        feature_flags = <optimized out>
```

hughsie, 2024-03-11:
Did you press enter after 'n'?

paulmenzel, 2024-03-11:
> Did you press enter after 'n'?

Oh, I can’t remember. I have to try later. That would be embarrassing.

PS: The Ctrl + c should have terminated the program, shouldn’t it?

hughsie, 2024-03-11:
> The Ctrl + c should have terminated the program, shouldn’t it?

We do set up a handler for SIGINT, maybe FuConsole should watch the GCancellable too.

hughsie, 2025-05-03:
@superm1 added support for readline :)

closed 2025-05-03 by hughsie

paulmenzel, 2025-05-05:
Awesome! That is merge/pull request https://github.com/fwupd/fwupd/pull/8659.