fwupd/fwupd #6911: EspLocation option isn't recognized

reliant8307, 2024-03-13:
**Describe the bug**

Since I don't use udisks, I prefer to use the EspLocation option, but it isn't being recognized.
```
$ mount|grep boot/efi
/dev/nvme0n1p2 on /boot/efi type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)

$ sudo cat /etc/fwupd/fwupd.conf 
[fwupd]
# use `man 5 fwupd.conf` for documentation
EspLocation=/boot/efi

$ sudo fwupdtool esp-list
Loading…                 [************************************** ]
No valid 'EspLocation' specified in /etc/fwupd/fwupd.conf
```

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
```
```
$ fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.15
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.15
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.15
runtime   org.kernel                    6.8.0

```
Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Source.

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

```
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Avell High Performance A75 RTX MUV/ G1575 RTX MUV
│
├─Unknown Device:
│     Device ID:          57e35fbedc32c4e5e69c55e3d7810f7ae6f1427a
│     Serial Number:      4K8X78AG078L
│     GUID:               7c8e1e1a-a54f-5461-81e1-62bb71a78896 ← DRM\VEN_DEL&DEV_A0EC
│   
├─Unknown Device:
│     Device ID:          d324166a5e48e27ba0360ac27eeaa89559930203
│     Serial Number:      4K8X78AG078L
│     GUID:               b03e58c9-c548-5309-9c50-f3d7338d2fc6 ← DRM\VEN_DEL&DEV_A0EA
│   
├─CoffeeLake-H GT2 [UHD Graphics 630]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    02
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              db677673-1102-5481-a11e-6f397e123ced ← PCI\VEN_8086&DEV_3E9B
[... 664 lines cut ...]
|       |-- videotest_checksum.mod
|       |-- wrmsr.mod
|       |-- xfs.mod
|       |-- xnu.mod
|       |-- xnu_uuid.mod
|       |-- xnu_uuid_test.mod
|       |-- xzio.mod
|       |-- zfs.mod
|       |-- zfscrypt.mod
|       |-- zfsinfo.mod
|       `-- zstd.mod
|-- vmlinuz
`-- vmlinuz.old

52 directories, 499 files
```

**Additional questions**

- Operating system and version: Linux 6.8
- Have you tried rebooting? Yes
- Is this a regression? No
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

hughsie, 2024-03-13:
> Since I don't use udisks, I prefer to use the EspLocation option, but it isn't being recognized

You need udisks now for the UEFI capsule plugin I'm afraid. The error message does look weird; @superm1?

superm1, 2024-03-13:
Yeah udisks is no longer optional. Please install it.

closed 2024-03-13 by superm1