fwupd/fwupd #6913: 1.9.15: meson fails because messy checking dependencies

kloczek, 2024-03-13:
**Describe the bug**
Looks like instead checking dependencies on meson layer new po/test-deps script has been introduced.
😞 
**Steps to Reproduce**
Just run meson without installed gobject-introspection.

**Expected behavior**
All  dependency checks shuld be done by meson.

**fwupd version information**
1.9.15

**Additional questions**
- Operating system and version: Linux x86/64
- Have you tried rebooting? N/A
- Is this a regression? N/A

Without installed gobject-introspection meson fails with enigmatic
```console
Configuring simple_client.py using configuration
Configuring fwupd.spec.in using configuration
Configuring 35_fwupd using configuration
Run-time dependency fwupd-efi found: YES 1.4

plugins/uefi-capsule/meson.build:123:2: ERROR: Command `/usr/bin/python3 /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.15/po/test-deps /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.15/po/LINGUAS` failed with status 1.

A full log can be found at /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.15/x86_64-redhat-linux-gnu/meson-logs/meson-log.txt
```

kloczek, 2024-03-13:
Just checked po/test-deps
```python
try:
    import gi
except ImportError:
    print("Error: missing dependency python gobject introspection (python3-gi)")
    err = 1
try:
    gi.require_version("Pango", "1.0")
    from gi.repository import Pango
except ValueError:
    print("Error: missing pango gobject introspection library")
    err = 1
try:
    gi.require_version("PangoCairo", "1.0")
    from gi.repository import PangoCairo
except ValueError:
    print("Error: missing pangocairo gobject introspection library")
    err = 1

try:
    gi.require_version("cairo", "1.0")
    from gi.repository import cairo
except ValueError:
    print("Error: missing cairo gobject introspection library")
    err = 1

try:
    import cairo
except ImportError:
    print("Error: missing dependency python cairo (python3-cairo)")
    err = 1
```
All above build requirements have proper pkgconfig interface so looks like that script can be removed and replaced by few `dependency ()` calls in main meson.build.

hughsie, 2024-03-13:
>  Looks like instead checking dependencies on meson layer new po/test-deps script has been introduced.

Yup, in 2017...

> so looks like that script can be removed and replaced by few dependency () calls in main meson.build

Patches certainly welcome.

PR #6936 (2024-03-22): trivial: show po/test-deps messages
Rather than saying it fails, show the message that po/test-deps emitted.

Fixes: #6913

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 14b60250 (2024-03-22): trivial: show po/test-deps messages

Rather than saying it fails, show the message that po/test-deps emitted.

Fixes: #6913
commit f4f74843 (2024-03-22): trivial: show po/test-deps messages

Rather than saying it fails, show the message that po/test-deps emitted.

Fixes: #6913
(cherry picked from commit 14b60250f3299c308b0fc82d4780c32ecbfe782f)
closed 2024-03-22 by superm1