fwupd/fwupd #6914: fwupd blocks fastboot flash

jm355, 2024-03-13:
**Describe the bug**
When fwupd is running, I can't use fastboot flash on my pixel 5

**Steps to Reproduce**
```
sudo systemctl start fwupd
fastboot flash <your image>
```

**Expected behavior**
fastboot should flash even when fwupd is running.

**fwupd version information**
fwupd version 1.9.15-1 on Arch linux

```shell
~ $ fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.15
compile   org.freedesktop.Passim        0.1.5
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.15
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.Passim        0.1.5
runtime   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.7.9-hardened1-1-hardened
runtime   org.freedesktop.fwupd         1.9.15
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Pacman

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?
Yes. According to https://grapheneos.org/install/cli#working-around-fwupd-bug-on-linux-distributions this should be fixed on fwupd 1.9.10 and later.

superm1, 2024-03-13:
This was supposed to be fixed. Can you please launch the daemon with `-vv` and then share the log from startup until you reproduce this issue?

jm355, 2024-03-13:
Is there any data I should redact? Skimming through I see it shows some info about my secureboot keys and such

hughsie, 2024-03-13:
> Is there any data I should redact

No, the log handler will remove any passwords automatically.

jm355, 2024-03-13:
And secureboot platformkey is safe to share? it has `FuDevice using <key> for pk` right under the issuer and subject of my platform key

hughsie, 2024-03-13:
> And secureboot platformkey is safe to share

Yes, but if you don't believe me just omit it, it's not interesting to debug this.

hughsie, 2024-03-13:
> and then share the log from startup until you reproduce this issue?

I don't see any fastboot activity at all. Is the fastboot USB device plugged in and in fastboot mode?

jm355, 2024-03-13:
I got the logs via `sudo /usr/lib/fwupd/fwupd -vv > fwupdlogs.txt 2>&1`, would you expect fwupd to try to connect to a fastboot device?

hughsie, 2024-03-13:
> would you expect fwupd to try to connect to a fastboot device?

No, but I was wondering if the DS-20 stuff is affecting anything, but it seems we don't touch that either:

	17:22:13.027 FuUsbDevice          ignoring: not available as bcdUSB 0x0200 <= 0x0200
	17:22:13.027 FuEngine             usb:06:00:01 added FuUsbDevice:
	  Guid:                 4980b2e3-f8f0-5434-8a41-031e35500953 <- USB\VID_18D1&PID_4EE0 ?
	  Flags:                none
	  VendorId:             USB:0x18D1
	  Version:              1.0
	  VersionFormat:        bcd
	  VersionRaw:           0x00000100
	  Guid[quirk]:          0357022b-3c79-5f77-aa9d-fc0ec3e8c934 <- USB\VID_18D1
	  Guid[quirk]:          4bd986bc-4aaf-5baf-8bc2-9f7928182387 <- USB\CLASS_FF
	  Guid[quirk]:          54f7aa0f-6450-5d0c-ac85-0302809a7f6b <- USB\CLASS_FF&SUBCLASS_42
	  Guid[quirk]:          e49bd1ea-2f0e-5709-a60c-63e72f607ba2 <- USB\CLASS_FF&SUBCLASS_42&PROT_03
	  PhysicalId:           usb:06:00:01
	  AcquiesceDelay:       2500
	  ParentPhysicalIds:    usb:06:00,usb:06
	  UsbDeviceClass:       interface-desc

hughsie, 2024-03-13:
I don't think fwupd is doing anything much to the device at all. Does it work if you stop fwupd? What's the fastboot connection error?

jm355, 2024-03-13:
It works when I stop fwupd, there is no error, it just sits without doing anything

hughsie, 2024-03-13:
Can you explain exactly how you're testing it now and tell me what error fastboot is giving you please.

hughsie, 2024-03-13:
@superm1 any ideas? @jm355 can you maybe try 1.9.10, 1.9.11, 1.9.12, 1.9.13, 1.9.14 and 1.9.15 to try and narrow it down?

jm355, 2024-03-13:
Testing it by booting into phone bootloader, disabling usbguard on my computer, connecting phone via usb, adding the latest version of fastboot to the path according to the instructions at https://grapheneos.org/install/cli#standalone-platform-tools (arch version is 34.0.4, version I'm using after this step is 35.0.0-11411520), and running the first line of flash-all.sh: `fastboot flash --slot=other bootloader bootloader-redfin-r3-0.6-10489834.img` (it's a test build for the pixel 5). When I run it, the phone screen increases brightness, but then nothing happens. Fastboot doesn't output anything and just does nothing indefinitely (when I first did it I waited 40 minutes cause I was doing other stuff and forgot about it). After a bit the phone brightness goes down again.

jm355, 2024-03-13:
It works as expected when I downgraded fwupd to 1.9.14-1, restarted the service, and unplugged and replugged in the phone

superm1, 2024-03-13:
That's quite bizarre. Do you think you can bisect from 1.9.14 to 1.9.15 then?

jm355, 2024-03-13:
I can try it out, just stop the fwupd service, build according to https://github.com/fwupd/fwupd/blob/main/docs/building.md, and run the built binary right? Is there an easy way to quickly go to the commit halfway between 1.9.14 and 1.9.15?

superm1, 2024-03-13:
> I can try it out, just stop the fwupd service, build according to https://github.com/fwupd/fwupd/blob/main/docs/building.md, and run the built binary right? Is there an easy way to quickly go to the commit halfway between 1.9.14 and 1.9.15?

Yeah if you do it in the fwupd development environment you can easily execute the binary that gets installed into `venv/dist`.

Just like a kernel bisect you would basically run this (rough) series of commands:

```
git bisect start --term-new=broken --term-old=fixed
git bisect fixed 1.9.14
git bisect broken 1.9.15
```

Then it will start in the middle and you mark it fixed/broken and keep going until you have the result.

jm355, 2024-03-13:
hmm. Well this is very weird, now it's working even with regular fwupd. I started the bisect and it didn't work but then I realized I hadn't rebuilt for the next one when it didn't work again, but then I unplugged and replugged my phone and it did work. So then I aborted the bisect and tried it on main and it worked on unplug/replug, so then I tried it with plain fwupd and it worked. I'm going to try rebooting and see if I can get it to work from a "clean state"

jm355, 2024-03-13:
hmm, yeah looks like it's working now. I have no idea why it wasn't working, I wonder if I plugged it in before disabling usbguard or something :thinking: Sorry for the false alarm

closed 2024-03-13 by jm355