fwupd/fwupd #6920: WD19TB - upgrade on macos

vcctm, 2024-03-14:
**Describe the bug**
So trying to upgrade the WD19TB firmware via macos happen's this on the log's for all the:

`sudo fwupdtool get-devices`

My log stucks specifically after detect the dock.

```
UsbDeviceClass:       interface-desc

  

10:37:26.812 FuPlugin             backend_device_added(dell_dock)

10:37:26.812 FuDevice             using 86ab8c54564f4086b9d82e9826f80917aa0f4a85 for usb:03:02:03:05:hub

10:37:26.814 FuHidDevice          HID::SetReport [wValue=0x0200, wIndex=0]:

       │ 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 

───────┼────────────────────────────────────────────────────────────────────────────────────────────────

0x0000 │ c0 09 00 00 00 00 0c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

0x0020 │ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

0x0040 │ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

0x0060 │ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

0x0080 │ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

0x00a0 │ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

10:37:26.814 FuConfig             /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf changed, reloading all configs

10:37:26.814 FuConfig             skipping mode check for /opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf as not writable

10:37:26.814 FuConfig             trying to load config values from /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf

10:37:26.814 FuCommon             mapped file /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf of size 0x33

10:37:26.814 FuConfig             trying to load config values from /opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf

10:37:26.814 FuConfig             Failed to open file “/opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf”: No such file or directory

10:37:26.814 FuConfig             ::configuration changed

10:37:26.814 FuIdle               setting timeout to 300s

10:37:26.814 FuConfig             /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf changed, reloading all configs

10:37:26.814 FuConfig             skipping mode check for /opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf as not writable

10:37:26.814 FuConfig             trying to load config values from /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf

10:37:26.814 FuCommon             mapped file /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf of size 0x33

10:37:26.814 FuConfig             trying to load config values from /opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf

10:37:26.814 FuConfig             Failed to open file “/opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf”: No such file or directory

10:37:26.814 FuConfig             ::configuration changed

10:37:26.814 FuIdle               setting timeout to 300s

10:37:26.814 FuConfig             /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf changed, reloading all configs

```

hughsie, 2024-03-14:
Do you know why the `::configuration changed` events keep happening? Also, can you give us the full log of `sudo sudo fwupdtool get-devices -vv` also -- thanks.

hughsie, 2024-03-15:
I'm also trying to install Ventura on my super old macBook -- if that actually works I'll try to reproduce the problem myself.

closed 2024-03-15 by hughsie

FabioEight, 2024-03-19:
I have the same issue on macOS and WD22TB4:
```
10:08:00.038 FuConfig             Failed to open file “/opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf”: No such file or directory
10:08:00.038 FuConfig             ::configuration changed
10:08:00.038 FuConfig             /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf changed, reloading all configs
10:08:00.038 FuConfig             skipping mode check for /opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf as not writable
10:08:00.038 FuConfig             trying to load config values from /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf
10:08:00.038 FuCommon             mapped file /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf of size 0x33
10:08:00.038 FuConfig             trying to load config values from /opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf
10:08:00.038 FuConfig             Failed to open file “/opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf”: No such file or directory
10:08:00.038 FuConfig             ::configuration changed
10:08:00.038 FuConfig             /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf changed, reloading all configs
10:08:00.038 FuConfig             skipping mode check for /opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf as not writable
10:08:00.038 FuConfig             trying to load config values from /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf
10:08:00.038 FuCommon             mapped file /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf of size 0x33
10:08:00.038 FuConfig             trying to load config values from /opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf
10:08:00.038 FuConfig             Failed to open file “/opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf”: No such file or directory
10:08:00.038 FuConfig             ::configuration changed
10:08:00.038 FuConfig             /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf changed, reloading all configs
10:08:00.038 FuConfig             skipping mode check for /opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf as not writable
10:08:00.038 FuConfig             trying to load config values from /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf
10:08:00.038 FuCommon             mapped file /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf of size 0x33
10:08:00.038 FuConfig             trying to load config values from /opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf
10:08:00.038 FuConfig             Failed to open file “/opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf”: No such file or directory
10:08:00.038 FuConfig             ::configuration changed
10:08:00.038 FuConfig             /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf changed, reloading all configs
10:08:00.038 FuConfig             skipping mode check for /opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf as not writable
10:08:00.038 FuConfig             trying to load config values from /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf
10:08:00.038 FuCommon             mapped file /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf of size 0x33
10:08:00.038 FuConfig             trying to load config values from /opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf
10:08:00.038 FuConfig             Failed to open file “/opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf”: No such file or directory
10:08:00.038 FuConfig             ::configuration changed
10:08:00.038 FuConfig             /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf changed, reloading all configs
10:08:00.038 FuConfig             skipping mode check for /opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf as not writable
10:08:00.038 FuConfig             trying to load config values from /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf
10:08:00.039 FuCommon             mapped file /opt/homebrew/Cellar/fwupd/1.9.15/etc/fwupd/fwupd.conf of size 0x33
10:08:00.039 FuConfig             trying to load config values from /opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf
10:08:00.039 FuConfig             Failed to open file “/opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf”: No such file or directory
10:08:00.039 FuConfig             ::configuration changed
```
I'm using fwupd installed via brew

FabioEight, 2024-03-19:
@hughsie I see the commits for this bugfix, is there a compiled binary to test on macOS? Thank you!

FabioEight, 2024-03-19:
As workaround, I removed "/opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf" completely.

vcctm, 2024-03-19:
> As workaround, I removed "/opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf" completely.

It works after this ? Will try here too, my problem was the fan noise of WTB, was turning on and never turning down, since I create this issue I removed the fan! Now is quiet, will try to update!

FabioEight, 2024-03-19:
> > As workaround, I removed "/opt/homebrew/Cellar/fwupd/1.9.15/var/etc/fwupd/fwupd.conf" completely.
> 
> It works after this ? Will try here too, my problem was the fan noise of WTB, was turning on and never turning down, since I create this issue I removed the fan! Now is quiet, will try to update!

I just raised #6941 with the latests!