fwupd/fwupd #6925: fwupdmgr security reports disabled SPI write protection

lunarlettuce, 2024-03-15:
**Describe the bug**
fwdupmgr security reports that SPI write protection is disabled, when it should be enabled. Additionally, CET OS support is marked as "Not supported".

**Steps to Reproduce**
Run `fwupdmgr security`

**Expected behavior**
It is expected that SPI write protection is enabled.

**fwupd version information**

```compile   org.freedesktop.fwupd         1.9.15
compile   com.hughsie.libxmlb           0.3.15
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.15
runtime   org.kernel                    6.7.9-200.fc39.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Fedora Silverblue Flatpak Repository
<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```LENOVO 20XF004RUS
│
├─Lenovo USB-C Mini Dock:
│ │   Device ID:          da77984c82b59c6fc69516431f467fd9a8d39a7f
│ │   Summary:            USB 3.x hub
│ │   Current version:    4.154
│ │   Vendor:             VIA Labs, Inc. (USB:0x17EF)
│ │   Install Duration:   15 seconds
│ │   GUIDs:              fd4b20d3-2612-5743-ad85-5c3065361c51
│ │                       f281c1df-c3d5-5f8a-984d-e9548ffc95fe ← USB\VID_17EF&PID_3094
│ │                       ce8b3f6c-9ddd-5d50-b3f8-e87e72d2aacc ← USB\VID_17EF&PID_3094&HUB_0012
│ │                       e62c5403-daa6-5482-9e9e-74666884ce43 ← USB\VID_17EF&PID_3094&SPI_C223
│ │                       75b11f2d-86b6-5ecc-912e-a2a649f334d5 ← USB\VID_17EF&PID_3094&SPI_C223&REV_04F4
│ │   Device Flags:       • Updatable
│ │                       • Cryptographic hash verification is available
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Unsigned Payload
│ │ 
│ ├─Lenovo USB-C Mini Dock:
│ │ │   Device ID:        983c3cffc6fd36d32b00b62928d30721eaeb93db
│ │ │   Summary:          USB 3.x hub
│ │ │   Current version:  4.154
│ │ │   Vendor:           VIA Labs, Inc. (USB:0x17EF)
│ │ │   Install Duration: 15 seconds
│ │ │   GUIDs:            fd4b20d3-2612-5743-ad85-5c3065361c51 ← USB\VID_17EF&PID_3095
│ │ │                     2b337b4f-fc17-520d-8d93-095a9bfd6ba8 ← USB\VID_17EF&PID_3095&HUB_32
│ │ │                     152db1ae-acd6-5b6d-aad2-178ec2af5199 ← USB\VID_17EF&PID_3095&SPI_C223
│ │ │                     8ce1ac09-39f9-51a5-9468-74433dfa575f ← USB\VID_17EF&PID_3095&SPI_C223&REV_04F4
│ │ │   Device Flags:     • Updatable
│ │ │                     • Cryptographic hash verification is available
│ │ │                     • Device stages updates
│ │ │                     • Device can recover flash failures
│ │ │                     • Unsigned Payload
│ │ │ 
│ │ ├─Lenovo USB-C Mini Dock:
│ │ │     Device ID:      d0950b8556ed65b4b8e8bfa3809fdb849005f298
│ │ │     Summary:        USB 3.x hub
│ │ │     Current version:4.93
│ │ │     Vendor:         VIA Labs, Inc. (USB:0x17EF)
[... 259 lines cut ...]
      GUIDs:              fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2 ← NVME\VEN_15B7&DEV_5006
                          12c86995-0b90-5ec5-98f3-7a6ed4ca50e0 ← NVME\VEN_15B7&DEV_5006&SUBSYS_15B75006
                          06b4e2aa-91af-508b-b06e-65e3b3189e97 ← WDC PC SN730 SDBQNTY-512G-1001
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
────────────────────────────────────────────────
Devices that have been updated successfully:
 • System Firmware (0.1.27 → 0.1.28)
 • UEFI dbx (371 → 371)
Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.

```

</details>

**Additional questions**

- Operating system and version: Fedora Silverblue 39
- Have you tried rebooting? Yes, and I have tried resetting BIOS to defaults, and resetting secure boot keys to defaults.
- Is this a regression? Unsure if this is an issue with the laptop, or with fwupd.

lunarlettuce, 2024-03-15:
fwupd security report:

```Host Security ID: HSI:1! (v1.9.15)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ Fused platform:                Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled

HSI-2
✔ BIOS rollback protection:      Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid
✘ SPI write protection:          Disabled

HSI-3
✔ SPI replay protection:         Enabled
✔ CET Platform:                  Supported
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✔ Processor rollback protection: Enabled
✔ Encrypted RAM:                 Encrypted
✔ SMAP:                          Enabled

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ Linux kernel:                  Untainted
✘ CET OS Support:                Not supported

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix
```

hughsie, 2024-03-15:
What hardware is that? Both failures look legitimate to me..

lunarlettuce, 2024-03-15:
> What hardware is that? Both failures look legitimate to me..

It is a Thinkpad T14s Gen 2 (AMD), with model code 20XF004RUS.

hughsie, 2024-03-18:
@lunarlettuce can you attach us the full `sudo fwupdtool security -vv` output please.

lunarlettuce, 2024-03-18:
@hughsie 
[output.txt](https://github.com/fwupd/fwupd/files/14642141/output.txt)

superm1, 2024-03-18:
> 20:48:54.761 FuPluginPciPsp       ROM armor not enforced
Looks like fwupd isn't doing anything wrong.  [fwupd](https://github.com/fwupd/fwupd/blob/main/plugins/pci-psp/fu-pci-psp-device.c#L285
) is a messenger for the [kernel](https://github.com/torvalds/linux/blob/bf3a69c6861ff4dc7892d895c87074af7bc1c400/Documentation/ABI/testing/sysfs-driver-ccp#L78) which is a messenger for what the hardware reports.

Looks to me that this particular security feature is not enabled.