fwupd/fwupd #6926: About how to use Computer Hardware ID (CHID) restriction

vincent-tang1, 2024-03-15:
**Precondition**
1. host OS Ubuntu 23.10
2. fwupd using Redfish plugin
3. Using snap to install fwupd

**Describe the question**
When CAB file is uploaded from LVFS WebUI, we can add CHID restriction by modified metadata online as below
[image]

in our host os, can see system UUID from SMBIOS tabe as
```
# dmidecode --type system
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.5.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: Advantech
        Product Name: FWA-6172
        Version: A101-1
        Serial Number: AKA1234567
        UUID: 894feb65-2ca8-4378-a5d3-41c78ff5b59f
        Wake-up Type: Power Switch
        SKU Number: FWA-6172
        Family: Server

```
About correct CHID value usage, not sure if should fill this system UUID or other ? Thanks

**fwupd version information**
```
# fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.6
compile   io.snapcraft.fwupd            5749
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.9.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.5.0-10-generic
runtime   org.freedesktop.fwupd         1.9.14

hughsie, 2024-03-15:
> About correct CHID value usage, not sure if should fill this system UUID or other

The CHIDs are generated the same was as "ComputerHardwareIds.exe" on Windows:

	$ sudo fwupdtool hwids
	Computer Information
	--------------------
	BiosVendor: LENOVO
	BiosVersion: N2VET45W (1.30 )
	BiosMajorRelease: 1
	BiosMinorRelease: 30
	FirmwareMajorRelease: 01
	FirmwareMinorRelease: 0b
	Manufacturer: LENOVO
	Family: ThinkPad P1 Gen 3
	ProductName: 20TJS2F44A
	ProductSku: LENOVO_MT_20TJ_BU_Think_FM_ThinkPad P1 Gen 3
	EnclosureKind: a
	BaseboardManufacturer: LENOVO
	BaseboardProduct: 20TJS2F44A
	Hardware IDs
	------------
	{687b4b40-14cf-5a9e-b9a2-84d93a5f99c9}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
	{d6dffdd4-f8c5-56b2-8a54-29d4554a81b7}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
	{2f6d9243-39b3-561b-acb5-33c664ac656e}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
	{bf5fea93-df40-5336-922f-a2fa3ae67e5f}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
	{5ac99727-ff1a-5d8a-aef4-8a03d8f5cfe3}   <- Manufacturer + Family + ProductName + ProductSku
	{742d6de4-82fc-56b8-8079-963759c68dea}   <- Manufacturer + Family + ProductName
	{ff4db33f-0e86-52e0-b531-5ecf58c194c1}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
	{3969d15b-9209-5350-9f13-966ee621eb4b}   <- Manufacturer + ProductSku
	{a4d618e0-6988-5177-800c-64405dbbcefa}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
	{2dc64c9e-97c4-59bf-b0be-892546d7e655}   <- Manufacturer + ProductName
	{536c0c4c-e739-5bc7-8f93-042c73d961ef}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
	{3667f735-639c-5ca5-9ae7-a464d4b49e27}   <- Manufacturer + Family
	{e093d715-70f7-51f4-b6c8-b4a7e31def85}   <- Manufacturer + EnclosureKind
	{eb8b729b-355c-5144-b23c-90f24e542a8c}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
	{6de5d951-d755-576b-bd09-c5cf66b27234}   <- Manufacturer

Does that help?

vincent-tang1, 2024-03-15:
very helpful information and the clarification, thank you!

vincent-tang1, 2024-03-15:
just double confirm if my understanding is correct.
About the CHID restriction in metadata, following the windows format standard,
we only can restrict a group computers, but we can not limit only a computer to be used, right ?

hughsie, 2024-03-15:
> we only can restrict a group computers, but we can not limit only a computer which can be used, right

Correct, there's no serial number in any of the CHIDs.