fwupd/fwupd #6937: About unique computer GUID restriction

vincent-tang1, 2024-03-18:
**Is your feature request related to a problem? Please describe.**
Considering a scenario, in server product development stage or after MP, we might provide a testing firmware to customer by uploading to LVFS testing remote such that customer can easily install it. However, we do not want to let other people to be also able to install this testing firmware. So, adding a unquie computer GUID limitation might be a choose to achieve it.  

**Describe the solution you'd like**
Not sure it is appropriate that we can limit a unquie computer GUID to access. (allow OR operation)

Thanks

hughsie, 2024-03-18:
I don't think this is something we want to support; having that component in the metadata downloaded over a million times per day for a single machine seems a huge amount of overhead and waste. The testing firmware can be uploaded into the private remote and then you can give the customer the URL to the cab file to be installed manually.

vincent-tang1, 2024-03-18:
Got it, aggree your consideration and many thinks for your suggestion.

closed 2024-03-18 by hughsie