fwupd/fwupd #6940: Setting ProtectClock blocks access to /dev/cpu/0/msr

RZR7332, 2024-03-18:
**Describe the question**
I am testing the systemd hardening options as I go and discovered today that setting ProtectClock=true prints out the below in the journald output:

11:00:30:0706 FuEngine             failed to add device /sys/devices/virtual/msr/msr0: failed to add device using on msr: failed to open /dev/cpu/0/msr: Operation not permitted

I am not convinced there is any actual impact here, but would like to verify or get some input as to this message. I toggled settings on and off until I found the culprit.

This was observed on an Ubuntu 22.04 LTS virtual machine; I replicated exactly the same configuration on a Pi 2 (same OS, same fwupd version) and ProtectClock did not generate such a message when enabled.

It is also perhaps worth noting that the version bundled with Ubuntu 22.04 LTS is quite old, so this may also be a factor.

Something to be concerned about? Cosmetic? 

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.

```shell
This program may only work correctly as root
fwupd version: 1.7.9
```

superm1, 2024-03-18:
I would say if that message is coming up it caused a problem and that specific one should be reverted.

RZR7332, 2024-03-18:
I would agree with you in principle, but the fact that I cannot replicate it on a physical device (albeit different architecture) and the fact that the setting prevents write access to the hardware/software clock bothers me. Turning this setting on should not affect anything.

From the man page:

Takes a boolean argument. If set, writes to the hardware clock or system clock will be denied. Defaults to off. Enabling this option removes CAP_SYS_TIME and CAP_WAKE_ALARM from the capability bounding set for this unit, installs a system call filter to block calls that can set the clock, and DeviceAllow=char-rtc r is implied. Note that the system calls are blocked altogether, the filter does not take into account that some of the calls can be used to read the clock state with some parameter combinations. Effectively, /dev/rtc0, /dev/rtc1, etc. are made read-only to the service. See [systemd.resource-control(5)](https://www.freedesktop.org/software/systemd/man/latest/systemd.resource-control.html#) for the details about DeviceAllow=.

I am uncertain if an explicit DeviceAllow=/dev/cpu/ would work. Figured I would check in first with the people who know the software best 😄 

I would rather attempt to replicate this first in a completely fresh Ubuntu VM without any special config before reverting the setting.

superm1, 2024-03-18:
I think it's worth experimenting with a daily image like Fedora 40 or Ubuntu 24.04 with this setting in place vs not in you VM.

Then it will have a more up to date systemd, more up to date fwupd etc.

RZR7332, 2024-03-18:
I tested now quickly on a Fedora 40 VM with ProtectClock enabled, no issue. I will set aside time in the coming days to build a 24.04 VM to test with, but I do suspect it is either system or version specific. Thanks for the idea!

RZR7332, 2024-03-26:
Closing as I cannot replicate this on the latest daily build of 24.04 with a newer version of fwupd (1.9.14), nor on Fedora 39/40.

closed 2024-03-26 by RZR7332