fwupd/fwupd #6941: Dell WD22TB4 upgrade issue

FabioEight, 2024-03-19:
**Describe the bug**
Failed to lock: another instance has locked /var/run/fwupdtool

**Steps to Reproduce**
fwupdtool refresh
fwupdtool update

**Expected behavior**
WD22TB4 docking firmware upgrade.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
sudo fwupdtool --version
Loading…                 [************************************** ]
runtime   org.freedesktop.fwupd         1.9.15
compile   com.hughsie.libxmlb           0.3.15
runtime   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    23.4.0
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.15
runtime   org.freedesktop.gusb          0.4.8
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): Brew

Host is macOS Macbook Pro with Sonoma.

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the external fwupd devices recognized in your system.

```shell
Apple Mac14,10
│
└─WD22TB4:
  │   Device ID:          b3db560e794a8ceb981cf5118b5818ac75d0eafd
  │   Summary:            High performance dock
  │   Current version:    01.01.00.07
  │   Vendor:             Dell Inc. (USB:0x413C)
  │   Install Duration:   1 minute
  │   Serial Number:      FQRVGT3/3168144382052025
  │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
  │   Device Flags:       • Updatable
  │                       • Supported on remote server
  │                       • Device stages updates
  │                       • Device can recover flash failures
  │                       • Device is usable for the duration of the update
  │                       • Signed Payload
  │
  ├─Package level of Dell dock:
  │     Device ID:        edff9adc305e2d3327882bb26704af735e74d704
  │     Summary:          A representation of dock update status
  │     Current version:  01.00.36.00
  │     Vendor:           Dell Inc. (USB:0x413C)
  │     Install Duration: 5 seconds
  │     GUIDs:            af2a3b39-98c4-59a9-9cbe-a6909a7d14a2
  │                       7fb217aa-2312-5c99-a853-56d865c76815 ← USB\VID_413C&PID_B06E&hub&salomon_mlk_status
  │     Device Flags:     • Updatable
  │                       • Supported on remote server
  │                       • Device can recover flash failures
  │                       • Device is usable for the duration of the update
  │                       • Unsigned Payload
  │
  ├─RTS5413 in Dell dock:
  │     Device ID:        0bcab2ad4789ecbcca1bf84409a79681c8c700aa
  │     Summary:          USB 3.1 Generation 1 Hub
  │     Current version:  01.22
  │     Vendor:           Dell Inc. (USB:0x413C)
  │     Install Duration: 14 seconds
  │     GUIDs:            b27d25f1-019d-5718-b41a-02ddaefe5577 ← USB\VID_413C&PID_B06F
  │                       ac5b774c-b49d-566b-9255-85f0f7f8a4ed ← USB\VID_413C&PID_B06F&hub
  │     Device Flags:     • Updatable
  │                       • Supported on remote server
  │                       • Device stages updates
  │                       • Device is usable for the duration of the update
  │                       • Signed Payload
  │
  ├─RTS5487 in Dell dock:
  │     Device ID:        4c2aca590d34d7122d000b0ed87e4b79750f255d
  │     Summary:          USB 3.1 Generation 2 Hub
  │     Current version:  01.57
  │     Vendor:           Dell Inc. (USB:0x413C)
  │     Install Duration: 3 seconds
  │     GUIDs:            acfcd89b-105d-55b9-b85b-08bf8508f38c ← USB\VID_413C&PID_B06E
  │                       568ffa1e-a0db-5287-9ea3-872b60f7730b ← USB\VID_413C&PID_B06E&hub
  │     Device Flags:     • Updatable
  │                       • Supported on remote server
  │                       • Device stages updates
  │                       • Device is usable for the duration of the update
  │                       • Signed Payload
  │
  ├─USB4 controller in Dell dock:
  │     Device ID:        71413822e87deae707da948bd4b7e537af463c19
  │     Summary:          USB4 controller
  │     Current version:  38.0
  │     Vendor:           Dell Inc. (USB:0x8087, TBT:0x00D4)
  │     Install Duration: 46 seconds
  │     GUID:             bd79ce60-525b-5f39-a3f6-c98c495039ff ← TBT-00d4b071
  │     Device Flags:     • Updatable
  │                       • Supported on remote server
  │                       • Device stages updates
  │                       • Device is usable for the duration of the update
  │                       • Signed Payload
  │
  └─VMM5331 in Dell dock:
        Device ID:        680e17701c22f091054a2445bb8946f72645d88d
        Summary:          Multi Stream Transport controller
        Current version:  05.07.04
        Vendor:           Dell Inc. (USB:0x413C)
        Install Duration: 6 minutes
        GUID:             89fec0b6-6b76-5008-b82c-5e5c6c164007 ← MST-panamera-vmm5331-259
        Device Flags:     • Updatable
                          • Supported on remote server
                          • Device stages updates
                          • Device is usable for the duration of the update
                          • Unsigned Payload
```

</details>

**Dock SKU**
Please mention which module is installed in your WD22.

- [ ] WD19 (Single-C)
- [X ] WD22TB4 (Thunderbolt)
- [ ] WD19DC (Dual-C)

**Peripherals connected to the dock**
Please describe all devices connected to the dock.  Be as specific as possible,
including USB devices, hubs, monitors, and downstream type-C devices.

2x USB hubs on Dell monitors
1x Ethernet RJ45
1x USB-C Monitor
1x Displayport Monitor
1x USB Headset

- Operating system and version: macOS Sonoma
- Have you tried unplugging the dock or any peripherals from your machine? Y
- Have you tried to power cycle the dock from the AC adapter? Y
- Is this a regression? N

vcctm, 2024-03-19:
Trying to remove the .conf file din't work, will keep posted once we have some fix!

hughsie, 2024-03-19:
This is fixed in https://github.com/fwupd/fwupd/commit/e835714257f5ddd0f67f18c04b4ebffb67679690 -- you probably need to build from --head until the next stable tarball I'm afraid.

vcctm, 2024-03-19:
> This is fixed in [e835714](https://github.com/fwupd/fwupd/commit/e835714257f5ddd0f67f18c04b4ebffb67679690) -- you probably need to build from --head until the next stable tarball I'm afraid.

Can you guide how do that on mac ? Sorry if you guys have some post already can you post here ?

superm1, 2024-03-19:
> This is fixed in https://github.com/fwupd/fwupd/commit/e835714257f5ddd0f67f18c04b4ebffb67679690 -- you probably need to build from --head until the next stable tarball I'm afraid.

Any chance we can capture artifacts for MacOS for people to test like we do for other builds?

FabioEight, 2024-03-19:
@hughsie I'm trying to follow the build process but it's failing early:
```

➜  fwupd git:(main) ./contrib/setup
Using OS profile darwin to setup
Install build dependencies? (y/N) y
Usage: brew install [options] formula|cask [...]

Install a formula or cask. Additional options specific to a formula may be
appended to the command.

Unless HOMEBREW_NO_INSTALLED_DEPENDENTS_CHECK is set, brew upgrade or brew
reinstall will be run for outdated dependents and dependents with broken
linkage, respectively.

Unless HOMEBREW_NO_INSTALL_CLEANUP is set, brew cleanup will then be run for
the installed formulae or, every 30 days, for all formulae.

Unless HOMEBREW_NO_INSTALL_UPGRADE is set, brew install formula will
upgrade formula if it is already installed but outdated.
....
  -h, --help                       Show this message.

Error: Invalid usage: This command requires at least 1 formula or cask argument.
Traceback (most recent call last):
  File "/Users/fcaminotto/Downloads/fwupd/./contrib/ci/fwupd_setup_helpers.py", line 265, in <module>
    install_packages(
  File "/Users/fcaminotto/Downloads/fwupd/./contrib/ci/fwupd_setup_helpers.py", line 204, in install_packages
    subprocess.check_call(installer)
  File "/opt/homebrew/Cellar/python@3.12/3.12.2_1/Frameworks/Python.framework/Versions/3.12/lib/python3.12/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['brew', 'install']' returned non-zero exit status 1.

```

hughsie, 2024-03-19:
Can you try again pls? I've made the deps arch-neutral now. I only have a very old macbook for testing.

FabioEight, 2024-03-19:
Working! I'm now struggling in building fwupd due to: 

```
(fwupd) ➜  fwupd git:(main) build-fwupd
ninja: Entering directory `/Users/user/Downloads/fwupd/venv/bin/../build'
ninja: error: loading 'build.ninja': No such file or directory
```
I tried to pip reinstall ninja but still nothing, digging..

hughsie, 2024-03-19:
Maybe try `rm -rf /Users/user/Downloads/fwupd/venv` then try again with `./contrib/setup`

FabioEight, 2024-03-19:
I removed the build folder after reinstalling ninja and it did the trick!

```
(fwupd) ➜  fwupd git:(main) sudo fwupdtool update
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Package level of Dell dock from 01.00.36.00 to 01.00.36.01?          ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ -Update the firmware package version of WD19/WD22 Dell docking station's     ║
║ composite devices to 01.00.36                                                ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Decompressing…           [*********************************      ]
Waiting…                 [***************************************]
The update will continue when the device USB cable has been unplugged.
The update will continue when the device USB cable has been unplugged.
The update will continue when the device USB cable has been unplugged.
The update will continue when the device USB cable has been unplugged.
Devices with the latest available firmware version:
 • RTS5413 in Dell dock
 • RTS5487 in Dell dock
 • Embedded Controller
 • VMM5331 in Dell dock
 • WD22TB4
(fwupd) ➜  fwupd git:(main)
```
Docking updated! @vcctm 
Thank you @hughsie @superm1

closed 2024-03-19 by superm1