fwupd/fwupd #6942: @hughsie I'm trying to follow the build process but it's failing early:

superm1, 2024-03-19:
@hughsie I'm trying to follow the build process but it's failing early:
```

➜  fwupd git:(main) ./contrib/setup
Using OS profile darwin to setup
Install build dependencies? (y/N) y
Usage: brew install [options] formula|cask [...]

Install a formula or cask. Additional options specific to a formula may be
appended to the command.

Unless HOMEBREW_NO_INSTALLED_DEPENDENTS_CHECK is set, brew upgrade or brew
reinstall will be run for outdated dependents and dependents with broken
linkage, respectively.

Unless HOMEBREW_NO_INSTALL_CLEANUP is set, brew cleanup will then be run for
the installed formulae or, every 30 days, for all formulae.

Unless HOMEBREW_NO_INSTALL_UPGRADE is set, brew install formula will
upgrade formula if it is already installed but outdated.
....
  -h, --help                       Show this message.

Error: Invalid usage: This command requires at least 1 formula or cask argument.
Traceback (most recent call last):
  File "/Users/fcaminotto/Downloads/fwupd/./contrib/ci/fwupd_setup_helpers.py", line 265, in <module>
    install_packages(
  File "/Users/fcaminotto/Downloads/fwupd/./contrib/ci/fwupd_setup_helpers.py", line 204, in install_packages
    subprocess.check_call(installer)
  File "/opt/homebrew/Cellar/python@3.12/3.12.2_1/Frameworks/Python.framework/Versions/3.12/lib/python3.12/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['brew', 'install']' returned non-zero exit status 1.

```

_Originally posted by @FabioEight in https://github.com/fwupd/fwupd/issues/6941#issuecomment-2006980677_

hughsie, 2024-03-19:
I think it's probably because I added the macOS deps as `package variant="x86_64"` which is only true for non ARM64 macbooks. I don't know what the variant is there, maybe `aarch64`? `arm64`? We could also maybe remove the variant completely and remove the clause in that case?

superm1, 2024-03-19:
> I think it's probably because I added the macOS deps as `package variant="x86_64"` which is only true for non ARM64 macbooks. I don't know what the variant is there, maybe `aarch64`? `arm64`? We could also maybe remove the variant completely and remove the clause in that case?

Yeah unless the deps are actually going to change in different architectures I'd say just no variant part.

closed 2024-03-19 by hughsie