fwupd/fwupd #6946: Cannot call fwupdmgr from within a classic snap

pieqq, 2024-03-20:
**Describe the bug**

On Ubuntu distributions, `fwupd` is delivered as a Debian package:

```
$ dpkg -S fwupdmgr
fwupd: /usr/bin/fwupdmgr
(...)
```

Classic snaps are allowed to access programs from the system. So it should be possible to access `fwupdmgr` from within a classic snap.

However, it is not the case, and the following error is returned:

```
$ fwupdmgr get-devices
Mismatched daemon and client, use fwupd.fwupdmgr instead
```

This is, I believe, because of the way fwupd [returns the associated systemd unit](https://github.com/fwupd/fwupd/blob/main/src/fu-tool.c#L217), based on the existence of the `$SNAP` env variable:

```c
fu_util_get_systemd_unit(void)
{
	if (g_getenv("SNAP") != NULL)
		return SYSTEMD_SNAP_FWUPD_UNIT;
	return SYSTEMD_FWUPD_UNIT;
}
```

My understanding is that this is to check if `fwupd` is run from a snap. Maybe a better way would be to also check if `SNAP_NAME` equals to `fwupd` before assuming the [systemd unit should be `snap.fwupd.fwupd.service`.
](https://github.com/fwupd/fwupd/blob/main/src/fu-tool.c#L52)

**Steps to Reproduce**

1. Install any classic snap (for instance the [Kate text editor](https://kate-editor.org/)):

```
$ sudo snap install kate --classic
```

2. Try to call `fwupdmgr` from within the snap:

```
$ snap run --shell kate
$ echo $SNAP
/snap/kate/171
$ echo $SNAP_NAME
kate
$ fwupdmgr get-devices
Mismatched daemon and client, use fwupd.fwupdmgr instead
```

A workaround is to unset the `SNAP` variable before calling `fwupdmgr`:

```
$ unset SNAP
$ fwupdmgr get-devices    # Command now works!
Dell Inc. Precision 3570
│
├─12th Gen Intel Core™ i7-1265U:
(...)
```

**Expected behavior**

It should be possible to call `fwupdmgr` from within a classic snap without having to unset the `SNAP` env var.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.5
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.9.5
runtime   org.kernel                    6.5.0-25-generic
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Installed by default with Ubuntu (in my case, Ubuntu 23.10)

**Additional questions**

- Operating system and version: Ubuntu 23.10
- Have you tried rebooting? N/A
- Is this a regression? Probably not :)

superm1, 2024-03-20:
Would you mind submitting a pull request with your suggested change?

It sounds sensible to me, as long as it keeps working for the confined snap too.

pieqq, 2024-03-20:
@superm1 Done! Please check #6947.

closed 2024-03-20 by superm1

superm1, 2024-03-20:
for the record, your "original" reported issue with `fwupdmgr` is already fixed in both [main](https://github.com/fwupd/fwupd/commit/b54d06f9fe361bfc41c6f7564a7c2b7e356016ac) and [1_9_X](https://github.com/fwupd/fwupd/commit/d70cd3ea174752a4962df1ccc942fb1ceb0843fc ).  It's in 1.9.15.

The fix you submitted is for `fwupdtool` to DTRT.  I'll backport that to 1_9_X as well.

PR #6947 (2024-03-20): Allow classic snaps to call fwupd
The systemd unit is set to the snap version (snap.fwupd.fwupd.service) if the name of the snap is fwupd. This allows classic snaps to call a version of fwupd installed from classic package repositories (deb or rpm).

Fix #6946

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [X] Code fix
- [ ] Feature
- [ ] Documentation

Tested by building fwupd, then:

Running the deamon in a terminal:

```shell
$ /home/pieq/dev/work/fwupd/venv/bin/fwupd
```

while running `fwupdmgr`  [... cut]
commit a5c2e641 (2024-03-20): Allow classic snaps to call fwupd

The systemd unit is set to the snap version (snap.fwupd.fwupd.service)
if the name of the snap is fwupd. This allows classic snaps to call a
version of fwupd installed from classic package repositories (deb or
rpm).

Fix #6946
commit 0b139633 (2024-03-20): Allow classic snaps to call fwupd

The systemd unit is set to the snap version (snap.fwupd.fwupd.service)
if the name of the snap is fwupd. This allows classic snaps to call a
version of fwupd installed from classic package repositories (deb or
rpm).

Fix #6946

(cherry picked from commit a5c2e64123e75f0e5d3c03c83be1cd4f86fd0336)