fwupd/fwupd #6954: redfish : LVFS web UI can not received error report occurence

vincent-tang1, 2024-03-21:
**Precondition**
- Use ubuntu 23.10 
- Use snap to install 'fwupd' with parameter "--channel=stable"
- redfish.quirk apply as below
```
[REDFISH\VENDOR_Advantech&ID_BMC]
Flags = manager-reset,no-manager-reset-request
RedfishResetPreDelay = 100000
```

**Describe the bug**
Using redfish plugin to update BMC FW, somehow i trigger manually a update fail during the fwupd update. After failed, console can show error message, but furthermore check "get-history" command, as result, have not seen error report. Thereforce, no error report is uploaded into remote LVFS that vendor manager in this LVFS web UI can not receive error report occurence.

Note : This issue is similar with another ticket #6879 

**Steps to Reproduce**
1. Get devices
```
├─FWA6172 Platform Management:
│     Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
│     Summary:            Redfish host bus adapter
│     Current version:    0.77.0000
│     Vendor:             Advantech (REDFISH:ADVANTECH)
│     GUIDs:              10598c6b-6f24-518d-9065-bb948baa4907 ← REDFISH\VENDOR_Advantech&SOFTWAREID_894feb65-2ca8-4378-a5d3-41c78ff5b59f-BMC
│                         69ec5e09-0466-5679-83f5-0dc75e3ff4d5 ← REDFISH\VENDOR_Advantech&ID_BMC
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
```

2. Upgrade image and maunally do firmware update fail, the console can show error message.
```
root@FWA-6172-efi:~/tmp/hughsie-redfish-poll-task/fwupd# fwupdmgr update
Devices with no available firmware updates:
 • BIOS
 • BL
 • BMCONF
 • FPGA
 • NVRAM
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade FWA6172 Platform Management from 0.77.0000 to 1.03.0000?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This unstable release fixes the following issues:                            ║
║                                                                              ║
║ • Fix bug for test 1                                                         ║
║ • Improve performance for test 1                                             ║
║                                                                              ║
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ • Add support for test 1                                                     ║
║                                                                              ║
║ FWA6172 Platform Management and all connected devices may not be usable      ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Waiting…                 [************************************** ] Less than one minute remaining…
Transfer of image 'firmware.bin' to 'BMC' failed.

==================================================
==================================================
As you can see, it return the error message and come from Exception state to be detected, as below as noted code

static gboolean
fu_redfish_device_poll_task_once(FuRedfishDevice *self, FuRedfishDevicePollCtx *ctx, GError **error)
{
    FuRedfishDevicePrivate *priv = GET_PRIVATE(self);
    JsonObject *json_obj;
    const gchar *message = "Unknown failure";
    const gchar *state_tmp;
[... 159 lines cut ...]
        Remote ID:        lvfs-testing
        Release ID:       82544
        Summary:          Firmware for the Advantech FWA6172 BMC
        Variant:          sku:10598c6b
        License:          Proprietary
        ...
```

**Summary**
Could you review if this is an appropriate solution ? if yes, you can modify into contents you would expect. Thanks

**fwupd version information**
```
fwupdmgr --version

compile   com.hughsie.libxmlb           0.3.6
compile   io.snapcraft.fwupd            5749
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.9.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.5.0-10-generic
runtime   org.freedesktop.fwupd         1.9.14
```

superm1, 2024-03-21:
Could you submit a pull request with your proposal for review?

vincent-tang1, 2024-03-21:
Sure, however, before that, i would like to confirm something with you
I have a situation, using compling build as below, i can execute **'fwupdtool'**, but **'fwupdmrg' NOT**
So, i can not use '**fwupdmgr'** to verify it.
https://github.com/fwupd/fwupd/blob/main/docs/building.md

Do you know whether in each time "fwupdmgr update" or similar command , FU_REDFISH_DEVICE_FLAG_MANAGER_RESET flag status can be recovery from redfish.quirk define ? If yes,  I think that this solution should be valid.
(Note: fupwd daemon always run in backgroup if using 'fwupdmgr' utility)

Thanks

superm1, 2024-03-21:
What you can do is start up two tabs.  In each tab run `source venv/bin/activate` and then in one tab run `fwupd -v` and then in the other tab you can run `fwupdgmr`.

vincent-tang1, 2024-03-21:
Ohh, thanks, after using this method, 'fwupdmgr' can be executed in compling build

> Do you know whether in each time "fwupdmgr update" or similar command , FU_REDFISH_DEVICE_FLAG_MANAGER_RESET flag status can be recovery from redfish.quirk define ? If yes, I think that this solution should be valid.

After verfied, i think **above solution from my side is not valid**, because in each time "fwupdmgr update", FU_REDFISH_DEVICE_FLAG_MANAGER_RESET flag status can not be recovery from redfish.quirk define (it still be memorize previous change status). **So, please ignore first solution.**

vincent-tang1, 2024-03-21:
I tried to seek another solution and verify it as below, 
```
--- a/plugins/redfish/fu-redfish-plugin.c
+++ b/plugins/redfish/fu-redfish-plugin.c
@@ -553,6 +553,13 @@ fu_redfish_plugin_cleanup(FuPlugin *plugin,
        g_autoptr(FuRedfishRequest) request = fu_redfish_backend_request_new(self->backend);
        g_autoptr(GPtrArray) devices = NULL;

+       /* check if existing update error, if yes, nothing to do */
+       if (g_error_matches(*error, FWUPD_ERROR, FWUPD_ERROR_INTERNAL) ||
+           g_error_matches(*error, FWUPD_ERROR, FWUPD_ERROR_NEEDS_USER_ACTION) ||
+           g_error_matches(*error, FWUPD_ERROR, FWUPD_ERROR_WRITE) ||
+           g_error_matches(*error, FWUPD_ERROR, FWUPD_ERROR_INVALID_FILE))
+               return TRUE;
+
        /* nothing to do */
        if (!fu_device_has_private_flag(device, FU_REDFISH_DEVICE_FLAG_MANAGER_RESET))
                return TRUE;
```
but seem like that 'error' message in write_firmware() callback can not delivery into above cleanup() callback, this is becasue below common code fu_engine_write_firmware(), it does not use global 'error' variable to delivery to fu_engine_cleanup()
```
static gboolean
fu_engine_write_firmware(FuEngine *self,
                         const gchar *device_id,
                         GInputStream *stream_fw,
                         FuProgress *progress,
                         FwupdInstallFlags flags,
                         GError **error)
{
        ....
        if (!fu_plugin_runner_write_firmware(plugin, device, stream_fw, progress, flags, error)) {
                ...
                g_autoptr(GError) error_cleanup = NULL;
                ...
                if (!fu_engine_cleanup(self, device_id, progress, flags, &error_cleanup)) {
                        g_warning("failed to update-cleanup after failed update: %s",
                                  error_cleanup->message);
                }
                return FALSE;
        }
        ...
```
So, the second solution seems like also can not work.

========================
Currently, i might have no other idea.
From your extensive experience in fwupd code structure, not sure if you have any other thoughts or directions? thanks

vincent-tang1, 2024-03-22:
> [trivial: redfish: don't reboot BMC if update failed](https://github.com/fwupd/fwupd/commit/e4beb032017dbb2ddcba3ad8d5522d47fb5a436e) 

Thanks a lot! but after verifed it, it can not work. And then after trace code from my side, the reason as below to let you know!

1. We can see that when fu_engine_install_blob() fail, it will set the deivce error state using fu_device_set_update_state()
```
gboolean
fu_engine_install_release(FuEngine *self,
                          FuRelease *release,
                          GInputStream *stream,
                          FuProgress *progress,
                          FwupdInstallFlags flags,
                          GError **error)
{
        ...
        if (!fu_engine_install_blob(self,
                                    device,
                                    stream_fw,
                                    progress,
                                    flags,
                                    feature_flags,
                                    &error_local)) {
                if (g_error_matches(error_local, FWUPD_ERROR, FWUPD_ERROR_AC_POWER_REQUIRED) ||
                    g_error_matches(error_local, FWUPD_ERROR, FWUPD_ERROR_BATTERY_LEVEL_TOO_LOW) ||
                    g_error_matches(error_local, FWUPD_ERROR, FWUPD_ERROR_NEEDS_USER_ACTION) ||
                    g_error_matches(error_local, FWUPD_ERROR, FWUPD_ERROR_BROKEN_SYSTEM)) {
                        fu_device_set_update_state(device_orig,
                                                   FWUPD_UPDATE_STATE_FAILED_TRANSIENT);
                } else {
                        fu_device_set_update_state(device, FWUPD_UPDATE_STATE_FAILED);
                }
                fu_device_set_update_error(device_orig, error_local->message);
                fu_device_set_update_error(device, error_local->message);
                g_propagate_error(error, g_steal_pointer(&error_local));
                return FALSE;
        }
        ...
``` 

2. but in fact, cleanup() callback function in fu_engine_install_blob() already be invoked first if update fail, so at this moment, the cleanup() can not identify if current device is in error status
```
gboolean
fu_engine_install_blob(FuEngine *self,
                       FuDevice *device,
                       GInputStream *stream_fw,
                       FuProgress *progress,
                       FwupdInstallFlags flags,
                       FwupdFeatureFlags feature_flags,
                       GError **error)
{
                ...
                if (!fu_engine_write_firmware(self,
                                              device_id,
                                              stream_fw,
                                              fu_progress_get_child(progress_local),
                                              flags,
                                              error))
                        return FALSE;
               ...
}

==> from below, you can see when fu_plugin_runner_write_firmware() fail, it will call fu_engine_cleanup(), but error state have not set into the device yet.

static gboolean
fu_engine_write_firmware(FuEngine *self,
                         const gchar *device_id,
                         GInputStream *stream_fw,
                         FuProgress *progress,
                         FwupdInstallFlags flags,
                         GError **error)
{
        ...
        if (!fu_plugin_runner_write_firmware(plugin, device, stream_fw, progress, flags, error)) {
                g_autoptr(GError) error_attach = NULL;
                g_autoptr(GError) error_cleanup = NULL;

                /* attach back into runtime then cleanup */
                fu_engine_set_install_phase(self, FU_ENGINE_INSTALL_PHASE_ATTACH);
                fu_progress_reset(progress);
                if (!fu_plugin_runner_attach(plugin, device, progress, &error_attach)) {
                        g_warning("failed to attach device after failed update: %s",
                                  error_attach->message);
                }
                fu_engine_set_install_phase(self, FU_ENGINE_INSTALL_PHASE_CLEANUP);
                fu_progress_reset(progress);
here=>          if (!fu_engine_cleanup(self, device_id, progress, flags, &error_cleanup)) { 
                        g_warning("failed to update-cleanup after failed update: %s",
                                  error_cleanup->message);
                }
                return FALSE;
        }
        ...
}

```

superm1, 2024-03-22:
Got it, thanks!  I added another commit that I think handles what you've pointed out.  Can you try again with that?

vincent-tang1, 2024-03-22:
Many thanks, after verified your branch superm1/redfish, please see below two experiments: 

**Experiment 1:** It can not work, this is because below commit to cause 
> [Keep polling in fu_redfish_device_poll_task() even for task errors](https://github.com/fwupd/fwupd/commit/67a911b62b71cabce72e3e68fe2c6de423d0d127) 

when firmware update expection occured, it still retry to request until timer expire 60*60 sec (1 hr)
```
static gboolean
fu_redfish_device_poll_task_cb(FuDevice *device, gpointer user_data, GError **error)
{
        FuRedfishDevicePollCtx *ctx = (FuRedfishDevicePollCtx *)user_data;
        FuRedfishDevice *self = FU_REDFISH_DEVICE(device);
        FuRedfishDevicePrivate *priv = GET_PRIVATE(self);
        JsonObject *json_obj;
        const gchar *message = "Unknown failure";
        const gchar *state_tmp;
        g_autoptr(FuRedfishRequest) request = fu_redfish_backend_request_new(priv->backend);
        ...
        if (g_strcmp0(state_tmp, "Exception") == 0 ||
            g_strcmp0(state_tmp, "UserIntervention") == 0) {
                g_set_error_literal(error, FWUPD_ERROR, ctx->error_code, message);
                return FALSE;
        }
       ...
```
as you can see, when update fail is detected (Exception occur), fu_redfish_device_poll_task_cb() will retrn false, but it still continue to be retried from below fu_device_retry_full() until 60 *60 (1 hr) second 
```
        return fu_device_retry_full(FU_DEVICE(self),
                                    fu_redfish_device_poll_task_cb,
                                    60 * 60, /* a long time */
                                    1000,
                                    ctx,
                                    error);
```
and you can see debug log
```
06:56:06.579 FuDevice             failed on try 1 of 3600: TaskState is Running
06:56:07.862 FuDevice             failed on try 2 of 3600: TaskState is Running
06:56:09.157 FuDevice             failed on try 3 of 3600: TaskState is Running
06:56:10.464 FuDevice             failed on try 4 of 3600: TaskState is Running
06:56:11.795 FuDevice             failed on try 5 of 3600: TaskState is Running
06:56:13.195 FuDevice             failed on try 6 of 3600: TaskState is Running
06:56:14.628 FuDevice             failed on try 7 of 3600: TaskState is Running
06:56:15.979 FuDevice             failed on try 8 of 3600: TaskState is Running
06:56:17.287 FuDevice             failed on try 9 of 3600: TaskState is Running
06:56:18.578 FuDevice             failed on try 10 of 3600: TaskState is Running
06:56:19.871 FuDevice             failed on try 11 of 3600: TaskState is Running
06:56:21.184 FuDevice             failed on try 12 of 3600: TaskState is Running
06:56:22.556 FuDevice             failed on try 13 of 3600: TaskState is Running
06:56:23.956 FuDevice             failed on try 14 of 3600: TaskState is Running
06:56:25.344 FuDevice             failed on try 15 of 3600: TaskState is Running
06:56:26.464 FuDevice             failed on try 16 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:27.555 FuDevice             failed on try 17 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:28.647 FuDevice             failed on try 18 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:29.737 FuDevice             failed on try 19 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:30.840 FuDevice             failed on try 20 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:31.940 FuDevice             failed on try 21 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:33.088 FuDevice             failed on try 22 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:34.180 FuDevice             failed on try 23 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:35.270 FuDevice             failed on try 24 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:36.361 FuDevice             failed on try 25 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:37.451 FuDevice             failed on try 26 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:38.543 FuDevice             failed on try 27 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:39.644 FuDevice             failed on try 28 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:40.760 FuDevice             failed on try 29 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:41.851 FuDevice             failed on try 30 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:42.942 FuDevice             failed on try 31 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:44.032 FuDevice             failed on try 32 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:45.122 FuDevice             failed on try 33 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
06:56:46.229 FuDevice             failed on try 34 of 3600: Transfer of image 'firmware.bin' to 'BMC' failed.
...
...
```

**Experiment 2** : Also based on your branch superm1/redfish, but i reverted this patch
> [Keep polling in fu_redfish_device_poll_task() even for task errors](https://github.com/fwupd/fwupd/commit/67a911b62b71cabce72e3e68fe2c6de423d0d127) 

After did the verification, it can work. and finnally we can received error report from LVFS web UI
[image]

About the above merged commit, more consideration, please see (https://github.com/fwupd/fwupd/commit/67a911b62b71cabce72e3e68fe2c6de423d0d127) 
https://github.com/fwupd/fwupd/issues/6879#issuecomment-2014498658

hughsie, 2024-03-22:
> [Keep polling in fu_redfish_device_poll_task() even for task errors](https://github.com/fwupd/fwupd/commit/67a911b62b71cabce72e3e68fe2c6de423d0d127)

I've reverted this.

superm1, 2024-03-22:
closed via https://github.com/fwupd/fwupd/pull/6955

closed 2024-03-22 by superm1

superm1, 2024-03-22:
> [REDFISH\VENDOR_Advantech&ID_BMC]
Flags = manager-reset,no-manager-reset-request
RedfishResetPreDelay = 100000

BTW - do you want your quirk added by default upstream?

vincent-tang1, 2024-03-25:
> BTW - do you want your quirk added by default upstream?

It already be added into upstream in redfish.quirk file before, thanks ^^

vincent-tang1, 2024-03-27:
Not sure if these soultion commits will be put into 1.9.16 release tag ? thanks

hughsie, 2024-03-27:
Yup, they're in the branch now: https://github.com/fwupd/fwupd/blob/1_9_X/plugins/redfish/redfish.quirk#L26

vincent-tang1, 2024-03-27:
> [Set update state in fu_engine_write_firmware() for failures](https://github.com/fwupd/fwupd/commit/c6803f1a1295cc982d39677d627f5b36837dab5d)

> [Revert "Actually set the created property when creating FuDevice"](https://github.com/fwupd/fwupd/commit/6dd77dac06e054ce49441e60cc4984b0b6008e8c)

Thanks, not sure if these two commits in main branch will be planned to put it into 1.9.16 release tag ?

hughsie, 2024-03-27:
> Thanks, not sure if these two commits in main branch will be planned to put it into 1.9.16 release tag ?

I think the first is in the 1_9_X branch - and the second never made it into the backport list.

vincent-tang1, 2024-03-27:
Thanks, so any release tag is generated by 1_9_X branch, not main branch, my understanding is correct ?

hughsie, 2024-03-27:
> so any release tag is generated by 1_9_X branch

The 1.9.x tarballs come from 1_9_X, the 1.8.x tarballs come from 1_8_X etc. The main branch is what will become 2.0.x

vincent-tang1, 2024-03-27:
got it, many thanks!