fwupd/fwupd #6956: HP Elitebook 865 G10 upgrade issue

darkbasic, 2024-03-22:
**Describe the bug**

I've tried updating the UEFI firmware of my HP Elitebook 865 G10 from 1.3.9 to 1.4.10 but at the reboot nothing happens: it boots straight into zfsbootmenu and starts Arch Linux as usual.

```
[niko@arch-phoenix ~]$ fwupdmgr get-updates
Devices with no available firmware updates: 
 • ELAN07A9:00 04F3:31ED
 • HP 5MP Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • USB3.0 Hub
 • WD BLACK SN850X 4000GB
Devices with the latest available firmware version:
 • Prometheus
 • Prometheus IOTA Config
 • Thunderbolt Dock G4
 • UEFI dbx
HP HP EliteBook 865 16 inch G10 Notebook PC
│
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Summary:            UEFI ESRT device
  │   Current version:    0x01030900
  │   Minimum Version:    0x00000001
  │   Vendor:             HP (DMI:HP)
  │   Update State:       Success
  │   GUID:               8cf4e0ec-a249-4c1f-9cdf-bbe85a9096e4
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │   Device Requests:    • Message
  │ 
  └─EliteBook 8x5 G10 Notebook PC V82 System Update:
        New version:      0x01040a00
        Remote ID:        lvfs
        Release ID:       86733
        Summary:          System Firmware for HP Laptop, family V82
        License:          Proprietary
        Size:             42.2 MB
        Created:          2024-03-01
        Urgency:          High
        Vendor:           HP
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:      
        Fixes and enhancements in System firmware 1.4.10.0:
        
        • Fixes an issue where incorrect characters are generated when the internal keyboard is used.
        • Fixes an issue where the USB keyboard does not function properly on the BitLocker PIN screen when fast boot is enabled.
        • Fixes an issue where the Wi-Fi Password does not appear in the Preboot Wi-Fi Profile variable.
        • Fixes an issue where certain options that are present in UefiBIOSConfig file are not present in the HpSetup.txt file.
        • Fixes an issue where a system using only DC power while flashing the BIOS through the Firmware Update and Recovery Utility boots to the operating system and then auto restarts unexpectedly.
        
        Please refer hp.com.
        Issue:            CVE-2023-5410
        Checksum:         51fd079072369738a0a1d6215ac5bb96352fb846d54e7bf8e3fae7fe52d5ff3c
```

```
[niko@arch-phoenix ~]$ journalctl -b -u fwupd.service
Mar 22 10:08:05 arch-phoenix systemd[1]: Starting Firmware update daemon...
Mar 22 10:08:06 arch-phoenix fwupd[2736]: 09:08:06.737 FuEngine             failed to update history database: failed to do post-reboot cleanup: no mountpoint for ESP
Mar 22 10:08:06 arch-phoenix fwupd[2736]: 09:08:06.840 FuMain               Daemon ready for requests (locale en_US.UTF-8)
Mar 22 10:08:06 arch-phoenix systemd[1]: Started Firmware update daemon.
[... 341 lines cut ...]
│       │   ├── zfsbootmenu-release-vmlinuz-x86_64.EFI
│       │   └── zfsbootmenu-release-vmlinuz-x86_64.rc.EFI
│       ├── zbm.back
│       │   ├── zfsbootmenu-recovery-vmlinuz-x86_64.EFI
│       │   ├── zfsbootmenu-release-vmlinuz-x86_64.EFI
│       │   └── zfsbootmenu-release-vmlinuz-x86_64.rc.EFI
│       └── zbm.back.back
│           ├── zfsbootmenu-recovery-vmlinuz-x86_64.EFI
│           ├── zfsbootmenu-release-vmlinuz-x86_64.EFI
│           └── zfsbootmenu-release-vmlinuz-x86_64.rc.EFI
├── initramfs-linux-fallback.img
├── initramfs-linux.img
└── vmlinuz-linux

8 directories, 16 files
```

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? Yes
- Is this a regression? No idea, it's the first time that this laptop is being supported by fwupd
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

hughsie, 2024-03-22:
What's zfsbootmenu? What's the output of `sudo fwupdtool get-devices --plugins uefi_capsule -vv` please.

darkbasic, 2024-03-22:
It's a "bootloader" (which runs a patched linux kernel) capable of booting linux from a zfs root parition, doing rollbacks to previous snapshots etc: https://docs.zfsbootmenu.org/en/v2.3.x/
Basically my rootfs is zfs, plus the mandatory fat32 parition for efi.

<details>

<summary>sudo fwupdtool get-devices --plugins uefi_capsule -vv</summary>

```shell
[niko@arch-phoenix ~]$ sudo fwupdtool get-devices --plugins uefi_capsule -vv
10:11:19.411 FuDebug              verbose to debug (on console 1)
10:11:19.412 FuEngine             starting fwupd 1.9.15…
10:11:19.412 FuMain               locked /run/lock/fwupdtool
10:11:19.413 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
10:11:19.415 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…                 [ -                                     ]10:11:19.415 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
10:11:19.415 FuConfig             loading config /etc/fwupd/fwupd.conf
10:11:19.415 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
10:11:19.415 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
10:11:19.415 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
10:11:19.415 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
10:11:19.415 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
10:11:19.415 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
10:11:19.415 FuConfig             ::configuration loaded
10:11:19.416 FuRemoteList         loading /usr/share/fwupd/metainfo
10:11:19.416 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
10:11:19.416 XbSilo               file: c0c35f6e-444b-11bf-9fdb-6ce9c040c9b3, current:c0c35f6e-444b-11bf-9fdb-6ce9c040c9b3, cached: (null)
10:11:19.416 XbSilo               loading silo with file contents
10:11:19.416 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
10:11:19.416 FuRemoteList         ::remote_list changed
10:11:19.416 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
10:11:19.416 FuRemoteList         ::remote_list changed
10:11:19.416 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
10:11:19.416 FuRemoteList         ::remote_list changed
10:11:19.416 FuEngine             priority local and vendor-directory is not download remote, so bumping
10:11:19.416 FuRemoteList         ordering lvfs-testing=lvfs+1
10:11:19.416 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs-testing[1], lvfs[0]
Loading…                 [                                       ]10:11:19.416 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
10:11:19.416 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
10:11:19.416 Jcat                 loaded 1 certificates
10:11:19.416 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
10:11:19.416 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
10:11:19.416 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
10:11:19.416 Jcat                 loaded 1 certificates
10:11:19.416 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
10:11:19.416 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
10:11:19.416 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
10:11:19.416 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]10:11:19.416 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
10:11:19.417 FuHistory            got schema version of 12
Loading…                 [*                                      ]10:11:19.419 FuPlugin             load(/usr/lib/fwupd-1.9.15/libfu_plugin_modem_manager.so)
10:11:19.420 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
10:11:19.420 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
10:11:19.420 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
10:11:19.420 XbSilo               file: 1cde641d-bac4-a86c-ca2e-8d2385db79c6, current:1cde641d-bac4-a86c-ca2e-8d2385db79c6, cached: (null)
10:11:19.420 XbSilo               loading silo with file contents
Loading…                 [*                                      ]10:11:19.421 FuConfig             loading config /etc/fwupd/fwupd.conf
10:11:19.421 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
10:11:19.421 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
10:11:19.421 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
10:11:19.421 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
10:11:19.421 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
10:11:19.421 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
10:11:19.421 FuConfig             ::configuration loaded
10:11:19.421 FuStruct             SmbiosEp64:
[... cut ...]
  VersionLowest:        1
  VersionFormat:        number
  VersionRaw:           0x37c42f06
  VersionLowestRaw:     0x00000001
  Created:              2024-03-22
  UpdateState:          success

10:11:21.005 FuPluginTest         destroy

10:11:21.005 FuProgress           raw timing data was { 0.054, 0.000, 0.003, 0.001, 0.004, 0.000 } -- steps were set as [ 63 1 8 1 26 1 ] but should have been [ 88 0 4 1 6 1 ] at ../fwupd-1.9.15/plugins/uefi-capsule/fu-uefi-capsule-plugin.c:1130
```

</details>

hughsie, 2024-03-22:
> Those show NVRAM devices, so we should have firmwar eupdate entry in `efibootmgr -v`. Can you deploy the update (with fwupdmgr update) but **don't** reboot yet and get the output of `efibootmgr -v`. Thanks.

darkbasic, 2024-03-22:
It shows:

```shell
[niko@arch-phoenix ~]$ efibootmgr -v
BootNext: 0000
BootCurrent: 0008
Timeout: 0 seconds
BootOrder: 0008,0005,0006,0003,0004,0001,0000
Boot0000* Linux Firmware Updater	HD(1,GPT,210d5992-254c-4c6c-8b40-93bbd3891ede,0x800,0x219800)/\EFI\arch\fwupdx64.efi
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 98 21 00 00 00 00 00 92 59 0d 21 4c 25 6c 4c 8b 40 93 bb d3 89 1e de 02 02 / 04 04 32 00 5c 00 45 00 46 00 49 00 5c 00 61 00 72 00 63 00 68 00 5c 00 66 00 77 00 75 00 70 00 64 00 78 00 36 00 34 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
Boot0001* Wi-Fi IPV6 Network	PciRoot(0x0)/Pci(0x2,0x2)/Pci(0x0,0x0)/MAC(00410e57e907,1)/Wi-Fi(00:00:00:00:3e:00:00:00:00:00:00:00:18:7c:7e:19:00:00:00:00:7a:21:d7:23:00:00:00:00:00:00:00:00)/IPv6([::],0,Static,[::],[::],64)4eac0881119f594d850ee21a522c59b21940000049535048
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 02 02 / 01 01 06 00 00 00 / 03 0b 25 00 00 41 0e 57 e9 07 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 / 03 1c 24 00 00 00 00 00 3e 00 00 00 00 00 00 00 18 7c 7e 19 00 00 00 00 7a 21 d7 23 00 00 00 00 00 00 00 00 / 03 0d 3c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 7f ff 04 00
    data: 4e ac 08 81 11 9f 59 4d 85 0e e2 1a 52 2c 59 b2 19 40 00 00 49 53 50 48
Boot0003  USB NETWORK BOOT:  	PciRoot(0x0)/Pci(0x8,0x3)/Pci(0x0,0x3)/USB(1,0)/MAC(606d3cfcdc43,0)/IPv6([::],0,Static,[::],[::],64)4eac0881119f594d850ee21a522c59b21b30000049535048
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 03 08 / 01 01 06 00 03 00 / 03 05 06 00 01 00 / 03 0b 25 00 60 6d 3c fc dc 43 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 0d 3c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 7f ff 04 00
    data: 4e ac 08 81 11 9f 59 4d 85 0e e2 1a 52 2c 59 b2 1b 30 00 00 49 53 50 48
Boot0004* Wi-Fi IPV4 Network	PciRoot(0x0)/Pci(0x2,0x2)/Pci(0x0,0x0)/MAC(00410e57e907,1)/Wi-Fi(00:00:00:00:3e:00:00:00:00:00:00:00:18:7c:7e:19:00:00:00:00:7a:21:d7:23:00:00:00:00:00:00:00:00)/IPv4(0.0.0.0,0,DHCP,0.0.0.0,0.0.0.0,0.0.0.0)4eac0881119f594d850ee21a522c59b21938000049535048
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 02 02 / 01 01 06 00 00 00 / 03 0b 25 00 00 41 0e 57 e9 07 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 / 03 1c 24 00 00 00 00 00 3e 00 00 00 00 00 00 00 18 7c 7e 19 00 00 00 00 7a 21 d7 23 00 00 00 00 00 00 00 00 / 03 0c 1b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 7f ff 04 00
    data: 4e ac 08 81 11 9f 59 4d 85 0e e2 1a 52 2c 59 b2 19 38 00 00 49 53 50 48
Boot0005  USB:  	PciRoot(0x0)/Pci(0x8,0x1)/Pci(0x0,0x3)4eac0881119f594d850ee21a522c59b20b18000049535048
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 01 08 / 01 01 06 00 03 00 / 7f ff 04 00
    data: 4e ac 08 81 11 9f 59 4d 85 0e e2 1a 52 2c 59 b2 0b 18 00 00 49 53 50 48
Boot0006  USB NETWORK BOOT:  	PciRoot(0x0)/Pci(0x8,0x3)/Pci(0x0,0x3)/USB(1,0)/MAC(606d3cfcdc43,0)/IPv4(0.0.0.0,0,DHCP,0.0.0.0,0.0.0.0,0.0.0.0)4eac0881119f594d850ee21a522c59b21b28000049535048
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 03 08 / 01 01 06 00 03 00 / 03 05 06 00 01 00 / 03 0b 25 00 60 6d 3c fc dc 43 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 0c 1b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 7f ff 04 00
    data: 4e ac 08 81 11 9f 59 4d 85 0e e2 1a 52 2c 59 b2 1b 28 00 00 49 53 50 48
Boot0008* ZFSBootMenu	HD(1,GPT,816992f9-6fe2-4c5b-a316-df6ec116d021,0x100,0x40000)/\EFI\zbm\zfsbootmenu-release-vmlinuz-x86_64.EFI0400000049535048
      dp: 04 01 2a 00 01 00 00 00 00 01 00 00 00 00 00 00 00 00 04 00 00 00 00 00 f9 92 69 81 e2 6f 5b 4c a3 16 df 6e c1 16 d0 21 02 02 / 04 04 64 00 5c 00 45 00 46 00 49 00 5c 00 7a 00 62 00 6d 00 5c 00 7a 00 66 00 73 00 62 00 6f 00 6f 00 74 00 6d 00 65 00 6e 00 75 00 2d 00 72 00 65 00 6c 00 65 00 61 00 73 00 65 00 2d 00 76 00 6d 00 6c 00 69 00 6e 00 75 00 7a 00 2d 00 78 00 38 00 36 00 5f 00 36 00 34 00 2e 00 45 00 46 00 49 00 00 00 / 7f ff 04 00
    data: 04 00 00 00 49 53 50 48
```

darkbasic, 2024-03-22:
By the way it looks like it doesn't put the updater into the efi partition:

```
Boot0000* Linux Firmware Updater	HD(1,GPT,210d5992-254c-4c6c-8b40-93bbd3891ede,0x800,0x219800)/\EFI\arch\fwupdx64.efi

Boot0008* ZFSBootMenu	HD(1,GPT,816992f9-6fe2-4c5b-a316-df6ec116d021,0x100,0x40000)/\EFI\zbm\zfsbootmenu-release-vmlinuz-x86_64.EFI0400000049535048
```

The partuuid is different and points to `/dev/zd0p1` which I have no idea what's supposed to be
```
[niko@arch-phoenix ~]$ sudo ls -al /dev/disk/by-partuuid/ | grep -i 93bbd3891ede
lrwxrwxrwx 1 root root  11 Mar 22 13:08 210d5992-254c-4c6c-8b40-93bbd3891ede -> ../../zd0p1
```

```
[niko@arch-phoenix ~]$ ls -al /dev/disk/by-partuuid/
total 0
drwxr-xr-x 2 root root 120 Mar 22 13:08 .
drwxr-xr-x 8 root root 160 Mar 22 13:08 ..
lrwxrwxrwx 1 root root  11 Mar 22 13:08 210d5992-254c-4c6c-8b40-93bbd3891ede -> ../../zd0p1
lrwxrwxrwx 1 root root  15 Mar 22 13:08 816992f9-6fe2-4c5b-a316-df6ec116d021 -> ../../nvme0n1p1
lrwxrwxrwx 1 root root  11 Mar 22 13:08 89bfbe9c-cfc0-44e2-be43-8517e6e1d0d4 -> ../../zd0p2
lrwxrwxrwx 1 root root  15 Mar 22 13:08 efc4aaae-5383-44ae-9bc2-2a477972a593 -> ../../nvme0n1p2
```

What's sure is that `fwupdx64.efi` is not in `/boot/efi`:

```
[niko@arch-phoenix ~]$ sudo find / 2> /dev/null | grep fwupdx64.efi
/usr/lib/fwupd/efi/fwupdx64.efi
[niko@arch-phoenix ~]$
```

darkbasic, 2024-03-22:
The laptop's UEFI boot menu doesn't show the entry for the updater either:

[image]

hughsie, 2024-03-22:
> Boot0000* Linux Firmware Updater	HD(1,GPT,210d5992-254c-4c6c-8b40-93bbd3891ede,0x800,0x219800)/\EFI\arch\fwupdx64.efic

That looks correct -- does rebooting remove this entry automatically?

> The partuuid is different and points to /dev/zd0p1 which I have no idea what's supposed to be

fwupd found two ESPs, and ordered them like:

	10:11:19.452 FuVolume             mounting /org/freedesktop/UDisks2/block_devices/zd0p1
	10:11:19.464 FuPluginUefiCapsule  found /run/media/root/2230-C418/EFI/ubuntu/grubx64.efi which indicates a Linux ESP, using /run/media/root/2230-C418
	10:11:19.464 FuVolume             unmounting /org/freedesktop/UDisks2/block_devices/zd0p1
	10:11:19.496 FuPluginUefiCapsule  not a Linux ESP: did not find grubx64.efi|shimx64.efi|systemd-bootx64.efi in /boot/efi
	10:11:19.496 FuPluginUefiCapsule  more than one ESP possible:
	 - 0x30433:	/org/freedesktop/UDisks2/block_devices/zd0p1
	 - 0x20400:	/org/freedesktop/UDisks2/block_devices/nvme0n1p1

superm1, 2024-03-22:
> failed to update history database: failed to do post-reboot cleanup: no mountpoint for ESP

At least after reboot I think something is wrong with the ESP detection.

hughsie, 2024-03-22:
@superm1 does the fu_volume_get_mount_point() need to come after the fu_volume_locker perhaps?

superm1, 2024-03-22:
I think so.

superm1, 2024-03-22:
But also we don't look at all ESP candidates after the reboot AFAICT.

hughsie, 2024-03-22:
I think we do, we seem to use `fu_uefi_capsule_plugin_get_default_esp()` to get `self->esp`

superm1, 2024-03-22:
That at least looks correct for post reboot cleanup, but I think the issue here is still the wrong ESP candidate being picked:

>   EspId:                /org/freedesktop/UDisks2/block_devices/zd0p1

Probably because our heuristic expects GRUB or shim, but zfsbootmenu doesn't use those.

hughsie, 2024-03-22:
> Probably because our heuristic expects GRUB or shim, but zfsbootmenu doesn't use those.

Ahh I see what you mean. We gave the score `+0x10000` because `EFI/ubuntu/grubx64.efi` exists -- which is a weird thing to have on a non-ESP. I also wonder what's on /boot/efi -- @darkbasic what does `sudo fwupdtool esp-list` show for your actual ESP please?

darkbasic, 2024-03-22:
> does rebooting remove this entry automatically?

Yes.

> what does sudo fwupdtool esp-list show for your actual ESP please?

There is a `tree /boot` in the original post, but here it is:

```shell
[niko@arch-phoenix ~]$ sudo fwupdtool esp-list
[sudo] password for niko: 
Loading…                 [************************************** ]13:22:34.407 FuEngine             failed to update history database: device ID a45df35ac0e948ee180fe216a5f703f32dda163f was not found

0.	Cancel
1.	/org/freedesktop/UDisks2/block_devices/zd0p1
2.	/org/freedesktop/UDisks2/block_devices/nvme0n1p1
Choose volume [0-2]: 2 
/boot/efi/EFI/zbm/zfsbootmenu-recovery-vmlinuz-x86_64.EFI
/boot/efi/EFI/zbm/zfsbootmenu-release-vmlinuz-x86_64.rc.EFI
/boot/efi/EFI/zbm/zfsbootmenu-release-vmlinuz-x86_64.EFI
/boot/efi/EFI/HP/DEVFW/Firmware.BIN
/boot/efi/EFI/HP/DEVFW/Realtek.bin
/boot/efi/EFI/HP/DEVFW/ClickPad.bin
/boot/efi/EFI/zbm.back/zfsbootmenu-recovery-vmlinuz-x86_64.EFI
/boot/efi/EFI/zbm.back/zfsbootmenu-release-vmlinuz-x86_64.rc.EFI
/boot/efi/EFI/zbm.back/zfsbootmenu-release-vmlinuz-x86_64.EFI
/boot/efi/EFI/zbm.back.back/zfsbootmenu-recovery-vmlinuz-x86_64.EFI
/boot/efi/EFI/zbm.back.back/zfsbootmenu-release-vmlinuz-x86_64.rc.EFI
/boot/efi/EFI/zbm.back.back/zfsbootmenu-release-vmlinuz-x86_64.EFI
```

Not sure what `failed to update history database: device ID a45df35ac0e948ee180fe216a5f703f32dda163f was not found` refers to.

hughsie, 2024-03-22:
@superm1 does that make any more sense to you than me? I'm also a bit confused why fwupd thinks zd0p1 is a ESP... @darkbasic can you attach `udisksctl dump` pls.

superm1, 2024-03-22:
If i'm following correctly then https://github.com/fwupd/fwupd/pull/6960 may help.

darkbasic, 2024-03-22:
> Ahh I see what you mean. We gave the score +0x10000 because EFI/ubuntu/grubx64.efi exists -- which is a weird thing to have on a non-ESP.

I have a single disk with just two partitions: the efi (which doesn't have `ubuntu/grubx64.efi`) and the zfs rootfs, which doesn't have it either. I still have a hard time figuring out what `/dev/zd0p1` is supposed to be.

EDIT: I know what it is! It's a zvol (basically a block device) that I used to hold an Ubuntu virtual machine to debug this zfs regression: https://github.com/openzfs/zfs/issues/15351

> If i'm following correctly then https://github.com/fwupd/fwupd/pull/6960 may help.

Yes, but there is a second issue at play here: fwupd shouldn't look for ESPs into zfs zvols. That's a recipe for disaster.

> can you attach udisksctl dump pls

Here it is:
```
/org/freedesktop/UDisks2/Manager:
  org.freedesktop.UDisks2.Manager:
    DefaultEncryptionType:      luks2
    SupportedEncryptionTypes:   luks1
                                luks2
    SupportedFilesystems:       ext2
                                ext3
                                ext4
                                xfs
                                vfat
                                ntfs
                                f2fs
                                nilfs2
                                exfat
                                btrfs
                                udf
                                swap
    Version:                    2.10.1
  org.freedesktop.UDisks2.Manager.NVMe:
    HostID:             
    HostNQN:            nqn.2014-08.org.nvmexpress:uuid:ee1da50e-0e3f-4398-8bef-29c938e3a3b3

/org/freedesktop/UDisks2/block_devices/nvme0n1:
  org.freedesktop.UDisks2.Block:
    Configuration:              []
    CryptoBackingDevice:        '/'
    Device:                     /dev/nvme0n1
    DeviceNumber:               66304
    Drive:                      '/org/freedesktop/UDisks2/drives/WD_BLACK_SN850X_4000GB_23273Q802388'
    HintAuto:                   false
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          true
    HintSymbolicIconName:       
    HintSystem:                 true
    Id:                         by-id-nvme-WD_BLACK_SN850X_4000GB_23273Q802388
    IdLabel:                    
    IdType:                     
    IdUUID:                     
    IdUsage:                    
    IdVersion:                  
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/nvme0n1
    ReadOnly:                   false
    Size:                       4000787030016
    Symlinks:                   /dev/disk/by-diskseq/1
                                /dev/disk/by-id/nvme-WD_BLACK_SN850X_4000GB_23273Q802388
                                /dev/disk/by-id/nvme-WD_BLACK_SN850X_4000GB_23273Q802388_1
                                /dev/disk/by-id/nvme-eui.e8238fa6bf530001001b448b4aa2b013
                                /dev/disk/by-path/pci-0000:02:00.0-nvme-1
    UserspaceMountOptions:      
  org.freedesktop.UDisks2.NVMe.Namespace:
    EUI64:                      001b448b4aa2b013
    FormatPercentRemaining:     0
[... 256 lines cut ...]
    Seat:                       seat0
    Serial:                     23273Q802388
    SiblingId:                  
    Size:                       4000787030016
    SortKey:                    00coldplug/00fixed/nvme0
    TimeDetected:               1711110013411295
    TimeMediaDetected:          1711110013411295
    Vendor:                     
    WWN:                        
  org.freedesktop.UDisks2.NVMe.Controller:
    ControllerID:                       8224
    FGUID:                              
    NVMeRevision:                       1.4
    SanitizePercentRemaining:           -1
    SanitizeStatus:                     never_sanitized
    SmartCriticalWarning:               
    SmartPowerOnHours:                  231
    SmartSelftestPercentRemaining:      -1
    SmartSelftestStatus:                success
    SmartTemperature:                   305
    SmartUpdated:                       1711113634
    State:                              live
    SubsystemNQN:                       nqn.2018-01.com.wdc:nguid:E8238FA6BF53-0001-001B448B4AA2B013
    UnallocatedCapacity:                0
```

hughsie, 2024-03-22:
> I know what it is! It's a zvol

But how does fwupd know that? The only helpful data we have is:

	/org/freedesktop/UDisks2/block_devices/zd0p1:
	  org.freedesktop.UDisks2.Block:
	    Device:                     /dev/zd0p1
	    IdType:                     vfat
	    IdUUID:                     2230-C418
	    IdUsage:                    filesystem
	    IdVersion:                  FAT32
	    Symlinks:                   /dev/disk/by-diskseq/2-part1
					/dev/disk/by-partuuid/210d5992-254c-4c6c-8b40-93bbd3891ede
					/dev/disk/by-uuid/2230-C418
					/dev/zvol/rpool/zvols/ubuntu-23.10-part1
	  org.freedesktop.UDisks2.Partition:
	    Type:               c12a7328-f81f-11d2-ba4b-00a0c93ec93b
	    UUID:               210d5992-254c-4c6c-8b40-93bbd3891ede

...and that looks a *lot* like a ESP to me.

hughsie, 2024-03-22:
I thought ignoring by symlink (e.g. a prefix of `/dev/zvol`) is probably the best we can do.

darkbasic, 2024-03-22:
> Can you please test that branch?

Sure.

> ...and that looks a lot like a ESP to me.

Because it is! It's the ESP of the virtual machine!

> Hmm, what do we get back from udisks to tell they're part of zfs?

Under `Symlinks` it starts by `/dev/zvol`

hughsie, 2024-03-22:
@superm1 want me to do that?

superm1, 2024-03-22:
> @superm1 want me to do that?

Go for it

darkbasic, 2024-03-22:
I cannot manage to build that branch. Initially I've tried `fwupd-git` from the AUR and it fails with the following error:

```shell
[niko@arch-phoenix fwupd-git]$ makepkg -s
==> Making package: fwupd-git 1.9.10.r450.g465c59c33-1 (Fri 22 Mar 2024 03:32:15 PM CET)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating fwupd-git git repo...
From https://github.com/fwupd/fwupd
 - [deleted]             (none)               -> hughsie/covscan
 - [deleted]             (none)               -> superm1/1-9-x-proposed
 - [deleted]             (none)               -> refs/pull/6952/merge
 - [deleted]             (none)               -> refs/pull/6959/merge
remote: Enumerating objects: 103, done.
remote: Counting objects: 100% (103/103), done.
remote: Compressing objects: 100% (36/36), done.
remote: Total 103 (delta 81), reused 73 (delta 67), pack-reused 0
Receiving objects: 100% (103/103), 82.70 KiB | 6.89 MiB/s, done.
Resolving deltas: 100% (81/81), completed with 60 local objects.
   88f7dc397..bdcb233fa  1_9_X                -> 1_9_X
   1b8f1b5dc..d36c352e9  main                 -> main
 + 2033946a6...f982a4737 refs/pull/6936/merge -> refs/pull/6936/merge  (forced update)
 + 7956f96d8...4eac33420 refs/pull/6959/head  -> refs/pull/6959/head  (forced update)
 + cc9b1241a...8863c0f1f refs/pull/6960/merge -> refs/pull/6960/merge  (forced update)
 * [new ref]             refs/pull/6961/head  -> refs/pull/6961/head
 * [new ref]             refs/pull/6962/head  -> refs/pull/6962/head
==> Validating source files with sha512sums...
    fwupd-git ... Skipped
==> Extracting sources...
  -> Creating working copy of fwupd-git git repo...
Cloning into 'fwupd-git'...
done.
Switched to a new branch 'makepkg'
==> Starting prepare()...
+ exec meson setup --prefix /usr --libexecdir lib --sbindir bin --buildtype plain --auto-features enabled --wrap-mode nodownload -D b_pie=true -D python.bytecompile=1 /home/niko/devel/fwupd-git-arch/src/fwupd-git /home/niko/devel/fwupd-git-arch/src/build -D passim=disabled -D b_lto=false -D docs=disabled -D man=true -D plugin_powerd=disabled -D supported_build=enabled -D launchd=disabled -D efi_binary=false
The Meson build system
Version: 1.4.0
Source dir: /home/niko/devel/fwupd-git-arch/src/fwupd-git
Build dir: /home/niko/devel/fwupd-git-arch/src/build
Build type: native build
Project name: fwupd
Project version: 2.0.0
C compiler for the host machine: cc (gcc 13.2.1 "cc (GCC) 13.2.1 20230801")
C linker for the host machine: cc ld.bfd 2.42.0
Host machine cpu family: x86_64
Host machine cpu: x86_64
Program git found: YES (/usr/bin/git)
Compiler for C supports arguments -Wfatal-errors: YES 
Compiler for C supports arguments -Waggregate-return: YES 
Compiler for C supports arguments -Wunused: YES 
Compiler for C supports arguments -Warray-bounds: YES 
Compiler for C supports arguments -Wcast-align: YES 
Compiler for C supports arguments -Wclobbered: YES 
Compiler for C supports arguments -Wdeclaration-after-statement: YES 
Compiler for C supports arguments -Wdiscarded-qualifiers: YES 
Compiler for C supports arguments -Wduplicated-branches: YES 
Compiler for C supports arguments -Wduplicated-cond: YES 
Compiler for C supports arguments -Wempty-body: YES 
Compiler for C supports arguments -Wfloat-equal: YES 
Compiler for C supports arguments -Wformat=2: YES 
Compiler for C supports arguments -Wformat-nonliteral: YES 
Compiler for C supports arguments -Wformat-security: YES 
Compiler for C supports arguments -Wformat-signedness: YES 
Compiler for C supports arguments -Wignored-qualifiers: YES 
Compiler for C supports arguments -Wimplicit-function-declaration: YES 
Compiler for C supports arguments -Wimplicit-int: YES 
Compiler for C supports arguments -Wincompatible-pointer-types: YES 
Compiler for C supports arguments -Winit-self: YES 
Compiler for C supports arguments -Wint-conversion: YES 
[... 2403 lines cut ...]
Configuring fwupdtool.test using configuration
Configuring fwupd.test using configuration
Configuring fwupdmgr-p2p.sh using configuration
Configuring fwupdtool.sh using configuration
Configuring fwupd.sh using configuration
Configuring fwupd-tests.conf using configuration
Running command: /usr/bin/python3 /home/niko/devel/fwupd-git-arch/src/fwupd-git/generate-build/unittest_inspector.py /home/niko/devel/fwupd-git-arch/src/fwupd-git/data/tests/fwupd_test.py
--- stdout ---

--- stderr ---
Traceback (most recent call last):
  File "/home/niko/devel/fwupd-git-arch/src/fwupd-git/generate-build/unittest_inspector.py", line 35, in <module>
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/niko/devel/fwupd-git-arch/src/fwupd-git/data/tests/fwupd_test.py", line 12, in <module>
    import dbusmock
ModuleNotFoundError: No module named 'dbusmock'

fwupd-git/data/tests/meson.build:113:6: ERROR: Command `/usr/bin/python3 /home/niko/devel/fwupd-git-arch/src/fwupd-git/generate-build/unittest_inspector.py /home/niko/devel/fwupd-git-arch/src/fwupd-git/data/tests/fwupd_test.py` failed with status 1.
```

</details>

Then I've tried with the fwupd 1.9.15 PKGBUILD and it works. So I've created a diff from `1.9.15` to `superm1/esp-list` and applied it to the (previously working) 1.9.15 PKGBUILD. I also had to comment out `-D plugin_intel_spi=true` but I still get the same error as `fwupd-git` from the AUR.

superm1, 2024-03-22:
You're missing the python-dbusmock package AFAICT.

superm1, 2024-03-22:
FWIW you don't need to replace your system package to do the test - you can do it all in the fwupd development environment.

https://fwupd.github.io/libfwupdplugin/building.html

darkbasic, 2024-03-22:
<details>

<summary>sudo fwupdtool get-devices --plugins uefi_capsule -vv with superm1/esp-list</summary>

```shell
[niko@arch-phoenix ~]$ sudo fwupdtool get-devices --plugins uefi_capsule -vv
14:50:49.606 FuDebug              verbose to debug (on console 1)
14:50:49.606 FuEngine             starting fwupd 2.0.0…
14:50:49.606 FuMain               locked /run/lock/fwupdtool
14:50:49.608 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
Loading…                 [ -                                     ]14:50:49.612 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
14:50:49.612 FuConfig             loading config /etc/fwupd/fwupd.conf
14:50:49.612 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
14:50:49.612 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
14:50:49.612 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
14:50:49.612 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
14:50:49.612 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
14:50:49.612 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
14:50:49.612 FuConfig             ::configuration loaded
14:50:49.613 FuRemoteList         loading /usr/share/fwupd/metainfo
14:50:49.613 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
14:50:49.613 XbSilo               file: 7219fb74-f275-0692-379a-f6d95f6f7a98, current:7219fb74-f275-0692-379a-f6d95f6f7a98, cached: (null)
14:50:49.613 XbSilo               loading silo with file contents
14:50:49.613 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
14:50:49.613 FuRemoteList         ::remote_list changed
14:50:49.613 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
14:50:49.613 FuRemoteList         ::remote_list changed
14:50:49.613 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
14:50:49.613 FuRemoteList         ::remote_list changed
14:50:49.613 FuEngine             priority local and vendor-directory is not download remote, so bumping
14:50:49.613 FuRemoteList         ordering lvfs-testing=lvfs+1
14:50:49.613 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…                 [                                       ]14:50:49.613 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
14:50:49.613 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
14:50:49.613 Jcat                 loaded 1 certificates
14:50:49.613 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
14:50:49.613 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
14:50:49.613 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
14:50:49.613 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
14:50:49.613 Jcat                 loaded 1 certificates
14:50:49.613 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
14:50:49.613 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
14:50:49.613 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]14:50:49.613 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
14:50:49.613 FuHistory            got schema version of 12
Loading…                 [*                                      ]14:50:49.617 FuPlugin             load(/usr/lib/fwupd-2.0.0/libfu_plugin_modem_manager.so)
14:50:49.618 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
14:50:49.618 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
14:50:49.619 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
14:50:49.619 XbSilo               file: 6415fd21-f348-6f90-1a32-dd431324c33d, current:6415fd21-f348-6f90-1a32-dd431324c33d, cached: (null)
14:50:49.619 XbSilo               loading silo with file contents
Loading…                 [*                                      ]14:50:49.621 FuConfig             loading config /etc/fwupd/fwupd.conf
14:50:49.621 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
14:50:49.621 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
14:50:49.621 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
14:50:49.621 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
14:50:49.621 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
14:50:49.621 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
14:50:49.621 FuConfig             ::configuration loaded
14:50:49.621 FuStruct             FuStructSmbiosEp64:
  anchor_str: _SM3_
  entry_point_csum: 0xad
  entry_point_len: 0x18
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x4
  smbios_docrev: 0x0
  entry_point_rev: 0x1
  reserved0: 0x0
  structure_table_len: 0x7d1
  structure_table_addr: 0x26347000
[... 1245 lines cut ...]

14:50:50.712 FuMain               FuUefiNvramDevice:
  DeviceId:             e97608be1da5b9d742670ca8c3c4f515bc10646a
[... cut ...]
  Version:              935603974
  VersionLowest:        1
  VersionFormat:        number
  VersionRaw:           0x37c42f06
  VersionLowestRaw:     0x00000001
  Created:              2024-03-22
  UpdateState:          success

14:50:50.712 FuPluginTest         destroy

14:50:50.713 FuProgress           raw timing data was { 0.000, 0.003, 0.001, 0.010, 0.000 } -- steps were set as [ 1 8 64 26 1 ] but should have been [ 0 19 6 71 3 ] at ../fwupd-git/plugins/uefi-capsule/fu-uefi-capsule-plugin.c:1086
```

</details>

superm1, 2024-03-22:
Looks like it's getting the right volume.
```
  EspId:                /org/freedesktop/UDisks2/block_devices/nvme0n1p1
  EspPath:              /boot/efi
  EspKind:              c12a7328-f81f-11d2-ba4b-00a0c93ec93b
```

hughsie, 2024-03-22:
That looks a bit nicer. I think that's one problem down -- now I wonder why the Boot0000 entry is being deleted. Is there anything in the BIOS about boot order lock or anything else that's "recovering" the boot options?

superm1, 2024-03-22:
> It's weird, isn't it supposed to need https://github.com/fwupd/fwupd/pull/6962 for that?

The branch just got rebased onto main to fixup some feedback.  So the test has both commits in place.  You can revert that other one locally to confirm the behavior without it.

superm1, 2024-03-22:
> That looks a bit nicer. I think that's one problem down -- now I wonder why the Boot0000 entry is being deleted. Is there anything in the BIOS about boot order lock or anything else that's "recovering" the boot options?

I at least have a theory.  Perhaps because the file wasn't installed to the correct place the BIOS looked at all entries and saw an -ENOENT condition and deleted the entry.

darkbasic, 2024-03-22:
> now I wonder why the Boot0000 entry is being deleted

I had this issue alot with `sedutil`: basically most UEFI bios automatically get rid of entries if their path doesn't exist during the boot process. This is a big issue with self encrypting drives because they show a different layout when encrypted in order to boot the decrypting payload and at the next boot it shows the real decrypted disk layout, but at that point the uefi entry is already gone and so have no way to boot your system. Somehow some distros managed to work around that issue, for example Fedora: https://github.com/Drive-Trust-Alliance/sedutil/issues/409
Ironically when I need persisting entries on buggy bios I install fedora first so I have a persisting entry. No idea how they managed to achieve that.

 > You can revert that other one locally to confirm the behavior without it.

I will.

darkbasic, 2024-03-22:
I've compiled main minus https://github.com/fwupd/fwupd/pull/6962/commits/19865ebfebd84c92cd034275536038a7e63ea10e which I have reverted. Here is the result:

<details>

<summary>sudo fwupdtool get-devices --plugins uefi_capsule -vv with main plus Revert "Ignore ZFS zvols when finding the default ESP"</summary>

```shell
[niko@arch-phoenix ~]$ sudo fwupdtool get-devices --plugins uefi_capsule -vv
15:14:08.179 FuDebug              verbose to debug (on console 1)
15:14:08.180 FuEngine             starting fwupd 2.0.0…
15:14:08.180 FuMain               locked /run/lock/fwupdtool
15:14:08.181 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
15:14:08.183 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…                 [ -                                     ]15:14:08.183 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
15:14:08.183 FuConfig             loading config /etc/fwupd/fwupd.conf
15:14:08.183 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
15:14:08.183 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
15:14:08.183 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
15:14:08.183 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
15:14:08.183 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
15:14:08.183 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
15:14:08.183 FuConfig             ::configuration loaded
15:14:08.183 FuRemoteList         loading /usr/share/fwupd/metainfo
15:14:08.184 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
15:14:08.184 XbSilo               file: 7219fb74-f275-0692-379a-f6d95f6f7a98, current:14836b2a-0679-64c1-608d-b7133f14d629, cached: (null)
15:14:08.217 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
15:14:08.217 FuRemoteList         ::remote_list changed
15:14:08.217 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
15:14:08.217 FuRemoteList         ::remote_list changed
15:14:08.217 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
15:14:08.217 FuRemoteList         ::remote_list changed
15:14:08.217 FuEngine             priority local and vendor-directory is not download remote, so bumping
15:14:08.217 FuRemoteList         ordering lvfs-testing=lvfs+1
15:14:08.217 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…                 [                                       ]15:14:08.217 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:14:08.217 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
15:14:08.217 Jcat                 loaded 1 certificates
15:14:08.217 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
15:14:08.217 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
15:14:08.217 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:14:08.217 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
15:14:08.217 Jcat                 loaded 1 certificates
15:14:08.217 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
15:14:08.217 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
15:14:08.217 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]15:14:08.217 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
15:14:08.218 FuHistory            got schema version of 12
Loading…                 [*                                      ]15:14:08.221 FuPlugin             load(/usr/lib/fwupd-2.0.0/libfu_plugin_modem_manager.so)
15:14:08.222 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
15:14:08.222 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
15:14:08.222 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
15:14:08.222 XbSilo               file: 6415fd21-f348-6f90-1a32-dd431324c33d, current:522f0467-a55b-3417-975c-8ce370cdec32, cached: (null)
Loading…                 [*                                      ]15:14:08.234 FuConfig             loading config /etc/fwupd/fwupd.conf
15:14:08.234 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
15:14:08.234 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
15:14:08.234 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
15:14:08.234 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
15:14:08.234 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
15:14:08.234 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
15:14:08.234 FuConfig             ::configuration loaded
15:14:08.234 FuStruct             FuStructSmbiosEp64:
  anchor_str: _SM3_
  entry_point_csum: 0xad
  entry_point_len: 0x18
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x4
  smbios_docrev: 0x0
  entry_point_rev: 0x1
  reserved0: 0x0
[... cut ...]
/run/media/root/2230-C418/EFI/ubuntu/grubx64.efi
/run/media/root/2230-C418/EFI/ubuntu/shimx64.efi
/run/media/root/2230-C418/EFI/ubuntu/mmx64.efi
/run/media/root/2230-C418/EFI/ubuntu/BOOTX64.CSV
/run/media/root/2230-C418/EFI/ubuntu/grub.cfg
/run/media/root/2230-C418/EFI/BOOT/BOOTX64.EFI
/run/media/root/2230-C418/EFI/BOOT/fbx64.efi
/run/media/root/2230-C418/EFI/BOOT/mmx64.efi
/run/media/root/2230-C418/EFI/arch/fw/fwupd-8cf4e0ec-a249-4c1f-9cdf-bbe85a9096e4.cap
/run/media/root/2230-C418/EFI/arch/fwupdx64.efi
```

darkbasic, 2024-03-22:
> I wonder if we forced another index it would work?

No idea honestly.

superm1, 2024-03-22:
```
15:14:08.604 FuPluginUefiCapsule  found /run/media/root/2230-C418/EFI/ubuntu/grubx64.efi which indicates a Linux ESP, using /run/media/root/2230-C418
15:14:08.604 FuVolume             unmounting /org/freedesktop/UDisks2/block_devices/zd0p1
15:14:08.625 FuPluginUefiCapsule  found /boot/efi/EFI/zbm/zfsbootmenu-recovery-vmlinuz-x86_64.EFI which indicates a Linux ESP, using /boot/efi
```

They both now got marked as ESPs, so yes I would call that a success.  We don't have a way to tell which one is more valuable, so https://github.com/fwupd/fwupd/commit/19865ebfebd84c92cd034275536038a7e63ea10e is then correct to exclude the ZFS one.

Thanks for the check!

I've got a PR opened to backport all the relevant stuff to 1_9_X branch.

hughsie, 2024-03-22:
> No idea honestly.

If the firmware still removes the entry, could you try this on top of the other two please:

	diff --git a/plugins/uefi-capsule/fu-uefi-bootmgr.c b/plugins/uefi-capsule/fu-uefi-bootmgr.c
	index 769286f83..dd13ee168 100644
	--- a/plugins/uefi-capsule/fu-uefi-bootmgr.c
	+++ b/plugins/uefi-capsule/fu-uefi-bootmgr.c
	@@ -210,7 +210,7 @@ fu_uefi_setup_bootnext_with_loadopt(FuEfiLoadOption *loadopt,
			/* create a new one */
		} else {
			g_autofree gchar *boot_next_name = NULL;
	-               for (guint16 value = 0; value < G_MAXUINT16; value++) {
	+               for (guint16 value = 1; value < G_MAXUINT16; value++) {
				if (set_entries[value])
					continue;
				boot_next = value;

darkbasic, 2024-03-22:
@hughsie are you sure you want me to add that patch on top of the other PRs? If I do so it will point to the correct esp and thus 
the entry won't probably be deleted. Would be better to test your patch on top of 1.9.15 so it will end up using the zvol instead.

superm1, 2024-03-22:
Test it as is right now please without adding the patch to change the index to use. I would like to see if putting the file in the right place was enough to fix everything.

superm1, 2024-03-22:
I've got a bad feeling this might be a separate regression in main from some new systems lockdown stuff added very recently.

darkbasic, 2024-03-22:
Can you please point me to the 1.x backport branch so I can rule this out?

superm1, 2024-03-22:
1_9_X branch has all the backports.

```
diff --git a/data/fwupd.service.in b/data/fwupd.service.in
index 2a0187f09..2e8ce9708 100644
--- a/data/fwupd.service.in
+++ b/data/fwupd.service.in
@@ -18,7 +18,6 @@ MemoryDenyWriteExecute=yes
 NoNewPrivileges=no
 PrivateDevices=no
 PrivateTmp=true
-ProcSubset=pid
 ProtectClock=yes
 ProtectControlGroups=yes
 ProtectHome=yes
```

I think that fixes the regression in main hopefully.

darkbasic, 2024-03-22:
The regression seems to be fixed:
[image]

Unfortunately now the laptop is stuck at this screen since several minutes, I have no idea what it is doing and I fear to brick it if I turn it off and it's doing something:

[image]

hughsie, 2024-03-22:
Give it 30 minutes; it could be you're missing gnu-efi fixes or something.

superm1, 2024-03-22:
> The regression seems to be fixed with your patch:

OK great.  We just merged that to main, thanks.

darkbasic, 2024-03-22:
The good news it that I didn't brick anything. The bad news is that it didn't do anything either :)
I've tried to manually open the boot menu via the F9 key and I selected the Linux Firmware Updater entry but this time I've got just a black screen. I have no idea how to debug this issue further.

> it could be you're missing gnu-efi fixes or something.

Could you please elaborate?

superm1, 2024-03-22:
> The good news it that I didn't brick anything.
:smile: 

>  I have no idea how to debug this issue further.

See this part of the conffile:
https://fwupd.github.io/libfwupdplugin/fwupd.conf.html#uefi_capsule-parameters

We have a key you can add `EnableEfiDebugging=true` to a `[uefi_capsule]` section and this will capture debug logging for the EFI binary when you trigger an update with it.

So add that key, restart the daemon and then trigger an update.
On the next boot `fwupdmgr get-results` should hopefully have more debugging information for us on what exactly transpired.

darkbasic, 2024-03-22:
```shell
[niko@arch-phoenix ~]$ sudo cat /etc/fwupd/fwupd.conf 
[sudo] password for niko: 
[fwupd]

[uefi_capsule]
EnableEfiDebugging=true
# use `man 5 fwupd.conf` for documentation
```

```shell
[niko@arch-phoenix ~]$ fwupdmgr get-results
0.	Cancel
1.	0ea18213457780129dd1fbc1201577be52c0f20d (ELAN07A9:00 04F3:31ED)
2.	1382b870dac684d48d4e1444cb42027c2f0a62af ((null))
3.	c04dba7a46022e62363e93839c86a9bcf1cb708a ((null))
4.	10feb8f8641be9b4f44d52078baa6338209a53ef ((null))
5.	4bde70ba4e39b28f9eab1628f9dd6e6244c03027 (AMD Ryzen 9 PRO 7940HS w/ Radeon 780M Graphics)
6.	936bcc6bcdeab99e67ccb956fa03abec80577314 (DMC)
7.	f685512aa07369c9e77742acef941d779d31e766 (GPIO controller)
8.	983c3cffc6fd36d32b00b62928d30721eaeb93db (HP 5MP Camera)
9.	23ec719b6aabc2d2dac5176c232f0da7a21881b0 (Prometheus)
10.	1c9da4ef563b0ccf0e36e409e3281a1ac62309af (Prometheus IOTA Config)
11.	c54ab0237d7a8db8c717b68e0be78e4374a2a079 (Secure Processor)
12.	307a00f856126b8dca325acd8b542cd28c9002c3 (System Firmware)
13.	db0330716216c629bb2c07256e5d018f499eb6ce (System Management Unit (SMU))
14.	c6a80ac3a22083423992a3cb15018989f37834d6 (TPM)
15.	7a23304c3da50930ca73b689adb34f75dd3ede2e (Thunderbolt Dock G4)
16.	904f9e062b08a03a7337f7f6cbff53b799344c3e (UEFI Device Firmware)
17.	b970edafffcf075cf953e5efebbf2595bf98f1db (UEFI Device Firmware)
18.	e97608be1da5b9d742670ca8c3c4f515bc10646a (UEFI Device Firmware)
19.	6924110cde4fa051bfdc600a60620dc7aa9d3c6a (UEFI Platform Key)
20.	362301da643102b9f38477387e2193e57abaa590 (UEFI dbx)
21.	48cdf390f0a1768e3fb0bcb10af8fcfb38e5d89e (USB3.0 Hub)
22.	03281da317dccd2b18de2bd1cc70a782df40ed7e (WD BLACK SN850X 4000GB)
Choose device [0-22]: 12
System Firmware:
  Device ID:            307a00f856126b8dca325acd8b542cd28c9002c3
  Previous version:     0x01030900
  Update State:         Failed
  Update Error:         failed to run update on reboot: expected 0x01040a00 and got 0x01030900
  Last modified:        2024-03-22 17:47
  GUID:                 8cf4e0ec-a249-4c1f-9cdf-bbe85a9096e4
  Device Flags:         • Internal device
                        • Updatable
                        • System requires external power source
                        • Supported on remote server
                        • Needs a reboot after installation
                        • Cryptographic hash verification is available
                        • Device is usable for the duration of the update
```

```shell
[niko@arch-phoenix ~]$ fwupdmgr get-updates
Devices with no available firmware updates: 
 • ELAN07A9:00 04F3:31ED
 • HP 5MP Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • USB3.0 Hub
 • WD BLACK SN850X 4000GB
Devices with the latest available firmware version:
 • Prometheus
 • Prometheus IOTA Config
 • Thunderbolt Dock G4
 • UEFI dbx
────────────────────────────────────────────────
Devices that were not updated correctly:
 • System Firmware (0x01030900 → 0x01040a00)
[... 108 lines cut ...]
        Remote ID:        lvfs
        Release ID:       86733
        Summary:          System Firmware for HP Laptop, family V82
        License:          Proprietary
        Size:             42.2 MB
        Created:          2024-03-01
        Urgency:          High
        Vendor:           HP
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:      
        Fixes and enhancements in System firmware 1.4.10.0:
        
        • Fixes an issue where incorrect characters are generated when the internal keyboard is used.
        • Fixes an issue where the USB keyboard does not function properly on the BitLocker PIN screen when fast boot is enabled.
        • Fixes an issue where the Wi-Fi Password does not appear in the Preboot Wi-Fi Profile variable.
        • Fixes an issue where certain options that are present in UefiBIOSConfig file are not present in the HpSetup.txt file.
        • Fixes an issue where a system using only DC power while flashing the BIOS through the Firmware Update and Recovery Utility boots to the operating system and then auto restarts unexpectedly.
        
        Please refer hp.com.
        Issue:            CVE-2023-5410
        Checksum:         51fd079072369738a0a1d6215ac5bb96352fb846d54e7bf8e3fae7fe52d5ff3c
```

By the way I've switched to the 1.9.x backports branch.

superm1, 2024-03-22:
Hmm, that's missing the EFI debug log variable.  Or it really didn't launch.  @hughsie isn't that how it's supposed to work?

superm1, 2024-03-23:
See https://github.com/fwupd/fwupd/issues/6976

Can you check if downgrading fwupd-efi or upgrading gnu-efi helps?

darkbasic, 2024-03-23:
gnu-efi isn't even installed on my system, so I guess it's a build-time dependency? I can try to downgrade fwupd-efi 1.5 to 1.4, but first I'll try to rebuild it against a newer gnu-efi.

superm1, 2024-03-23:
Sounds good.
Yeah it's a build time dependency.

darkbasic, 2024-03-23:
Upgrading gnu-efi to 3.0.18 and rebuilding fwupd-efi 1.5 against it did the trick:

[image]

Unfortunately that doesn't mean the fourth issue was the last one because the new firmware update somehow broke compatibility with the ethernet on my Thunderbolt Dock G4:

[image]

But that's not your fault if HP uploads untested crap on LVFS. I will try powering off both the laptop and the dock for a couple of minutes and see if it helps, it's not the first time I've seen such weird interactions after firmware upgrades.

darkbasic, 2024-03-23:
I will let you know if I manage to get the ethernet back working but for the moment I'm closing this, thanks for your help!

closed 2024-03-23 by darkbasic

darkbasic, 2024-03-24:
As I suspected detaching the power cord from the dock and powering off the laptop was enough to fix the issue with the network card. Last time I updated the firmware it was displayport multi-stream which didn't work, it's really weird how it somehow keeps track of the previous state and becomes incompatible with the new firmware until a full power cycle.

darkbasic, 2024-03-31:
[image]

Do you have any insight why 1.4.10.0 has been removed from LVFS?

darkbasic, 2024-03-31:
Weird, I can still see it here: https://fwupd.org/lvfs/devices/com.hp.laptop.system.V82.firmware

I would have bet that Gnome Firmware showed a `Reinstall` button before.
Is it possible that some firmware can be reinstalled (like the following) and others not?

[image]

superm1, 2024-03-31:
The firmware might not allow a downgrade.
This policy is decided by the OEM.
This usually happens with security issues.

darkbasic, 2024-03-31:
You can downgrade (there is a bios config which allows you to do so). I am sure because I've recently done so to debug a regression which ended up being a bug in Arch Linux's mkinitcpio. Also as you can see the button to downgrade is not graded out, but the reinstall button is missing altogether. Maybe system firmware cannot be reinstalled to the same version?

superm1, 2024-03-31:
Can you please show output for `fwupdmgr get-releases`? That should help clarify what's actually going on.

superm1, 2024-03-31:
Also does `fwupdmgr reinstall` not work? If it works there this is a GNOME firmware bug not a fwupd bug.

[18 more replies]