fwupd/fwupd #6976: Stuck when rebooting and updating firmware `failed to run update on reboot: expected` [T480s/P14s Gen4 AMD]

saikocat, 2024-03-23:
**Describe the bug**
I was attempting to install updates on my 2 machines  [T480s/P14s Gen4 AMD], but both of them exhibits the same issue with different updating version. For e.g my P14s Gen4 AMD:
` Update Error:       failed to run update on reboot: expected 0.1.18 and got 0.1.16`

**Steps to reproduce**
* `fwupdmgr get-updates`
* `fwupdmgr update` / `fwupdmgr update <device-id>`
* Reboot

After selecting yes for rebooting, the screen was just stuck the LENOVO Logo for 30 minutes.  I have to hold down the power button to restart and get back to my OS.

**fwupd version information**
```
fwupdmgr --version                                                                                                                                           [0]
compile   com.hughsie.libxmlb           0.3.15
compile   org.freedesktop.Passim        0.1.5
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.15
runtime   org.freedesktop.fwupd-efi     1.5
compile   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.Passim        0.1.5
runtime   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.8.1-arch1-1
runtime   org.freedesktop.fwupd         1.9.15
```

Installed using pacman

**Output of `fwupdmgr get-history`**

```
─System Firmware:
  │   Device ID:          f95c9218acd12697af946874bfe4239587209232
  │   Previous version:   0.1.16
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot: expected 0.1.18 and got 0.1.16
  │   Last modified:      2024-03-23 04:32
  │   GUID:               18cced7e-d108-41c4-9189-b0355c41450d
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T14/P14s Gen 4 T16/P16s Gen 2 AMD System Update:
        New version:      0.1.18
        Remote ID:        lvfs
        Release ID:       81520
        Summary:          Lenovo ThinkPad T14/P14s Gen 4 T16/P16s Gen 2 AMD System Firmware
        License:          Proprietary
        Size:             68.3 MB
        Created:          2021-07-13
        Urgency:          High
        Vendor:           Lenovo
```

**System UEFI configuration**
Please provide the output of the following commands:

```
$ efibootmgr -v
<Command not found cos I am using systemd-boot>

$ efivar -l | grep fw
<EMPTY>
```

**Boot tree / systemd-boot**
```
$ tree /boot/
/boot/
├── amd-ucode.img
├── EFI
│   ├── arch
│   │   ├── fw
│   │   └── fwupdx64.efi
│   ├── BOOT
│   │   └── BOOTX64.EFI
│   ├── Linux
│   └── systemd
│       └── systemd-bootx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── loader
│   ├── entries
│   │   └── arch.conf
│   ├── entries.srel
│   ├── loader.conf
│   └── random-seed
└── vmlinuz-linux
```

**Additional questions**
 * Is this a regression?
    Yes, I was able to install firmwares last December 2023
* Are you using an NVMe disk?
    Yes
* Is secure boot enabled?
    No
*  Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
    No

superm1, 2024-03-23:
By chance did you have a USB key or SD card plugged in?

saikocat, 2024-03-23:
> By chance did you have a USB key or SD card plugged in?

No, I didn't have anything plugged in except the usb-c power port

hughsie, 2024-03-23:
> runtime   org.freedesktop.fwupd-efi     1.5

If you downgrade to fwupd-efi 1.4 does it work? If so, what gnu-efi version are you using?

saikocat, 2024-03-23:
> > runtime   org.freedesktop.fwupd-efi     1.5
> 
> If you downgrade to fwupd-efi 1.4 does it work? If so, what gnu-efi version are you using?

I don't have `gnu-efi` installed.

Downgrade to `fwupd-efi 1.4` solved the problem @hughsie . Installation went smoothly! So I guess it's a regression of 1.5 version then.

hughsie, 2024-03-23:
Nah, it's probably that gnu-efi needs to be upgraded as a build requirement.

saikocat, 2024-03-23:
I see. I think Arch build is using gnu-efi `3.0.17-3`, which is 9 months. And `3.0.18` was just release 24 hours at sourceforge.

darkbasic, 2024-03-23:
> Nah, it's probably that gnu-efi needs to be upgraded as a build requirement.

That did the trick for me.

darkbasic, 2024-03-23:
I will open a bug report on Arch Linux.

darkbasic, 2024-03-23:
https://gitlab.archlinux.org/archlinux/packaging/packages/fwupd-efi/-/issues/1

saikocat, 2024-03-24:
Fixed upstream in Arch today

closed 2024-03-24 by saikocat

hughsie, 2024-03-24:
@superm1 I'm thinking we should probably pull out all the fallback stuff from fwupd-efi (and bump the dep) now there's a nice new 3.0.18 tarball release with all the right bits in.

superm1, 2024-03-24:
What I'm wondering is _why_ the fallback stuff isn't working. Wasn't the point of it to avoid this?

If we can't make it work it's going to make things like RHEL harder to upgrade.

But otherwise yeah I think that's logical.

hughsie, 2024-03-25:
> What I'm wondering is why the fallback stuff isn't working. Wasn't the point of it to avoid this?

Yes, I think we're missing at least https://github.com/fwupd/fwupd-efi/pull/55 although there appears to be one more bit that I don't understand yet. Between the Fedora gnu-efi "fork", the fwupd-efi fork (it's diverged quite a bit now!) and upstream it's all quite a mess.

> If we can't make it work it's going to make things like RHEL harder to upgrade.

If I rebase fwupd-efi in RHEL 10 I'd do it on the precondition of gnu-efi too. I think we just have to remove all the forks and have one canonical (if you pardon the pun) location.