fwupd/fwupd #6977: incorrect 'New version' value from command 'get-history' command if using fwupd edge version

vincent-tang1, 2024-03-25:
**Describe the bug**

**Using "snap refresh fwupd --channel=edge" version**
* there is a issue : incorrect 'new version' infomation from command 'get-history' command after successfully updated
```
# snap list
fwupd   1.9.10-604-g239b34e1a    5852   latest/edge      richard-hughes  -
```

**Using "snap refresh fwupd --channel=stable" version (1.9.15 version)**
* no issue : can show correctly 'new version' infomation from command 'get-history' command after successfully updated
```
# snap list
fwupd   1.9.15                   5793   latest/stable    richard-hughes  -
```

**Steps to Repro duce**
Steps to reproduce the behavior.
1. get current device version, as 1.03 for example
```
# fwupdmgr get-devices
├─FWA6172 Platform Management:
│     Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
│     Summary:            Redfish host bus adapter
│     Current version:    1.03.0000
│     Vendor:             Advantech (REDFISH:ADVANTECH)
│     GUIDs:              10598c6b-6f24-518d-9065-bb948baa4907 ← REDFISH\VENDOR_Advantech&SOFTWAREID_894feb65-2ca8-4378-a5d3-41c78ff5b59f-BMC
│                         69ec5e09-0466-5679-83f5-0dc75e3ff4d5 ← REDFISH\VENDOR_Advantech&ID_BMC
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
...
```
2. update device version from 1.03 to 1.01
```
# fwupdmgr install
WARNING: This package has not been validated, it may not work properly.
Selected device: FWA6172 Platform Management
0.      Cancel
1.      1.03.0000
2.      1.01.0000
3.      0.99.0000
4.      0.97.0000
Choose release [0-4]: 2
╔══════════════════════════════════════════════════════════════════════════════╗
║ Downgrade FWA6172 Platform Management from 1.03.0000 to 1.01.0000?           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This unstable release fixes the following issues:                            ║
║                                                                              ║
║ • Fix bug for test 1                                                         ║
║ • Improve performance for test 1                                             ║
║                                                                              ║
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ • Add support for test 1                                                     ║
║                                                                              ║
║ FWA6172 Platform Management and all connected devices may not be usable      ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Waiting…                 [***************************************] Less than one minute remaining…
Successfully installed firmware
```
3. after successully updated, use 'get-history' command, and can see incorrect 'New version' value
```
# fwupdmgr get-history
WARNING: This package has not been validated, it may not work properly.
Advantech FWA-6172
│
└─FWA6172 Platform Management:
  │   Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
  │   Previous version:   1.03.0000
  │   Update State:       Success
  │   GUID:               10598c6b-6f24-518d-9065-bb948baa4907
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Reported to remote server
  │
  └─FWA6172 Platform Management BMC Update:
        New version:      1.03.0000
        Remote ID:        lvfs-testing
        Release ID:       82544
        Summary:          Firmware for the Advantech FWA6172 BMC
        Variant:          sku:10598c6b
        License:          Proprietary
```

4. However, use "get-devices" command, it still can show correct version
```
# fwupdmgr get-devices
...
├─FWA6172 Platform Management:
│     Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
│     Summary:            Redfish host bus adapter
│     Current version:    1.01.0000
│     Vendor:             Advantech (REDFISH:ADVANTECH)
│     Update State:       Success
│     GUIDs:              10598c6b-6f24-518d-9065-bb948baa4907 ← REDFISH\VENDOR_Advantech&SOFTWAREID_894feb65-2ca8-4378-a5d3-41c78ff5b59f-BMC
│                         69ec5e09-0466-5679-83f5-0dc75e3ff4d5 ← REDFISH\VENDOR_Advantech&ID_BMC
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server

```

**Expected behavior**
Follow above step, it should show correct 'New version' value as below (Note: stable version 1.9.15 => no this issue)
```
# fwupdmgr get-history
WARNING: This package has not been validated, it may not work properly.
Advantech FWA-6172
│
└─FWA6172 Platform Management:
  │   Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
  │   Previous version:   1.03.0000
  │   Update State:       Success
  │   GUID:               10598c6b-6f24-518d-9065-bb948baa4907
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Reported to remote server
  │
  └─FWA6172 Platform Management BMC Update:
        New version:      1.01.0000
        Remote ID:        lvfs-testing
        Release ID:       82544
        Summary:          Firmware for the Advantech FWA6172 BMC
        Variant:          sku:10598c6b
        License:          Proprietary
        ...
```

Seems like using 1.9.15, no this issue. But using latest edge version, will appear this issue

hughsie, 2024-03-25:
On a hunch, can you try the hughsie/history-release-flags branch please. It's included in https://github.com/fwupd/fwupd/pull/6979

vincent-tang1, 2024-03-26:
Seems like it still have same issue as above "Steps to Reproduce" after using hughsie/history-release-flags branch

```
(fwupd) root@FWA-6172-efi:~/third-party/fwupd-branch-history-release-flags/fwupd# git branch
* hughsie/history-release-flags
(fwupd) root@FWA-6172-efi:~/third-party/fwupd-branch-history-release-flags/fwupd# fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         2.0.0
runtime   org.freedesktop.fwupd-efi     2.0.0
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.5.0-10-generic
runtime   org.freedesktop.fwupd         2.0.0
```

hughsie, 2024-03-26:
> # snap list
> fwupd   1.9.10-604-g239b34e1a    5852   latest/edge      richard-hughes  -

@superm1 latest/edge is 1.9.10?!

>  └─FWA6172 Platform Management BMC Update:
>        New version:      1.03.0000
>        Remote ID:        lvfs-testing
>        Release ID:       82544

@vincent-tang1 -- that's indeed the correct component ID for the 1.03.0000 update. The way fwupd matches the history release to the remote release is with the checksum of the archive. On the broken system, can you try:

 * stopping fwupd
 * sudo rm -rf /var/lib/fwupd/pending.db (or whatever the path is on snap)
 * do the downgrade from 1.03.0000 to 1.01.0000
 * attach the pending.db file

Thanks!

superm1, 2024-03-26:
No, that's just because we don't have a tag on main since 1.9.10. it's what `git describe` says.

vincent-tang1, 2024-03-27:
> * stopping fwupd
> * sudo rm -rf /var/lib/fwupd/pending.db (or whatever the path is on snap)
> * do the downgrade from 1.03.0000 to 1.01.0000
> * attach the pending.db file

Here is pending.db file (from /var/snap/fwupd/common/var/lib/fwupd)
[pending.db.txt](https://github.com/fwupd/fwupd/files/14767090/pending.db.txt)
 (Please recovery file with "# mv pending.db.txt  pending.db")

```
======================= New command
root@FWA-6172-efi:/var/snap/fwupd/common/var/lib/fwupd# snap list
fwupd   1.9.10-616-g63ae2620d    5859   latest/edge      richard-hughes  -

======================= New command
root@FWA-6172-efi:/var/snap/fwupd/common/var/lib/fwupd# fwupdmgr downgrade
WARNING: This package has not been validated, it may not work properly.
Selected device: FWA6172 Platform Management
0.      Cancel
1.      1.01.0000
2.      0.99.0000
3.      0.97.0000
Choose release [0-3]: 1
╔══════════════════════════════════════════════════════════════════════════════╗
║ Downgrade FWA6172 Platform Management from 1.03.0000 to 1.01.0000?           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This unstable release fixes the following issues:                            ║
║                                                                              ║
║ • Fix bug for test 1                                                         ║
║ • Improve performance for test 1                                             ║
║                                                                              ║
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ • Add support for test 1                                                     ║
║                                                                              ║
║ FWA6172 Platform Management and all connected devices may not be usable      ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Waiting…                 [***************************************] Less than one minute remaining…
Successfully installed firmware
Successfully uploaded 1 report

======================= New command
root@FWA-6172-efi:/var/snap/fwupd/common/var/lib/fwupd# fwupdmgr get-history
WARNING: This package has not been validated, it may not work properly.
Advantech FWA-6172
│
└─FWA6172 Platform Management:
  │   Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
  │   Previous version:   1.03.0000
  │   Update State:       Success
  │   GUID:               10598c6b-6f24-518d-9065-bb948baa4907
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Reported to remote server
  │
  └─FWA6172 Platform Management BMC Update:
        New version:      1.03.0000
        Remote ID:        lvfs-testing
        Release ID:       82544
        Summary:          Firmware for the Advantech FWA6172 BMC
        Variant:          sku:10598c6b
        License:          Proprietary
        Size:             14.7 MB

======================= New command
root@FWA-6172-efi:/var/snap/fwupd/common/var/lib/fwupd# fwupdmgr get-devices
├─FWA6172 Platform Management:
│     Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
│     Summary:            Redfish host bus adapter
│     Current version:    1.01.0000
│     Vendor:             Advantech (REDFISH:ADVANTECH)
│     Update State:       Success
│     GUIDs:              10598c6b-6f24-518d-9065-bb948baa4907 ← REDFISH\VENDOR_Advantech&SOFTWAREID_894feb65-2ca8-4378-a5d3-41c78ff5b59f-BMC
│                         69ec5e09-0466-5679-83f5-0dc75e3ff4d5 ← REDFISH\VENDOR_Advantech&ID_BMC
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server

======================= New command
root@FWA-6172-efi:/var/snap/fwupd/common/var/lib/fwupd# ipmitool hpm check

PICMG HPM.1 Upgrade Agent 1.0.9:

-------Target Information-------
Device Id          : 0xcb
Device Revision    : 0x81
Product Id         : 0x0000
Manufacturer Id    : 0x2839 (Advantech)

--------------------------------------------------------------------------
|ID  | Name        |                     Versions                        |
|    |             |     Active      |     Backup      |      Deferred   |
--------------------------------------------------------------------------
|   0| BL          |   0.30 00000000 | ---.-- -------- | ---.-- -------- |
|   1| BMC         |   1.01 00000000 |   1.03 00000000 | ---.-- -------- |
|   2| BMCONF      |   1.02 00000000 | ---.-- -------- | ---.-- -------- |
|*  3| FPGA        |   0.12 00000000 | ---.-- -------- | ---.-- -------- |
|*  4| BIOS        |   1.00 00000000 |   1.00 00000000 | ---.-- -------- |
|*  5| NVRAM       |   3.00 00000000 | ---.-- -------- | ---.-- -------- |
--------------------------------------------------------------------------
(*) Component requires Payload Cold Reset

```

hughsie, 2024-03-28:
Looking at your pending db:

    checksum='2ee448b1f2aaa3e2eb8c38675d71abe36ee7b062'

weirdly, this is the checksum of the 1.01.0000 downgrade -- which is correct. 

> Release ID:       82544

This is 1.03.0000...

I've spent quite a bit of time trying to reproduce this on a local device; but it's all worked perfectly so far.

If the release ID is being set correctly, then that maybe suggests that fu_engine_get_component_for_checksum() is returning the wrong XbNode component, as I don't see how else that remote ID would be known.

I'm nearly out of time before the easter holiday, but if you can't work it out before I get back I'll continue debugging in a few days.

superm1, 2024-03-28:
Something potentially similar I noticed was that the version that fwupdmgr wants to report for a composite device (dell-dock when I was doing the other PR for composite emulation) is totally wrong.  It's taking the version from one component and saying it should report all of them that version which tells me something got messed up in the history database.

I suspect it's fallout from https://github.com/fwupd/fwupd/pull/6979

hughsie, 2024-04-05:
@vincent-tang1 on a hunch, if you do `fwupdmgr get-history --json`  what do you see? Two releases perhaps?

hughsie, 2024-04-05:
Okay, I can reproduce with your exact pending database -- with --json it's obvious:

          "Checksum" : [
            "2ee448b1f2aaa3e2eb8c38675d71abe36ee7b062",
            "99e2466662c341a8aaccb61083e2d9f37b196e1a",
            "7b20ac6db5cf8307013736d6d111b0b7488b19aa589961680bb5ae4cbd80fbf3"
          ],

debugging now.

hughsie, 2024-04-05:
Worked it out! Thanks for all the debugging, especially for sharing the pending.db -- much appreciated.

closed 2024-04-05 by hughsie