fwupd/fwupd #6994: Failed to download due to server limit

Schlentzen, 2024-03-27:
**Describe the question**
I am trying to update a pool of Devices but every time i get the Error:
Failed to download due to server limit

**fwupd version information**
```
fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.15
compile   org.freedesktop.Passim        0.1.5
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.15
runtime   org.freedesktop.fwupd-efi     1.5
compile   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.Passim        0.1.5
runtime   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.8.1-arch1-1
runtime   org.freedesktop.fwupd         1.9.15
```

The Problem stays the same no matter what device i use

hughsie, 2024-03-27:
What's the public IP you're using to download please?

Schlentzen, 2024-03-27:
> What's the public IP you're using to download please?

193.175.84.201

hughsie, 2024-03-27:
Th tl;dr: is:

[image]

[image]

I've deleted the ACL that blocked updates -- but please also be careful of the 250/day limit for the same firmware file.

closed 2024-03-27 by hughsie

hughsie, 2025-11-05:
> 71.127.49.162

Blocked 20411 times, last seen 12 hours ago. It's downloading the same Dell Precision 7X30 firmware over and over again.

[image]

hughsie, 2025-11-05:
> I believe I have corrected the problem

Out of morbid curiosity, what was the problem please?

proximous, 2025-11-05:
:) I created a firewall rule to block all traffic to fwupd.org and log the IP address. Unfortunately we have around 10 identical systems needing the update, so I'm not yet sure who is the offender, but at least we will stop spamming the site, and the logs should point me to the offender.

hughsie, 2025-11-05:
Okay, ACL deleted -- please try again gently.