fwupd/fwupd #7009: Warn users needing to `fwupdmgr activate` before uploading signed reports

rishabh-chromium, 2024-03-29:
**Is your feature request related to a problem? Please describe.**
We recently tried upload signed reports for WD19 Dock but didn't see it on fwupd.org. After chat with Richard we realized that we need to run `fwupdmgr activate` for some peripherals before uploading signed reports

**Describe the solution you'd like**
Since only few devices need activation before uploading signed reports, it would be ideal to give a warning or message to do so before upload signed reports.

superm1, 2024-04-01:
Wd19 is a bit of a special butterfly. It runs the update when unplugged nominally.

Activate is just the way to force the update to take effect while it's plugged in.

I don't know if it's worth "generalizing" this behavior.

rishabh-chromium, 2024-04-01:
Are there other popular devices that need to be activated before uploading signed reports? If not, then you are right, it's probably not worth investing into getting this behaviour

superm1, 2024-04-01:
@hughsie any others you know of? Dell dock handles wd19, wd22, and one other. So at least from a design perspective you will have all those products that use that plugin.

I'm sure that @CragW will try to keep the same design for follow on products.

Back to the original ask - from a fwupd success vs failure perspective why does it matter if activation is done or not for the signed report?

rishabh-chromium, 2024-04-01:
Not sure why it works when we `fwupdmgr activate`. This is the response from Richard when we emailed him about this:
```
so what might be happening here is that the [old] Dell dock is
weird in that it needs to be "activated" before the firmware gets
deployed. If you manually call "fwupdmgr activate" before doing the
report upload, does that help?
```

CragW, 2024-04-01:
WD19 needs to `activate` the staged updates or physically unplug the dock typec cable to complete the work. Very likely some devices in a state of `needs-activation` at the time you upload the report.

superm1, 2024-04-02:
But why does that block an upload of a report? I would think we should still allow reports at that stage.

hughsie, 2024-04-03:
> But why does that block an upload of a report

The version isn't set to the "new" version and therefore the report doesn't show the result as success.

superm1, 2024-04-03:
> > But why does that block an upload of a report
> 
> The version isn't set to the "new" version and therefore the report doesn't show the result as success.

Got it.

hughsie, 2024-04-04:
@rishabh-chromium can you grab me the output of the device after the update, but **before** the activate step please? i.e. a `fwupdmgr get-devices --json` please.

rishabh-chromium, 2024-04-04:
```
{                                                                                                                                                                                                                  
  "Devices" : [                                                                                                                                                                                                    
    {                                                                                                                                                                                                              
      "Name" : "WD19S",                                                                                                                                                                                            
      "DeviceId" : "981a41a15b4c9689a2d75cc26b87bd4dbee87b67",                                                                                                                                                     
      "InstanceIds" : [                                                                                                                                                                                            
        "USB\\VID_413C&PID_B06E&hub&embedded"                                                                                                                                                                      
      ],                                                                                                                                                                                                           
      "Guid" : [                                                                                                                                                                                                   
        "cd357cf1-40b2-5d87-b8df-bb2dd82774aa"                                                                                                                                                                     
      ],                                                                                                                                                                                                           
      "Serial" : "9DB74S3/3120590235710182",                                                                                                                                                                       
      "Summary" : "High performance dock",                                                                                                                                                                         
      "Plugin" : "dell_dock",                                                                                                                                                                                      
      "Protocol" : "com.dell.dock",                                                                                                                                                                                
      "Flags" : [                                                                                                                                                                                                  
        "supported",                                                                                                                                                                                               
        "registered",                                                                                                                                                                                              
        "needs-activation",                                                                                                                                                                                        
        "dual-image",                                                                                                                                                                                              
        "self-recovery",                                                                                                                                                                                           
        "usable-during-update",                                                                                                                                                                                    
        "updatable-hidden",                                                                                                                                                                                        
        "skips-restart",                                                                                                                                                                                           
        "signed-payload"                                                                                                                                                                                           
[... cut ...]
          "Uri" : "https://storage.googleapis.com/chromeos-localmirror/lvfs/03ff7f61284b6d3c6dcab2644f1cdf701c650ad4270b93590c5eb4f6d8571192-DellDockFirmwareUpdateLinux_01.00.09.cab",
          "Homepage" : "http://support.dell.com/",
          "Vendor" : "Dell",
          "Flags" : [
            "trusted-metadata",
            "is-downgrade"
           ],
          "InstallDuration" : 3
        }
      ]
    }
  ]
}
```
This is the output of `fwupdmgr get-devices --json` after runnig `fwupdmgr update` without unplugging the device

closed 2024-04-05 by hughsie