fwupd/fwupd #7010: HP Elitebook 865 G10 fails to offer release for reinstall

superm1, 2024-04-01:
```
[niko@arch-phoenix ~]$ fwupdmgr get-releases
0.	Cancel
1.	23ec719b6aabc2d2dac5176c232f0da7a21881b0 (Prometheus)
2.	1c9da4ef563b0ccf0e36e409e3281a1ac62309af (Prometheus IOTA Config)
3.	307a00f856126b8dca325acd8b542cd28c9002c3 (System Firmware)
4.	7a23304c3da50930ca73b689adb34f75dd3ede2e (Thunderbolt Dock G4)
5.	362301da643102b9f38477387e2193e57abaa590 (UEFI dbx)
Choose device [0-5]: 3
HP HP EliteBook 865 16 inch G10 Notebook PC
│
└─EliteBook 8x5 G10 Notebook PC V82 System Update:
      New version:        0x01030900
      Remote ID:          lvfs
      Release ID:         76184
      Summary:            System Firmware for HP Laptop, family V82
      License:            Proprietary
      Size:               42.2 MB
      Created:            2023-12-11
      Urgency:            High
      Vendor:             HP
      Release Flags:      • Trusted metadata
                          • Is downgrade
      Description:        
      Fixes and enhancements in System firmware 1.3.9.0:
      
      • Add Gaming Optimize Mode.
      Checksum:           f3fb5a7377bf553436856190b9238e608209f5c549cc82ad416039abd541dadf
```

```
[niko@arch-phoenix ~]$ fwupdmgr reinstall
0.	Cancel
1.	23ec719b6aabc2d2dac5176c232f0da7a21881b0 (Prometheus)
2.	1c9da4ef563b0ccf0e36e409e3281a1ac62309af (Prometheus IOTA Config)
3.	307a00f856126b8dca325acd8b542cd28c9002c3 (System Firmware)
4.	7a23304c3da50930ca73b689adb34f75dd3ede2e (Thunderbolt Dock G4)
5.	362301da643102b9f38477387e2193e57abaa590 (UEFI dbx)
Choose device [0-5]: 3
Unable to locate release 0x01040a00 for System Firmware
```

I fear the issue is not in Gnome Firmware.

_Originally posted by @darkbasic in https://github.com/fwupd/fwupd/issues/6956#issuecomment-2029212096_

superm1, 2024-04-01:
I'm presuming @darkbasic this is 1.9.15 fwupd release right?

Can you please share `sudo fwupdtool get-releases -vv` output?

darkbasic, 2024-04-01:
Should be 1.9.16ish from the backport branch. I'll post the output later today.

superm1, 2024-04-01:
I tried to reproduce it using a different system (Framework 13) on fwupd 1_9_X commit ad2aec9b but that has a firmware on a non standard remote.  I can't reproduce.

```
$ fwupdmgr --version  | grep fwupd
Idle…: 0%
compile   org.freedesktop.fwupd         1.9.16
runtime   org.freedesktop.fwupd-efi     1.4
runtime   org.freedesktop.fwupd         1.9.16
```

```
$ fwupdmgr get-devices --filter can-verify
WARNING: This package has not been validated, it may not work properly.
Framework Laptop 13 (AMD Ryzen 7040Series)
│
├─System Firmware:
│     Device ID:          1e4fa9cea0b89e613370cf9657ffa0b6d3f37fec
│     Summary:            UEFI System Resource Table device (Updated via caspule-on-disk)
│     Current version:    0.0.3.5
│     Minimum Version:    0.0.0.1
│     Vendor:             Framework (DMI:INSYDE Corp.)
│     Update State:       Success
│     GUID:               b5f7dcc1-568c-50f8-a4dd-e39d1f93fda1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
```

```
$ fwupdmgr reinstall 
WARNING: This package has not been validated, it may not work properly.
0.	Cancel
1.	23ec719b6aabc2d2dac5176c232f0da7a21881b0 (Fingerprint Sensor)
2.	1e4fa9cea0b89e613370cf9657ffa0b6d3f37fec (System Firmware)
3.	df8674a3eb6e83570a477b2a0d4a3d181cb93b2e (USB-C Dock G5)
4.	87843528e58918327b0fa67fb460a77b50243d42 (Unifying Receiver)
Choose device [0-4]: 2
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall System Firmware to 0.0.3.5?                            
```

```
$ fwupdmgr get-releases --filter can-verify 
WARNING: This package has not been validated, it may not work properly.
Selected device: System Firmware
Framework Laptop 13 (AMD Ryzen 7040Series)
│
├─Laptop 13 AMD Ryzen 7040 System Update:
│     New version:        0.0.3.5
│     Remote ID:          <redacted>
│     Release ID:         88484
│     Summary:            Framework Laptop 13 System Firmware for AMD Ryzen 7040 Mainboards
│     License:            Proprietary
│     Size:               36.1 MB
│     Created:            2024-03-28
│     Urgency:            High
│     Vendor:             Framework
│     Duration:           2 minutes
│     Release Flags:      • Trusted metadata
│     Description:        
│     <redacted>
│     Checksum:           246a9bcc6b6c0c7d47aabd371a173c9179bd58fe704c3420707367e4473d46fc
```

darkbasic, 2024-04-01:
I'm pretty sure the reinstall button was being rendered before in Gnome Software, otherwise I wouldn't have risked downgrading to test a regression. I suspect HP might have changed something on LVFS.

superm1, 2024-04-01:
Maybe the GUID changed during the upgrade? We certainly need verbose logs to make sense of why things are happening

superm1, 2024-04-01:
Dumb question but are you sure you have the LVFS testing enabled?

Does it show up in `fwupdmgr get-remotes`?

darkbasic, 2024-04-01:
Yes it does.

superm1, 2024-04-01:
> Yes it does.

Specifically as `enabled` right?

superm1, 2024-04-01:
My hypothesis is that the firmware was in the stable channel and you had lvfs-testing disabled.  It rolled out some people installed it and there must have been some number of issues reported that forced it to automatically demote to `lvfs-testing`.  You have `lvfs-testing` disabled so now it doesn't show up.

My ACL at fwupd.org doesn't let me see the details for the HP firmware, so I don't know if this is true.

@hughsie can you check it?

And if it did get demoted - can we see if it was because of a bunch of people running fwupd-efi 1.5?  If so; it should probably be re-promoted because we know that fwupd-efi 1.5 issues are _caused_ by an old gnu-efi.

darkbasic, 2024-04-01:
You're completely right: I saw it was in the remotes but I didn't check if it was enabled because I enabled testing a long time ago and took it for granted. Somehow it got disabled (maybe after the temporary upgrade to 2.0?) and in the meantime the fw has been demoted to the testing branch.
I probably contributed to the demotion myself because I've reported several failed attempts due to the multiple issues we've encountered. With the addition of the far more common issues with gnu-efi it makes totally sense.

hughsie, 2024-04-03:
> can you check it?

[HP EliteBook 8x5 G10 Notebook PC V82 0x1030900](https://fwupd.org/downloads/f3fb5a7377bf553436856190b9238e608209f5c549cc82ad416039abd541dadf-V82_01030900.cab) still looks to be in stable to me -- it was moved from embargo to stable on 2024-02-06. 

> Unable to locate release

Weird. What do you get if you download the .cab file (link in this comment) and then do `sudo fwupdtool install *V82_01030900.cab -vv` -- can you upload the full log to this bug please. Also please the output of `sudo fwupdtool hwids` as well.

superm1, 2024-04-03:
1.4.10.0 is the release we're talking about tho.

hughsie, 2024-04-03:
Ohh my mistake! In that case it's easy; fwupd auto-demoted the firmware back to testing as it had 61% success.

The failures do seem to be of the form "failed to run update on reboot: expected 0x01040a00 and got 0x01030900" -- and both have `RuntimeVersion(org.freedesktop.fwupd-efi)	"1.5"`

Is fwupd-efi ready to be released? If so, I can tag a new build and then do an LVFS known issue for 1.5

superm1, 2024-04-03:
> Is fwupd-efi ready to be released? If so, I can tag a new build and then do an LVFS known issue for 1.5

Is gnu-efi 3.0.18 in Fedora yet?  If so; I think we should make a hard requirement on 3.0.18+ right now.  It was waiting for that.  That's the last thing I think needed.

hughsie, 2024-04-03:
> Is gnu-efi 3.0.18 in Fedora yet

Fedora 41 yes, Fedora 40 is in freeze. The latter I can buildroot-override tho, so go for the hard dep please.

darkbasic, 2024-04-05:
I confirm it works now:
[image]

superm1, 2024-04-16:
fwupd-efi 1.6 is tagged
https://github.com/fwupd/fwupd-efi/releases/tag/1.6

superm1, 2024-04-16:
@hughsie did you get the known issue on 1.5 added?

closed 2024-04-16 by superm1

hughsie, 2024-04-16:
@superm1 something like this?

[image]

hughsie, 2024-04-16:
link points to https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-fwupd%E2%80%90efi-1.5-and-old-gnu%E2%80%90efi if you want to wordsmith that a bit.

superm1, 2024-04-16:
Looks great thanks.

Can you also double check if there are any others that fit the pattern and got auto demoted?

hughsie, 2024-04-16:
That's harder, will do!

hughsie, 2024-04-16:
Only two matched, neither demoted.