fwupd/fwupd #7021: Query on @resources system call

RZR7332, 2024-04-03:
**Describe the question**
Before I go ahead and create a PR to drop the @resources system call, I figured I would first ask if the below are required, rather than potentially wasting everyone's time:

@resources
    # Alter resource settings
    ioprio_set
    mbind
    migrate_pages
    move_pages
    nice
    sched_setaffinity
    sched_setattr
    sched_setparam
    sched_setscheduler
    set_mempolicy
    setpriority
    setrlimit

This is the last one I am uncertain about (fairly confident reboot and swap can be safely dropped).

hughsie, 2024-04-04:
I think all those are probably fine -- and it's certainly not a waste of time -- I appreciate the help!

RZR7332, 2024-04-05:
Awesome - thanks! I shall submit a PR during the course of the day to drop the final three system calls. If merged, I will submit another PR next week to switch from blacklisting to whitelisting. After that, the only thing remaining to try and tackle will be the CapablityBoundingSet strings which may be a bit tricky.

Thanks for all the help and patience thus far, really appreciate it.

closed 2024-04-05 by RZR7332