fwupd/fwupd #7022: StarTech USB-C dock half-functional after update

agarof, 2024-04-03:
**Describe the bug**

Running `fwupdmgr update` with the [DK31C3MNCRUE](https://www.startech.com/en-eu/universal-laptop-docking-stations/dk31c3mncrue) StarTech dock flashes it with an incompatible firmware.

Its usb ports still work but its video outputs don't, and the front status LED stays off.

The dock is recognized by `fwupdmgr` as `Gingerbread` ([lvfs link](https://fwupd.org/lvfs/devices/tw.com.goodway.dock.gingerbread.firmware)), I have pasted its `get-history` entry below.

My guess is that this is due to the dock improperly reporting itself as the reference design it is based on.

**Steps to Reproduce**
Run `fwupdmgr update` with the dock attached

**fwupd version information**
```
compile   org.freedesktop.fwupd         1.9.15
compile   com.hughsie.libxmlb           0.3.15
compile   org.freedesktop.Passim        0.1.5
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.5
runtime   org.freedesktop.Passim        0.1.5
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.8.2-arch2-1
runtime   org.freedesktop.fwupd         1.9.15
```
Installed using `pacman`

<details>

<summary>**fwupd device information**</summary>

├─Gingerbread:
│ │   Device ID:          d5b8513b68f2137d1509b18ffaafd1e86d505892
│ │   Previous version:   2.0.10265
│ │   Update State:       Success
│ │   GUID:               95e8798d-bd6b-5a60-9fb5-79228aea43e6
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Is in bootloader mode
│ │                       • Device stages updates
│ │                       • Signed Payload
│ │ 
│ └─(null) Update:
│       New version:      2.0.11319
│       Remote ID:        lvfs
│       License:          Unknown
│       Description:      
│       The vendor did not supply any release notes.

</details>

**Additional questions**

- Operating system and version: ArchLinux 6.8.2

hughsie, 2024-04-04:
I've emailed Good Way and asked them to comment. Fingers crossed there is an easy fix. Apologies.

hughsie, 2024-04-11:
@agarof can you attach the output of `sudo fwupdtool get-devices --plugins cros-ec -vv` please

hughsie, 2024-04-11:
@bleungatchromium for visibility -- https://fwupd.org/lvfs/firmware/64999 has been demoted back to private while we work out what's going on. It's maybe that StarTech is using the same board name or VID/PID as the Gingerbread device -- but with different firmware.

agarof, 2024-04-11:
Here it is : [output.txt](https://github.com/fwupd/fwupd/files/14944691/output.txt)

hughsie, 2024-04-11:
@bleungatchromium so both devices match the `gingerbread` boardname -- that seems like an overly powerful and generic identifier! Fix incoming.

hughsie, 2024-04-11:
I've also been speaking at length with David at StarTech this morning. He's going to go back to his team and work out what's happened. I'll keep you updated with progress.

closed 2024-04-17 by hughsie

hughsie, 2024-04-18:
@agarof can you send me your email address please (my email is richard_at_hughsie_dot_com)? We might have some good news for you.

agarof, 2024-04-19:
I have made my email address public on my Github profile.