fwupd/fwupd #7030: fwupd.service fails to start or restart in systemd on Fedora 40 Beta

tristan957, 2024-04-04:
**Describe the bug**

`fwupd.service` fails to start and eventually times out. Restarting via `systemctl restart fwupd.service` seemingly hangs systemd. I have to SIGINT the process.

Here are some interesting logs that I have found:

```
-- Boot 6a846db8efc4476c97bf700c62fbcc11 --
Apr 03 14:27:34 gonk fwupd[3836]: 19:27:34.761 FuMain               Daemon ready for requests (locale en_US.UTF-8)
Apr 04 01:01:41 gonk fwupd[3836]: 06:01:41.241 FuPluginUpower       failed to query lid state
Apr 04 01:01:41 gonk fwupd[3836]: 06:01:41.246 FuPluginUpower       failed to query lid state
Apr 04 01:01:41 gonk fwupd[3836]: 06:01:41.276 FuMain               Received SIGTERM
```

**Steps to Reproduce**

Unsure. This only started happening after upgrade to Fedora 40 Beta. After a quick discussion, with the Fedora QA team, they said to report this upstream.

**Expected behavior**

Start the service.

**fwupd version information**

Version information can't be collected because the service doesn't start.

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `dnf`

I can't provide any device information because the service isn't running, but this device is a ThinkPad X1 Carbon (11th gen, I believe).

**Additional questions**

- Operating system and version: Fedora 40 Beta
- Have you tried rebooting? Yes
- Is this a regression? Yes

tristan957, 2024-04-04:
I'm happy to provide any extra information you might need.

superm1, 2024-04-04:
If you manually run the daemon (`/usr/libexec/fwupd/fwupd`) do you also reproduce it?  Or only in systemd?

tristan957, 2024-04-04:
Here is the output as me (regular user)
```
15:16:59.436 FuPluginUefiCapsule  SMBIOS BIOS Characteristics Extension Byte 2 is invalid -- UEFI Specification is unsupported, but /sys/firmware/efi exists: SMBIOS not supported
15:16:59.439 FuPluginTpm          failed to load eventlog: Failed to open file “/sys/kernel/security/tpm0/binary_bios_measurements”: Permission denied
15:16:59.491 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
15:16:59.491 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
15:16:59.491 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
15:16:59.491 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
15:16:59.492 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
15:16:59.492 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
15:16:59.492 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
15:16:59.492 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
15:16:59.492 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
15:16:59.492 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
15:16:59.492 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
15:16:59.492 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
15:16:59.492 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
15:16:59.492 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
15:16:59.492 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
15:16:59.494 FuUsbDevice          failed to load BOS descriptor from USB device: USB error on device 1d6b:0003 : Access denied (insufficient permissions) [-3]
```
With sudo, the process seems to hang, similar to what I was experiencing with `systemctl restart`. No output. Do you have any recommendations for debugging a superuser process? I'd be happy to get a backtrace to see what is going on.

superm1, 2024-04-04:
As a super user can you please run with -vv?

tristan957, 2024-04-04:
Sure. Here is the output.
[fwupd.txt](https://github.com/fwupd/fwupd/files/14873510/fwupd.txt)

tristan957, 2024-04-04:
I personally don't notice any issues there. Do you? I think it could be more systemd related.

Something that I think might be related to this problem, assuming systemd is part of the problem, is that my computer no longer fully shuts down. The power lights near the power button continue to stay on, and I'm forced to do a hard shut down.

superm1, 2024-04-04:
The file link doesn't work for some reason; but check and see if a client (like `fwupdmgr` or `gnome-firmware`) is working.  If it is then we need to debug on the systemd unit.

Add `-vv` to the `ExecStart` line do `systemctl daemon-reload` and try to start it again and then share the journal for `fwupd.service`.

tristan957, 2024-04-04:
Clients are not working due to the service not starting up. Here is `-vv` in the systemd unit.

https://paste.sr.ht/~tristan957/90f8470a660eaf7d78b8b6ea641d4eb5e4c2584b

tristan957, 2024-04-04:
[image]

tristan957, 2024-04-04:
@GeorgeSapkin I see you gave a thumbs-up. Are you experiencing a similar issue?

GeorgeSapkin, 2024-04-04:
> @GeorgeSapkin I see you gave a thumbs-up. Are you experiencing a similar issue?

Same issues with Fedora 40 Beta on T14 Gen 1 AMD. Not a fresh install, but an upgrade going back to whatever Fedora was latest in Fall 2020. All the other symptoms are exactly the same. AFAIR, fwupdmgr used to work just a few days back, so this is something recent.

superm1, 2024-04-04:
Can you please unplug all your USB devices and try again?

If that works can you please identify which one is causing the issue?

tristan957, 2024-04-04:
This problem happens to me with or without being plugged into my dock. @GeorgeSapkin I just tried booting into a 6.8.1 kernel, and didn't have the problem anymore. How about yourself? Right now I seem to only have the problem on 6.8.2.

tristan957, 2024-04-04:
Just rebooted into 6.8.1 to check and back to 6.8.2. 6.8.1 continues to be good. 6.8.2 continues to have the same problems.

superm1, 2024-04-04:
So let me see if I follow right - 6.8.1 fwupd service starts. 6.8.2 it hangs. 

Can you attach gdb and see where it's spinning?

This might need a kernel bisect if we can't tell from gdb.

GeorgeSapkin, 2024-04-04:
> I just tried booting into a 6.8.1 kernel, and didn't have the problem anymore. How about yourself? Right now I seem to only have the problem on 6.8.2.

I can confirm this is issue doesn't manifest itself with 6.8.1, but comes back with 6.8.2.

tristan957, 2024-04-04:
Yes. You are following correctly. Is there anything special I need to do to debug with gdb on a superuser process? Never tried it before.

tristan957, 2024-04-04:
```
#0  0x00007f85f5af3cfa in read () from target:/lib64/libc.so.6
#1  0x00007f85f504ced5 in read_sysfs_attr () from target:/lib64/libusb-1.0.so.0
#2  0x00007f85f504d13d in op_get_active_config_descriptor () from target:/lib64/libusb-1.0.so.0
#3  0x00007f85f504c188 in get_active_config_descriptor () from target:/lib64/libusb-1.0.so.0
#4  0x00007f85f5050d24 in libusb_get_active_config_descriptor () from target:/lib64/libusb-1.0.so.0
#5  0x00007f85f59a0a7a in g_usb_device_get_interfaces () from target:/lib64/libgusb.so.2
#6  0x00007f85f61368cf in fu_usb_device_probe () from target:/usr/lib64/fwupd-1.9.15/libfwupdplugin.so
#7  0x00007f85f60f1759 in fu_device_probe () from target:/usr/lib64/fwupd-1.9.15/libfwupdplugin.so
#8  0x00007f85f62b9112 in fu_engine_backend_device_added () from target:/usr/lib64/fwupd-1.9.15/libfwupdengine.so
#9  0x00007f85f62bc4c3 in fu_engine_load () from target:/usr/lib64/fwupd-1.9.15/libfwupdengine.so
#10 0x000055d97f7c67ef in fu_daemon_setup ()
#11 0x000055d97f7bfc00 in main ()
```

Here is the callstack, no debug symbols though.

tristan957, 2024-04-04:
Wait... That fwupd successfully restarted that time. Let me restart my computer.

GeorgeSapkin, 2024-04-04:
I switched to kernel 6.8.3-300.fc40.x86_64 from https://koji.fedoraproject.org/koji/buildinfo?buildID=2431057, and the issue is no longer there.

tristan957, 2024-04-04:
Restarted my computer. fwupd was stuck, attached gdb, same stacktrace, but when I detach, fwupd successfully starts according to systemd. Don't know what that means.

GeorgeSapkin, 2024-04-04:
@tristan957 do you happen to have an AX200 in your laptop? There seemed to be some regression with the Bluetooth in 6.8.2 according to https://bugzilla.redhat.com/show_bug.cgi?id=2271784

superm1, 2024-04-04:
Can you please add debug symbols and inspect that frame?

superm1, 2024-04-04:
> @tristan957 do you happen to have an AX200 in your laptop? There seemed to be some regression with the Bluetooth in 6.8.2 according to https://bugzilla.redhat.com/show_bug.cgi?id=2271784

Good theory.

tristan957, 2024-04-04:
```
Bus 003 Device 004: ID 8087:0033 Intel Corp. AX211 Bluetooth
```

Appears so

tristan957, 2024-04-04:
@GeorgeSapkin how can I install 6.8.3?

GeorgeSapkin, 2024-04-04:
> @GeorgeSapkin how can I install 6.8.3?

You can download the required packages directly for your arch from https://koji.fedoraproject.org/koji/buildinfo?buildID=2431057

You can do something like `dnf list --installed | grep 6.8.2` to see the ones you need.

tristan957, 2024-04-04:
Yeah that worked. I think we should probably close this then? Sorry for the wild goose chase @superm1. It is strange how that bluetooth bug manifested itself.

superm1, 2024-04-04:
Don't worry about it! Glad we got down to the bottom of the mystery!

closed 2024-04-04 by superm1

lpsantil, 2024-09-05:
This helped me fix an F40 machine with 6.10.x kernel.  Specifically, I needed to `dnf install kernel-modules-extra.x86_64`.