fwupd/fwupd #7035: Secure boot is enabled, but shim isn't installed to EFI/fedora/shimx64.efi

dafrady, 2024-04-05:
**Describe the bug**
══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.18 to 0.1.35?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:Fixed The unit show "No       ║
║ Wi-Fi networks found" after about 5 minutes of connecting to WiFi.           ║
║                                                                              ║
║ Some new functionality has also been added: Added AIM-T function.            ║
║                                                                              ║
║ 21K5001HUS must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Decompressing…           [                                       ]
Secure boot is enabled, but shim isn't installed to EFI/fedora/shimx64.efi

NOTE!! admin:///boot/efi/EFI/fedora.   <<<This is the location of shimx64.efi  Secure Boot is enabled and functions properly

**Steps to Reproduce**
Run Command fwupdmgr upgrade

**Expected behavior**
Install Firmware

**fwupd version information**
Please provide the version of the daemon and client.

compile   org.freedesktop.fwupd         1.9.15
compile   com.hughsie.libxmlb           0.3.15
compile   org.freedesktop.Passim        0.1.5
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.Passim        0.1.5
runtime   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd         1.9.15
runtime   org.kernel                    6.8.2-300.fc40.x86_64

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Preinstalled by system vendor - Fedora 40

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

LENOVO 21K5001HUS
│
├─ELAN0676:00 04F3:3195:
│     Device ID:          e0e5d26dd3fa9940f593886e80a7e1eb8f564836
│     Summary:            Touchpad
│     Current version:    0x0004
│     Bootloader Version: 0x0002
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              4e8cda57-37ba-5dc6-b114-7610e38714e2 ← HIDRAW\VEN_04F3&DEV_3195
│                         cc456594-6f7e-5d1b-82a6-b4558a813152 ← HIDRAW\VEN_04F3&DEV_3195&MOD_0006
│                         d15e804c-570f-511b-803a-b38a4eddae29 ← ELANTP\ICTYPE_14&MOD_0006
│                         296e64f7-9e08-5717-bff2-651e41b0477e ← ELANTP\ICTYPE_14&MOD_0006&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─AMD Ryzen 7 PRO 7840U w/ Radeon 780M Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a704103
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              2a1df9de-fe02-59c8-8af1-32c8bb011f94 ← CPUID\PRO_0&FAM_19&MOD_74
│ │                       5a4bca32-6547-584f-bb36-920d9adede25 ← CPUID\PRO_0&FAM_19&MOD_74&STP_1
│ │   Device Flags:       • Internal device
│ │ 
[... 236 lines cut ...]
# System Details Report
---

## Report details
- **Date generated:**                              2024-04-04 21:35:21

## Hardware Information:
- **Hardware Model:**                              Lenovo ThinkPad P14s Gen 4
- **Memory:**                                      32.0 GiB
- **Processor:**                                   AMD Ryzen™ 7 PRO 7840U w/ Radeon™ 780M Graphics × 16
- **Graphics:**                                    AMD Radeon™ Graphics
- **Disk Capacity:**                               (null)

## Software Information:
- **Firmware Version:**                            R2FET38W (1.18 )
- **OS Name:**                                     Fedora Linux 40 (Workstation Edition Prerelease)
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               46
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.8.2-300.fc40.x86_64

- Have you tried rebooting? Yes several times and it is plugged in with the manufacturer power supply 
- Is this a regression? No
-

superm1, 2024-04-05:
Can you please share output for `sudo fwupdtool esp-list`?

dafrady, 2024-04-05:
Sent email to list so this may show up twice

Did option and 2 (dual boot linux and windows)

[sudo] password for
Loading…                 [************************************** ]
0.    Cancel
1.    /org/freedesktop/UDisks2/block_devices/nvme0n1p5
2.    /org/freedesktop/UDisks2/block_devices/nvme0n1p1
Choose volume [0-2]: 1
/boot/efi/EFI/BOOT/BOOTIA32.EFI
/boot/efi/EFI/BOOT/BOOTX64.EFI
/boot/efi/EFI/BOOT/fbia32.efi
/boot/efi/EFI/BOOT/fbx64.efi
/boot/efi/EFI/fedora/grub.cfg
/boot/efi/EFI/fedora/BOOTIA32.CSV
/boot/efi/EFI/fedora/BOOTX64.CSV
/boot/efi/EFI/fedora/gcdia32.efi
/boot/efi/EFI/fedora/gcdx64.efi
/boot/efi/EFI/fedora/grubia32.efi
/boot/efi/EFI/fedora/grubx64.efi
/boot/efi/EFI/fedora/mmia32.efi
/boot/efi/EFI/fedora/mmx64.efi
/boot/efi/EFI/fedora/shim.efi
/boot/efi/EFI/fedora/shimia32.efi
/boot/efi/EFI/fedora/shimx64.efi
/boot/efi/System/Library/CoreServices/SystemVersion.plist
/boot/efi/mach_kernel
/boot/efi/System Volume Information/LightningSand.CFD

Loading…                 [************************************** ]
0.    Cancel
1.    /org/freedesktop/UDisks2/block_devices/nvme0n1p5
2.    /org/freedesktop/UDisks2/block_devices/nvme0n1p1
Choose volume [0-2]: 2
/run/media/root/SYSTEM/EFI/Microsoft/Boot/bg-BG/bootmgfw.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/bg-BG/bootmgr.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/kd_07_1415.dll
/run/media/root/SYSTEM/EFI/Microsoft/Boot/CIPolicies/Active/{5DAC656C-21AD-4A02-AB49-649917162E70}.cip
/run/media/root/SYSTEM/EFI/Microsoft/Boot/CIPolicies/Active/{82443e1e-8a39-4b4a-96a8-f40ddc00b9f3}.cip
/run/media/root/SYSTEM/EFI/Microsoft/Boot/CIPolicies/Active/{CDD5CB55-DB68-4D71-AA38-3DF2B6473A52}.cip
/run/media/root/SYSTEM/EFI/Microsoft/Boot/cs-CZ/bootmgr.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/cs-CZ/memtest.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/cs-CZ/bootmgfw.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/da-DK/bootmgr.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/da-DK/memtest.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/da-DK/bootmgfw.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/de-DE/bootmgr.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/de-DE/memtest.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/de-DE/bootmgfw.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/el-GR/bootmgr.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/el-GR/memtest.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/el-GR/bootmgfw.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/en-GB/bootmgfw.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/en-GB/bootmgr.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/en-US/bootmgr.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/en-US/memtest.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/en-US/bootmgfw.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/es-ES/bootmgr.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/es-ES/memtest.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/es-ES/bootmgfw.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/es-MX/bootmgfw.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/es-MX/bootmgr.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/et-EE/bootmgfw.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/et-EE/bootmgr.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/fi-FI/bootmgr.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/fi-FI/memtest.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/fi-FI/bootmgfw.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/fr-CA/bootmgfw.efi.mui
/run/media/root/SYSTEM/EFI/Microsoft/Boot/fr-CA/bootmgr.efi.mui
[... 100 lines cut ...]
/run/media/root/SYSTEM/EFI/Microsoft/Boot/BCD.LOG2
/run/media/root/SYSTEM/EFI/Microsoft/Boot/FveTcg_0.log
/run/media/root/SYSTEM/EFI/Microsoft/Boot/FveTcg_1.log
/run/media/root/SYSTEM/EFI/Microsoft/Boot/FveTcg_2.log
/run/media/root/SYSTEM/EFI/Microsoft/Boot/kd_02_8086.dll
/run/media/root/SYSTEM/EFI/Microsoft/Recovery/BCD
/run/media/root/SYSTEM/EFI/Microsoft/Recovery/BCD.LOG
/run/media/root/SYSTEM/EFI/Microsoft/Recovery/BCD.LOG1
/run/media/root/SYSTEM/EFI/Microsoft/Recovery/BCD.LOG2
/run/media/root/SYSTEM/EFI/Boot/LenovoBT.EFI
/run/media/root/SYSTEM/EFI/Boot/License.txt
/run/media/root/SYSTEM/EFI/Boot/ReadMe.txt
/run/media/root/SYSTEM/EFI/Boot/bootx64.efi
/run/media/root/SYSTEM/EFI/debian/shimx64.efi
/run/media/root/SYSTEM/EFI/debian/grubx64.efi
/run/media/root/SYSTEM/EFI/debian/mmx64.efi
/run/media/root/SYSTEM/EFI/debian/fbx64.efi
/run/media/root/SYSTEM/EFI/debian/BOOTX64.CSV
/run/media/root/SYSTEM/EFI/debian/grub.cfg
/run/media/root/SYSTEM/EFI/fedora/fw/fwupd-18cced7e-d108-41c4-9189-b0355c41450d.cap
/run/media/root/SYSTEM/BOOT/BOOT.SDI
/run/media/root/SYSTEM/System Volume Information/LightningSand.CFD
/run/media/root/SYSTEM/System Volume Information/WPSettings.dat
/run/media/root/SYSTEM/System Volume Information/IndexerVolumeGuid
/run/media/root/SYSTEM/$RECYCLE.BIN/desktop.ini

superm1, 2024-04-05:
Okay I'm suspecting it's picking the wrong esp.

Can up please share `sudo fwupdtool get-devices -vv` specifically as an attachment?

dafrady, 2024-04-05:
[data.txt](https://github.com/fwupd/fwupd/files/14879428/data.txt)

superm1, 2024-04-05:
Yup looks like there are two ESP candidates:
```
03:15:48.638 FuPluginUefiCapsule  found /boot/efi/EFI/fedora/grubx64.efi which indicates a Linux ESP, using /boot/efi
03:15:48.638 FuVolume             mounting /org/freedesktop/UDisks2/block_devices/nvme0n1p1
03:15:48.690 FuPluginUefiCapsule  found /run/media/root/SYSTEM/EFI/debian/shimx64.efi which indicates a Linux ESP, using /run/media/root/SYSTEM
03:15:48.690 FuVolume             unmounting /org/freedesktop/UDisks2/block_devices/nvme0n1p1
03:15:48.705 FuPluginUefiCapsule  more than one ESP possible:
 - 0x30104:	/org/freedesktop/UDisks2/block_devices/nvme0n1p1
 - 0x3005f:	/org/freedesktop/UDisks2/block_devices/nvme0n1p5
```

You can see it picked the same one you showed from `fwupdtool esp-list`:
```
  EspId:                /org/freedesktop/UDisks2/block_devices/nvme0n1p1
```

I don't see that file that is missing on that ESP.

You have a few options:

1. Clean up all the Linux files on that ESP so it's not picked as a candidate
2. Consolidate your ESPs
3. Manually copy the files needed to that ESP.
4. Set the preferred ESP in `/etc/fwupd/fwupd.conf` in the `EspLocation` [key](https://fwupd.github.io/libfwupdplugin/fwupd.conf.html).

dafrady, 2024-04-05:
ok, thanks for the help got to make up my mind what I want to do now, just may wipe the whole thing and start over.

dafrady, 2024-04-05:
So, I cheated I didn't feel like messing up the ESP or boot loader or EFI partition, and it was getting late, so I hide the drive partition I didn't want to use, rebooted and then it was able to do the upgrade, rebooted and un-hide the drive with the wrong Linux info on it.

closed 2024-04-05 by dafrady

dafrady, 2024-04-05:
Thanks Mario L for the help
So, I cheated I didn't feel like messing up the ESP or boot loader or EFI partition, and it was getting late, so I hide the drive partition I didn't want to use, rebooted and then it was able to do the upgrade, rebooted and un-hide the drive with the wrong Linux info on it.

D. Allen Frady 

 
 

   
   - Consolidate your ESPs