fwupd/fwupd #7040: fwupdmgr timeout

Germano0, 2024-04-05:
Clone of https://github.com/fwupd/fwupd/issues/6022 . Opened a new bugreport following the suggestion of Mario L.

I experienced that on
- fwupd-1.9.15-1.fc39.x86_64
- 6.7.11-200.fc39.x86_64
- Thinkpad T16 Gen 2
```
# fwupdmgr get-updates
Connessione al demone non riuscita: Errore nel chiamare StartServiceByName per org.freedesktop.fwupd: È stato raggiunto il timeout
```
journalctl
```
apr 04 22:00:00 T16-user systemd[1]: Starting fwupd.service - Firmware update daemon...
apr 04 22:03:00 T16-user systemd[1]: fwupd.service: start operation timed out. Terminating.
apr 04 22:03:00 T16-user systemd[1]: fwupd.service: Failed with result 'timeout'.
apr 04 22:03:00 T16-user systemd[1]: Failed to start fwupd.service - Firmware update daemon.
apr 04 22:03:00 T16-user audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=fwupd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? >
```

superm1, 2024-04-05:
Can you please share the full journal from a boot this occurred?

Also if you downgrade to an older kernel does it go away?

Germano0, 2024-04-05:
Problem solved with kernel `6.8.4-200.fc39.x86_64`

closed 2024-04-05 by Germano0

superm1, 2024-04-05:
Duplicate of #7030 then