fwupd/fwupd #7045: Failed to mount ESP: vfat/fat32 and RAID 1

laka, 2024-04-07:
**Describe the bug**
fwupd fails to mount ESP partition in a RAID 1 setup: 
`fwupd[10145]: 20:05:45.469 FuPluginUefiCapsule  failed to mount ESP: GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: No such interface "org.freedesktop.UDisks2.Filesystem" on object at path /org/freedesktop/UDisks2/block_devices/nvme0n1p2`

Setting the `esp` flag doesn't help:
```
Model: KINGSTON SFYRD2000G (nvme)
Disk /dev/nvme1n1: 2000GB
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags: 

Number  Start   End     Size    File system  Name  Flags
 1      1049kB  1001MB  1000MB                     raid
 2      1001MB  1631MB  629MB   fat32              boot, esp
 3      1631MB  2000GB  1999GB                     raid
```

**Steps to Reproduce**
Install Linux (Fedora) having /boot and /boot/efi in a RAID 1 setup.

**Expected behavior**
Not sure - maybe this is expected with this type of partition?

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.16
compile   com.hughsie.libxmlb           0.3.17
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.16
runtime   org.kernel                    6.8.4-200.fc39.x86_64

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
dnf
<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```
fwupdmgr get-devices --show-all-devices
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
ASUS System Product Name
│
├─Unknown Device:
│     Device ID:          0785bd3eac5b152d9fc34e5549136304ca025e4c
│     GUID:               bc9f6c6f-473f-5bc2-a0d1-3171c965d0d3 ← GPIO\ID_AMDIF031:00
│   
├─AMD Ryzen 9 7900 12-Core Processor:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a601206
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              16dd7d28-eade-5077-adb8-0775fdfbb2e5 ← CPUID\PRO_0&FAM_19&MOD_61
│ │                       6c9f4098-a058-5cd6-a1ce-36bfe27404fb ← CPUID\PRO_0&FAM_19&MOD_61&STP_2
│ │   Device Flags:       • Internal device
│ │ 
│ ├─Graphics Processing Unit (GPU):
│ │     Device ID:        20161e3870a2030fa932712e7b7dae8a2aec59c3
│ │     Summary:          Raphael Generic VBIOS
[... 195 lines cut ...]
│       ├── gcdia32.efi
│       ├── gcdx64.efi
│       ├── grub.cfg
│       ├── grubia32.efi
│       ├── grubx64.efi
│       ├── mmia32.efi
│       ├── mmx64.efi
│       ├── shim.efi
│       ├── shimia32.efi
│       └── shimx64.efi
├── mach_kernel
└── System
    └── Library
        └── CoreServices

```

**Additional questions**

- Operating system and version: Fedora 39
- Have you tried rebooting? Yes
- Is this a regression? Don't know.
- Are you using an NVMe disk? Yes, two in a RAID 1 setup.
- Is secure boot enabled? No.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No.

superm1, 2024-04-07:
Can up please provide output for `udisksctl dump`

laka, 2024-04-08:
```
/org/freedesktop/UDisks2/Manager:
  org.freedesktop.UDisks2.Manager:
    DefaultEncryptionType:      luks2
    SupportedEncryptionTypes:   luks1
                                luks2
    SupportedFilesystems:       ext2
                                ext3
                                ext4
                                xfs
                                vfat
                                ntfs
                                f2fs
                                nilfs2
                                exfat
                                btrfs
                                udf
                                swap
    Version:                    2.10.1
  org.freedesktop.UDisks2.Manager.NVMe:
    HostID:             
    HostNQN:            nqn.2014-08.org.nvmexpress:uuid:9b4d419e-23f8-7a42-e9ef-c87f54aebd5d

/org/freedesktop/UDisks2/block_devices/md124:
  org.freedesktop.UDisks2.Block:
    Configuration:              [('fstab', {'fsname': <b'UUID=e32bc6ae-0479-4426-82cd-b59bb37abde9'>, 'dir': <b'/backup'>, 'type': <b'ext4'>, 'opts': <b'defaults'>, 'freq': <0>, 'passno': <0>})]
    CryptoBackingDevice:        '/'
    Device:                     /dev/md124
    DeviceNumber:               2428
    Drive:                      '/'
    HintAuto:                   false
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          true
    HintSymbolicIconName:       
    HintSystem:                 true
    Id:                         by-id-md-name-trofast:backup
    IdLabel:                    backup
    IdType:                     ext4
    IdUUID:                     e32bc6ae-0479-4426-82cd-b59bb37abde9
    IdUsage:                    filesystem
    IdVersion:                  1.0
    MDRaid:                     '/org/freedesktop/UDisks2/mdraid/99f81e20_bb49b469_9dddf1fb_bbc25130'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/md/backup
    ReadOnly:                   false
    Size:                       479966789632
    Symlinks:                   /dev/disk/by-id/md-name-trofast:backup
                                /dev/disk/by-id/md-uuid-99f81e20:bb49b469:9dddf1fb:bbc25130
                                /dev/disk/by-label/backup
                                /dev/disk/by-uuid/e32bc6ae-0479-4426-82cd-b59bb37abde9
                                /dev/md/backup
    UserspaceMountOptions:      
  org.freedesktop.UDisks2.Filesystem:
    MountPoints:        /backup
    Size:               479966789632

/org/freedesktop/UDisks2/block_devices/md125:
  org.freedesktop.UDisks2.Block:
    Configuration:              [('fstab', {'fsname': <b'UUID=6E73-A26B'>, 'dir': <b'/boot/efi'>, 'type': <b'vfat'>, 'opts': <b'umask=0077,shortname=winnt'>, 'freq': <0>, 'passno': <2>})]
    CryptoBackingDevice:        '/'
    Device:                     /dev/md125
    DeviceNumber:               2429
    Drive:                      '/'
    HintAuto:                   false
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          true
[... 904 lines cut ...]
    SyncAction:                 idle
    SyncCompleted:              0.0
    SyncRate:                   0
    SyncRemainingTime:          0
    UUID:                       99f81e20:bb49b469:9dddf1fb:bbc25130

/org/freedesktop/UDisks2/mdraid/d36861fa_64a478e8_0b1f7917_3732d64b:
  org.freedesktop.UDisks2.MDRaid:
    ActiveDevices:              [('/org/freedesktop/UDisks2/block_devices/nvme0n1p2', 0, ['in_sync'], 0, {}), ('/org/freedesktop/UDisks2/block_devices/nvme1n1p2', 1, ['in_sync'], 0, {})]
    BitmapLocation:             -16
    ChildConfiguration:         []
    ChunkSize:                  0
    Degraded:                   0
    Level:                      raid1
    Name:                       localhost-live:efi
    NumDevices:                 2
    Running:                    true
    Size:                       629080064
    SyncAction:                 idle
    SyncCompleted:              0.0
    SyncRate:                   0
    SyncRemainingTime:          0
    UUID:                       d36861fa:64a478e8:0b1f7917:3732d64b

```

superm1, 2024-04-08:
AFAICT In your system we should ideally be picking `/org/freedesktop/UDisks2/block_devices/md125` and skipping the RAID array

Can you please try the PR I just pushed (it's linked to this bug)?

laka, 2024-04-08:
Definitely, thanks! 

I cloned the repo and built your branch, but not quite sure if the development environment requires something more than just invoking the fwupd binary?

```
(fwupd) [laka@trofast fwupd]$ git status
On branch superm1/ignore-raid-esp
Your branch is up to date with 'origin/superm1/ignore-raid-esp'.

nothing to commit, working tree clean
(fwupd) [laka@trofast fwupd]$ fwupd
13:44:33.836 FuPluginUefiCapsule  cannot find default ESP: No ESP or BDP found
13:44:34.131 FuMain               Daemon ready for requests (locale en_US.UTF-8)
```

superm1, 2024-04-08:
> Definitely, thanks!
>
> I cloned the repo and built your branch, but not quite sure if the development environment requires something more than just invoking the fwupd binary?
>
> ```
> (fwupd) [...]

Within the environment can you please post `fwupdtool esp-list --verbose --verbose` output with that branch?

It seems to fix the immediate issue of trying to mount a raid member but we're not finding the esp so I would like to understand more.

laka, 2024-04-08:
```
(fwupd) [laka@trofast fwupd]$ fwupdtool esp-list --verbose --verbose
[sudo] password for laka: 
19:02:08.895 FuDebug              verbose to debug (on console 1)
19:02:08.895 FuEngine             starting fwupd 2.0.0…
19:02:08.895 FuMain               locked /run/lock/fwupdtool
Loading…                 [ -                                     ]19:02:08.895 FuConfig             loading config /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
19:02:08.896 FuConfig             loading config /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
19:02:08.896 FuConfig             trying to load config values from /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
19:02:08.896 FuCommon             mapped file /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf of size 0x72
19:02:08.896 FuConfig             setting config [fwupd] EspLocation=/boot/efi
19:02:08.896 FuConfig             setting config [uefi] OverrideESPMountPoint=/boot/efi
19:02:08.896 FuConfig             trying to load config values from /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
19:02:08.896 FuCommon             mapped file /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf of size 0x72
19:02:08.896 FuConfig             setting config [fwupd] EspLocation=/boot/efi
19:02:08.896 FuConfig             setting config [uefi] OverrideESPMountPoint=/boot/efi
19:02:08.896 FuConfig             ::configuration loaded
Loading…                 [                                       ]19:02:08.896 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
19:02:08.896 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
19:02:08.896 Jcat                 reading /home/laka/fwupd/venv/bin/../dist/etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
19:02:08.896 Jcat                 loaded 1 certificates
19:02:08.896 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
19:02:08.896 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
19:02:08.896 Jcat                 reading /home/laka/fwupd/venv/bin/../dist/etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
19:02:08.896 Jcat                 loaded 1 certificates
19:02:08.896 Jcat                 reading /home/laka/fwupd/venv/bin/../dist/lib/fwupd/pki/secret.key with 2459 bytes
19:02:08.896 Jcat                 reading /home/laka/fwupd/venv/bin/../dist/lib/fwupd/pki/client.pem with 1383 bytes
19:02:08.896 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]19:02:08.896 FuHistory            trying to open database '/home/laka/fwupd/venv/bin/../dist/lib/fwupd/pending.db'
19:02:08.896 FuHistory            got schema version of 13
Loading…                 [*                                      ]19:02:08.899 FuPlugin             load(/home/laka/fwupd/venv/bin/../dist/lib64/fwupd-2.0.0/libfu_plugin_modem_manager.so)
19:02:08.900 FuQuirks             loading quirks from /home/laka/fwupd/venv/bin/../dist/share/fwupd/quirks.d
19:02:08.900 FuQuirks             loading quirks from /home/laka/fwupd/venv/bin/../dist/lib/fwupd/quirks.d
19:02:08.900 XbSilo               attempting to load /home/laka/fwupd/venv/dist/cache/fwupd/quirks.xmlb
19:02:08.900 XbSilo               file: 4b3b8c0b-d259-f389-bb28-f98a036721a5, current:4b3b8c0b-d259-f389-bb28-f98a036721a5, cached: (null)
19:02:08.900 XbSilo               loading silo with file contents
Loading…                 [*                                      ]19:02:08.900 FuConfig             loading config /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
19:02:08.900 FuConfig             loading config /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
19:02:08.900 FuConfig             trying to load config values from /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
19:02:08.900 FuCommon             mapped file /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf of size 0x72
19:02:08.900 FuConfig             setting config [fwupd] EspLocation=/boot/efi
19:02:08.900 FuConfig             setting config [uefi] OverrideESPMountPoint=/boot/efi
19:02:08.900 FuConfig             trying to load config values from /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
19:02:08.900 FuCommon             mapped file /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf of size 0x72
19:02:08.900 FuConfig             setting config [fwupd] EspLocation=/boot/efi
19:02:08.900 FuConfig             setting config [uefi] OverrideESPMountPoint=/boot/efi
19:02:08.900 FuConfig             ::configuration loaded
19:02:08.900 FuStruct             FuStructSmbiosEp64:
  anchor_str: _SM3_
  entry_point_csum: 0xa8
  entry_point_len: 0x18
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x5
  smbios_docrev: 0x0
  entry_point_rev: 0x1
  reserved0: 0x0
  structure_table_len: 0xc86
  structure_table_addr: 0x794b5000
19:02:08.900 FuStruct             FuStructSmbiosStructure:
  type: 0x0
  length: 0x1a
  handle: 0x0
19:02:08.900 FuStruct             FuStructSmbiosStructure:
  type: 0x1
  length: 0x1b
  handle: 0x1
19:02:08.901 FuStruct             FuStructSmbiosStructure:
  type: 0x2
  length: 0xf
  handle: 0x2
[... 767 lines cut ...]
19:02:08.936 FuVolume             ignoring linux_raid_member device /org/freedesktop/UDisks2/block_devices/nvme0n1p2
[... cut ...]
19:02:08.956 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
19:02:08.957 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
19:02:08.958 FuContext            no volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
No ESP or BDP found
19:02:08.964 FuMain               No ESP or BDP found

19:02:08.964 FuPlugin             finalize(modem_manager)
19:02:08.964 FuPlugin             finalize(flashrom)
19:02:08.964 FuPluginTest         destroy

```

hughsie, 2024-04-08:
@superm1 does the firmware support ESP on raid? IIRC @vathpela says it's a bit of a disaster putting the ESP on any kind of RAID, but he might share more.

superm1, 2024-04-08:
> @superm1 does the firmware support ESP on raid? IIRC @vathpela says it's a bit of a disaster putting the ESP on any kind of RAID, but he might share more.

In the case of raid 1 I suspect that the firmware just sees it as two ESP partitions.

laka, 2024-04-08:
> > @superm1 does the firmware support ESP on raid? IIRC @vathpela says it's a bit of a disaster putting the ESP on any kind of RAID, but he might share more.
> 
> In the case of raid 1 I suspect that the firmware just sees it as two ESP partitions.

It indeed does. From the dnf installed version:

```
fwupdtool esp-list --verbose --verbose
...
0.      Cancel
1.      /org/freedesktop/UDisks2/block_devices/nvme0n1p2
2.      /org/freedesktop/UDisks2/block_devices/nvme1n1p2
Choose volume [0-2]: 1
20:33:17.806 FuVolume             mounting /org/freedesktop/UDisks2/block_devices/nvme0n1p2
20:33:17.807 FuVolume             unmounting /org/freedesktop/UDisks2/block_devices/nvme0n1p2
20:33:17.807 FuDeviceLocker       ignoring close error on aborted open: GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: No such interface “org.freedesktop.UDisks2.Filesystem” on object at path /org/freedesktop/UDisks2/block_devices/nvme0n1p2
GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: No such interface “org.freedesktop.UDisks2.Filesystem” on object at path /org/freedesktop/UDisks2/block_devices/nvme0n1p2
WARNING: This program may only work correctly as root
20:33:17.807 FuPlugin             finalize(modem_manager)
20:33:17.807 FuPlugin             finalize(flashrom)
20:33:17.807 FuPluginTest         destroy

```

superm1, 2024-04-08:
OK, have another try with https://github.com/fwupd/fwupd/pull/7053

If that still doesn't work, please give the same debug output for `fwupdtool`.

laka, 2024-04-09:
```
(fwupd) [laka@trofast fwupd]$ fwupdtool esp-list --verbose --verbose 
[sudo] password for laka: 
06:52:24.518 FuDebug              verbose to debug (on console 1)
06:52:24.518 FuEngine             starting fwupd 2.0.0…
06:52:24.518 FuMain               locked /run/lock/fwupdtool
Loading…                 [ -                                     ]06:52:24.518 FuConfig             loading config /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
06:52:24.518 FuConfig             loading config /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
06:52:24.518 FuConfig             trying to load config values from /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
06:52:24.518 FuCommon             mapped file /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf of size 0x49
06:52:24.518 FuConfig             setting config [fwupd] EspLocation=/boot/efi
06:52:24.518 FuConfig             trying to load config values from /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
06:52:24.518 FuCommon             mapped file /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf of size 0x49
06:52:24.518 FuConfig             setting config [fwupd] EspLocation=/boot/efi
06:52:24.518 FuConfig             ::configuration loaded
Loading…                 [                                       ]06:52:24.518 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
06:52:24.518 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
06:52:24.518 Jcat                 reading /home/laka/fwupd/venv/bin/../dist/etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
06:52:24.518 Jcat                 loaded 1 certificates
06:52:24.518 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
06:52:24.518 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
06:52:24.518 Jcat                 reading /home/laka/fwupd/venv/bin/../dist/etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
06:52:24.518 Jcat                 loaded 1 certificates
06:52:24.518 Jcat                 reading /home/laka/fwupd/venv/bin/../dist/lib/fwupd/pki/secret.key with 2459 bytes
06:52:24.518 Jcat                 reading /home/laka/fwupd/venv/bin/../dist/lib/fwupd/pki/client.pem with 1383 bytes
06:52:24.518 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]06:52:24.518 FuHistory            trying to open database '/home/laka/fwupd/venv/bin/../dist/lib/fwupd/pending.db'
06:52:24.518 FuHistory            got schema version of 13
Loading…                 [*                                      ]06:52:24.521 FuPlugin             load(/home/laka/fwupd/venv/bin/../dist/lib64/fwupd-2.0.0/libfu_plugin_modem_manager.so)
06:52:24.522 FuQuirks             loading quirks from /home/laka/fwupd/venv/bin/../dist/share/fwupd/quirks.d
06:52:24.522 FuQuirks             loading quirks from /home/laka/fwupd/venv/bin/../dist/lib/fwupd/quirks.d
06:52:24.522 XbSilo               attempting to load /home/laka/fwupd/venv/dist/cache/fwupd/quirks.xmlb
06:52:24.522 XbSilo               file: aaf65f75-3b86-2469-16ec-ffbd3b5ad721, current:aaf65f75-3b86-2469-16ec-ffbd3b5ad721, cached: (null)
06:52:24.522 XbSilo               loading silo with file contents
Loading…                 [*                                      ]06:52:24.523 FuConfig             loading config /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
06:52:24.523 FuConfig             loading config /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
06:52:24.523 FuConfig             trying to load config values from /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
06:52:24.523 FuCommon             mapped file /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf of size 0x49
06:52:24.523 FuConfig             setting config [fwupd] EspLocation=/boot/efi
06:52:24.523 FuConfig             trying to load config values from /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
06:52:24.523 FuCommon             mapped file /home/laka/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf of size 0x49
06:52:24.523 FuConfig             setting config [fwupd] EspLocation=/boot/efi
06:52:24.523 FuConfig             ::configuration loaded
06:52:24.523 FuStruct             FuStructSmbiosEp64:
  anchor_str: _SM3_
  entry_point_csum: 0xa8
  entry_point_len: 0x18
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x5
  smbios_docrev: 0x0
  entry_point_rev: 0x1
  reserved0: 0x0
  structure_table_len: 0xc86
  structure_table_addr: 0x794b5000
06:52:24.523 FuStruct             FuStructSmbiosStructure:
  type: 0x0
  length: 0x1a
  handle: 0x0
06:52:24.523 FuStruct             FuStructSmbiosStructure:
  type: 0x1
  length: 0x1b
  handle: 0x1
06:52:24.523 FuStruct             FuStructSmbiosStructure:
  type: 0x2
  length: 0xf
  handle: 0x2
06:52:24.523 FuStruct             FuStructSmbiosStructure:
  type: 0x3
  length: 0x16
  handle: 0x3
[... 765 lines cut ...]
06:52:24.562 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: linux_raid_member
06:52:24.562 FuVolume             ignoring linux_raid_member device /org/freedesktop/UDisks2/block_devices/nvme1n1p2
06:52:24.562 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
[... cut ...]
06:52:24.579 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
06:52:24.580 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
06:52:24.581 FuContext            no volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
No ESP or BDP found
06:52:24.588 FuMain               No ESP or BDP found

06:52:24.588 FuPlugin             finalize(modem_manager)
06:52:24.588 FuPlugin             finalize(flashrom)
06:52:24.588 FuPluginTest         destroy

```

superm1, 2024-04-09:
Can you please confirm what git hash you built that log from? I force pushed the branch twice, want to make sure you had the latest version because I don't see the md devices I expected.

laka, 2024-04-09:
```
(fwupd) [laka@trofast fwupd]$ git status
On branch superm1/proxy-part-optional
Your branch is up to date with 'origin/superm1/proxy-part-optional'.

nothing to commit, working tree clean
(fwupd) [laka@trofast fwupd]$ git pull
Already up to date.
(fwupd) [laka@trofast fwupd]$ git log -5 --pretty=oneline 
72dc2e92dd434312922417f84c55fd70b1aa0430 (HEAD -> superm1/proxy-part-optional, origin/superm1/proxy-part-optional) trivial: add an extra message to show why fu_volume_new_by_kind() is called
fb1584e89d1b2a517a7db3209d99db89fd162f54 trivial: Allow ESP's in mdraid arrays to be included as a volume
6e1164c00e218f38abcc3321f751b857f7173cfb trivial: upgrade messages about ESP selection to INFO
f3d2e981872a45cf2aec887072f39acec5352c2b trivial: Fix oss-fuzz CI target
3dd6ebf3fef861f078bf36fed9712e74c48637b0 trivial: ignore RAID block devices for building ESP candidates
(fwupd) [laka@trofast fwupd]$ 

```

superm1, 2024-04-09:
Ah dang. I don't know why it's missing those block devices then. Let me think some more.

laka, 2024-04-09:
Seems to work fine?

```
WARNING: This package has not been validated, it may not work properly.
15:06:01.816 FuVolume             Looking for volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b
15:06:01.817 FuVolume             Got MDRaid: /
15:06:01.817 FuVolume             device /org/freedesktop/UDisks2/block_devices/sdb1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:06:01.818 FuVolume             Got MDRaid: /
15:06:01.818 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:06:01.819 FuVolume             Got MDRaid: /org/freedesktop/UDisks2/mdraid/8fe63910_a5ecd9fc_36cf2e03_4f283357
15:06:01.819 FuVolume             device /org/freedesktop/UDisks2/block_devices/md127, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: btrfs
15:06:01.819 FuVolume             Got MDRaid: /org/freedesktop/UDisks2/mdraid/99f81e20_bb49b469_9dddf1fb_bbc25130
15:06:01.819 FuVolume             device /org/freedesktop/UDisks2/block_devices/md124, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: ext4
15:06:01.820 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: linux_raid_member
15:06:01.820 FuVolume             ignoring linux_raid_member device /org/freedesktop/UDisks2/block_devices/nvme0n1p2
15:06:01.820 FuVolume             Got MDRaid: /
15:06:01.821 FuVolume             Got MDRaid: /org/freedesktop/UDisks2/mdraid/219611a2_b6584b70_1483e8bc_9c1e1ac0
15:06:01.821 FuVolume             device /org/freedesktop/UDisks2/block_devices/md126, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: ext4
15:06:01.821 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: linux_raid_member
15:06:01.821 FuVolume             ignoring linux_raid_member device /org/freedesktop/UDisks2/block_devices/nvme1n1p2
15:06:01.822 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:06:01.822 FuVolume             Got MDRaid: /
15:06:01.823 FuVolume             device /org/freedesktop/UDisks2/block_devices/sda1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:06:01.823 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:06:01.824 FuVolume             Got MDRaid: /org/freedesktop/UDisks2/mdraid/d36861fa_64a478e8_0b1f7917_3732d64b
15:06:01.824 FuVolume             device /org/freedesktop/UDisks2/block_devices/md125, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
15:06:01.824 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:06:01.825 FuVolume             Got MDRaid: /
15:06:01.830 FuVolume             Looking for volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
15:06:01.831 FuVolume             Got MDRaid: /
15:06:01.831 FuVolume             device /org/freedesktop/UDisks2/block_devices/sdb1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:06:01.832 FuVolume             Got MDRaid: /
15:06:01.833 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:06:01.833 FuVolume             Got MDRaid: /org/freedesktop/UDisks2/mdraid/8fe63910_a5ecd9fc_36cf2e03_4f283357
15:06:01.833 FuVolume             device /org/freedesktop/UDisks2/block_devices/md127, type: ebd0a0a2-b9e5-4433-87c0-68b6b72699c7, internal: 1, fs: btrfs
15:06:01.833 FuVolume             Got MDRaid: /org/freedesktop/UDisks2/mdraid/99f81e20_bb49b469_9dddf1fb_bbc25130
15:06:01.833 FuVolume             device /org/freedesktop/UDisks2/block_devices/md124, type: ebd0a0a2-b9e5-4433-87c0-68b6b72699c7, internal: 1, fs: ext4
15:06:01.834 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: linux_raid_member
15:06:01.834 FuVolume             Got MDRaid: /
15:06:01.835 FuVolume             Got MDRaid: /org/freedesktop/UDisks2/mdraid/219611a2_b6584b70_1483e8bc_9c1e1ac0
15:06:01.835 FuVolume             device /org/freedesktop/UDisks2/block_devices/md126, type: ebd0a0a2-b9e5-4433-87c0-68b6b72699c7, internal: 1, fs: ext4
15:06:01.835 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: linux_raid_member
15:06:01.836 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:06:01.836 FuVolume             Got MDRaid: /
15:06:01.837 FuVolume             device /org/freedesktop/UDisks2/block_devices/sda1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:06:01.837 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:06:01.838 FuVolume             Got MDRaid: /org/freedesktop/UDisks2/mdraid/d36861fa_64a478e8_0b1f7917_3732d64b
[... cut ...]
/boot/efi/EFI/fedora/gcdia32.efi
/boot/efi/EFI/fedora/gcdx64.efi
/boot/efi/EFI/fedora/grubia32.efi
/boot/efi/EFI/fedora/grubx64.efi
/boot/efi/EFI/fedora/mmia32.efi
/boot/efi/EFI/fedora/mmx64.efi
/boot/efi/EFI/fedora/shim.efi
/boot/efi/EFI/fedora/shimia32.efi
/boot/efi/EFI/fedora/shimx64.efi
/boot/efi/System/Library/CoreServices/SystemVersion.plist
/boot/efi/mach_kernel
15:06:03.587 FuPlugin             finalize(modem_manager)
15:06:03.587 FuPlugin             finalize(flashrom)
15:06:03.587 FuPluginTest         destroy
```

superm1, 2024-04-09:
OK, it's fixed in main in https://github.com/fwupd/fwupd/commit/7c1df3d1ee032434b6d2fac978f0139b4a27d585 and we'll backport to 1_9_X as well.  It will be part of 1.9.17 release in a few weeks.

closed 2024-04-09 by superm1

laka, 2024-04-09:
Great work, thanks a bunch.

minnoce, 2024-05-21:
Hi, still same problem here on Fedora 39 fully updated (fwupd v. 1.9.19) and RAID.