fwupd/fwupd #7050: AMD/ATI Failed to verify Firmware for AMD Radeon HD 7800 Series Graphics Card

CaptainMorgan12, 2024-04-08:
Environment: Ubuntu 23.10, MacPro 2013 trashcan, AMD Radeon™ HD 7800 Series, Current Version 113-C3861LB-029, Vendor IDs PCI:0x1002, PCI:0x8086, Gnome Firmware v46

Steps to Repeat:
1. Attestation Unknown
2. Select Verify
3. Failed to verify firmware, failed to read firmware: Error reading from file: Input/output error

Flags: Cryptographic hash verification is available, Internal Device

superm1, 2024-04-08:
Can you please re-run your daemon in verbose mode and share the logs?  Also please share your dmesg.

CaptainMorgan12, 2024-04-08:
How do you restart daemon in verbose mode?

superm1, 2024-04-08:
> How do you restart daemon in verbose mode?

Add `--verbose --verbose` to the `ExecStart` command in the systemd unit and then run `systemctl daemon-reload` and `systemctl restart fwupd.service`

superm1, 2024-04-08:
Is the GPU even working?  Given those above kernel logs.

CaptainMorgan12, 2024-04-08:
Yes works fine

superm1, 2024-04-08:
Please modify `/lib/systemd/system/fwupd.service` to modify the `ExecStart` command.

CaptainMorgan12, 2024-04-08:
I did that but no log shows. I tried:

> flatpak run --command=sh --devel org.gnome.Firmware -v
[📦 org.gnome.Firmware ~]$ gdb /var/lib/flatpak/runtime/org.gnome.Firmware.Debug/x86_64/stable/21d2816eb01e3b906d9ea54ae57b9016e311702f014888b485b93a13bb556027/files/bin/gnome-firmware.debug
gdb /var/lib/flatpak/runtime/org.gnome.Firmware.Debug/x86_64/stable/21d2816eb01e3b906d9ea54ae57b9016e311702f014888b485b93a13bb556027/files/bin/gnome-firmware.debug
GNU gdb (GDB) 14.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-unknown-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
/var/lib/flatpak/runtime/org.gnome.Firmware.Debug/x86_64/stable/21d2816eb01e3b906d9ea54ae57b9016e311702f014888b485b93a13bb556027/files/bin/gnome-firmware.debug: No such file or directory.
(gdb) 

but even though the binary is in that folder it says no such file

superm1, 2024-04-08:
Let's take gnome-firmware out of the picture.  Reproduce using `fwupdmgr` and restart the fwupd daemon manually.  The logs will be part of `journalctl -u fwupd.service -b0`

CaptainMorgan12, 2024-04-08:
[fwupd.txt](https://github.com/fwupd/fwupd/files/14908894/fwupd.txt)

There are 12 mentions of verify since I selected it a few times. Sorry for the long log.

superm1, 2024-04-08:
> Apr 08 10:25:08 amiga fwupd[17260]: 17:25:08.851 FuPluginOptionrom    ROM returned 0xffffffff bytes

As this is a mac, are you sure there really is an oprom?  There very well might not be.
If anything this to me looks like a kernel bug that it exports one.

CaptainMorgan12, 2024-04-08:
Not entirely sure according to:
https://www.techpowerup.com/download/ati-atiflash/ there is a flash utility that can flash bios firmware.

Also I can revise and override the clock frequency and power states manually on the card. So some settings are possible not sure those are part of a oprom?

superm1, 2024-04-08:
Those settings are done at runtime, there doesn't have to be an OpRom for them to work.

Macs don't boot with "regular" uefi, so they don't have a "need" for things like OpRom.

To me it sounds like there is a firmware flashed to a SPI chip but not that the system exports as an OpRom.

@hughsie agree?

hughsie, 2024-04-11:
>  If anything this to me looks like a kernel bug that it exports one.

It looks like the OPROM is just blank to me. I'm not sure what the "fix" is here? Maybe we can remove the can-verify flag when running on apple hardware perhaps?

superm1, 2024-04-11:
> >  If anything this to me looks like a kernel bug that it exports one.
> 
> It looks like the OPROM is just blank to me. I'm not sure what the "fix" is here? Maybe we can remove the can-verify flag when running on apple hardware perhaps?

Sounds reasonable to me

hughsie, 2024-04-11:
@CaptainMorgan12 can you grab us the output of `sudo fwupdtool hwids` please.

CaptainMorgan12, 2024-04-11:
`Computer Information
--------------------
BiosVendor: Apple Inc.
BiosVersion: 132.0.0.0.0
BiosMajorRelease: 0
BiosMinorRelease: 1
FirmwareMajorRelease: ff
FirmwareMinorRelease: ff
Manufacturer: Apple Inc.
Family: MacPro
ProductName: MacPro6,1
ProductSku: 
EnclosureKind: 2
BaseboardManufacturer: Apple Inc.
BaseboardProduct: Mac-F60DEB81FF30ACF6
Hardware IDs
------------
{93ca75cf-58a6-538f-a668-59808991dbb6}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{07f859b0-da58-5427-9d64-2605c571e74d}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{bd1eb59b-38a1-5028-94dc-af3fff6ef694}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{43456f36-fa78-5c0b-a367-0e459ccfb608}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{c39b3346-a2c2-5e59-b2ff-643224b685db}   <- Manufacturer + Family + ProductName + ProductSku
{6d3c0cc9-516e-5b2a-800b-2a8fcb139dc6}   <- Manufacturer + Family + ProductName
{1dcd2a1e-869d-5335-a0eb-6d080e7d4e1a}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{3d30c516-f24b-52e5-ac1b-8f2280953b17}   <- Manufacturer + ProductSku
{cf256612-d5fa-5a47-946d-c2b48565354a}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{0f702b49-0ccd-5840-a113-20a81d880278}   <- Manufacturer + ProductName
{47e9e831-6810-5922-9577-058506909103}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{03e745c6-d351-5e12-baba-f55a0d7affb1}   <- Manufacturer + Family
{5ea55873-7575-5e8c-8f82-c4fb707aa0ef}   <- Manufacturer + EnclosureKind
{6bcadbd0-d22a-52c0-9674-439c0a504a6c}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{80f95c96-a739-5ef5-8482-3d65cb39ff55}   <- Manufacturer`

closed 2024-04-16 by hughsie