fwupd/fwupd #7051: Finding multiple leaks in `fu-self-test.c`

rishabh-chromium, 2024-04-08:
**Describe the bug**
Getting over 400 bytes of leaks when running `fu-self-test.c`. Here is the log message found in our builds:
```
* ==3==ERROR: LeakSanitizer: detected memory leaks
 * 
 * Direct leak of 36 byte(s) in 1 object(s) allocated from:
 *     #0 0x5ea908fe1d2e in __interceptor_malloc ??:0:0
 *     #1 0x7ba74702ece7 in g_malloc ??:0:0
 *     #2 0x7ba747044efe in g_strconcat ??:0:0
 *     #3 0x7ba7470ff93c in ?? ??:0
 *     #4 0x7ba747100524 in ?? ??:0
 *     #5 0x7ba746d3ea38 in ?? ??:0
 *     #6 0x7ba746d3db84 in ?? ??:0
 *     #7 0x7ba746d3d96f in ?? ??:0
 *     #8 0x7ba746d1e9b9 in ?? ??:0
 *     #9 0x5ea909028565 in fu_engine_downgrade_func /build/amd64-generic/tmp/portage/sys-apps/fwupd-1.9.15-r295/work/fwupd-1.9.15-build/../fwupd-1.9.15/src/fu-self-test.c:2067:8
 *     #10 0x7ba74704c69e in g_test_queue_destroy ??:?
 *     #11 0x7ba74704c91d in g_test_queue_destroy ??:?
 *     #12 0x7ba74704b5b5 in g_test_run_suite ??:0:0
 *     #13 0x7ba74704b460 in g_test_run ??:0:0
 *     #14 0x5ea909016036 in main /build/amd64-generic/tmp/portage/sys-apps/fwupd-1.9.15-r295/work/fwupd-1.9.15-build/../fwupd-1.9.15/src/fu-self-test.c:6238:9
 *     #15 0x7ba746611705 in __libc_start_call_main ??:0:0
 *     #16 0x7ba7466117c1 in __libc_start_main@GLIBC_2.2.5 ??:0:0
 *     #17 0x5ea908f4f260 in _start ??:0:0
 * 
 * Direct leak of 36 byte(s) in 1 object(s) allocated from:
 *     #0 0x5ea908fe1d2e in __interceptor_malloc ??:0:0
 *     #1 0x7ba74702ece7 in g_malloc ??:0:0
 *     #2 0x7ba747044efe in g_strconcat ??:0:0
 *     #3 0x7ba7470ff93c in ?? ??:0
 *     #4 0x7ba747100524 in ?? ??:0
 *     #5 0x7ba746d3ea38 in ?? ??:0
 *     #6 0x7ba746d3db84 in ?? ??:0
 *     #7 0x7ba746d3d96f in ?? ??:0
 *     #8 0x7ba746d1e9b9 in ?? ??:0
 *     #9 0x5ea909031cc3 in fu_engine_generate_md_func /build/amd64-generic/tmp/portage/sys-apps/fwupd-1.9.15-r295/work/fwupd-1.9.15-build/../fwupd-1.9.15/src/fu-self-test.c:248:8
 *     #10 0x7ba74704c69e in g_test_queue_destroy ??:?
 *     #11 0x7ba74704c91d in g_test_queue_destroy ??:?
 *     #12 0x7ba74704b5b5 in g_test_run_suite ??:0:0
 *     #13 0x7ba74704b460 in g_test_run ??:0:0
 *     #14 0x5ea909016036 in main /build/amd64-generic/tmp/portage/sys-apps/fwupd-1.9.15-r295/work/fwupd-1.9.15-build/../fwupd-1.9.15/src/fu-self-test.c:6238:9
 *     #15 0x7ba746611705 in __libc_start_call_main ??:0:0
 *     #16 0x7ba7466117c1 in __libc_start_main@GLIBC_2.2.5 ??:0:0
 *     #17 0x5ea908f4f260 in _start ??:0:0
 * 
 * Direct leak of 36 byte(s) in 1 object(s) allocated from:
 *     #0 0x5ea908fe1d2e in __interceptor_malloc ??:0:0
 *     #1 0x7ba74702ece7 in g_malloc ??:0:0
 *     #2 0x7ba747044efe in g_strconcat ??:0:0
 *     #3 0x7ba7470ff93c in ?? ??:0
 *     #4 0x7ba747100524 in ?? ??:0
 *     #5 0x7ba746d3ea38 in ?? ??:0
 *     #6 0x7ba746d3db84 in ?? ??:0
 *     #7 0x7ba746d3d96f in ?? ??:0
 *     #8 0x7ba746d1e9b9 in ?? ??:0
 *     #9 0x5ea909029613 in fu_engine_md_verfmt_func /build/amd64-generic/tmp/portage/sys-apps/fwupd-1.9.15-r295/work/fwupd-1.9.15-build/../fwupd-1.9.15/src/fu-self-test.c:2210:8
 *     #10 0x7ba74704c69e in g_test_queue_destroy ??:?
 *     #11 0x7ba74704c91d in g_test_queue_destroy ??:?
 *     #12 0x7ba74704b5b5 in g_test_run_suite ??:0:0
 *     #13 0x7ba74704b460 in g_test_run ??:0:0
 *     #14 0x5ea909016036 in main /build/amd64-generic/tmp/portage/sys-apps/fwupd-1.9.15-r295/work/fwupd-1.9.15-build/../fwupd-1.9.15/src/fu-self-test.c:6238:9
 *     #15 0x7ba746611705 in __libc_start_call_main ??:0:0
 *     #16 0x7ba7466117c1 in __libc_start_main@GLIBC_2.2.5 ??:0:0
 *     #17 0x5ea908f4f260 in _start ??:0:0
 * 
 * Direct leak of 36 byte(s) in 1 object(s) allocated from:
 *     #0 0x5ea908fe1d2e in __interceptor_malloc ??:0:0
 *     #1 0x7ba74702ece7 in g_malloc ??:0:0
 *     #2 0x7ba747044efe in g_strconcat ??:0:0
 *     #3 0x7ba7470ff93c in ?? ??:0
 *     #4 0x7ba747100524 in ?? ??:0
 *     #5 0x7ba746d3ea38 in ?? ??:0
 *     #6 0x7ba746d3db84 in ?? ??:0
 *     #7 0x7ba746d3d96f in ?? ??:0
 *     #8 0x5ea909016ed3 in fu_remote_list_repair_func /build/amd64-generic/tmp/portage/sys-apps/fwupd-1.9.15-r295/work/fwupd-1.9.15-build/../fwupd-1.9.15/src/fu-self-test.c:5856:8
 *     #9 0x7ba74704c69e in g_test_queue_destroy ??:?
 *     #10 0x7ba74704c91d in g_test_queue_destroy ??:?
 *     #11 0x7ba74704b5b5 in g_test_run_suite ??:0:0
 *     #12 0x7ba74704b460 in g_test_run ??:0:0
 *     #13 0x5ea909016036 in main /build/amd64-generic/tmp/portage/sys-apps/fwupd-1.9.15-r295/work/fwupd-1.9.15-build/../fwupd-1.9.15/src/fu-self-test.c:6238:9
 *     #14 0x7ba746611705 in __libc_start_call_main ??:0:0
 *     #15 0x7ba7466117c1 in __libc_start_main@GLIBC_2.2.5 ??:0:0
 *     #16 0x5ea908f4f260 in _start ??:0:0
 *
```

**Expected behavior**
Should not leak

**fwupd version information**
1.9.15

hughsie, 2024-04-08:
>*     #3 0x7ba7470ff93c in ?? ??:0

...can you install some debuginfo and get the trace again (or maybe try `valgrind --leak-check=full --show-leak-kinds=definite` ? I think it's something ChromeOS specific, as I see:

	LEAK SUMMARY:
	   definitely lost: 0 bytes in 0 blocks
	   indirectly lost: 0 bytes in 0 blocks
	     possibly lost: 10,202 bytes in 60 blocks
	   still reachable: 346,131 bytes in 5,042 blocks
	        suppressed: 0 bytes in 0 blocks

....from valgrind using 1.9.15, and `meson configure -Db_sanitize=leak` and `-Db_sanitize=address,undefined` don't show anything for me either. Thanks.

rishabh-chromium, 2024-04-11:
Weirdly fixed in 1.9.16. Will probably uprev to that version soon so not spending time bisecting. Thanks for looking into this

closed 2024-04-11 by rishabh-chromium

rishabh-chromium, 2024-04-11:
FYI: It was [this](https://github.com/fwupd/fwupd/commit/f7215add8f5ca023c0a3b24720a9386ff79985cf) commit