fwupd/fwupd #7059: fwupd is attempting to download .gz metadata regardless URL given in conff

CragW, 2024-04-09:
**Describe the bug**
fwupd is not using the `MetadataURI` url given in the conff file at `/etc/fwupd/remotes.d/oem-embargo.conf`.  The conff has configured `MetadataURI=https://fwupd.org/downloads/firmware-hash.xml.xz` however in fwupd daemon log it is downloading from `gz` variant.

**Steps to Reproduce**
`sudo fwupdtool refresh --force`

**Expected behavior**
fwupd is reading the embargo.conf and respect to the MetadataURI for metadata sources.

**fwupd version information**

```shell
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.5
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.9.5
runtime   org.kernel                    6.5.0-26-generic
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apt

**Additional questions**

- Operating system and version: Ubuntu 23.10
- Have you tried rebooting? Yes
- Is this a regression? No

hughsie, 2024-04-09:
Is `/etc/fwupd/remotes.d/oem-embargo.conf` still set as .xz or did fwupd "upgrade" it to `gz`? If you set it to `zst` instead does that work as a workaround?

CragW, 2024-04-09:
I've just tested this issue with fwupd 1.9.16, the metadata persist with `zst` regardless the `MetadataURI` in conff is given to `xz`.

It looks like fwupd 1.9.5 persists with `gz`, while 1.9.16 persists with `zst`.  Hereunder is a small slice from fwupd 1.9.16 runtime log.

```
$ sudo fwupdtool refresh --force -vv 2>&1
...
12:44:30.541 Fwupd                changing metadata URI from https://fwupd.org/downloads/firmware-hashtag.xml.zst to https://fwupd.org/downloads/firmware-16598-hashtag.xml.zst
12:44:30.541 Fwupd                downloading https://fwupd.org/downloads/firmware-16598-hashtag.xml.zst
...
```

superm1, 2024-04-09:
I suspect that this is just a bug in older fwupd 1.9.x releases, there shouldn't be anything we can or will do about it.

CragW, 2024-04-09:
Sounds like an intentional enforcement to a predefined postfix file format regardless `metadatauri`, as `gz` is in older fwupd.  If that is the case then this is probably a won't fix issue.

superm1, 2024-04-09:
> It looks like fwupd is broken now on Fedora 39:
> 
> ```
> $ fwupdmgr refresh
> Updating lvfs
> Downloading…             [************************************** ]
> Successfully downloaded new metadata: 0 local devices supported
> ```
> 
> It downloads metadata in new format but can't find any supported devices.
> 
> ```
> $ fwupdmgr --version
> compile   org.freedesktop.fwupd         1.9.16
> compile   com.hughsie.libxmlb           0.3.17
> compile   com.hughsie.libjcat           0.2.1
> runtime   org.freedesktop.fwupd-efi     1.4
> compile   org.freedesktop.gusb          0.4.8
> runtime   com.hughsie.libjcat           0.2.1
> runtime   org.freedesktop.gusb          0.4.8
> runtime   org.freedesktop.fwupd         1.9.16
> runtime   org.kernel                    6.8.4-200.fc39.x86_64
> ```
> 
> The last working version: fwupd-1.9.15-1.fc39

Might be same issue as #7560

hughsie, 2024-04-11:
> Successfully downloaded new metadata: 0 local devices supported

@xvitaly Grab the new libxmlb in updates-testing please.

hughsie, 2024-04-11:
@CragW on 1.9.5 what does `sudo fwupdtool refresh -vv` say when the metadata ends with xz?

hughsie, 2024-04-11:
Okay, weird -- I can't reproduce the .xz to gz thing at all with the proper dell embargo remote and 1.9.5.

superm1, 2024-04-11:
Libxmlb version?

CragW, 2024-04-11:
> Libxmlb version?

Sorry missed this question, the runtime libxmlb is 0.3.10.

```
$ dpkg -l | grep libxmlb
ii  gir1.2-xmlb-2.0:amd64                         0.3.10-2                                amd64        GObject introspection data for libxmlb
ii  libxmlb-dev:amd64                             0.3.10-2                                amd64        Development files for libxmlb
ii  libxmlb2:amd64                                0.3.10-2                                amd64        Binary XML library
```

hughsie, 2025-05-04:
@CragW do you still see this?

closed 2025-05-07 by hughsie