fwupd/fwupd #7060: No testing BIOS update available for Framework 13

Lee245, 2024-04-09:
**Describe the bug**
On my Framework 13 AMD running Fedora 39 with BIOS 3.03, the 3.05 beta BIOS is not available. `lvfs-testing` is enabled. The laptop is plugged into AC.

**Steps to Reproduce**

1. `fwupdmgr enable-remote lvfs-testing`
2. `fwupdmgr refresh --force`
3. `fwupdmgr get-updates`
4. `fwupdmgr update`

**Expected behavior**
3.05 beta BIOS showing up as an update.

**fwupd version information**
```fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.17
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.16
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.16
runtime   org.kernel                    6.8.4-200.fc39.x86_64```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

I believe it was installed OOTB in Fedora 39. Otherwise, via dnf.

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```
fwupdmgr get-devices --show-all-devices
Framework Laptop 13 (AMD Ryzen 7040Series)
│
├─AMD Ryzen 5 7640U w/ Radeon 760M Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a704103
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              2a1df9de-fe02-59c8-8af1-32c8bb011f94 ← CPUID\PRO_0&FAM_19&MOD_74
│ │                       5a4bca32-6547-584f-bb36-920d9adede25 ← CPUID\PRO_0&FAM_19&MOD_74&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ ├─Graphics Processing Unit (GPU):
│ │ │   Device ID:        83eba1fa14bdefee1f6b4bcf06069b298e1c338f
│ │ │   Summary:          AMD AMD_PHOENIX_GENERIC
│ │ │   Current version:  022.012.000.024.000001
│ │ │   Vendor:           Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │ │   GUID:             6c23af4f-d6cd-5ffc-a502-0b85e472e7cb ← AMD\113-PHXGENERIC-001
│ │ │   Device Flags:     • Internal device
│ │ │ 
│ │ └─NE135FBM-N41:
│ │       Device ID:      aec1a869eb0df71b7cea6b3ac71d39b830faf164
│ │       GUID:           ace3bb34-1bbb-5a5f-9694-73c4cede9692 ← DRM\VEN_BOE&DEV_0BCA
│ │       Device Flags:   • Internal device
│ │     
│ ├─Secure Processor:
│ │     Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│ │     Current version:  00.2d.00.74
│ │     Bootloader Version:00.2d.00.74
│ │     Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
│ │     GUIDs:            9eb6a793-7f97-5fb5-b49c-139d2ce3ee46 ← PCI\VEN_1022&DEV_15C7
│ │                       2e4eb727-a7c8-5eba-80cd-553b8d796d80 ← PCI\VEN_1022&DEV_15C7&SUBSYS_F1110006
│ │     Device Flags:     • Internal device
│ │   
│ ├─System Management Unit (SMU):
│ │     Device ID:        db0330716216c629bb2c07256e5d018f499eb6ce
│ │     Summary:          Microcontroller used within CPU/APU program 0
[... 174 lines cut ...]
│       ├── f430515d1e424da0aa37ee30fe09544e-0-rescue.conf
│       ├── f430515d1e424da0aa37ee30fe09544e-6.7.10-200.fc39.x86_64.conf
│       ├── f430515d1e424da0aa37ee30fe09544e-6.7.11-200.fc39.x86_64.conf
│       └── f430515d1e424da0aa37ee30fe09544e-6.8.4-200.fc39.x86_64.conf
├── lost+found
├── symvers-6.7.10-200.fc39.x86_64.xz -> /lib/modules/6.7.10-200.fc39.x86_64/symvers.xz
├── symvers-6.7.11-200.fc39.x86_64.xz -> /lib/modules/6.7.11-200.fc39.x86_64/symvers.xz
├── symvers-6.8.4-200.fc39.x86_64.xz -> /lib/modules/6.8.4-200.fc39.x86_64/symvers.xz
├── System.map-6.7.10-200.fc39.x86_64
├── System.map-6.7.11-200.fc39.x86_64
├── System.map-6.8.4-200.fc39.x86_64
├── vmlinuz-0-rescue-f430515d1e424da0aa37ee30fe09544e
├── vmlinuz-6.7.10-200.fc39.x86_64
├── vmlinuz-6.7.11-200.fc39.x86_64
└── vmlinuz-6.8.4-200.fc39.x86_64
```

**Additional questions**

- Operating system and version: Fedora 39
- Have you tried rebooting? Yes, multiple times
- Is this a regression? No
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?

superm1, 2024-04-09:
You're missing all the "interesting" output from `fwupdtool get-updates -vv`.  It's on stderr, can you redirect that and capture it too?

Lee245, 2024-04-09:
> You're missing all the "interesting" output from `fwupdtool get-updates -vv`. It's on stderr, can you redirect that and capture it too?

Woops, should be there now.

superm1, 2024-04-09:
OK, one more thing.  Can you please run `sudo fwupdtool refresh -vv` and share?

superm1, 2024-04-09:
And can you please attach `/var/lib/fwupd/metadata/lvfs-testing/metadata.xml.zst` and `/var/lib/fwupd/metadata/lvfs-testing/metadata.xml.zst.jcat` after as well?

Lee245, 2024-04-09:
[refresh.log](https://github.com/fwupd/fwupd/files/14922450/refresh.log)
[metadata.zip](https://github.com/fwupd/fwupd/files/14922496/metadata.zip)

Here you go, my good sir.

superm1, 2024-04-09:
Hmm, looks like no components in the silo to me.  This seems pretty bad @hughsie.
We should have seen
```
	g_info("%u components now in silo", components->len);
```

superm1, 2024-04-09:
Seems to be tied specifically to zstd..

hughsie, 2024-04-09:
Seems to be something funky about decoding the file:

    ninja && G_MESSAGES_DEBUG=all ./src/xb-tool compile test.xmlb metadata.xml.zst -v -p
    ignoring invalid file /home/hughsie/Downloads/metadata.xml.zst:ctime=1712688114.854182: Mismatched XML

The last chunk of decoded data seems to be `org.uefi.capsule</value></custom><releases><release id="` which is obviously not correct.

superm1, 2024-04-09:
It's specifically with libxmlb's decompression.  Running `unzstd metadata.xml.zst` and then `G_MESSAGES_DEBUG=all  xb-tool compile test.xmlb metadata.xml -v -p` seems to work

hughsie, 2024-04-09:
I've got an idea; one sec.

superm1, 2024-04-09:
Regression from https://github.com/hughsie/libxmlb/commit/20eeebd4b385da1c7165cf3e9d94acd449d7a2bc perhaps?

hughsie, 2024-04-09:
I think so; https://github.com/hughsie/libxmlb/pull/170 is probably the right fix and seems to work with everything I have to throw at it.

hughsie, 2024-04-09:
If that works I can do a new libxmlb tarball on Thursday.

superm1, 2024-04-09:
Yeah that works for me.

closed 2024-04-09 by hughsie

joakim-tjernlund, 2024-04-11:
hmm, updated to libxmlb-0.3.18 and my fwupdmgr still shows:
Successfully downloaded new metadata: 0 local devices supported

joakim-tjernlund, 2024-04-11:
```
fwupdmgr get-devices
LENOVO 21HES2J20G
│
├─13th Gen Intel Core™ i7-1370P:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0000411c
│     Vendor:             Intel
│     GUIDs:              c3f97ae6-7da2-54cc-ad26-3225f9517203 ← CPUID\PRO_0&FAM_06&MOD_BA
│                         557a4f44-6981-5889-8676-684b9727b697 ← CPUID\PRO_0&FAM_06&MOD_BA&STP_2
│     Device Flags:       • Internal device
│   
├─Fingerprint Sensor:
│     Device ID:          1f914e619b232dc4da68ec3389b98b20163003a2
│     Summary:            Match-On-Chip fingerprint sensor
│     Current version:    01000352
│     Vendor:             Goodix (USB:0x27C6)
│     Install Duration:   10 seconds
│     Serial Number:      UID0D72C39E_XXXX_MOC_B0
│     GUID:               14450e82-8667-54c3-98e1-098d8c6dc3b8 ← USB\VID_27C6&PID_6594
│     Device Flags:       • Updatable
│                         • Device can recover flash failures
│                         • Signed Payload
│   
├─Integrated RGB Camera:
│     Device ID:          23ec719b6aabc2d2dac5176c232f0da7a21881b0
│     Current version:    68.8
│     Vendor:             SunplusIT Inc (USB:0x5986)
│     Serial Number:      01.00.00
│     GUID:               d9b95caa-1706-5d2d-8ea4-e19fdfb246a7 ← USB\VID_5986&PID_2142
│     Device Flags:       • Updatable
│   
├─MZVL2512HDJD-00BLL:
│     Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:            NVM Express solid state drive
│     Current version:    6L2QGXD7
│     Vendor:             Samsung (NVME:0x144D)
│     Serial Number:      S760NE0W812423
│     GUIDs:              bec63ed7-a95f-54fe-b8cc-8e9fee64ba5a ← NVME\VEN_144D&DEV_A80A
│                         310f81b5-6fce-501e-acfb-487d10501e78 ← NVME\VEN_144D&DEV_A80A&SUBSYS_144DA801
│                         8aa4f523-f64a-57b5-9e3e-623df117ed3f ← SAMSUNG MZVL2512HDJD-00BLL
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─Raptor Lake-P [Iris Xe Graphics]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    04
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              0e8b1f1a-8c2b-53b6-89d7-d1a05dbc039d ← PCI\VEN_8086&DEV_A7A0
│                         b4d3d7b2-aba5-529e-ae7c-a5f1d765ca04 ← PCI\VEN_8086&DEV_A7A0&SUBSYS_17AA230E
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─System Firmware:
│ │   Device ID:          398af2f706c8d17f9c58b4837c8322dfcd3c7c32
│ │   Summary:            UEFI System Resource Table device (updated via NVRAM)
│ │   Current version:    65580
│ │   Vendor:             LENOVO (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUID:               a1e2c5f2-a296-4603-9a8b-61737d4e3ea6
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message
[... 246 lines cut ...]
│     Current version:    18548864
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               4e78ce68-5389-4a95-bf10-e3568c30caf8
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
├─UEFI Device Firmware:
│     Device ID:          bede098fde861317d555511f082f412aa2285b09
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    3344935
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               023a338b-246f-47e0-b4e5-304203184ec1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
```

superm1, 2024-04-11:
Have you restarted the service since upgrading and refreshed metadata since then?

superm1, 2024-04-11:
And also what version of `libxmlb` was the daemon compiled _against_? Perhaps can we see a build log?

joakim-tjernlund, 2024-04-11:
Yes to both Qs  above. Rebuilt fwupd against new libxlmb too(on Gentoo)

superm1, 2024-04-11:
Okay, please open another issue and attach daemon logs with `-vv` added to the exec command.

From the client run the refresh cycle and let's see what it shows.

Please also attach build logs for fwupd.

joakim-tjernlund, 2024-04-11:
```
 fwupdmgr refresh -vv --force 
(fwupdmgr:6082): GLib-DEBUG: 19:15:26.147: setenv()/putenv() are not thread-safe and should not be used after threads are created
(pkttyagent:6084): GLib-GIO-DEBUG: 19:15:26.149: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:6082): GLib-GIO-DEBUG: 19:15:26.153: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:6082): GLib-GIO-DEBUG: 19:15:26.154: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:6082): dconf-DEBUG: 19:15:26.154: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:6082): dconf-DEBUG: 19:15:26.154: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:6082): dconf-DEBUG: 19:15:26.154: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:6082): dconf-DEBUG: 19:15:26.154: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:6082): dconf-DEBUG: 19:15:26.154: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:6082): dconf-DEBUG: 19:15:26.154: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:6082): dconf-DEBUG: 19:15:26.154: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:6082): dconf-DEBUG: 19:15:26.154: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:6082): dconf-DEBUG: 19:15:26.154: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:6082): dconf-DEBUG: 19:15:26.154: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:6082): GLib-GIO-DEBUG: 19:15:26.154: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:6082): GLib-GIO-DEBUG: 19:15:26.154: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:6082): dconf-DEBUG: 19:15:26.154: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:6082): dconf-DEBUG: 19:15:26.154: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:6082): dconf-DEBUG: 19:15:26.154: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:6082): dconf-DEBUG: 19:15:26.155: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:6082): dconf-DEBUG: 19:15:26.155: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:6082): GLib-GIO-DEBUG: 19:15:26.155: _g_io_module_get_default: Found default implementation gvfs (GDaemonVfs) for ‘gio-vfs’
(fwupdmgr:6082): pxbackend-DEBUG: 19:15:26.155: px_config_sysconfig_set_config_file: Could not read file /etc/sysconfig/proxy
(fwupdmgr:6082): pxbackend-DEBUG: 19:15:26.155: Active config plugins:
(fwupdmgr:6082): pxbackend-DEBUG: 19:15:26.155:  - config-env
(fwupdmgr:6082): pxbackend-DEBUG: 19:15:26.155:  - config-sysconfig
(fwupdmgr:6082): GLib-GIO-DEBUG: 19:15:26.155: Failed to initialize portal (GNetworkMonitorPortal) for gio-network-monitor: Not using portals
(fwupdmgr:6082): GLib-GIO-DEBUG: 19:15:26.156: _g_io_module_get_default: Found default implementation networkmanager (GNetworkMonitorNM) for ‘gio-network-monitor’
(fwupdmgr:6082): pxbackend-DEBUG: 19:15:26.156: px_manager_on_network_changed: Network connection changed, clearing pac data
(fwupdmgr:6082): pxbackend-DEBUG: 19:15:26.156: px_manager_constructed: Up and running
(fwupdmgr:6082): GLib-GIO-DEBUG: 19:15:26.156: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:6082): Fwupd-DEBUG: 19:15:26.157: Emitting ::status-changed() [idle]
Updating lvfs-testing
Fwupd-INFO: 19:15:26.159: downloading https://cdn.fwupd.org/downloads/firmware-testing.xml.zst.jcat
(fwupdmgr:6082): pxbackend-DEBUG: 19:15:26.159: px_manager_get_proxies_sync: url=https://cdn.fwupd.org/downloads/firmware-testing.xml.zst.jcat online=1
(fwupdmgr:6082): pxbackend-DEBUG: 19:15:26.159: px_manager_get_proxies_sync: Proxy[0] = direct://
(fwupdmgr:6082): Fwupd-DEBUG: 19:15:26.159: Emitting ::status-changed() [downloading]
Downloading…             [      \                                ]Fwupd-INFO: 19:15:26.398: download progress: 100%
(fwupdmgr:6082): Fwupd-DEBUG: 19:15:26.398: Emitting ::status-changed() [idle]
Fwupd-INFO: 19:15:26.398: status-code was 200
Fwupd-INFO: 19:15:26.399: changing metadata URI from https://cdn.fwupd.org/downloads/firmware-testing.xml.zst to https://cdn.fwupd.org/downloads/firmware-05534-testing.xml.zst
Fwupd-INFO: 19:15:26.399: metadata signature of lvfs-testing is unchanged, skipping

Updating lvfs
Fwupd-INFO: 19:15:26.399: downloading https://cdn.fwupd.org/downloads/firmware.xml.zst.jcat
(fwupdmgr:6082): pxbackend-DEBUG: 19:15:26.399: px_manager_get_proxies_sync: url=https://cdn.fwupd.org/downloads/firmware.xml.zst.jcat online=1
(fwupdmgr:6082): pxbackend-DEBUG: 19:15:26.399: px_manager_get_proxies_sync: Proxy[0] = direct://
(fwupdmgr:6082): Fwupd-DEBUG: 19:15:26.399: Emitting ::status-changed() [downloading]
Downloading…             [***************************************]Fwupd-INFO: 19:15:26.636: download progress: 0%
Fwupd-INFO: 19:15:26.636: download progress: 100%
Downloading…             [***************************************](fwupdmgr:6082): Fwupd-DEBUG: 19:15:26.636: Emitting ::status-changed() [idle]
Fwupd-INFO: 19:15:26.636: status-code was 200
Fwupd-INFO: 19:15:26.636: changing metadata URI from https://cdn.fwupd.org/downloads/firmware.xml.zst to https://cdn.fwupd.org/downloads/firmware-07166-stable.xml.zst
Fwupd-INFO: 19:15:26.636: metadata signature of lvfs is unchanged, skipping
```

joakim-tjernlund, 2024-04-11:
oh, new issue. OK will do

superm1, 2024-04-11:
> oh, new issue. OK will do

Yes please

>  fwupdmgr refresh -vv --force 

Not the client; the daemon.  like modify `fwupd.service.` and then reload systemd units.

joakim-tjernlund, 2024-04-11:
https://github.com/fwupd/fwupd/issues/7078

aenertia, 2024-04-16:
Affects FC40 as well (0.3.18 has not been pushed to testing).

> `aenertia@emiemi:~$ dnf info libxmlb
> Last metadata expiration check: 0:00:29 ago on Tue 16 Apr 2024 15:00:13.
> Installed Packages
> Name         : libxmlb
> Version      : 0.3.17
> Release      : 1.fc40
> Architecture : x86_64
> Size         : 288 k
> Source       : libxmlb-0.3.17-1.fc40.src.rpm
> Repository   : @System
> From repo    : updates-testing
> Summary      : Library for querying compressed XML metadata
> URL          : https://github.com/hughsie/libxmlb
> License      : LGPL-2.1-or-later
> Description  : XML is slow to parse and strings inside the document cannot be memory mapped as
>              : they do not have a trailing NUL char. The libxmlb library takes XML source, and
>              : converts it to a structured binary representation with a deduplicated string
>              : table -- where the strings have the NULs included.
>              : 
>              : This allows an application to mmap the binary XML file, do an XPath query and
>              : return some strings without actually parsing the entire document. This is all
>              : done using (almost) zero allocations and no actual copying of the binary data.
>

superm1, 2024-04-16:
> Affects FC40 as well (0.3.18 has not been pushed to testing).

From an upstream perspective it's finished.  I suggest advocating for that change in the matching RH bugzilla.

tristan957, 2024-04-16:
This still doesn't seem to work for me. I installed the RPM from the link above, and fwupdmgr still doesn't seem to recognize the update. I have lvfs-testing enabled. I made sure to restart after installing the libxmlb update.

```
$ fwupdmgr --force refresh
Updating lvfs-testing
Downloading…             [      \                                ]
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 0 local devices supported

$ fwupdmgr get-updates
Devices with no available firmware updates:
 • System Firmware
 • WD BLACK SN850X 1000GB
 • Fingerprint Sensor
 • UEFI dbx
No updatable device
```

superm1, 2024-04-16:
Try manually deleting the cache in `/var/cache/fwupd`.  The automatic invalidation we talked about won't work until the next libxmlb version.

tristan957, 2024-04-16:
I deleted it, but still not seeing any changes with respect to the behavior. I have an empty `/var/cache/fwupd` after `sudo rm -rf /var/cache/fwupd/*`

superm1, 2024-04-16:
And then restart the service `sudo systemctl restart fwupd.service` followed by refreshing again `fwupdmgr refresh`

aenertia, 2024-04-16:
> That was it. Should've followed the linked issue a little better. Thanks
> for your support once again 😄. fwupd/lvfs are truly life changing pieces
> of software.

It's in updates-testing not updates ; yes my bad I thought I had enabled
testing on this machine.