fwupd/fwupd #7078: fwupdmgr refresh --force finds 0 local devices supported

joakim-tjernlund, 2024-04-11:
**Describe the bug**
fwupd does not find any support FW on my ThinkPad , 1.9.14 had no such problem.
Note, updated to the new libxmlb as well

**Steps to Reproduce**
fwupdmgr refresh --force

**Expected behavior**
Should find a bunch of supported devices like 1.9.14 did

**fwupd version information**
libxmlb-0.3.18 

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.16
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.5
compile   org.freedesktop.gusb          0.4.7
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.7
runtime   org.kernel                    6.6.25-x86_64
runtime   org.freedesktop.fwupd         1.9.16

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
emerge sys-apps/fwupd

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
LENOVO 21HES2J20G
│
├─Unknown Device:
│     Device ID:          348dee50b7a5bea9eb8cf66e1e1d42290f02cb13
│     Serial Number:      41125293NB
│     GUID:               062c1c51-dde4-58c5-93dc-485644c5e1d2 ← DRM\VEN_NEC&DEV_68D6
│   
├─Unknown Device:
│     Device ID:          6b5a06645e27c4e94b6e67d5185c46d11ec9ad67
│     GUID:               c96751f4-16e8-59ea-975d-69d3dac88d7d ← GPIO\ID_INTC1055:00
│   
├─13th Gen Intel Core™ i7-1370P:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0000411c
│     Vendor:             Intel
│     GUIDs:              c3f97ae6-7da2-54cc-ad26-3225f9517203 ← CPUID\PRO_0&FAM_06&MOD_BA
│                         557a4f44-6981-5889-8676-684b9727b697 ← CPUID\PRO_0&FAM_06&MOD_BA&STP_2
│     Device Flags:       • Internal device
│   
├─Fingerprint Sensor:
│     Device ID:          0d5d05911800242bb1f35287012cdcbd9b381148
│     Summary:            Match-On-Chip fingerprint sensor
│     Current version:    01000352
│     Vendor:             Goodix (USB:0x27C6)
│     Install Duration:   10 seconds
│     Serial Number:      UID0D72C39E_XXXX_MOC_B0
│     GUID:               14450e82-8667-54c3-98e1-098d8c6dc3b8 ← USB\VID_27C6&PID_6594
│     Device Flags:       • Updatable
│                         • Device can recover flash failures
│                         • Signed Payload
│   
├─Integrated RGB Camera:
│     Device ID:          3fa281ddf80d8a06b8ee5d8beb48d38ee95a9627
│     Current version:    68.8
│     Vendor:             SunplusIT Inc (USB:0x5986)
[... 307 lines cut ...]
│     Current version:    3344935
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               023a338b-246f-47e0-b4e5-304203184ec1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Lenovo Ltd. PK CA 2012
      Vendor:             Lenovo Ltd.
      GUID:               71599d14-9b31-5270-b3bd-74c494585820 ← UEFI\CRT_9AEF2123F4DE7C19AFABD909BB2C8CAC4411E07E
```

</details>

**Additional questions**

- Operating system and version: Gentoo
- Have you tried rebooting? Yes
- Is this a regression? Yes

joakim-tjernlund, 2024-04-11:
more info:
```
fwupdmgr refresh -vv --force 
(fwupdmgr:8640): GLib-DEBUG: 19:25:52.795: setenv()/putenv() are not thread-safe and should not be used after threads are created
(pkttyagent:8642): GLib-GIO-DEBUG: 19:25:52.797: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:8640): GLib-GIO-DEBUG: 19:25:52.801: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:8640): GLib-GIO-DEBUG: 19:25:52.802: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:8640): dconf-DEBUG: 19:25:52.802: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:8640): dconf-DEBUG: 19:25:52.802: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:8640): dconf-DEBUG: 19:25:52.802: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:8640): dconf-DEBUG: 19:25:52.802: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:8640): dconf-DEBUG: 19:25:52.802: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:8640): dconf-DEBUG: 19:25:52.802: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:8640): dconf-DEBUG: 19:25:52.802: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:8640): dconf-DEBUG: 19:25:52.802: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:8640): dconf-DEBUG: 19:25:52.802: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:8640): dconf-DEBUG: 19:25:52.802: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:8640): GLib-GIO-DEBUG: 19:25:52.802: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:8640): GLib-GIO-DEBUG: 19:25:52.802: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:8640): dconf-DEBUG: 19:25:52.803: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:8640): dconf-DEBUG: 19:25:52.803: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:8640): dconf-DEBUG: 19:25:52.803: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:8640): dconf-DEBUG: 19:25:52.803: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:8640): dconf-DEBUG: 19:25:52.803: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:8640): GLib-GIO-DEBUG: 19:25:52.803: _g_io_module_get_default: Found default implementation gvfs (GDaemonVfs) for ‘gio-vfs’
(fwupdmgr:8640): pxbackend-DEBUG: 19:25:52.803: px_config_sysconfig_set_config_file: Could not read file /etc/sysconfig/proxy
(fwupdmgr:8640): pxbackend-DEBUG: 19:25:52.803: Active config plugins:
(fwupdmgr:8640): pxbackend-DEBUG: 19:25:52.804:  - config-env
(fwupdmgr:8640): pxbackend-DEBUG: 19:25:52.804:  - config-sysconfig
(fwupdmgr:8640): GLib-GIO-DEBUG: 19:25:52.804: Failed to initialize portal (GNetworkMonitorPortal) for gio-network-monitor: Not using portals
(fwupdmgr:8640): GLib-GIO-DEBUG: 19:25:52.804: _g_io_module_get_default: Found default implementation networkmanager (GNetworkMonitorNM) for ‘gio-network-monitor’
(fwupdmgr:8640): pxbackend-DEBUG: 19:25:52.804: px_manager_on_network_changed: Network connection changed, clearing pac data
(fwupdmgr:8640): pxbackend-DEBUG: 19:25:52.804: px_manager_constructed: Up and running
(fwupdmgr:8640): GLib-GIO-DEBUG: 19:25:52.804: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:8640): Fwupd-DEBUG: 19:25:52.805: Emitting ::status-changed() [idle]
Updating lvfs-testing
Fwupd-INFO: 19:25:52.808: downloading https://cdn.fwupd.org/downloads/firmware-testing.xml.zst.jcat
(fwupdmgr:8640): pxbackend-DEBUG: 19:25:52.808: px_manager_get_proxies_sync: url=https://cdn.fwupd.org/downloads/firmware-testing.xml.zst.jcat online=1
(fwupdmgr:8640): pxbackend-DEBUG: 19:25:52.808: px_manager_get_proxies_sync: Proxy[0] = direct://
(fwupdmgr:8640): Fwupd-DEBUG: 19:25:52.808: Emitting ::status-changed() [downloading]
Downloading…             [       \                               ]Fwupd-INFO: 19:25:53.057: download progress: 100%
(fwupdmgr:8640): Fwupd-DEBUG: 19:25:53.057: Emitting ::status-changed() [idle]
Fwupd-INFO: 19:25:53.057: status-code was 200
Fwupd-INFO: 19:25:53.057: changing metadata URI from https://cdn.fwupd.org/downloads/firmware-testing.xml.zst to https://cdn.fwupd.org/downloads/firmware-05534-testing.xml.zst
Fwupd-INFO: 19:25:53.057: metadata signature of lvfs-testing is unchanged, skipping

Updating lvfs
Fwupd-INFO: 19:25:53.058: downloading https://cdn.fwupd.org/downloads/firmware.xml.zst.jcat
(fwupdmgr:8640): pxbackend-DEBUG: 19:25:53.058: px_manager_get_proxies_sync: url=https://cdn.fwupd.org/downloads/firmware.xml.zst.jcat online=1
(fwupdmgr:8640): pxbackend-DEBUG: 19:25:53.058: px_manager_get_proxies_sync: Proxy[0] = direct://
(fwupdmgr:8640): Fwupd-DEBUG: 19:25:53.058: Emitting ::status-changed() [downloading]
Downloading…             [***************************************](fwupdmgr:8640): Fwupd-DEBUG: 19:25:53.298: Emitting ::status-changed() [idle]
Fwupd-INFO: 19:25:53.298: status-code was 200
Fwupd-INFO: 19:25:53.298: changing metadata URI from https://cdn.fwupd.org/downloads/firmware.xml.zst to https://cdn.fwupd.org/downloads/firmware-07166-stable.xml.zst
Fwupd-INFO: 19:25:53.298: metadata signature of lvfs is unchanged, skipping

Successfully downloaded new metadata: 0 local devices supported
```

superm1, 2024-04-11:
I did try to reproduce this with Fedora 40 with:
* libxmlb-0.3.18-1.fc40.x86_64
* fwupd-1.9.16-1.fc40.x86_64

I get plenty of devices from `fwupdmgr get-devices --filter supported`

superm1, 2024-04-11:
And from your build of libxmlb what happens with:

```
G_MESSAGES_DEBUG=all xb-tool compile test.xmlb metadata.xml.zst -v -p
```

joakim-tjernlund, 2024-04-11:
[fwupd-vv-log.txt](https://github.com/fwupd/fwupd/files/14949416/fwupd-vv-log.txt)

joakim-tjernlund, 2024-04-11:
> G_MESSAGES_DEBUG=all xb-tool compile test.xmlb metadata.xml.zst -v -p

```
_MESSAGES_DEBUG=all xb-tool compile test.xmlb /var/lib/fwupd/metadata/lvfs/metadata.xml.zst -v -p
(xb-tool:12019): GLib-DEBUG: 19:39:14.510: setenv()/putenv() are not thread-safe and should not be used after threads are created
(xb-tool:12019): GLib-GIO-DEBUG: 19:39:14.512: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(xb-tool:12019): GLib-GIO-DEBUG: 19:39:14.514: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
(xb-tool:12019): XbSilo-DEBUG: 19:39:14.515: attempting to load /var/lib/fwupd/metadata/lvfs/test.xmlb
(xb-tool:12019): XbSilo-DEBUG: 19:39:14.515: failed to load silo: Failed to open file “/var/lib/fwupd/metadata/lvfs/test.xmlb”: open() failed: No such file or directory
94.42ms     compile /var/lib/fwupd/metadata/lvfs/metadata.xml.zst:ctime=1712852648.224157
99.50ms     get size nodetab
6.57ms      adding strtab element
6.17ms      adding strtab attr name
7.46ms      adding strtab attr value
22.49ms     adding strtab text
4.90ms      adding strtab tokens
11.32ms     writing nodetab
9.28ms      fixing ->parent and ->next
1.42ms      appending strtab
0.01ms      parse blob
3.18ms      save file
0.08ms      parse blob
0.13ms      loaded file
```

joakim-tjernlund, 2024-04-11:
> /var/lib/fuwpd/metadata/lvfs/

cd /var/lib/fuwpd/metadata/lvfs
-bash: cd: /var/lib/fuwpd/metadata/lvfs: No such file or directory

superm1, 2024-04-11:
okay so it looks like it's compiling.  I wonder if it's a caching problem.  Can you please backup `/var/cache/fwupd` (so we can examine it if this helps) and then remove the contents of it?

superm1, 2024-04-11:
> > /var/lib/fuwpd/metadata/lvfs/
> 
> cd /var/lib/fuwpd/metadata/lvfs -bash: cd: /var/lib/fuwpd/metadata/lvfs: No such file or directory

Typo sorry, corrected it.

joakim-tjernlund, 2024-04-11:
> okay so it looks like it's compiling. I wonder if it's a caching problem. Can you please backup `/var/cache/fwupd` (so we can examine it if this helps) and then remove the contents of it?

```
rm -rf *
se-jocke9-lx fwupd # fwupdmgr refresh 
Metadata is up to date; use --force to refresh again.
se-jocke9-lx fwupd # fwupdmgr refresh --force
Updating lvfs-testing
Downloading…             [      \                                ]
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 0 local devices supported
se-jocke9-lx fwupd # pwd
/var/cache/fwupd
```

superm1, 2024-04-11:
What version of zstd do you have?

joakim-tjernlund, 2024-04-11:
> Can you please attach the content from `/var/lib/fwupd/metadata/lvfs/*`?

github wont let me

joakim-tjernlund, 2024-04-11:
> What version of zstd do you have?

app-arch/zstd 1.5.5-r1

lukasgierth, 2024-04-11:
Same situation on Fedora Silverblue 39. Tested with Thinkpad X280, Thinkpad T14 AMD G1.

> compile   org.freedesktop.fwupd         1.9.16
compile   com.hughsie.libxmlb           0.3.17
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.16
runtime   org.kernel                    6.8.4-200.fc39.x86_64

superm1, 2024-04-11:
> Same situation on Fedora Silverblue 39. Tested with Thinkpad X280, Thinkpad T14 AMD G1.
>
> compile   com.hughsie.libxmlb           0.3.17
> compile   com.hughsie.libjcat           0.2.1
> runtime [...]

For your case please update to libxmlb 0.3.18, should be in testing iirc.

superm1, 2024-04-11:
> > Can you please attach the content from `/var/lib/fwupd/metadata/lvfs/*`?
> 
> github wont let me

Can you zip it up and attach the zip?

lukasgierth, 2024-04-11:
> For your case please update to libxmlb 0.3.18, should be in testing iirc.

Did an "override replace" in Silverblue and now the version is 0.3.18.
Did not change the outcome however. But "fwupdmgr --version" still shows "0.3.17". Is it compiled into fwupd or why the difference?

hughsie, 2024-04-11:
> fwupdmgr --version" still shows "0.3.17".

Ya, that's the compile time version, nothing to worry about. You'll need to restart the fwupd daemon (or restart the computer) for the new libxmlb to take effect.

hughsie, 2024-04-11:
> USE=lzma zlib -lz4 -static-libs

If you're using static libs you'll need to recompile fwupd against the new libxmlb, no?

lukasgierth, 2024-04-11:
> > fwupdmgr --version" still shows "0.3.17".
> 
> Ya, that's the compile time version, nothing to worry about. You'll need to restart the fwupd daemon (or restart the computer) for the new libxmlb to take effect.

Did restart the system (only safe way to apply the update in a rpm-ostree based system), I never use "rpm-ostree apply-live", is still way too buggy.

So the newer libxmlb version did not change the outcome, still 0 local devices supported.

hughsie, 2024-04-11:
> So the newer libxmlb version did not change the outcome

Ahh, did you refresh? You'll either need to `fwupdmgr refresh --force` or wait 1 day for the next metadata refresh.

joakim-tjernlund, 2024-04-11:
> > USE=lzma zlib -lz4 -static-libs
> 
> If you're using static libs you'll need to recompile fwupd against the new libxmlb, no?

no, "-static-libs" means no static libs and I did recompile fwupd as well

joakim-tjernlund, 2024-04-11:
I just did one more refresh:
```
fwupdmgr refresh --force
Updating lvfs-testing
Downloading…             [      \                                ]
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 5 local devices supported
```
Now there are 5 devices

I haven't changed anything, it just worked.

lukasgierth, 2024-04-11:
> > So the newer libxmlb version did not change the outcome
> 
> Ahh, did you refresh? You'll either need to `fwupdmgr refresh --force` or wait 1 day for the next metadata refresh.

[image]

Does not work for me. Is the Thinkpad x280, there should be supported devices.

hughsie, 2024-04-11:
Either wait 24h or do `sudo rm /var/lib/fwupd/metadata/lvfs/*`, restart fwupd and then `fwupdmgr refresh`.

joakim-tjernlund, 2024-04-11:
Same for me,  seems something did not refresh the cache.
I guess by tmw the cache would update anyway so maybe no harm ?

superm1, 2024-04-11:
I wonder if we want to invalidate caches when libxmlb versions changes.  That's what all the browsers (try) to do when mesa changes for very similar types of problems.

hughsie, 2024-04-11:
We invalidate it when the fwupd version changes; we need to do the same for libxmlb...

joakim-tjernlund, 2024-04-11:
> Either wait 24h or do `sudo rm /var/lib/fwupd/metadata/lvfs/*`, restart fwupd and then `fwupdmgr refresh`.

What threw me was that I figured refresh --force would override caches et. al

hughsie, 2024-04-11:
e.g. https://github.com/hughsie/libxmlb/pull/172

closed 2024-04-13 by superm1