fwupd/fwupd #7082: [Ubuntu 24.04][redfish plugin] no find devices with "fwupdmgr refresh --force" command

vincent-tang1, 2024-04-12:
**Describe the bug**
using ubuntu 24.04, this linux distrubution adapoted built-in fwupd 1.9.16, as result, no device found using "refresh" option.
but using **sanp fwupd 1.9.16** instead, no this issue happened

**Steps to Reproduce**
fwupdmgr refresh --force
```
vincent@vincent:~$ sudo fwupdmgr refresh --force
Updating lvfs-testing
Downloading…             [*******************                    ]
Updating lvfs
Downloading…             [*********************************      ]
Successfully downloaded new metadata: 0 local devices supported
```

**Expected behavior**
should find the device after runing "refresh" command

**Other finding**
if installing fwupd with same version 1.9.16 by **snap** in ubuntu 24.04 as below, it can normally work
```
vincent@vincent:~$ snap list
Name         Version    Rev    Tracking          Publisher       Notes
...
fwupd        1.9.16     5907   latest/stable     richard-hughes  -
snapd        2.62       21465  latest/candidate  canonical✓      snapd

vincent@vincent:~$ sudo fwupdmgr refresh --force
Updating lvfs-testing
Downloading…             [***************************************]
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 1 local device supported
```

**fwupd version information**
```
vincent@vincent:~$ fwupdmgr --version

compile   org.freedesktop.fwupd         1.9.16
compile   com.hughsie.libxmlb           0.3.17
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.8.0-22-generic
runtime   org.freedesktop.fwupd         1.9.16
```

superm1, 2024-04-12:
Upgrade libxmlb from proposed, it's fixed there.

superm1, 2024-04-13:
It promoted out of proposed, run your updates and things are fine now.

https://launchpad.net/ubuntu/+source/libxmlb/0.3.18-1

closed 2024-04-13 by superm1

vincent-tang1, 2024-04-18:
Sorry for late reply,

not sure if below version is correct before i have not upgraded libxmlb yet?
```
vincent@vincent:/lib$ sudo apt-cache policy libxmlb*
libxmlbeans-java:
  Installed: (none)
  Candidate: 4.0.0-2
  Version table:
     4.0.0-2 500
        500 http://tw.archive.ubuntu.com/ubuntu noble/universe amd64 Packages
libxmlb2:
  Installed: 0.3.18-1
  Candidate: 0.3.18-1
  Version table:
 *** 0.3.18-1 500
        500 http://tw.archive.ubuntu.com/ubuntu noble/main amd64 Packages
        100 /var/lib/dpkg/status
libxmlb-utils:
  Installed: (none)
  Candidate: 0.3.18-1
  Version table:
     0.3.18-1 500
        500 http://tw.archive.ubuntu.com/ubuntu noble/universe amd64 Packages
libxmlbird1t64:
  Installed: (none)
  Candidate: 1.2.15-1.1build1
  Version table:
     1.2.15-1.1build1 500
        500 http://tw.archive.ubuntu.com/ubuntu noble/universe amd64 Packages
libxmlbird-dev:
  Installed: (none)
  Candidate: 1.2.15-1.1build1
  Version table:
     1.2.15-1.1build1 500
        500 http://tw.archive.ubuntu.com/ubuntu noble/universe amd64 Packages
libxmlb-tests:
  Installed: (none)
  Candidate: 0.3.18-1
  Version table:
     0.3.18-1 500
        500 http://tw.archive.ubuntu.com/ubuntu noble/universe amd64 Packages
libxmlb-dev:
  Installed: (none)
  Candidate: 0.3.18-1
  Version table:
     0.3.18-1 500
        500 http://tw.archive.ubuntu.com/ubuntu noble/main amd64 Packages
libxmlbeans-maven-plugin-java:
  Installed: (none)
  Candidate: 2.3.3-4
  Version table:
     2.3.3-4 500
        500 http://tw.archive.ubuntu.com/ubuntu noble/universe amd64 Packages
libxmlbird1:
  Installed: (none)
  Candidate: (none)
  Version table:
```

superm1, 2024-04-18:
If it's still failing after upgrading that please delete the cache from /var/cache/fwupd and restart the service and try again.