fwupd/fwupd #7084: Plugin algoltek-usb has a coverage score of 5%

hughsie, 2024-04-12:
The algoltek-usb  fwupd plugin has a very low coverage score. This means it’s not being adequately tested during CI and pre-release testing, and might mean we release the plugin to end users with an accidental and unfortunate regression. The best way to make sure this doesn’t happen is to provide a device-test, and there are a lot of examples in https://github.com/fwupd/fwupd/tree/main/data/device-tests for existing hardware.

The emulated device test in CI downloads a “_device emulation data_” from the LVFS which sets up a virtual USB device that we can control from the daemon. Then we download a cab archive of the actual firmware for the device (also from the LVFS) and then install the _real_ firmware archive on the _virtual_ device. This usually tests out a lot of the plugin functionality, and is enough to make the plugin coverage rise up to **70-90%** typically – which is great, and means we can refactor the common daemon code without risk of introducing a regression. All new plugins going into fwupd now must have an emulated test for CI, and we’re now trying to get the existing plugins up to the same standard.

We’re of course happy to help with this task, but we don’t have all the physical hardware needed to generate an emulation of the update being applied. This is why I’ve assigned the plugin owner to this issue :) What I’d like you to do is:

* Read https://fwupd.github.io/libfwupdplugin/device-emulation.html
* Make sure there is at least one firmware on the LVFS that works with this hardware (ideally two) – but it doesn’t have to be in a `embargo`, `testing`, or `stable` remote – it can remain in private if needed.
* Plug in the device, and follow the steps in https://github.com/fwupd/fwupd/blob/main/docs/device-emulation.md
* Attach to this issue:
  * The `fwupdmgr get-devices` output when the device is in the normal runtime mode
  * The link to the firmware file on the LVFS, e.g. [https://fwupd.org/downloads/hash-firmware.cab](https://fwupd.org/hash-firmware.cab)
  * The zip file(s) for each of the cab archives you’ve emulated

We can then test the emulation works on our system (without physical hardware), build the JSON file and add it to the fwupd repo for you. Although one firmware will work, ideally we want to test upgrading from 1.0.2 to 1.0.3 and downgrading from 1.0.3 to 1.0.2 (with example version numbers) rather than just reinstalling 1.0.3 onto itself.

If you have any questions, queries or worries please let us know, either here or by email. Thanks!

lingc24, 2024-05-06:
This is the information required :

* The fwupdmgr get-devices output when the device is in the normal runtime mode
```
└─Video Adaptor:
      Device ID:          b0a78eb71f4eeea7df8fb114522556ba8ce22074
      Current version:    00.09.14_03.00.00
      Vendor:             USB C (USB:0x25A4)
      Serial Number:      000000000001
      GUID:               b5edd38c-b9f3-5917-819a-0947612e4d7e ← USB\VID_25A4&PID_9421
      Device Flags:       • Updatable
                          • Unsigned Payload
```
*  The link to the firmware file on the LVFS
1. https://fwupd.org/downloads/60518fa0f9a78a38afda311e8a08f21995dc21020f16cd3b959c159f1ee360ca-algoltek-ag9421-00.09.07_02.18.05.cab
2. https://fwupd.org/downloads/6984d9c2e03ddc9f12879565587c2f42ded40209d0434f2f269208213f052dfd-algoltek-ag9421-00.09.14_03.00.00.cab

* The zip file(s) for each of the cab archives
[algoltek-usb_00.09.07_02.18.05.zip](https://github.com/fwupd/fwupd/files/15218983/algoltek-usb_00.09.07_02.18.05.zip)
[algoltek-usb_00.09.14_03.00.00.zip](https://github.com/fwupd/fwupd/files/15218984/algoltek-usb_00.09.14_03.00.00.zip)

Hope this will help.
Thanks!

hughsie, 2024-05-06:
@lingc24 nearly! I had to use 33333 to avoid the plugin segfaulting on me (so testing FTW) but I then got:

    data clear failure: no matching event for ControlTransfer:Direction=0x01,RequestType=0x02,Recipient=0x01,Request=0x19,Value=0x201f,Idx=0x0000,Data=AxkAAAAAAAAAAOQ=,Length=0xb

I'm a bit confused as in the emulation I only see one call to ERS:

        {
          "Id" : "ControlTransfer:Direction=0x01,RequestType=0x02,Recipient=0x01,Request=0x19,Value=0x0000,Idx=0x0000,Data=AxkAAAAAAAAAAOQ=,Length=0xb",
          "Data" : "AxkAAAAAAAAAAOQ="
        },

but the code clearly two:

	if (!fu_algoltek_usb_device_ers(self, 0x20, AG_IDENTIFICATION_128K_ADDR, error))
		return FALSE;
	if (!fu_algoltek_usb_device_ers(self, 0x20, AG_IDENTIFICATION_256K_ADDR, error))
		return FALSE;

Did you use git main branch to generate these emulations?

lingc24, 2024-05-07:
Here is the new information:

* The fwupdmgr get-devices output when the device is in the normal runtime mode
```
└─Video Adaptor:
      Device ID:          b0a78eb71f4eeea7df8fb114522556ba8ce22074
      Current version:    00.09.14_03.00.00
      Vendor:             USB C (USB:0x25A4)
      Serial Number:      000000000001
      GUID:               b5edd38c-b9f3-5917-819a-0947612e4d7e ← USB\VID_25A4&PID_9421
      Device Flags:       • Updatable
                          • Device stages updates
                          • Unsigned Payload
```
*  The link to the firmware file on the LVFS
1. https://fwupd.org/downloads/60518fa0f9a78a38afda311e8a08f21995dc21020f16cd3b959c159f1ee360ca-algoltek-ag9421-00.09.07_02.18.05.cab
2. https://fwupd.org/downloads/6984d9c2e03ddc9f12879565587c2f42ded40209d0434f2f269208213f052dfd-algoltek-ag9421-00.09.14_03.00.00.cab

* The zip file(s) for each of the cab archives

[algoltek-usb_00.09.07_02.18.05_gitMain.zip](https://github.com/fwupd/fwupd/files/15234040/algoltek-usb_00.09.07_02.18.05_gitMain.zip)
[algoltek-usb_00.09.14_03.00.00_gitMain.zip](https://github.com/fwupd/fwupd/files/15234041/algoltek-usb_00.09.14_03.00.00_gitMain.zip)

Thanks!

closed 2024-05-07 by hughsie