fwupd/fwupd #7088: Plugin elanfp has a coverage score of 5%

hughsie, 2024-04-12:
The elanfp fwupd plugin has a very low coverage score: https://coveralls.io/github/fwupd/fwupd -- This means it’s not being adequately tested during CI and pre-release testing, and might mean we release the plugin to end users with an accidental and unfortunate regression. The best way to make sure this doesn’t happen is to provide a device-test, and there are a lot of examples in https://github.com/fwupd/fwupd/tree/main/data/device-tests for existing hardware.

The emulated device test in CI downloads a “_device emulation data_” from the LVFS which sets up a virtual USB device that we can control from the daemon. Then we download a cab archive of the actual firmware for the device (also from the LVFS) and then install the _real_ firmware archive on the _virtual_ device. This usually tests out a lot of the plugin functionality, and is enough to make the plugin coverage rise up to **70-90%** typically – which is great, and means we can refactor the common daemon code without risk of introducing a regression. All new USB plugins going into fwupd now must have an emulated test for CI, and we’re now trying to get the existing plugins up to the same standard.

We’re of course happy to help with this task, but we don’t have all the physical hardware needed to generate an emulation of the update being applied. This is why I’ve assigned the plugin owner to this issue :) What I’d like you to do is:

* Read https://fwupd.github.io/libfwupdplugin/device-emulation.html
* Make sure there is at least one firmware on the LVFS that works with this hardware (ideally two) – but it doesn’t have to be in a `embargo`, `testing`, or `stable` remote – it can remain in private if needed.
* Plug in the device, and follow the steps in https://github.com/fwupd/fwupd/blob/main/docs/device-emulation.md
* Attach to this issue:
  * The `fwupdmgr get-devices` output when the device is in the normal runtime mode
  * The link to the firmware file on the LVFS, e.g. [https://fwupd.org/downloads/hash-firmware.cab](https://fwupd.org/hash-firmware.cab)
  * The zip file(s) for each of the cab archives you’ve emulated

We can then test the emulation works on our system (without physical hardware), build the JSON file and add it to the fwupd repo for you. Although one firmware will work, ideally we want to test upgrading from 1.0.2 to 1.0.3 and downgrading from 1.0.3 to 1.0.2 (with example version numbers) rather than just reinstalling 1.0.3 onto itself. If you'd like to write the device-test JSON file yourself and submit a PR that's even better.

If you have any questions, queries or worries please let us know, either here or by email. Thanks!

hughsie, 2024-11-11:
29% now. @MichaelCheng04 can you remind me if I have any hardware already, or maybe send me some hardware (which is supported on the LVFS) or provide some emulation data please? e.g. https://github.com/fwupd/fwupd/blob/main/docs/device-emulation.md

MichaelCheng04, 2024-11-13:
Hi @hughsie we can send the module to you for the test, could you provide the shipping address? Thanks!!

MichaelCheng04, 2025-01-07:
@hughsie the fw upgrade rule is that only accept fw version increase (ex: 0413 -> 0414) even if install the same version. Unfortunately there is no GPIO to disable version check on the dev board.

hughsie, 2025-01-13:
> Can you maybe dediprog old version for this purpose?

Nah, it's stored on the little SoC embedded in the epoxy. I said a little prayer and the emulation data came out okay first try.

closed 2025-01-13 by hughsie

MichaelCheng04, 2025-01-14:
@hughsie  and @superm1 I am cheking with internal, will upate to you later.