fwupd/fwupd #7094: Plugin nordic-hid has a coverage score of 3%

hughsie, 2024-04-12:
The nordic-hid fwupd plugin has a very low coverage score: https://coveralls.io/github/fwupd/fwupd -- This means it’s not being adequately tested during CI and pre-release testing, and might mean we release the plugin to end users with an accidental and unfortunate regression. The best way to make sure this doesn’t happen is to provide a device-test, and there are a lot of examples in https://github.com/fwupd/fwupd/tree/main/data/device-tests for existing hardware.

The emulated device test in CI downloads a “_device emulation data_” from the LVFS which sets up a virtual USB device that we can control from the daemon. Then we download a cab archive of the actual firmware for the device (also from the LVFS) and then install the _real_ firmware archive on the _virtual_ device. This usually tests out a lot of the plugin functionality, and is enough to make the plugin coverage rise up to **70-90%** typically – which is great, and means we can refactor the common daemon code without risk of introducing a regression. All new USB plugins going into fwupd now must have an emulated test for CI, and we’re now trying to get the existing plugins up to the same standard.

We’re of course happy to help with this task, but we don’t have all the physical hardware needed to generate an emulation of the update being applied. This is why I’ve assigned the plugin owner to this issue :) What I’d like you to do is:

* Read https://fwupd.github.io/libfwupdplugin/device-emulation.html
* Make sure there is at least one firmware on the LVFS that works with this hardware (ideally two) – but it doesn’t have to be in a `embargo`, `testing`, or `stable` remote – it can remain in private if needed.
* Plug in the device, and follow the steps in https://github.com/fwupd/fwupd/blob/main/docs/device-emulation.md
* Attach to this issue:
  * The `fwupdmgr get-devices` output when the device is in the normal runtime mode
  * The link to the firmware file on the LVFS, e.g. [https://fwupd.org/downloads/hash-firmware.cab](https://fwupd.org/hash-firmware.cab)
  * The zip file(s) for each of the cab archives you’ve emulated

We can then test the emulation works on our system (without physical hardware), build the JSON file and add it to the fwupd repo for you. Although one firmware will work, ideally we want to test upgrading from 1.0.2 to 1.0.3 and downgrading from 1.0.3 to 1.0.2 (with example version numbers) rather than just reinstalling 1.0.3 onto itself. If you'd like to write the device-test JSON file yourself and submit a PR that's even better.

If you have any questions, queries or worries please let us know, either here or by email. Thanks!

gwonpy, 2024-06-11:
Hi there, I have an nrf52840 DK and I wonder if it is possible to upgrade my own application firmware via fwupd tool on Ubuntu.I want to successfully upgrade the development kit on unbutu using fwupd.But I find that there is no actual tutorial that shows me how to upgrade step by step.What firmware should I burn so that unbutu recognizes that this is a hid device

I refer to the following links but to no avail.

 https://devzone.nordicsemi.com/f/nordic-q-a/111310/fwupd-on-ubuntu 

[developer.nordicsemi.com/.../fwupd.html](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/applications/nrf_desktop/fwupd.html)

[fwupd.org/.../search](https://fwupd.org/lvfs/search?value=nordic)

MarekPieta, 2024-06-19:
Hi @gwonpy,
This plugin uses configuration channel DFU as underlying transport. The protocol is integrated to nRF Desktop application available in the NCS (https://docs.nordicsemi.com/bundle/ncs-latest/page/nrf/applications/nrf_desktop/README.html). You can program release configuration of the application to your nRF52840 DK and test with firmware updates available in the LVFS (https://fwupd.org/lvfs/search?value=nordic). Device should basically be recognized by `fwupdmgr` (or `fwupdtool`) out-of-the box.

The nRF Desktop application is flexible and you can configure it to match your needs. Adding support for a custom boards there is also quite simple (see documentation for detailed list of steps). 

Using `nordic-hid` plugin with a custom HID application would require to support configuration channel protocol over HID feature reports (https://docs.nordicsemi.com/bundle/ncs-latest/page/nrf/applications/nrf_desktop/doc/config_channel.html) and configuration channel DFU (in nRF Desktop implemented by: https://docs.nordicsemi.com/bundle/ncs-latest/page/nrf/applications/nrf_desktop/doc/dfu.html). Apart from this, keep in mind that `nordic-hid` plugin quirk allowlists only devices using Nordic VID: https://github.com/fwupd/fwupd/blob/main/plugins/nordic-hid/nordic-hid.quirk.

Please let me know if I could provide more help.

hughsie, 2024-10-01:
Fixed in https://github.com/fwupd/fwupd/pull/7810/commits/b106817d8ef0b73c5f295a44b414db327b390ff0

closed 2024-10-01 by hughsie