fwupd/fwupd #7095: Plugin pixart-rf has a coverage score of 11%

hughsie, 2024-04-12:
The pixart-rf fwupd plugin has a very low coverage score: https://coveralls.io/github/fwupd/fwupd -- This means it’s not being adequately tested during CI and pre-release testing, and might mean we release the plugin to end users with an accidental and unfortunate regression. The best way to make sure this doesn’t happen is to provide a device-test, and there are a lot of examples in https://github.com/fwupd/fwupd/tree/main/data/device-tests for existing hardware.

The emulated device test in CI downloads a “_device emulation data_” from the LVFS which sets up a virtual USB device that we can control from the daemon. Then we download a cab archive of the actual firmware for the device (also from the LVFS) and then install the _real_ firmware archive on the _virtual_ device. This usually tests out a lot of the plugin functionality, and is enough to make the plugin coverage rise up to **70-90%** typically – which is great, and means we can refactor the common daemon code without risk of introducing a regression. All new USB plugins going into fwupd now must have an emulated test for CI, and we’re now trying to get the existing plugins up to the same standard.

We’re of course happy to help with this task, but we don’t have all the physical hardware needed to generate an emulation of the update being applied. This is why I’ve assigned the plugin owner to this issue :) What I’d like you to do is:

* Read https://fwupd.github.io/libfwupdplugin/device-emulation.html
* Make sure there is at least one firmware on the LVFS that works with this hardware (ideally two) – but it doesn’t have to be in a `embargo`, `testing`, or `stable` remote – it can remain in private if needed.
* Plug in the device, and follow the steps in https://github.com/fwupd/fwupd/blob/main/docs/device-emulation.md
* Attach to this issue:
  * The `fwupdmgr get-devices` output when the device is in the normal runtime mode
  * The link to the firmware file on the LVFS, e.g. [https://fwupd.org/downloads/hash-firmware.cab](https://fwupd.org/hash-firmware.cab)
  * The zip file(s) for each of the cab archives you’ve emulated

We can then test the emulation works on our system (without physical hardware), build the JSON file and add it to the fwupd repo for you. Although one firmware will work, ideally we want to test upgrading from 1.0.2 to 1.0.3 and downgrading from 1.0.3 to 1.0.2 (with example version numbers) rather than just reinstalling 1.0.3 onto itself. If you'd like to write the device-test JSON file yourself and submit a PR that's even better.

If you have any questions, queries or worries please let us know, either here or by email. Thanks!

hughsie, 2024-04-12:
@sam412081go

sam412081go, 2024-05-24:
@hughsie 
Hi Hughsie, I encountered an issue while implementing emulation device data.

I follow this page [Device Emulation](https://fwupd.github.io/libfwupdplugin/device-emulation.html)  to record behavior of my device, but have problem when I use `fwupdmgr emulation-save`, it shows `failed to save emulation data: no emulation data, perhaps no devices have been added?`

Below is my command history : 
```
root@210fdcccba1a:/github/workspace/testfw# fwupdmgr modify-config AllowEmulation true
Restart the daemon to make the change effective? [y|N]: y
Successfully modified configuration value

root@210fdcccba1a:/github/workspace/testfw# fwupdmgr emulation-tag 5a4e9ebd-9c40-5e7f-b7ff-424030dfcc66
0.	Cancel
1.	745fdf5b7ba2d86e9dcd525f1060d91b6a851ee8 (HP HP 65x/45x Series dongle)
2.	71d3042554d6d110f4268c92de0c117b53023512 (HP HP 65x/45x Series dongle)
3.	6e39378578f82f62f53a7d9b528d44ac304c053b (HP HP 65x/45x Series dongle)
Choose device [0-3]: 1
Action Required: Unplug and replug the device, then install the firmware.
```
I unplug my device, then replug it.
```
root@210fdcccba1a:/github/workspace/testfw# fwupdmgr get-devices --filter emulation-tag
Dell Inc. Latitude 5300
?

root@210fdcccba1a:/github/workspace/testfw# fwupdmgr download https://fwupd.org/downloads/498e534e8d28b6c835459ef20881ef1cae21a7b2aea5a76cc17794cb53504899-test_v1.cab
Downloading?             [******************************         ]

root@210fdcccba1a:/github/workspace/testfw# fwupdmgr install ./498e534e8d28b6c835459ef20881ef1cae21a7b2aea5a76cc17794cb53504899-test_v1.cab --allow-reinstall
Installing on HP HP 65x/45x Series dongle?                       ] Less than one minute remaining?
Installing on HP HP 65x/45x Series dongle?*********              ] Less than one minute remaining?
Waiting?                 [***************************************] Less than one minute remaining?
Successfully installed firmware

root@210fdcccba1a:/github/workspace/testfw# fwupdmgr emulation-save dongle_data.zip
failed to save emulation data: no emulation data, perhaps no devices have been added?
```

I am using docker with os ubuntu 24.04
```
root@210fdcccba1a:/github/workspace/testfw# cat /etc/os-release 
PRETTY_NAME="Ubuntu 24.04 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo
```

Below is information of my device:
```
root@210fdcccba1a:/github/workspace/testfw# fwupdtool get-devices --plugins=pixart_rf
Loading?                 [************************************** ]
Dell Inc. Latitude 5300
?
??HP HP 65x/45x Series dongle:
? ?   Device ID:          745fdf5b7ba2d86e9dcd525f1060d91b6a851ee8
? ?   Current version:    01.10.5
? ?   Vendor:             USB:0x093A|HIDRAW:0x03F0
? ?   GUIDs:              e8affe5f-f308-5388-ae26-8a4e956bef0b ? HIDRAW\VEN_03F0&DEV_6841
? ?                       20c3bda1-5b57-5eed-b8cb-f1c33bc9ba2a ? HIDRAW\VEN_03F0&DEV_6841&NAME_HP_HP_65X/45X_SERIES_DONGLE
? ?                       5a4e9ebd-9c40-5e7f-b7ff-424030dfcc66 ? HIDRAW\VEN_03F0&DEV_6841&MODEL_PDONGLE
? ?   Device Flags:       ? Updatable
? ?                       ? Unsigned Payload
? ? 
? ??PMOUSE:
?       Device ID:        08544d58926d515b867a48231b8efb261d25a753
?       Current version:  01.10.6
?       Vendor:           USB:0x093A
?       GUID:             14663864-3a54-56ab-ba07-66db9e97dee9 ? HIDRAW\VEN_03F0&DEV_6841&MODEL_PMOUSE
?       Device Flags:     ? Updatable
?                         ? Unsigned Payload
?     
??HP HP 65x/45x Series dongle:
? ?   Device ID:          71d3042554d6d110f4268c92de0c117b53023512
? ?   Current version:    01.10.5
? ?   Vendor:             USB:0x093A|HIDRAW:0x03F0
? ?   GUIDs:              e8affe5f-f308-5388-ae26-8a4e956bef0b ? HIDRAW\VEN_03F0&DEV_6841
? ?                       20c3bda1-5b57-5eed-b8cb-f1c33bc9ba2a ? HIDRAW\VEN_03F0&DEV_6841&NAME_HP_HP_65X/45X_SERIES_DONGLE
? ?                       5a4e9ebd-9c40-5e7f-b7ff-424030dfcc66 ? HIDRAW\VEN_03F0&DEV_6841&MODEL_PDONGLE
? ?   Device Flags:       ? Updatable
? ?                       ? Unsigned Payload
? ? 
? ??PMOUSE:
?       Device ID:        6a294ef329c04b79faef0be04c3a031f82794198
?       Current version:  01.10.6
?       Vendor:           USB:0x093A
?       GUID:             14663864-3a54-56ab-ba07-66db9e97dee9 ? HIDRAW\VEN_03F0&DEV_6841&MODEL_PMOUSE
?       Device Flags:     ? Updatable
?                         ? Unsigned Payload
?     
??HP HP 65x/45x Series dongle:
  ?   Device ID:          6e39378578f82f62f53a7d9b528d44ac304c053b
  ?   Current version:    01.10.5
  ?   Vendor:             USB:0x093A|HIDRAW:0x03F0
  ?   GUIDs:              e8affe5f-f308-5388-ae26-8a4e956bef0b ? HIDRAW\VEN_03F0&DEV_6841
  ?                       20c3bda1-5b57-5eed-b8cb-f1c33bc9ba2a ? HIDRAW\VEN_03F0&DEV_6841&NAME_HP_HP_65X/45X_SERIES_DONGLE
  ?                       5a4e9ebd-9c40-5e7f-b7ff-424030dfcc66 ? HIDRAW\VEN_03F0&DEV_6841&MODEL_PDONGLE
  ?   Device Flags:       ? Updatable
  ?                       ? Unsigned Payload
  ? 
  ??PMOUSE:
        Device ID:        1170804da958d74eff6b40d0b409b13a2467bde0
        Current version:  01.10.6
        Vendor:           USB:0x093A
        GUID:             14663864-3a54-56ab-ba07-66db9e97dee9 ? HIDRAW\VEN_03F0&DEV_6841&MODEL_PMOUSE
        Device Flags:     ? Updatable
                          ? Unsigned Payload
```

sam412081go, 2024-05-24:
Actually, these three Device IDs all come from the same Dongle because the Dongle declares three Interfaces. However, this does not affect the firmware update; It can be selected any one of them to update the Dongle.

superm1, 2024-05-24:
@hughsie same problem with composite devices maybe?

sam412081go, 2024-05-29:
So fwupdmgr emulation function is not support composite devices?

hughsie, 2024-06-03:
Sorry for the delay. I don't think it makes sense to show 6 devices for a dingle and a single mouse! :) Can we look at the interface field and ignore 2/3 of them perhaps? It's going to really confuse the various UIs we have for fwupd to see three devices when one is plugged in.

sam412081go, 2024-06-03:
@hughsie Sure, I will modify the device code so that it only displays one interface, and then test if it works properly.

sam412081go, 2024-06-05:
@hughsie , I have modified the device to display a single interface, but the emulation-save feature is still not working.

fwupdmgr get-devices shows as below, can see just only one device `HP HP 65x/45x Series dongle` in list
```
root@b3fd48506a96:~# fwupdmgr get-devices
Dell Inc. Latitude 5300
│
├─AMT [unprovisioned]:
│     Device ID:          8d5470e73fd9a31eaa460b2b6aea95483fe3f14c
│     Summary:            Hardware and firmware technology for remote out-of-band management
│     Current version:    12.0.47.1524
│     Bootloader Version: 12.0.47.1524
│     Vendor:             Intel Corporation (MEI:0x8086)
│     GUIDs:              12f80028-b4b7-4b2d-aca8-46e0ff65814c
│                         1f512ab8-bc3a-59e6-83a7-7a0da80c253e ← MEI\VEN_8086&DEV_9DE0
│                         b6ce6b9a-01f1-5f83-8320-60b81aad6a85 ← MEI\VEN_8086&DEV_9DE0&SUBSYS_102808B7
│     Device Flags:       • Internal device
│   
├─Core™ i5-8365U CPU @ 1.60GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f8
│     Vendor:             Intel
│     GUIDs:              561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         d5092916-1868-5182-8c70-12f83dab6372 ← CPUID\PRO_0&FAM_06&MOD_8E&STP_C
│     Device Flags:       • Internal device
│   
├─HP HP 65x/45x Series dongle:
│ │   Device ID:          71d3042554d6d110f4268c92de0c117b53023512
│ │   Current version:    01.10.5
│ │   Vendor:             HP, Inc (USB:0x093A, HIDRAW:0x03F0)
│ │   Update State:       Failed
│ │   GUIDs:              e8affe5f-f308-5388-ae26-8a4e956bef0b ← HIDRAW\VEN_03F0&DEV_6841
│ │                       20c3bda1-5b57-5eed-b8cb-f1c33bc9ba2a ← HIDRAW\VEN_03F0&DEV_6841&NAME_HP_HP_65X/45X_SERIES_DONGLE
│ │                       5a4e9ebd-9c40-5e7f-b7ff-424030dfcc66 ← HIDRAW\VEN_03F0&DEV_6841&MODEL_PDONGLE
│ │   Device Flags:       • Updatable
│ │                       • Unsigned Payload
│ │ 
│ └─PMOUSE:
│       Device ID:        6a294ef329c04b79faef0be04c3a031f82794198
│       Current version:  01.10.6
│       Vendor:           HP, Inc (USB:0x093A)
│       GUID:             14663864-3a54-56ab-ba07-66db9e97dee9 ← HIDRAW\VEN_03F0&DEV_6841&MODEL_PMOUSE
│       Device Flags:     • Updatable
│                         • Unsigned Payload
│     
├─Micron 2200S NVMe 512GB:
│     Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:            NVM Express solid state drive
│     Current version:    22001050
│     Vendor:             Micron Technology Inc (NVME:0x1344)
│     Serial Number:      200726878482
│     GUIDs:              b29d3ccf-500b-5b69-9bb9-e46b4832225f ← STORAGE-DELL-108489
│                         d63070ee-eeed-11e8-8eb2-f2801f1b9fd1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─TPM:
│     Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│     Current version:    74.8.17568.5511
│     Vendor:             ST Microelectronics (TPM:STM)
│     GUIDs:              84df3581-f896-54d2-bd1a-372602f04c32 ← TPM\VEN_STM&DEV_0001
│                         bfaed10a-bbc1-525b-a329-35da2f63e918 ← TPM\VEN_STM&MOD_
│                         70b7b833-7e1a-550a-a291-b94a12d0f319 ← TPM\VEN_STM&DEV_0001&VER_2.0
│                         06f005e9-cb62-5d1a-82d9-13c534c53c48 ← TPM\VEN_STM&MOD_&VER_2.0
│                         670b5381-3fd0-5653-8610-f446d2bedd10 ← 08b7-2.0
│     Device Flags:       • Internal device
│                         • Updatable
[... 46 lines cut ...]
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.16
runtime   org.kernel                    5.15.0-107-generic

```

OS version in Docker container:
```
root@b3fd48506a96:/testfw/hpac_dongle/v01# cat /etc/os-release 
PRETTY_NAME="Ubuntu 24.04 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo

```

hughsie, 2024-06-05:
Okay, so for some reason it's either not saving the "emulation-tag" flag for the device (seems unlikely) or it's not matching up the device on re-plug for some reason. Can you attach the output of `sudo fwupdtool get-devices --plugins=pixart_rf -vv` for clues please?

Also, do I have any pixart hardware here? e.g. would something like the HP910 work in the same way?

sam412081go, 2024-06-06:
[1device.txt](https://github.com/user-attachments/files/15613226/1device.txt)
[1device_vv.txt](https://github.com/user-attachments/files/15613227/1device_vv.txt)

@hughsie Attached are output files,
 `fwupdtool get-devices --plugins=pixart_rf` => 1device.txt
 `fwupdtool get-devices --plugins=pixart_rf -vv` => 1device_vv.txt

My device still has three interfaces(index 0~2) in the usb config description, so there are still three devices appearing in /dev.
When fwupd queries the device, I will only respond to interface 1 and ignore the others. 
So you will see like this :`failed to add device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:03F0:6841.0113/hidraw/hidraw5: failed to OTA init new: ioctl error: Broken pipe [32]` in log. Because I won't response to interface0 & interface 2.

> Also, do I have any pixart hardware here? e.g. would something like the HP910 work in the same way?

I will discuss internally to see how we can prepare a hardware for you.

hughsie, 2024-06-06:
Hardware would be great, thanks. Same address as last time.

I have a suspicion: If you run `fwupdtool get-devices --plugins=pixart_rf -vv`, then unplug, then replug, then run `fwupdtool` again, do you get a different `BackendId`?

e.g. I see `/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.1/0003:03F0:6841.010E/hidraw/hidraw6` but I wonder if the hidraw number is incrementing -- which means fwupd can't match up the new device to the old one. PR incoming.

sam412081go, 2024-06-12:
> Also, do I have any pixart hardware here? e.g. would something like the HP910 work in the same way?

@hughsie 
We will prepare a hardware for you, how should I give it to you? send it to your address?
If yes, you can provide your address to this mail:
sam_chen@pixart.com

closed 2024-10-01 by superm1