fwupd/fwupd #7098: Plugin synaptics-cape has a coverage score of 3%

hughsie, 2024-04-12:
The synaptics-cape fwupd plugin has a very low coverage score: https://coveralls.io/github/fwupd/fwupd -- This means it’s not being adequately tested during CI and pre-release testing, and might mean we release the plugin to end users with an accidental and unfortunate regression. The best way to make sure this doesn’t happen is to provide a device-test, and there are a lot of examples in https://github.com/fwupd/fwupd/tree/main/data/device-tests for existing hardware.

The emulated device test in CI downloads a “_device emulation data_” from the LVFS which sets up a virtual USB device that we can control from the daemon. Then we download a cab archive of the actual firmware for the device (also from the LVFS) and then install the _real_ firmware archive on the _virtual_ device. This usually tests out a lot of the plugin functionality, and is enough to make the plugin coverage rise up to **70-90%** typically – which is great, and means we can refactor the common daemon code without risk of introducing a regression. All new USB plugins going into fwupd now must have an emulated test for CI, and we’re now trying to get the existing plugins up to the same standard.

We’re of course happy to help with this task, but we don’t have all the physical hardware needed to generate an emulation of the update being applied. This is why I’ve assigned the plugin owner to this issue :) What I’d like you to do is:

* Read https://fwupd.github.io/libfwupdplugin/device-emulation.html
* Make sure there is at least one firmware on the LVFS that works with this hardware (ideally two) – but it doesn’t have to be in a `embargo`, `testing`, or `stable` remote – it can remain in private if needed.
* Plug in the device, and follow the steps in https://github.com/fwupd/fwupd/blob/main/docs/device-emulation.md
* Attach to this issue:
  * The `fwupdmgr get-devices` output when the device is in the normal runtime mode
  * The link to the firmware file on the LVFS, e.g. [https://fwupd.org/downloads/hash-firmware.cab](https://fwupd.org/hash-firmware.cab)
  * The zip file(s) for each of the cab archives you’ve emulated

We can then test the emulation works on our system (without physical hardware), build the JSON file and add it to the fwupd repo for you. Although one firmware will work, ideally we want to test upgrading from 1.0.2 to 1.0.3 and downgrading from 1.0.3 to 1.0.2 (with example version numbers) rather than just reinstalling 1.0.3 onto itself. If you'd like to write the device-test JSON file yourself and submit a PR that's even better.

If you have any questions, queries or worries please let us know, either here or by email. Thanks!

hughsie, 2024-11-11:
26% now. @CNXT-Simon can you remind me if I have any hardware already, or maybe send me some hardware (which is supported on the LVFS) or provide some emulation data please? e.g. https://github.com/fwupd/fwupd/blob/main/docs/device-emulation.md

JohnAZoidberg, 2025-02-19:
Looks like the plugin is broken.
I bisected and the last good commit is 63a69456923e70db06f75b49acb9ac4127a2ebe0
In d2b5dcef027d99538a99bf836ef5aefde41083e6 it stopped building.

Now in 2.0.6 I can't detect the device anymore.

Actually sometimes I can't even build anymore at all. I think something in the build system doesn't properly clean some generated files.
After `sudo git clean -dfx` and `./contrib/setup` I can build again.
```
../../plugins/ccgx-dmc/fu-ccgx-dmc-devx-device.c: In function ‘fu_ccgx_dmc_devx_device_finalize’:
../../plugins/ccgx-dmc/fu-ccgx-dmc-devx-device.c:358:17: error: implicit declaration of function ‘fu_struct_ccgx_dmc_devx_status_unref’; did you mean ‘fu_struct_ccgx_dmc_devx_status_parse’? [-Wimplicit-function-declaration]
  358 |                 fu_struct_ccgx_dmc_devx_status_unref(self->status);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                 fu_struct_ccgx_dmc_devx_status_parse
```

JohnAZoidberg, 2025-02-19:
This is the error on 2.0.6:

```
>  fwupdtool --plugins synaptics_cape get-devices
Loading…                 [**************************             ]10:03:30.280 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:08.3/0000:c3:00.3/usb5/5-1: failed to setup: failed to get firmware version info: unknown error -22
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
No detected devices
```

I actually don't have an update for our Audio Expansion card, which used to be able to detect the version with old versions of fwupd.
Can I record just `get-devices` and provide it to you?

superm1, 2025-02-19:
> Actually sometimes I can't even build anymore at all. I think something in the build system doesn't properly clean some generated files.

Yeah I've seen this too.  I'm suspecting the rustgen generator doesn't know when to clear an old file.

JohnAZoidberg, 2025-02-20:
> Yeah I've seen this too. I'm suspecting the rustgen generator doesn't know when to clear an old file.

What's the recomended way to clean and rebuild those files?
My bisect didn't yield useful results because builds failed for this unrelated reason.

superm1, 2025-02-20:
I think we basically want the generator equivalent of `build_always_stale` (not sure if that exists).

hughsie, 2025-05-04:
Issue was ignored, closing.

closed 2025-05-04 by hughsie