fwupd/fwupd #7108: LENOVO_MT_30GS_BU_Think_FM_ThinkStation P3 Tower fails to update new FW

joakim-tjernlund, 2024-04-15:
**Describe the bug**
fwupdmgr update finds and prepares FW to update but reboot does not update FW

**Steps to Reproduce**
fwupdmgr update and choose either of:
```
Upgrade Intel Management Engine from 16.1.25.2091 to 16.1.30.2307?
Upgrade System Firmware from 0.1.53 to 0.1.97?
```
reboot and watch monitor, there is only an Lenovo Logon visible for a few secs
and then normal boot continues.
**Expected behavior**
Expect to see FW installing when rebooting

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.14
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.5
compile   org.freedesktop.gusb          0.4.7
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.7
runtime   org.kernel                    6.6.26-x86_64
runtime   org.freedesktop.fwupd         1.9.14
```
fwupd-1.9.16 has the same problem

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

emerge sys-apps/fwupd

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
LENOVO 30GS001SMT
│
├─Unknown Device:
│     Device ID:          b5881e53bf55f4e454a88a354e1f9432e5780820
│     Serial Number:      26108815NB
│     GUID:               55a028df-5478-5dd0-9978-427c1e45e70b ← DRM\VEN_NEC&DEV_6865
│   
├─13th Gen Intel Core™ i7-13700K:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0000011e
│     Vendor:             Intel
│     GUIDs:              3d4645d8-13fe-541a-ac5c-7d9d944e4500 ← CPUID\PRO_0&FAM_06&MOD_B7
│                         988c4cf1-ace4-5eb5-b26c-905ff236f412 ← CPUID\PRO_0&FAM_06&MOD_B7&STP_1
│     Device Flags:       • Internal device
│   
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    16.1.25.2091
│     Minimum Version:    16.1.25.2091
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               dae3cd3d-9440-469a-be69-872613cd5626
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
├─Raptor Lake-S GT1 [UHD Graphics 770]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    04
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              d2e34a7c-ed41-5be7-bc67-304c11008b77 ← PCI\VEN_8086&DEV_A780
│                         02becf05-ae04-53f8-9d64-cf2b41ce6c45 ← PCI\VEN_8086&DEV_A780&SUBSYS_17AA1064
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─SKHynix HFS001TEJ9X102N:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express solid state drive
│     Current version:    51730A10
│     Vendor:             SK hynix (NVME:0x1C5C)
│     Serial Number:      SDC4T002211704959
│     GUIDs:              bcfb6135-158b-5c97-a08c-a76de6c90dc4 ← NVME\VEN_1C5C&DEV_1959
│                         021c33a7-8b6a-5ec2-a388-4c472df55c3f ← NVME\VEN_1C5C&DEV_1959&SUBSYS_1C5C1959
│                         2ed2cb8d-2e90-5ff1-ba71-44fd321f82fc ← SKHynix_HFS001TEJ9X102N
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.53
│ │   Minimum Version:    0.1.53
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUID:               f8fe29c6-a9a8-4ff7-86e7-aa8502c858f1
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  371
│       Minimum Version:  371
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            25c2b9af-7c95-564e-9b07-eccceacb46e8 ← UEFI\CRT_B4731FEF902AA3DA869F0803C84D732D790B0C23D095EDB2CCCCCB304FB00C53&ARCH_X64
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
├─UEFI Device Firmware:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    65547
│     Minimum Version:    65547
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               220ea54e-6741-44b2-9f2d-974b512f1d89
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
├─UEFI Device Firmware:
│     Device ID:          f95c9218acd12697af946874bfe4239587209232
│     Summary:            UEFI ESRT device
│     Current version:    373407744
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               3b19d722-d168-4114-ac7e-bd6fe1262c2e
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            LENOVO
      GUID:               23c29378-a1c3-5986-85af-668844423ec2 ← UEFI\CRT_F6F9EDD6A38C65F3F3FDD58C569AED88E91BE845

```

</details>

**Additional questions**

- Operating system and version: Gentoo
- Have you tried rebooting? Yes
- Is this a regression? Yes, the previous update went well

joakim-tjernlund, 2024-04-15:
> Try fwupd EFI main branch. We had a recreation against new gnu EFI

Will do, just need  gnu-efi-3.0.18 in Gentoo first

joakim-tjernlund, 2024-04-15:
I coul dnot wait so I installed  gnu-efi-3.0.18 myself and then built fwupd-efi main.
Now it worked :) Only system BIOS was installed and my BIOS settings was reset, that is
a concern as the system will not boot with default setting here.

No on to install Intel Management Engine from 16.1.25.2091 to 16.1.30.2307 as well

superm1, 2024-04-15:
@hughsie can we tag a new fwupd-efi?

joakim-tjernlund, 2024-04-15:
Moving on to Intel Management Engine from 16.1.25.2091 to 16.1.30.2307 did NOT work.
Just  displays the Lenovo logo for a few secs then it boot normally :(

superm1, 2024-04-16:
New fwupd-efi 1.6 is tagged. https://github.com/fwupd/fwupd-efi/releases/tag/1.6

I suggest opening up a new issue against https://github.com/fwupd/firmware-lenovo/ for your ME issue so they can dig into it.

closed 2024-04-16 by superm1