fwupd/fwupd #7111: T480 fails to flash everything, list devices, thunderbolt fix not installing, Laptop is stuck with 2018 firmware + UEFI

avgsurfman, 2024-04-15:
**Describe the bug**

I'm at my wits end. Tried updating the system firmware due to Thunderbolt bug affecting the T480 laptops.
I've had issues with Thunderbolt before, but it seems that my system fails to update most basic drivers.
I have been unable to flash the firmware, make fwupd aware of my system's peripherals, and it's also refusing to 
update my firmware that might be related to the bug that effectively kills thunderbolt in all Thinkpads models unless
flashed with new firmware.

**Steps to Reproduce**

1.  Get the stock EFI (0.1.18)

2. Install fwpd with the following useflags:

```vim /etc/portage/package.use/fwupd 
>=sys-apps/fwupd-1.9.16 bluetooth elogind nvme policykit tpm uefi gnutls spi lzma
```
3. Use fwpd as in the README

**Expected behavior**
Fwupd finds my devices, gets the update list, installs the updates, Thunderbolt starts working, my mood improves.

**Received behavior**
Fwupd refuses to find my devices,, partially installs the package
via cap files while refusing to install the other caps, some of the devices become available, are updated from the command line, update fails after a reboot, my mood doesn't improve.
Tried changing EFI options but they had little effect.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.16
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.6
runtime   org.kernel                    6.8.5-gentoo-r1-x86_64
runtime   org.freedesktop.fwupd         1.9.16
(with default use-flags)

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

```
doas emerge -a fwupmgr 
```

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
LENOVO 20L6SAYX00
│
├─B140QAN02.0:
│     Device ID:          10feb8f8641be9b4f44d52078baa6338209a53ef
│     GUID:               7acbf86a-d492-500f-b674-e6ffcfb6d8c9 ← DRM\VEN_AUO&DEV_2036
│     Device Flags:       • Internal device
│   
├─Core™ i7-8550U CPU @ 1.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             Intel
│     GUIDs:              561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         54bdc429-0179-5019-9cb6-8a343324c97c ← CPUID\PRO_0&FAM_06&MOD_8E&STP_A
│     Device Flags:       • Internal device
│   
[... 625 lines cut ...]
├── System.map-6.8.5-gentoo-r1-x86_64
├── vmlinuz-6.2.0-gentoo-x86_64
├── vmlinuz-6.2.0-gentoo-x86_64.old
├── vmlinuz-6.7.3-gentoo-x86_64
└── vmlinuz-6.8.5-gentoo-r1-x86_64
```

**Additional questions**

- Operating system and version:
Gentoo Linux 2.7
- Have you tried rebooting?
About a dozen times.
- Is this a regression?
No. I updated the software from 19.3 to 19.16 but the issue persists.
- Are you using an NVMe disk?
Yes, with Full-disk Encryption.
- Is secure boot enabled?
No.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
I don't know/there is no such option available in the BIOS.
- Is this a hardware issue?
God I hope not.
- Your modprobe?
[Here](https://linux-hardware.org/?probe=fc73e6bb02)

superm1, 2024-04-15:
Try fwupd EFI 1.6

avgsurfman, 2024-04-15:
New gnu-efi as is not available in my distro's repos yet, I'll try to install it from source.

hughsie, 2024-04-16:
Could you also try Fedora? We don't really do any QA with gentoo.

avgsurfman, 2024-04-16:
I'll try to get fedora and a stick and see how it goes.  I'll report back when I'm done.
I'm worried that might be a hardware failure that somehow crept in with the faulty firmware.

joakim-tjernlund, 2024-04-17:
You can copy sys-apps/fwupd-efi-1.5 to sys-apps/fwupd-efi-9999 and that will build from git

joakim-tjernlund, 2024-04-17:
> New gnu-efi as is not available in my distro's repos yet, I'll try to install it from source.

Monitor/comment in https://bugs.gentoo.org/929216 for gnu-efi

joakim-tjernlund, 2024-04-22:
you can also try downgrade to sys-boot/gnu-efi-3.0.15 and sys-apps/fwupd-efi-1.4, that worked for me when
upgrading Intel ME

avgsurfman, 2024-04-26:
I managed to install the EFI update with live fedora from the USB stick. Either it's a distro-specific problem or I'm just stupid.
It didn't work the first time even booting from Fedora. I think it started working when I deleted an empty windows boot entry from the EFI Menu - the laptop originally came with windows preinstalled, which I promptly removed after getting the laptop.
[This could be related to this post](https://bbs.archlinux.org/viewtopic.php?pid=1874494#p1874494)
Still no luck with installing (or even displaying) the Thunderbolt Controller but I have a feeling it's a hardware fault at this point.

joakim-tjernlund, 2024-04-26:
I guess you tried with latest gnu-efi/fwup-efi? They are in Gentoo now

joakim-tjernlund, 2024-04-26:
OK, I know from past that every now and then you have to do a complete power off/on

avgsurfman, 2024-04-26:
Okay, good news and bad news:
After a poweroff, I was able to find my thunderbolt controller with lspci. The UEFI update must have helped.
The bad news:
For whatever reason, refreshing devices with `fwupdmgr get-device` hides my thunderbolt controller from lspci.
Kernel spews out a lot of error messages afterwards.
[kernel.txt](https://github.com/fwupd/fwupd/files/15133993/kernel.txt)

I have no clue what do to.

hughsie, 2024-11-11:
Does it work with the latest release of Fedora or Ubuntu? We don't do any QA on Gentoo and don't have any experience there I'm afraid.

avgsurfman, 2024-11-11:
Hi, the update worked, though I needed to delete an old Windows entry from the boot entries.  It flashed from Fedora and it works from Gentoo. If you have a wiki, you might want to add this information for someone who might have the same problem.
As for the Thunderbolt chip it very recently failed, along with the USB controller.

closed 2024-11-11 by avgsurfman