fwupd/fwupd #7119: Cannot build fwupd and need "gnu-efi" dependency when executing subproject fwupd-efi (version 1.6)

chinyakao, 2024-04-17:
**Describe the bug**
After setting up the development environment, and try `build-fwupd`, it turns out `subprojects/fwupd-efi/meson.build:15:0: ERROR: Dependency "gnu-efi" not found, tried pkgconfig`

**Steps to Reproduce**
1. Setup dev env and enter virtualenv
2. `build-fwupd`

**Expected behavior**
Successfully build the fwupd in venv

**Fail log**

```
Receiving objects: 100% (2308/2308), 524.92 KiB | 2.40 MiB/s, done.
Resolving deltas: 100% (1487/1487), done.
HEAD is now at 58a2928 Release fwupd-efi 1.6

Executing subproject fwupd-efi 

fwupd-efi| Project name: fwupd-efi
fwupd-efi| Project version: 1.6
fwupd-efi| C compiler for the host machine: cc (gcc 11.4.0 "cc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
fwupd-efi| C linker for the host machine: cc ld.bfd 2.38
fwupd-efi| Program objcopy found: YES (/usr/bin/objcopy)
fwupd-efi| Run-time dependency gnu-efi found: NO (tried pkgconfig)

subprojects/fwupd-efi/meson.build:15:0: ERROR: Dependency "gnu-efi" not found, tried pkgconfig

A full log can be found at /home/u/work/fwupd/venv/build/meson-logs/meson-log.txt
```

**Additional questions**

- Operating system and version: Linux u-Dell-G16-7630 6.5.0-1019-oem #20-Ubuntu SMP PREEMPT_DYNAMIC Mon Mar 18 17:38:55 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
- Have you tried rebooting? yes
- Is this a regression?

superm1, 2024-04-17:
Ah this is true. Maybe we should default to EFI binary disabled for development environment I wonder.

closed 2024-04-17 by superm1

jvitous81, 2024-07-25:
Not building the EFI binary is not a solution, when the EFI binary is what you need to build.

hughsie, 2024-07-25:
> when the EFI binary is what you need to build

Most developers don't need the EFI binary as they're not using the UEFI plugin. And the ones that do, are probably using the distro-supplied *signed* one that's going to work with secure boot turned on.