fwupd/fwupd #7122: FuPluginLogitechHidpp failed to ping Logi K650 B: failed to receive: failed to receive: timeout

soyersoyer, 2024-04-17:
**Describe the bug**
The Software app waits a lot while refreshing the update tab because one of fwupd timeout.
If I kill the fwupd, it responds.

fwupd[3367]: 10:30:26.400 FuPluginLogitechHidpp failed to ping Logi K650 B: failed to receive: failed to receive: timeout

or after killing/stopping/disabling the fwupd service, the fwupd restarts, and I can see this instead:

fwupd[18429]: 10:32:30.496 FuEngine             failed to add device /sys/devices/virtual/misc/uhid/0005:046D:B032.0004/hidraw/hidraw3: failed to receive: failed to receive: timeout

**Steps to Reproduce**
Stay away from the bluetooth keyboard.

Start the software app
Click on the Updates
Click on the refresh button.

You have to wait many minutes

**Expected behavior**
You should see the updates in <5s.

**fwupd version information**
1.9.16-1.fc39 

```shell
fwupdmgr --version
```
First run:
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
journal:
fwupd[19308]: 10:49:00.465 FuEngine             failed to add device /sys/devices/virtual/misc/uhid/0005:046D:B032.0004/hidraw/hidraw3: failed to receive: failed to receive: timeout

second run:
compile   com.hughsie.libxmlb           0.3.17
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.16
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.16
runtime   org.kernel                    6.9.0-rc4

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
dnf
<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version: Fedora 39
- Have you tried rebooting? y
- Is this a regression? Probably yes

robotparty, 2024-06-21:
Similar issue... my mouse freezes and this is logged. Happens very couple hours:
`Jun 21 15:59:33 chrispop fwupd[5027]: 22:59:33.189 FuPluginLogitechHidpp failed to get pending read: connection request failed`

robotparty, 2024-06-21:
Bunch more just now, mouse froze for 30s.. 
```
Jun 21 16:03:19 chrispop fwupd[5027]: 23:03:19.172 FuPluginLogitechHidpp failed to ping MX MCHNCL M: connection request failed
Jun 21 16:03:23 chrispop fwupd[5027]: 23:03:23.188 FuPluginLogitechHidpp failed to get pending read: connection request failed
Jun 21 16:03:34 chrispop fwupd[5027]: 23:03:34.174 FuPluginLogitechHidpp failed to ping MX Master 3S: connection request failed
Jun 21 16:03:38 chrispop fwupd[5027]: 23:03:38.189 FuPluginLogitechHidpp failed to get pending read: connection request failed
Jun 21 16:03:44 chrispop code.desktop[1818415]: [1818415:0621/160344.995749:ERROR:check.cc(319)] Check failed: false. NOTREACHED log messages are omitted in official builds. Sorry!
Jun 21 16:03:45 chrispop code.desktop[1818415]: [1818415:0621/160345.005050:ERROR:check.cc(319)] Check failed: false. NOTREACHED log messages are omitted in official builds. Sorry!
Jun 21 16:03:45 chrispop code.desktop[1818415]: [1818415:0621/160345.328001:ERROR:check.cc(319)] Check failed: false. NOTREACHED log messages are omitted in official builds. Sorry!
Jun 21 16:03:45 chrispop code.desktop[1818415]: [1818415:0621/160345.328127:ERROR:check.cc(319)] Check failed: false. NOTREACHED log messages are omitted in official builds. Sorry!
Jun 21 16:03:45 chrispop code.desktop[1818415]: [1818415:0621/160345.328191:ERROR:check.cc(319)] Check failed: false. NOTREACHED log messages are omitted in official builds. Sorry!
Jun 21 16:03:45 chrispop code.desktop[1818415]: [1818415:0621/160345.328239:ERROR:check.cc(319)] Check failed: false. NOTREACHED log messages are omitted in official builds. Sorry!
Jun 21 16:03:45 chrispop code.desktop[1818415]: [1818415:0621/160345.332183:ERROR:check.cc(319)] Check failed: false. NOTREACHED log messages are omitted in official builds. Sorry!
Jun 21 16:03:45 chrispop code.desktop[1818415]: [1818415:0621/160345.364904:ERROR:check.cc(319)] Check failed: false. NOTREACHED log messages are omitted in official builds. Sorry!
Jun 21 16:03:46 chrispop xdg-desktop-por[1107369]: gtk_at_context_set_accessible_role: assertion '!self->realized' failed

```

hughsie, 2024-11-11:
Can you try with fwupd 2.0.x please? We fixed a bunch of things with polling of unifying devices there.

soyersoyer, 2024-11-11:
Logs are a bit different, but the operation is similar.

```
Nov 11 19:39:32 fedora kernel: Bluetooth: hci0: ACL packet for unknown connection handle 3837
...
Nov 11 19:39:42 fedora fwupd[2680]: 18:39:42.672 FuDevice             disabling polling: failed to subclass open: failed to open /dev/hidraw1: No such file or directory
...
Nov 11 19:40:17 fedora fwupd[2680]: 18:40:17.701 FuPluginLogitechHidpp failed to ping Logi K650 B: failed to receive: failed to receive: timeout
...
Nov 11 19:40:47 fedora fwupd[2680]: 18:40:47.708 FuPluginLogitechHidpp failed to ping Logi K650 B: failed to receive: failed to receive: timeout
...
Nov 11 19:41:17 fedora fwupd[2680]: 18:41:17.718 FuPluginLogitechHidpp failed to ping Logi K650 B: failed to receive: failed to receive: timeout
```
```
Name            : fwupd
Epoch           : 0
Version         : 2.0.1
Release         : 1.fc42
```

rkubik-hostersi, 2025-04-02:
The same is happening to me: the MX mouse is lagging when connected via Logi Bolt.

```
kwi 02 08:12:38 rkpc fwupd[9943]: 06:12:38.193 FuPluginLogitechHidpp failed to get pending read: invalid address
kwi 02 08:13:03 rkpc fwupd[9943]: 06:13:03.192 FuPluginLogitechHidpp failed to update paired device status: cannot rescan paired device: failed to open /dev/hidraw3
```

closed 2025-05-04 by hughsie