fwupd/fwupd #7129: HP Thunderbolt Dock G4: downgrade to 1.4.20.0 failed

d4s, 2024-04-18:
**Describe the bug**
VID/PID: 03F0/0488
Initial FW Version: 1.5.22.0
Target FW Version: 1.4.20.0

The FW version did not auto/manual downgrade from 1.5.22.0 and the status.log shows "FAIL" and "Command returned non-zero exit status" info.

The output is:
```
  stdout:
  {
    "Error" : {
      "Domain" : "FwupdError",
      "Code" : 10,
      "Message" : "invalid fwct analysis failed with status 0x05[(null)]"
    }
  }
```

Found `fwct` mention only in this issue: https://github.com/fwupd/fwupd/issues/5202

**Steps to Reproduce**
Steps to reproduce the behavior.

```
CACHE_DIRECTORY='/var/cache/fwupd' fwupdmgr local-install --json --allow-older --allow-reinstall https://fwupd.org/downloads/04a53b0567c68a8500e0ce2a5d3642e794ed8ad79c7c055df893c3a0c1627c19-Thunderbolt_Dock_G4_V1.4.20.0.cab 07aa90953cc32d1e08a45f292f61fc3a338c667a
```

**Expected behavior**
Downgrade should happen.

**fwupd version information**
Not provided. Should be 1.9.13 (might be 1.9.15).

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
ChromeOS R125-15833.0.0

superm1, 2024-04-18:
By chance is USB4/TBT3 controller part of the update? If so I wonder if it's got a blocked downgrade (security version bump or similar).

d4s, 2024-04-18:
@superm1 I don't think so tbh.
The issue comes from the automated testing system, and the exact command used in the test is above.
I see it tries to downgrade the exact device with DMC F/W.

hughsie, 2024-04-18:
So, 0x5 is unexpected.

    enum FuCcgxDmcFwctAnalysisStatus {
        InvalidFwct = 0,
        InvalidDockIdentity,
        InvalidCompositeVersion,
        AuthenticationFailed,
        InvalidAlgorithm,
    }

@IfxBrent can you tell us what the other error codes are?

IfxBrent, 2024-04-19:
For dock status, it should map to  this table. 
enum FuCcgxDmcDeviceStatus {
    Idle = 0,
    UpdateInProgress,
    UpdatePartial,
    UpdateCompleteFull,
    UpdateCompletePartial,
    UpdatePhase1Complete,
    FwDownloadedUpdatePend,
    FwDownloadedPartialUpdatePend,
    Phase2UpdateInProgress = 0x81,
.....
    UpdateFail = 0xff,
}

And here is updated definitions of this table for your reference.
enum FuCcgxDmcDeviceStatus {
    Idle = 0,
    UpdatePhase1InProgress,
    UpdatePhase1Partial,
    _Reserved,_
    UpdatePhase1CompletePartial,
    UpdatePhase1CompleteFullPhase2NotDone,
   Phase2UpdateInProgress = 0x81,
.....
    UpdateFail = 0xff,
}

0x05 here, means UpdatePhase1CompleteFullPhase2NotDone.

closed 2024-04-19 by hughsie