fwupd/fwupd #7133: Control-flow Enforcement Technology is reported as supported when it is not

andreldmonteiro, 2024-04-19:
**Describe the bug**
Control-flow Enforcement Technology is reported as supported in HSI-3 Tests and unsupported on Runtime Tests, according to AMD my Ryzen 5 7600X processor doesn't have support for this technology.

**Steps to Reproduce**
Just run fwupd.

**Expected behavior**
Should return correct values.

**fwupd version information**
fwupd-1.9.15-1.fc40.x86_64

```shell
fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.15
compile   org.freedesktop.Passim        0.1.5
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.15
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.Passim        0.1.5
runtime   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.8.5-301.fc40.x86_64
runtime   org.freedesktop.fwupd         1.9.15
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Pre-installed
<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
ASUS System Product Name
│
├─Dispositivo desconhecido:
│     ID do dispositivo:  60825d8c28117a7abde8e55e2826d555e76a74c0
│     GUID:               bc9f6c6f-473f-5bc2-a0d1-3171c965d0d3 ← GPIO\ID_AMDIF031:00
│   
├─Dispositivo desconhecido:
│     ID do dispositivo:  c280f385ae00c1689fbf6893b99977df01464f5f
│     GUID:               74da2946-dc9a-52d2-add0-4f4d84cfa737 ← GPIO\ID_AMDIF031:01
│   
├─AMD Ryzen 5 7600X 6-Core Processor:
│ │   ID do dispositivo:  4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Versão atual:       0x0a601206
│ │   Fornecedor:         Advanced Micro Devices, Inc.
│ │   GUIDs:              16dd7d28-eade-5077-adb8-0775fdfbb2e5 ← CPUID\PRO_0&FAM_19&MOD_61
│ │                       6c9f4098-a058-5cd6-a1ce-36bfe27404fb ← CPUID\PRO_0&FAM_19&MOD_61&STP_2
│ │   Opções do dispositivo:• Dispositivo interno
│ │ 
│ └─Secure Processor:
│       ID do dispositivo:c54ab0237d7a8db8c717b68e0be78e4374a2a079
│       Versão atual:     00.42.00.13
│       Versão do gestor de arranque:00.42.00.13
│       Fornecedor:       Advanced Micro Devices, Inc. (PCI:0x1022)
│       GUIDs:            9844da3e-1df2-52fe-9413-d4378af6221e ← PCI\VEN_1022&DEV_1649
│                         f3ba4376-77c1-5b2d-bf2e-a6a144b809c1 ← PCI\VEN_1022&DEV_1649&SUBSYS_10438877
│       Opções do dispositivo:• Dispositivo interno
│     
├─GPIO controller:
│     ID do dispositivo:  f685512aa07369c9e77742acef941d779d31e766
│     GUID:               37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│   
├─Mass storage controller:
[... 108 lines cut ...]
│                         • Unsigned Payload
│   
└─WD BLACK SN850X 1000GB:
      ID do dispositivo:  04e17fcf7d3de91da49a163ffe4907855c3648be
      Resumo:             NVM Express solid state drive
      Versão atual:       620331WD
      Fornecedor:         Sandisk Corp (NVME:0x15B7)
      Número de série:    223603800398
      GUIDs:              69749ddc-cd7a-542e-981f-620656c59918 ← NVME\VEN_15B7&DEV_5030
                          53d3c0f5-42b2-5cc9-8a52-fa350528dc02 ← NVME\VEN_15B7&DEV_5030&SUBSYS_15B75030
                          0680d54c-3981-5294-90fd-decf472ab2db ← WD_BLACK SN850X 1000GB
      Opções do dispositivo:• Dispositivo interno
                          • Atualizável
                          • O sistema exige uma fonte de alimentação externa
                          • Precisa de uma reinício após a instalação
                          • O dispositivo pode ser usado durante a atualização
```

</details>

**Additional questions**

- Operating system and version: Fedora 40
- Have you tried rebooting? Yes
- Is this a regression? Yes

superm1, 2024-04-19:
What this is telling you is the hardware supports it (HSI 3) but the software stack (compiler, kernel, software) doesn't.

andreldmonteiro, 2024-04-19:
Hi, actually I really don't think it really supports it because HSI 3 always reported failure until upgrading the kernel to a major release a few releases ago.

superm1, 2024-04-19:
No, Ryzen hardware really does support shadow stack (cet). We changed the reporting in fwupd a few releases ago.

superm1, 2024-04-19:
https://github.com/fwupd/fwupd/commit/a335faca1f24ccc3ee28b0b7170655f0255ec058

andreldmonteiro, 2024-04-19:
Alright, I though on the desktop it was only available in the PRO parts, that tells something, why I can enable cryptographic RAM on the BIOS and it works, on a desktop part.

closed 2024-04-19 by andreldmonteiro