fwupd/fwupd #7144: fwupd[4935]: Failed to load daemon: syscall filtering is configured but not working

pfactum, 2024-04-22:
**Describe the bug**

`fwupd` fails to start with the following message:

```
fwupd[4935]: Failed to load daemon: syscall filtering is configured but not working
```

**Steps to Reproduce**

1. update from 1.9.16 to 1.9.17
2. run `sudo fwupdmgr refresh`

**Expected behavior**

`fwupd` should start.

**fwupd version information**

```shell
fwupdmgr --version
```

Cannot run `^^` due to the error. This is Arch Linux, latest and greatest, `fwupd` is installed via `pacman`.

<details>

<summary>**fwupd device information**</summary>

```shell
fwupdmgr get-devices --show-all-devices
```

N/A due to error.

</details>

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? yes
- Is this a regression? yes

hughsie, 2024-04-22:
What does /usr/lib/systemd/system/fwupd.service have on arch?

pfactum, 2024-04-22:
```ini
[Unit]
Description=Firmware update daemon
Documentation=https://fwupd.org/
After=dbus.service
Before=display-manager.service
ConditionVirtualization=!container

[Service]
Environment="FWUPD_SYSCALL_FILTER=systemd"
Type=dbus
TimeoutSec=180
RuntimeDirectory=motd.d
RuntimeDirectoryPreserve=yes
BusName=org.freedesktop.fwupd
ExecStart=/usr/lib/fwupd/fwupd
KeyringMode=private
LockPersonality=yes
MemoryDenyWriteExecute=yes
NoNewPrivileges=no
PrivateDevices=no
PrivateTmp=yes
ProtectClock=yes
ProtectControlGroups=yes
ProtectHome=yes
ProtectHostname=yes
ProtectKernelLogs=yes
ProtectKernelModules=yes
ProtectKernelTunables=yes
ProtectProc=invisible
ProtectSystem=full
RestrictNamespaces=yes
RestrictRealtime=yes
RestrictSUIDSGID=yes
SystemCallArchitectures=native
SystemCallErrorNumber=EPERM
ProtectControlGroups=yes
ProtectKernelModules=yes
RestrictRealtime=yes
ReadWritePaths=-/boot/efi -/efi/EFI -/boot/EFI -/boot/grub
ConfigurationDirectory=fwupd
StateDirectory=fwupd
CacheDirectory=fwupd
RestrictAddressFamilies=AF_NETLINK AF_UNIX AF_INET AF_INET6
SystemCallFilter=@system-service @raw-io
SystemCallFilter=~@resources
```

rpigott, 2024-04-22:
Seems like the fwupd test uses mlock:

https://github.com/fwupd/fwupd/blob/9e3c37646184792e61e783923f4498e269d78662/src/fu-daemon.c#L2407

but this is explicitly allowed by the SystemCallFilter setting:
```
$ systemctl cat fwupd | grep SystemCallFilter
SystemCallFilter=@system-service @raw-io
SystemCallFilter=~@resources
$ print -l $(systemctl show -P SystemCallFilter fwupd) | grep mlock
mlock
mlock2
mlockall
```

(mlock is a member of `@system-service`, and not a member of `@resources` according to `systemd-analyze syscall-filter`.

hughsie, 2024-04-22:
> SystemCallFilter=@system-service @raw-io
> SystemCallFilter=~@resources

That's not what upstream was meant to have; I forgot to backport c4af843d2274d051088730ffd4fbc4237f498969

Can you include c4af843d2274d051088730ffd4fbc4237f498969 and see if that works? If so, I'll do a new tarball tomorrow.

hughsie, 2024-04-22:
I'm also surprised too. Maybe it's something to do with it running in a container?

pfactum, 2024-04-22:
Yup, I can confirm c4af843d2274d051088730ffd4fbc4237f498969 fixes this:

```ini
[Unit]
Description=Firmware update daemon
Documentation=https://fwupd.org/
After=dbus.service
Before=display-manager.service
ConditionVirtualization=!container

[Service]
Environment="FWUPD_SYSCALL_FILTER=systemd"
Type=dbus
TimeoutSec=180
RuntimeDirectory=motd.d
RuntimeDirectoryPreserve=yes
BusName=org.freedesktop.fwupd
ExecStart=/usr/lib/fwupd/fwupd
KeyringMode=private
LockPersonality=yes
MemoryDenyWriteExecute=yes
NoNewPrivileges=no
PrivateDevices=no
PrivateTmp=yes
ProtectClock=yes
ProtectControlGroups=yes
ProtectHome=yes
ProtectHostname=yes
ProtectKernelLogs=yes
ProtectKernelModules=yes
ProtectKernelTunables=yes
ProtectProc=invisible
ProtectSystem=full
RestrictNamespaces=yes
RestrictRealtime=yes
RestrictSUIDSGID=yes
SystemCallArchitectures=native
SystemCallErrorNumber=EPERM
ProtectControlGroups=yes
ProtectKernelModules=yes
RestrictRealtime=yes
ReadWritePaths=-/boot/efi -/efi/EFI -/boot/EFI -/boot/grub
ConfigurationDirectory=fwupd
StateDirectory=fwupd
CacheDirectory=fwupd
RestrictAddressFamilies=AF_NETLINK AF_UNIX AF_INET AF_INET6
SystemCallFilter=@basic-io @file-system @io-event @ipc @network-io @process @sync ioctl uname @raw-io
```

```
$ fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.19
compile   org.freedesktop.Passim        0.1.7
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.17
runtime   org.freedesktop.fwupd-efi     1.6
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.19
runtime   org.freedesktop.Passim        0.1.7
runtime   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.8.0-pf6
runtime   org.freedesktop.fwupd         1.9.17
```

hughsie, 2024-04-22:
@pfactum same output from "fwupdmgr get-devices" from 1.9.16 and 1.9.17?

pfactum, 2024-04-22:
Almost:

```diff
--- 16	2024-04-22 22:33:03.163409020 +0200
+++ 17	2024-04-22 22:32:39.683324862 +0200
@@ -68,8 +68,8 @@
 │ └─UEFI dbx:
 │       Device ID:        362301da643102b9f38477387e2193e57abaa590
 │       Summary:          UEFI revocation database
-│       Current version:  423
-│       Minimum Version:  423
+│       Current version:  371
+│       Minimum Version:  371
 │       Vendor:           UEFI:Linux Foundation
 │       Install Duration: 1 second
 │       GUIDs:            f8ff0d50-c757-5dc3-951a-39d86e16f419 ← UEFI\CRT_D7F66BE77CEF858C174BF4338A99263C8795B74E02026411F5F532F716AE3263&ARCH_X64
@@ -77,6 +77,7 @@
 │                         0c7691e1-b6f2-5d71-bc9c-aabee364c916 ← UEFI\CRT_ED1FE72CB9CA31C9AF5B757AFCD733323D675825032E6CED7FE1AE9EB767998C&ARCH_X64
 │       Device Flags:     • Internal device
 │                         • Updatable
+│                         • Supported on remote server
 │                         • Needs a reboot after installation
 │                         • Device is usable for the duration of the update
 │                         • Only version upgrades are allowed
@@ -136,5 +137,6 @@
       GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                           279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
       Device Flags:       • Updatable
+                          • Supported on remote server
                           • Signed Payload
```

pfactum, 2024-04-22:
(_FWIW, I didn't do a resync between jumping across the version; the list of actual devices is the same_)

hughsie, 2024-04-22:
@pfactum if you `fwupdmgr refresh` do you get the old list back?

michaellass, 2024-04-22:
https://github.com/fwupd/fwupd/commit/c4af843d2274d051088730ffd4fbc4237f498969 confirmed as a fix here as well.

`fwupdmgr get-devices` gives me different results, though:

```diff
--- 1.9.16
+++ 1.9.17
@@ -28,6 +28,7 @@
 ? ? ?                     181168b9-f5c8-51c4-a8f2-958daab8af2d ? USB\VID_17EF&PID_A392&SPI_C220
 ? ? ?                     b8ce86b1-5d74-5876-8d02-a9fd61b0a9ee ? USB\VID_17EF&PID_A392&SPI_C220&REV_0D24
 ? ? ?   Device Flags:     ? Updatable
+? ? ?                     ? Supported on remote server
 ? ? ?                     ? Cryptographic hash verification is available
 ? ? ?                     ? Device stages updates
 ? ? ?                     ? Device can recover flash failures
@@ -42,25 +43,12 @@
 ? ?   ?   GUIDs:          7a216856-8a97-550c-882e-8233751c7cf2 ? USB\VID_17EF&PID_A394
 ? ?   ?                   36a4a5cc-b78b-5419-a2ba-3a692148e9f5 ? USB\VID_17EF&PID_A394&HUB_08
 ? ?   ?   Device Flags:   ? Updatable
+? ?   ?                   ? Supported on remote server
 ? ?   ?                   ? Cryptographic hash verification is available
 ? ?   ?                   ? Device stages updates
 ? ?   ?                   ? Device can recover flash failures
 ? ?   ?                   ? Unsigned Payload
 ? ?   ? 
-? ?   ??ThinkPad USB-C Dock Gen2 PD Controller:
-? ?   ?     Device ID:    6a7820672f46c254983efcd4908583cb19141617
-? ?   ?     Current version:0.0.31
-? ?   ?     Vendor:       Cypress Semiconductor (USB:0x04B4)
-? ?   ?     Install Duration:2 minutes
-? ?   ?     GUIDs:        ebd96940-fa49-5cb1-b1a1-aa2d4dea1416 ? USB\VID_04B4&PID_521A
-? ?   ?                   5108e726-44a6-5eca-bd1a-fbbdabbe7bc1 ? USB\VID_04B4&PID_521A&SID_1F00&APP_6D64
-? ?   ?                   2717d94a-8560-5008-884f-59e5c35a72b3 ? USB\VID_04B4&PID_521A&SID_1F00&APP_6D64&MODE_fw2
-? ?   ?     Device Flags: ? Updatable
-? ?   ?                   ? System requires external power source
-? ?   ?                   ? Device stages updates
-? ?   ?                   ? Device can recover flash failures
-? ?   ?                   ? Unsigned Payload
-? ?   ?   
 ? ?   ??ThinkPad USB-C Dock Gen2 USB Audio:
 ? ?         Device ID:    86d082eb55867f3b57b29971f95664475fe7e20c
 ? ?         Summary:      CX21985 USB audio device
@@ -83,6 +71,7 @@
 ?                         8b2439a7-f319-5583-950a-236eeecfe918 ? USB\VID_17EF&PID_A393
 ?                         41802a3a-da33-52c9-92d0-b3e4c8fc45d8 ? USB\VID_17EF&PID_A393&HUB_0008
 ?       Device Flags:     ? Updatable
+?                         ? Supported on remote server
 ?                         ? Cryptographic hash verification is available
 ?                         ? Device stages updates
 ?                         ? Device can recover flash failures
@@ -122,6 +111,7 @@
 ? ?   Serial Number:      219227106458969
 ? ?   GUID:               8088f861-6318-5b1e-9ce4-fbddbedb09ac ? USB\VID_06CB&PID_00BD
 ? ?   Device Flags:       ? Updatable
+? ?                       ? Supported on remote server
 ? ?                       ? Cryptographic hash verification is available
 ? ?                       ? Signed Payload
 ? ? 
@@ -134,19 +124,21 @@
 ?       GUIDs:            7c5a1e62-38fa-5859-9337-09dbac6377e4 ? USB\VID_06CB&PID_00BD-cfg
 ?                         21a62213-7012-552c-9f1a-a51d07c87242 ? USB\VID_06CB&PID_00BD&CFG1_3462&CFG2_0
 ?       Device Flags:     ? Updatable
+?                         ? Supported on remote server
 ?                         ? Only version upgrades are allowed
 ?                         ? Signed Payload
 ?     
 ??System Firmware:
 ? ?   Device ID:          e253645641291c164e71d1690c5ce2d7e5f80fe5
 ? ?   Summary:            UEFI System Resource Table device (updated via NVRAM)
-? ?   Current version:    65582
[... 21 lines cut ...]
 ?       Install Duration: 1 second
 ?       GUIDs:            5971a208-da00-5fce-b5f5-1234342f9cf7 ? UEFI\CRT_A9087D1044AD18F7A94916D284CBC01827CF23CD8F60B79072C9CAA1FEF4D649&ARCH_X64
 ?                         f8ba2887-9411-5c36-9cee-88995bb39731 ? UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
 ?       Device Flags:     ? Internal device
 ?                         ? Updatable
+?                         ? Supported on remote server
 ?                         ? Needs a reboot after installation
 ?                         ? Device is usable for the duration of the update
 ?                         ? Only version upgrades are allowed
@@ -195,6 +188,7 @@
 ?                         cf1bdd9f-29bc-5dab-9d14-dad7a4e06d16 ? MST-panamera-vmm5322-596
 ?                         fa28f9c4-c5da-57d8-802b-c644e06159b4 ? MST-panamera-596
 ?     Device Flags:       ? Updatable
+?                         ? Supported on remote server
 ?                         ? Device stages updates
 ?                         ? Unsigned Payload
 ?     Device Requests:    ? Message
@@ -285,6 +279,7 @@
       Device Flags:       ? Internal device
                           ? Updatable
                           ? System requires external power source
+                          ? Supported on remote server
                           ? Needs a reboot after installation
                           ? Device is usable for the duration of the update
```

hughsie, 2024-04-22:
> -? ?   ??ThinkPad USB-C Dock Gen2 PD Controller:

So that's a slight worry; can you get the diff of the logs of `sudo fwupdtool get-devices --no-timestamp -vv` for old and new please.

pfactum, 2024-04-22:
For me, with full `/var/lib/fwupd` wipe between version jumps there's no difference in `fwupdmgr get-devices` whatsoever.

michaellass, 2024-04-22:
I think it was a caching issue here as well. I cannot reproduce the difference and when wiping /var/lib/fwupd after a version change and afterwards doing a `--refresh`, there is no difference whatsoever in the output.

hughsie, 2024-04-22:
Did the aarch64 self tests work on aarch64? They failed on Fedora.

superm1, 2024-04-22:
> Did the aarch64 self tests work on aarch64? They failed on Fedora.

We don't run aarch64 for CI I thought

hughsie, 2024-04-22:
> We don't run aarch64 for CI I thought

The failure was in koji... :( https://kojipkgs.fedoraproject.org//work/tasks/578/116740578/build.log

michaellass, 2024-04-30:
The issue was fixed in 1.9.18, so I think this can be closed.

closed 2024-05-03 by superm1

ambasta, 2024-05-11:
I am facing this issue on fwupd 1.9.18

```
[ebuild   R    ] sys-apps/fwupd-1.9.18::gentoo  USE="amdgpu archive bash-completion bluetooth fastboot flashrom gnutls gusb lzma nvme policykit spi sqlite systemd tpm uefi -amt -cbor (-elogind) -gtk-doc -introspection -logitech -minimal -modemmanager -synaptics -test -test-full" PYTHON_SINGLE_TARGET="python3_12 -python3_10 -python3_11" 0 KiB
```
but the service fails to start
```
# systemctl restart fwupd
Job for fwupd.service failed because the control process exited with error code.
See "systemctl status fwupd.service" and "journalctl -xeu fwupd.service" for details.
# systemctl status fwupd
× fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static)
     Active: failed (Result: exit-code) since Sat 2024-05-11 15:24:19 IST; 5s ago
       Docs: https://fwupd.org/
    Process: 83767 ExecStart=/usr/libexec/fwupd/fwupd (code=exited, status=1/FAILURE)
   Main PID: 83767 (code=exited, status=1/FAILURE)
        CPU: 25ms

May 11 15:24:19 lander systemd[1]: Starting Firmware update daemon...
May 11 15:24:19 lander fwupd[83767]: Failed to load daemon: syscall filtering is configured but not working
May 11 15:24:19 lander systemd[1]: fwupd.service: Main process exited, code=exited, status=1/FAILURE
May 11 15:24:19 lander systemd[1]: fwupd.service: Failed with result 'exit-code'.
May 11 15:24:19 lander systemd[1]: Failed to start Firmware update daemon.
```

superm1, 2024-05-11:
Install 1.9.19, it has fixes for this.

ambasta, 2024-11-13:
Exists again on 2.0.1?
```
[ebuild   R    ] sys-apps/fwupd-2.0.1::gentoo  USE="amdgpu archive bash-completion bluetooth fastboot flashrom gnutls lzma nvme policykit spi sqlite systemd tpm uefi -amt -cbor (-elogind) -gtk-doc -introspection -logitech -minimal -modemmanager -synaptics -test" PYTHON_SINGLE_TARGET="python3_13 -python3_10 -python3_11 -python3_12" 0 KiB
```

```
systemctl status fwupd
× fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static)
     Active: failed (Result: exit-code) since Wed 2024-11-13 10:04:46 IST; 4s ago
 Invocation: 30081fbe1c61400c90288767fb1c49f4
       Docs: https://fwupd.org/
    Process: 34417 ExecStart=/usr/libexec/fwupd/fwupd (code=exited, status=1/FAILURE)
   Main PID: 34417 (code=exited, status=1/FAILURE)
   Mem peak: 2.4M
        CPU: 25ms

Nov 13 10:04:46 lander systemd[1]: Starting Firmware update daemon...
Nov 13 10:04:46 lander fwupd[34417]: Failed to load daemon: syscall filtering is configured but not working
Nov 13 10:04:46 lander systemd[1]: fwupd.service: Main process exited, code=exited, status=1/FAILURE
Nov 13 10:04:46 lander systemd[1]: fwupd.service: Failed with result 'exit-code'.
Nov 13 10:04:46 lander systemd[1]: Failed to start Firmware update daemon.
```

superm1, 2024-11-13:
The code is [still there](https://github.com/fwupd/fwupd/blob/2.0.1/src/fu-daemon.c#L154), so this is probably a different regression.  Can you please open a new bug?