fwupd/fwupd #7157: Pre-boot DMA Protection setting not detected

perroboc, 2024-04-23:
**Describe the bug**
On an Asus TUF Gaming X670E-PLUS WIFI MOBO, enabling Pre-boot DMA Protection on UEFI is not detected by `fwupdtool`.

[image]

<details>

<summary>Log: <code>fwupdtool security --force --verbose</code></summary>

```console
$ sudo fwupdtool security --force --verbose
[sudo] password for perro: 
21:29:49.128 FuDebug              verbose to info (on console 1)
21:29:49.128 FuEngine             starting fwupd 1.9.16…
21:29:49.129 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…                 [ -                                     ]21:29:49.130 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
21:29:49.130 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
21:29:49.130 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
21:29:49.130 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…                 [                                       ]21:29:49.130 FuEngine             client certificate now exists: nothing to do
Loading…                 [*                                      ]21:29:49.134 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
21:29:49.134 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
Loading…                 [*                                      ]21:29:49.134 FuContext            SMBIOS Manufacturer=ASUS
21:29:49.134 FuContext            SMBIOS EnclosureKind=3
21:29:49.134 FuContext            SMBIOS Family=To be filled by O.E.M.
21:29:49.134 FuContext            SMBIOS ProductName=System Product Name
21:29:49.134 FuContext            SMBIOS ProductSku=SKU
21:29:49.134 FuContext            SMBIOS BiosVendor=American Megatrends Inc.
21:29:49.134 FuContext            SMBIOS BiosVersion=2613
21:29:49.134 FuContext            SMBIOS BiosMajorRelease=1a
21:29:49.134 FuContext            SMBIOS BiosMinorRelease=0d
21:29:49.134 FuContext            SMBIOS FirmwareMajorRelease=ff
21:29:49.134 FuContext            SMBIOS FirmwareMinorRelease=ff
21:29:49.134 FuContext            SMBIOS BaseboardManufacturer=ASUSTeK COMPUTER INC.
21:29:49.134 FuContext            SMBIOS BaseboardProduct=TUF GAMING X670E-PLUS WIFI
21:29:49.134 FuContext            failed to load fdt: cannot find /sys/firmware/fdt or override /var/lib/fwupd/system.dtb
21:29:49.134 FuContext            added udev subsystem watch of firmware-attributes
Loading…                 [**                                     ]21:29:49.135 FuEngine             loading metadata for remote 'vendor-directory'
21:29:49.135 FuEngine             ignoring: /usr/share/fwupd/remotes.d/vendor/firmware/README.md
21:29:49.135 FuEngine             ignoring: Error opening directory “/var/lib/fwupd/local.d”: No such file or directory
21:29:49.135 FuEngine             ignoring: Error opening directory “/usr/share/fwupd/local.d”: No such file or directory
21:29:49.135 FuEngine             2541 components now in silo
Loading…                 [***                                    ]21:29:49.144 FuContext            added udev subsystem watch of platform
21:29:49.144 FuContext            added udev subsystem watch of pci
21:29:49.144 FuContext            added udev subsystem watch of block for plugin android_boot
21:29:49.144 FuContext            added udev subsystem watch of i2c
21:29:49.144 FuContext            added udev subsystem watch of i2c-dev
21:29:49.144 FuContext            added udev subsystem watch of hidraw
21:29:49.144 FuContext            added udev subsystem watch of gpio for plugin gpio
21:29:49.144 FuContext            added udev subsystem watch of mei
21:29:49.144 FuContext            added udev subsystem watch of iommu for plugin iommu
21:29:49.144 FuContext            added udev subsystem watch of drm
21:29:49.144 FuContext            added udev subsystem watch of drm_dp_aux_dev for plugin kinetic_dp
21:29:49.144 FuContext            added udev subsystem watch of video4linux
21:29:49.144 FuContext            added udev subsystem watch of msr for plugin msr
21:29:49.145 FuContext            added udev subsystem watch of mtd for plugin mtd
Loading…                 [***                                    ]21:29:49.145 FuContext            added udev subsystem watch of nvme for plugin nvme
21:29:49.145 FuContext            added udev subsystem watch of serio
21:29:49.145 FuContext            added udev subsystem watch of thunderbolt
21:29:49.145 FuContext            added udev subsystem watch of tpm for plugin tpm
21:29:49.145 FuEngine             plugins disabled: test, test_ble
Loading…                 [***                                    ]21:29:49.145 FuContext            battery threshold now 10
Loading…                 [-                                      ]21:29:49.145 FuEngine             disabling plugin because: failed to startup using flashrom: no HwIDs found
21:29:49.145 FuEngine             disabling plugin because: failed to startup using lenovo_thinklmi: unsupported manufacturer
21:29:49.146 FuEngine             disabling plugin because: failed to startup using redfish: no username and password specified, and no vendor quirk for 'ipmi-create-user'
Loading…                 [****                                   ]21:29:49.147 FuEngine             disabling plugin because: failed to startup using dell: firmware updating not supported: no structures with type de
21:29:49.147 FuEngine             disabling plugin because: failed to coldplug using acpi_phat: Failed to open file “/sys/firmware/acpi/tables/PHAT”: No such file or directory
21:29:49.147 FuDeviceList         ::added 4bde70ba4e39b28f9eab1628f9dd6e6244c03027 [AMD Ryzen 9 7950X 16-Core Processor]
Loading…                 [****                                   ]21:29:49.148 FuPluginMm           ModemManager 1.22.0-3.fc40 is available
Loading…                 [*****                                  ]21:29:49.148 FuEngine             disabling plugin because: failed to coldplug using superio: not supported when kernel locked down
[... 729 lines cut ...]
│     GUID:               54ef26b1-94d9-51d5-9e46-91732769a3c2 ← UEFI\CRT_6F52EF400A8634497D726A7A7D469FCA5905BADD
│   
└─WDS250G3X0C-00SJG0:
      Device ID:          8e92eda1ab953dd9f27d11cd320c36e5d1127030
      Summary:            NVM Express solid state drive
      Current version:    102000WD
      Vendor:             Sandisk Corp (NVME:0x15B7)
      Serial Number:      19171B802381
      GUIDs:              ff2112dc-038c-596d-90ca-d43c5077c6ec ← NVME\VEN_15B7&DEV_5002
                          c528df4b-7972-5880-8cb1-330415e2dc6a ← NVME\VEN_15B7&DEV_5002&SUBSYS_15B75002
                          bcf288b7-13bb-5c90-9362-3b0fa7917a83 ← WDS250G3X0C-00SJG0
      Device Flags:       • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
    

```

</details>

**Additional questions**

- Operating system and version: Fedora 40
- Have you tried rebooting?: Yes
- Is this a regression?: I don't know

superm1, 2024-04-23:
Can you please attach the acpi ivrs table

perroboc, 2024-04-23:
I get an error while trying to do so:
```console
$ acpidump
Could not open table file: /sys/firmware/acpi/tables/SSDT4
Could not get ACPI tables, AE_ACCESS
```

Is there any other way to get them?

superm1, 2024-04-23:
/sys/firmware/acpi/tables/IVRS

perroboc, 2024-04-23:
Had to run it as root, my bad!

https://gist.github.com/perroboc/a5ffae1619fad6ebec47431be27dc06e

superm1, 2024-04-24:
```
04:04:58.751 FuPlugin             add_security_attrs(acpi_ivrs)
04:04:58.751 FuCommon             mapped file /tmp/IVRS of size 0xc8
04:04:58.751 FuStruct             AcpiTable:
  signature: IVRS
  length: 0xc8
  revision: 0x2
  checksum: 0x4d
  oem_id: AMD  
  oem_table_id: AmdTable
  oem_revision: 0x1000000
04:04:58.752 FuPluginAcpiIvrs     Flags: 0x41
```

I don't see any problem with what fwupd is parsing from this table.  It seems to me that it's not properly enabled by BIOS.  Maybe you need to change "Kernel DMA protection indicator"?

perroboc, 2024-04-24:
Thank you! It was, In fact, 3 things:

```
Advanced\AMD CBS\IOMMU: Enabled
Advanced\AMD CBS\NBIO Common Options\Pre-boot DMA Protection Indicator: Enabled
Advanced\AMD CBS\NBIO Common Options\Kernel DMA Protection Indicator: Enabled
```

Now, to keep tuning the UEFI to not have HSI runtime issues >_<

closed 2024-04-24 by perroboc

hughsie, 2024-04-24:
@superm1 Could we set those using the sysfs api?

superm1, 2024-04-24:
> @superm1 Could we set those using the sysfs api?

No sysfs API that I know of for Asus motherboards.

I'm surprised that they actually exported those options out to an end user, they're usually "hidden", but it's fortunate they did so you can actually fix this!

To be honest I think this should be raised to Asus support to pick more secure bios defaults.

hughsie, 2024-04-24:
@perroboc can we leave that to you please?

perroboc, 2024-04-24:
Sure, I'll send an email to Asus Support :)

Also, just enabled these settings:
```
Host Security Events
  2024-04-24 14:17:55:  ✔ Encrypted RAM changed: Not supported → Encrypted
```
[image]

Can't find these, though:
```
✘ SPI replay protection:         Not supported
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled
✘ Processor rollback protection: Disabled
✘ CET OS Support:                Not supported
```

superm1, 2024-04-24:
> ✘ SPI replay protection:         Not supported

This requires some hardware design consideration in the type of SPI part that is used.  You might not be able to affect it just by BIOS settings if it's the wrong kind of part.  But if they export the relevant BIOS setting it should be RPMC.

> ✘ Suspend-to-idle:               Disabled
> ✘ Suspend-to-ram:                Enabled

_Most_ AMD desktop SoCs don't use suspend to idle today, I don't think you'll be able to do much about this today.  They might export a BIOS setting called "Modern Standby" you can try to enable but 🐉 ahead as a result.

> ✘ Processor rollback protection: Disabled

This is something much more core in the BIOS than you can set.

> ✘ CET OS Support:                Not supported

This requires a new enough compiler, correct kernel options and the software stack to be compiled with CET support.  I don't think _anyone_ is meeting this yet in general purpose distros.

dominikegert, 2024-07-06:
> > ✘ CET OS Support:                Not supported
> 
> This requires a new enough compiler, correct kernel options and the software stack to be compiled with CET support. I don't think _anyone_ is meeting this yet in general purpose distros.

Actually, I have this recognised as enabled/Supported

HSI-1
✔ BIOS firmware updates:         Enabled
✔ csme manufacturing mode:       Locked
✔ csme override:                 Locked
✔ csme v0:16.1.30.2264:          Valid
✔ Platform debugging:            Disabled
✔ SPI write:                     Disabled
✔ SPI lock:                      Enabled
✔ SPI BIOS region:               Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked

HSI-2
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard OTP fuse:      Valid
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✘ Intel BootGuard ACM protected: Invalid
✘ Intel BootGuard verified boot: Invalid
✘ TPM PCR0 reconstruction:       Invalid

HSI-3
✔ CET Platform:                  Supported
✔ Suspend-to-ram:                Disabled
✘ Intel BootGuard error policy:  Invalid
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled

HSI-4
✔ SMAP:                          Enabled
✘ Encrypted RAM:                 Disabled

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ CET OS Support:                Supported
✔ Linux kernel:                  Untainted
✘ Linux kernel lockdown:         Unknown
✘ Linux swap:                    Unencrypted
✘ UEFI secure boot:              Disabled